lvgl/lvgl #3982: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'

wolfgangwayland7, 2023-02-12:
The "lvgl-release-v8.3" {either-> down loaded as a zip only or cloned} still dose not have 
"clip_area" defined inside the {lv_obj_draw_part_dsc_t   [struct] }  as of the  of  2-11-2023.

The struct is int the "lv_obj_draw.h" header inside the "lvgl-release-v8.3/src/core"  folder...
Thanks

kisvegabor, 2023-02-14:
Exactly what is not working? We have CI running on each commit testing the demos too so they should work on the `release/v8.3` branch.

modelvincent, 2023-02-19:
Hi, I have the same issue here with:
 - LVGL 8.3.5
 - lv_examples 8.1.0
 - Arduino 1.8.19

kisvegabor, 2023-02-20:
There isn't `clip_area` in `lv_obj_draw_part_dsc_t` as it has [`draw_ctx`](https://github.com/lvgl/lvgl/blob/release/v8.3/src/core/lv_obj_draw.h#L43) which contains the current `clip_area`.

But I don't understand why/where/how causes it issues?

closed 2023-03-15 by lvgl-bot

mverch67, 2023-07-05:
Had same issue and could fix it:

When defining LV_USE_DEMO_WIDGETS 1 in lv_conf.h in order to try the lv_demo_widgets() demo function you'll get a bunch of compile errors in lv_demo_widgets.c:

```
.pio/libdeps/WT32-SC01-PLUS/lv_demos/src/lv_demo_widgets/lv_demo_widgets.c:1284:41: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'; did you mean 'draw_area'?
             lv_draw_rect(&bg_area, dsc->clip_area, &rect_dsc);
.pio/libdeps/WT32-SC01-PLUS/lv_demos/src/lv_demo_widgets/lv_demo_widgets.c:1290:43: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'; did you mean 'draw_area'?
             lv_draw_label(&txt_area, dsc->clip_area, &label_dsc, buf, NULL);
.pio/libdeps/WT32-SC01-PLUS/lv_demos/src/lv_demo_widgets/lv_demo_widgets.c:1338:57: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'; did you mean 'draw_area'?
                 _lv_area_intersect(&obj_clip_area, dsc->clip_area, &obj->coords);
```

The correct parameters should read:
1284/1400/1451/1457/1462:  lv_draw_rect(dsc->draw_ctx, &rect_dsc, &bg_area);
1290/1406:  lv_draw_label(dsc->draw_ctx, &label_dsc, &txt_area, buf, NULL);
1338: _lv_area_intersect(dsc->draw_ctx, &obj_clip_area, &obj->coords);

kisvegabor, 2024-04-02:
That version list was really outdated. I've just updated it: 
[image]

I've removed the `v8 (latest minor)` option as it wasn't really used and it's better to be explicit about the version, once it's selected.

cybekRT, 2024-04-02:
That should solve the misunderstanding about version which the examples are for. Thanks!

ArsalanAliMujtaba, 2024-10-23:
in the double ended slider I replaced this

            lv_draw_label(&label_area, dsc->clip_area, &label_draw_dsc, buf, NULL);
            
 by this 

      lv_draw_label(dsc->draw_ctx, &label_draw_dsc, &label_area, buf, NULL);
      
and it worked. I think its just a stupid programming mistake by lvgl prorgammers in lvgl 8.0 as I am using those cheap yellow displays. And I am just sharing this so that someone couldnt have to grill their minds as I did.