lvgl/lvgl #3984: Can't turn off performance and memory monitor

edo44, 2023-02-12:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
In lv_conf.h LV_USE_USER_DATA 0 and LV_USE_MEM_MONITOR 0 don't disable the little monitor windows.

The include path is correct. Every other define works fine.

This is the display library I am using with an ESP32-3248S035C
https://github.com/rzeldent/esp32-smartdisplay

### To Reproduce
It's the same with every kind of project so I just attached my lv_conf.h

### Expected behavior
Monitors should not be displayed.

### Screenshots or video

[lv_conf.txt](https://github.com/lvgl/lvgl/files/10717477/lv_conf.txt)

kisvegabor, 2023-02-14:
These are NOT disabled in the `lv_conf` file you have attached:
[image]

`LV_USE_USER_DATA` is not related. You need to set
```c
/*1: Show CPU usage and FPS count*/
#define LV_USE_PERF_MONITOR 0
...

/*1: Show the used memory and the memory fragmentation
 * Requires LV_MEM_CUSTOM = 0*/
#define LV_USE_MEM_MONITOR 0
```

edo44, 2023-02-14:
> These are NOT disabled in the `lv_conf` file you have attached: [image]
>
> `LV_USE_USER_DATA` is not related. You need to set
>
> ```c
> /*1: Show CPU usage and FPS count*/
> #define LV_USE_PERF_MONITOR 0 [...]

I'm sorrry that was the wrong config file

Anyway I found the problem, I had two config files, one in the library folder and one in the src folder. I copied the file because the compiler was giving me errors.

Thank you by the way, amazing work.

kisvegabor, 2023-02-14:
I close this issue then :)

closed 2023-02-14 by kisvegabor