lvgl/lvgl #3985: conflict between semaphore and lv_timer_t

jiabuda, 2023-02-13:
### Perform all steps below and tick them with [x]
- [ Y ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ Y ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ Y ] Update lvgl to the latest version
- [ Y ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Hi, i am using LVGL 8.3.5 and ESP32-C3 to make a small demo, which screen A countdown 3 seconds and enter screen B, i try to use lv_timer_t instead of mcu's builtin timer function to finish it . Yet after I use the semaphore around the lv_task_handler, i found the screen stuck when i am using lv_timer_t. Here is the peudo code below. After I read the lv_timer_t doc, it seem the timer will execute inside the lv_task_handler, it means that the semaphore will be taken twice and get stuck.

So is any good solution to do this countdown and go into next screen?

```c
 while (1)
    {
        /* Delay 1 tick (assumes FreeRTOS tick is 10ms */
        vTaskDelay(pdMS_TO_TICKS(10));

        /* Try to take the semaphore, call lvgl related function on success */
        if (pdTRUE == xSemaphoreTake(xGuiSemaphore, portMAX_DELAY))
        {
            lv_task_handler();
            xSemaphoreGive(xGuiSemaphore);
        }
    }

-----------------------------------------------

void countdown_cb()
{
    countdown--;
    if (countdown <= 0)
    {
        showScreenB();
    }
}

showScreenA()
{
    if (pdTRUE == xSemaphoreTake(xGuiSemaphore, portMAX_DELAY))
    {
        LV_OBJ_CREATE
        xSemaphoreGive(xGuiSemaphore);
    }
    lv_timer_t *t = lv_timer_create(countdown_cb, 1000, NULL);
}

destroyScreenA()
{
    if (pdTRUE == xSemaphoreTake(xGuiSemaphore, portMAX_DELAY))
    {
        LV_OBJ_DEL
        xSemaphoreGive(xGuiSemaphore);
    }
}

showScreenB()
{
    if (pdTRUE == xSemaphoreTake(xGuiSemaphore, portMAX_DELAY))
    {
        LV_OBJ_CREATE
        xSemaphoreGive(xGuiSemaphore);
    }
}

destroyScreenB()
{
    if (pdTRUE == xSemaphoreTake(xGuiSemaphore, portMAX_DELAY))
    {
        LV_OBJ_DEL
        xSemaphoreGive(xGuiSemaphore);
    }
}
```

kisvegabor, 2023-02-14:
The simple solution is to not use a semaphore in `showScreenB`. As you pointed out `lv_timer`s are called from `lv_timer_handler` which is already protected. 

You don't need a semaphore either if you do something in events (e.g. a button's click event) as the events are also called from `lv_timer_handler`.

jiabuda, 2023-02-14:
thanks, i understand the semaphore thing. but sometimes the showScreenB will be trigger by some gpio event, so the best solution is to create another entry called `showScreenBWithoutSemaphore`, and in different situation i should judge which one to use, right?

closed 2023-02-14 by jiabuda

kisvegabor, 2023-02-15:
>y but sometimes the showScreenB will be trigger by some gpio event, so the best solution is to create another entry called showScreenBWithoutSemaphore, and in different situation i should judge which one to use, right?

It depends on how you organize the code. 

E.g. you can decide to poll the GPIO in an `lv_timer` and from the timer you simple create the new screen.

jiabuda, 2023-02-16:
Got it, thanks.