lvgl/lvgl #3986: Entire circle won't be showed when outline set too large

Pingye007, 2023-02-14:
### Describe the bug
I set outline of an obj and change it dynamiclly.Sometimes it shows normal but sometimes not.

### To Reproduce
```c
your code here
```
void test_effects1_pressing_cb(lv_event_t* e)
{
	lv_obj_t* obj = lv_event_get_current_target(e);
	lv_style_t* s = (lv_style_t*)lv_event_get_user_data(e);
	static int start_time = 0;
	lv_style_set_outline_pad(s, lv_rand(0, 25));
	//lv_obj_invalidate(obj);
	lv_obj_refresh_self_size(obj);
}

void test_effects1()
{
	lv_obj_t* obj = lv_obj_create(lv_scr_act());
	lv_obj_set_size(obj, 150, 150);
	lv_obj_set_style_radius(obj, LV_RADIUS_CIRCLE, 0);
	lv_obj_set_style_bg_color(obj, lv_color_black(), 0);
	lv_obj_center(obj);

	lv_obj_t* label = lv_label_create(obj);
	lv_label_set_text(label, "Clicked");
	lv_obj_set_style_text_color(label, lv_color_white(), 0);
	lv_obj_center(label);

	static lv_style_t s;
	lv_style_init(&s);
	lv_style_set_border_color(&s, lv_palette_main(LV_PALETTE_YELLOW));
	lv_style_set_border_width(&s, 2);
	lv_style_set_border_side(&s, LV_BORDER_SIDE_FULL);
	lv_style_set_outline_width(&s, 2);
	lv_style_set_outline_pad(&s, 15);
	lv_style_set_outline_color(&s, lv_palette_main(LV_PALETTE_CYAN));
	 lv_obj_add_style(obj, &s, 0);
	lv_obj_add_event_cb(obj, test_effects1_pressing_cb, LV_EVENT_PRESSING, &s);
}
-->

### Expected behavior
It should be showed entire circular outline no matter what value is set to obj.
I've checked the original code in folder `sw` and `lv_sw_draw_rect`.I see drawing outline there but I'm not sure that is the key.

### Screenshots or video

This is a normal outline showing.
[image]

And that is abnormal one.
[image]

closed 2023-03-09 by lvgl-bot

kisvegabor, 2023-03-13:
Hi,

Sorry, I've completely missed this issue somehow.

The problem in your code is that you are updating a style but the widgets using that style are not notified about the style change.

You can do 2 things:
1. Report the style change with `lv_obj_report_style_change(s)` or `lv_obj_refresh_style(obj, LV_PART_ANY, LV_STATE_ANY)`
2. Use a local style which automatically updates the related object `lv_obj_set_style_outline_pad(obj, lv_rand(0, 25), 0);`

Pingye007, 2023-03-15:
@kisvegabor 
Don't say sorry about this and it is just not really an issue.You guys created a great,useful and magnificent framework for embedded system,I can imagine the difficuties you encountered when constructing structures,processing events and handling all the features and bugs.So,just do what is really urgent for you and it will be fine to figure out the little problems like this when you're spare :).

kisvegabor, 2023-03-15:
Thank you for your understanding. Sometimes it's really not easy to manage everything.  :sweat_smile:   

Also thank you for the kind words! Finally did my suggestion solve your issue?

kisvegabor, 2023-03-16:
Amazing! :blush: