lvgl/lvgl #3987: it seems memory leakage when using many animations

jiabuda, 2023-02-14:
### Perform all steps below and tick them with [x]
- [ Y ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ Y ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ Y ] Update lvgl to the latest version
- [ Y ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I am using LVGL 8.35, MCU is ESP32C3, i try to make a animation like big square is smashed into small squares and fly away, please refer to the code below. On the mcu and pc simulator, both i can see the memory increasing when i exit and enter the screen again and again, i mean continue call the `showPage` and `exitPage` in the code with GPIO. After the memory usage run above 80%, the animation screen never show and the watchdog will throw out errors. I am confused that all the var in my code is in the stack, no heap area, it seem the core function of LVGL problem? Or i use the wrong way to create the animation? Or i need to delete the animations before exit the page? Please give me a hand, thanks!

### To Reproduce
```c
#include "firework_test_page.h"

#define TAG "FIREWORK"

static lv_obj_t *fireworkTestScreen = NULL;

#define BLOCK_ROW_COUNT 4
#define BLOCK_COUNT (BLOCK_ROW_COUNT * BLOCK_ROW_COUNT)
#define BLOCK_SIZE 10
static lv_obj_t *fireBlock[BLOCK_COUNT];
static lv_anim_t fireBlockAnim[BLOCK_COUNT * 3];

static void anim_x_cb(void *var, int32_t v)
{
	lv_obj_set_x(var, v);
}

static void anim_y_cb(void *var, int32_t v)
{
	lv_obj_set_y(var, v);
}

static void anim_bg_opa_cb(void *var, int32_t v)
{
	lv_obj_set_style_bg_opa(var, v, 0);
}

static void pageCreate(lv_obj_t *parent)
{
	fireworkTestScreen = lv_obj_create(parent);
	lv_obj_clear_flag(fireworkTestScreen, LV_OBJ_FLAG_SCROLLABLE);
	lv_obj_set_size(fireworkTestScreen, LV_HOR_RES_MAX, LV_VER_RES_MAX);
	lv_obj_set_style_pad_all(fireworkTestScreen, 0, 0);

	lv_anim_timeline_t *anim_timeline = lv_anim_timeline_create();
	for (int i = 0; i < BLOCK_COUNT; i++)
	{
		fireBlock[i] = lv_obj_create(fireworkTestScreen);
		int x = 120 - BLOCK_ROW_COUNT / 2 * BLOCK_SIZE + (i % BLOCK_ROW_COUNT) * BLOCK_SIZE;
		int y = 120 - BLOCK_ROW_COUNT / 2 * BLOCK_SIZE + (i / BLOCK_ROW_COUNT) * BLOCK_SIZE;
		lv_obj_set_style_radius(fireBlock[i], 0, 0);
		lv_obj_set_style_border_width(fireBlock[i], 0, 0);
		lv_obj_set_style_outline_width(fireBlock[i], 0, 0);
		lv_obj_set_style_pad_all(fireBlock[i], 0, 0);
		lv_obj_set_size(fireBlock[i], BLOCK_SIZE, BLOCK_SIZE);
		lv_obj_set_pos(fireBlock[i], x, y);
		lv_obj_set_style_bg_color(fireBlock[i], lv_color_make(128, 0, 255), 0);

		int relateX = (x - 120 + BLOCK_SIZE / 2) * 3 + 120 - BLOCK_SIZE / 2;
		lv_anim_init(&fireBlockAnim[i * 3]);
		lv_anim_set_var(&fireBlockAnim[i * 3], fireBlock[i]);
		lv_anim_set_values(&fireBlockAnim[i * 3], x, relateX);
		lv_anim_set_exec_cb(&fireBlockAnim[i * 3], anim_x_cb);
		lv_anim_set_path_cb(&fireBlockAnim[i * 3], lv_anim_path_linear);
		lv_anim_set_time(&fireBlockAnim[i * 3], 1000);
		lv_anim_set_repeat_count(&fireBlockAnim[i * 3], LV_ANIM_REPEAT_INFINITE);
		lv_anim_timeline_add(anim_timeline, 0, &fireBlockAnim[i * 3]);

		int relateY = (y - 120 + BLOCK_SIZE / 2) * 3 + 120 - BLOCK_SIZE / 2;
		lv_anim_init(&fireBlockAnim[i * 3 + 1]);
		lv_anim_set_var(&fireBlockAnim[i * 3 + 1], fireBlock[i]);
		lv_anim_set_values(&fireBlockAnim[i * 3 + 1], y, relateY);
		lv_anim_set_exec_cb(&fireBlockAnim[i * 3 + 1], anim_y_cb);
		lv_anim_set_path_cb(&fireBlockAnim[i * 3 + 1], lv_anim_path_linear);
		lv_anim_set_time(&fireBlockAnim[i * 3 + 1], 1000);
		lv_anim_set_repeat_count(&fireBlockAnim[i * 3 + 1], LV_ANIM_REPEAT_INFINITE);
		lv_anim_timeline_add(anim_timeline, 0, &fireBlockAnim[i * 3 + 1]);

		lv_anim_init(&fireBlockAnim[i * 3 + 2]);
		lv_anim_set_var(&fireBlockAnim[i * 3 + 2], fireBlock[i]);
		lv_anim_set_values(&fireBlockAnim[i * 3 + 2], LV_OPA_100, LV_OPA_0);
		lv_anim_set_exec_cb(&fireBlockAnim[i * 3 + 2], anim_bg_opa_cb);
		lv_anim_set_path_cb(&fireBlockAnim[i * 3 + 2], lv_anim_path_linear);
		lv_anim_set_time(&fireBlockAnim[i * 3 + 2], 1000);
		lv_anim_set_repeat_count(&fireBlockAnim[i * 3 + 2], LV_ANIM_REPEAT_INFINITE);
		lv_anim_timeline_add(anim_timeline, 0, &fireBlockAnim[i * 3 + 2]);
	}
	lv_anim_timeline_start(anim_timeline);
}

void showFireworkTestPage()
{
	if (pdTRUE == xSemaphoreTake(xGuiSemaphore, portMAX_DELAY))
	{
		pageCreate(basePageMountPoint);
		xSemaphoreGive(xGuiSemaphore);
	}
}

void exitFireworkTestPage()
{
	if (pdTRUE == xSemaphoreTake(xGuiSemaphore, portMAX_DELAY))
	{
		lv_obj_del(fireworkTestScreen);
		xSemaphoreGive(xGuiSemaphore);
	}
}
```

kisvegabor, 2023-02-15:
I've tried out you code and adding `lv_anim_timeline_del(anim_timeline);` to `exitFireworkTestPage` solved the issue. Of you course you need to make `lv_anim_timeline_t *anim_timelin` global.

jiabuda, 2023-02-16:
> I've tried out you code and adding `lv_anim_timeline_del(anim_timeline);` to `exitFireworkTestPage` solved the issue. Of you course you need to make `lv_anim_timeline_t *anim_timeline` global.

Oh~ Thanks for the help.

closed 2023-02-16 by jiabuda