lvgl/lvgl #3988: Possible divide by 0 exception in image widget

fvanroie, 2023-02-15:
## Description

We have hit an edge case in the image widget when all of the following are true:
- The image src failed to load _(e.g. file does not exist on disk, is corrupted or invalid, ...)_ &nbsp;&nbsp;  _**and**_
- its `img->w` or `img->h` are still zero  &nbsp;&nbsp; _**and**_
- you try to use `lv_image_set_offset_x` or `lv_image_set_offset_y` on the image object

then you get a divide by 0 exception.

## Debugging

There is no bound check before doing `x = x % img->w; ` or `y = y % img->h;` in the unlikely case when `img->w` or `img->h` are zero:
https://github.com/lvgl/lvgl/blob/b0fa0e286c2263c5cd4d9c941f167f2442799221/src/widgets/img/lv_img.c#L163-L185

## Expectation

Gracefully handle this edge case.

## Possible Solution

I think it would be best to do this modulo operation in the `lv_img_design` function, when `img->w` and `img->h` are _known_ to be valid. The draw operation is abandoned early if either is 0, so it is safe to use them as a divider.

Another advantage would be that the **original** offsets remain untouched. Whenever another image `src` is loaded later the original offset can _still_ be applied again at drawing time. The original value is not overwritten by moving the modulo operation out of  `lv_image_set_offset_x` and `lv_image_set_offset_y`.

I have taken a look at the draw code and in v7.11 this is what I've changed: https://github.com/HASwitchPlate/lvgl/commit/e586cca9abf8837d7df762497cae0e70fffabe79

Now the application won't crash anymore when this edge case is hit.
Is this the right approach? For v8.3 the drawing code is slightly different, but I can provide a PR if needed.

## Environment

ESP32 with LVGL 7.11 but the offending code is still there in 8.x and current master branch.

kisvegabor, 2023-02-15:
Hi @fvanroie,

Thanks for pointing it out.  Keeping the original offset and doing the calculations during drawing makes sense.

Could you send a PR for all three branches? (`release/v7`, `release/v8.3` and `master`)

fvanroie, 2023-02-15:
No problem, the 3 PRs have been created.

kisvegabor, 2023-02-15:
Thank you very much! I've merged all three.