lvgl/lvgl #3989: Border only one side has weird behavior.

lanistor, 2023-02-15:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

Border only one side has weird behavior: the other side will have transparent border.
```
lv_obj_set_style_border_color(view, LV_COLOR_MAKE(0xff, 0x00, 0x00), LV_PART_MAIN);
lv_obj_set_style_border_side(view, LV_BORDER_SIDE_RIGHT, LV_PART_MAIN);
lv_obj_set_style_border_width(view, 30, LV_PART_MAIN);
```

### Expected behavior

Other side have no transparent border, and not take up space。

### Screenshots or video

https://github.com/lanistor/assets/blob/master/imgs/2023-02-15%2012.45.00.png?raw=true

kisvegabor, 2023-02-15:
It works well for me on the latest `release/v8.3`.
[image]

```c
  lv_obj_t * view = lv_obj_create(lv_scr_act());
  lv_obj_set_size(view, 100, 100);
  lv_obj_set_style_border_color(view, lv_color_make(0x00, 0x00, 0x00), LV_PART_MAIN);
  lv_obj_set_style_border_side(view, LV_BORDER_SIDE_RIGHT, LV_PART_MAIN);
  lv_obj_set_style_border_width(view, 30, LV_PART_MAIN);

```

lanistor, 2023-02-15:
> ```c
> lv_obj_t * view = lv_obj_create(lv_scr_act());
>   lv_obj_set_size(view, 100, 100);
>   lv_obj_set_style_border_color(view, lv_color_make(0x00, 0x00, 0x00), LV_PART_MAIN); [...]

It seems to have another padding, but cannot removed by `lv_obj_set_style_pad_all`:

[image]

Code:
```c++
  lv_obj_t* box = lv_obj_create(view);
  lv_obj_set_size(box, 200, 200);
  lv_obj_set_style_bg_color(box, lv_color_make(0xff, 0x00, 0x00), LV_PART_MAIN);
  lv_obj_set_style_border_color(box, lv_color_make(0x00, 0x00, 0x00),
                                LV_PART_MAIN);
  lv_obj_set_style_border_side(box, LV_BORDER_SIDE_RIGHT, LV_PART_MAIN);
  lv_obj_set_style_border_width(box, 30, LV_PART_MAIN);
  lv_obj_set_style_pad_all(box, 0, LV_PART_MAIN);

  auto label = lv_label_create(box);
  lv_label_set_text(label, "AAAAABBBBBCCCCCDDDDD");
```

kisvegabor, 2023-02-15:
It should be fixed in `master`. `release/v8.3` still calculates with the border even if it's not shown on a side.

As workaround with v8.3 you can set `pad_left = -30` (equal to negative board width).

lanistor, 2023-02-16:
> It should be fixed in `master`. `release/v8.3` still calculates with the border even if it's not shown on a side.
> 
> As workaround with v8.3 you can set `pad_left = -30` (equal to negative board width).

Thanks a lot, hope this will fix on v8.x, upgrade to v9 have another some work to do.

kisvegabor, 2023-02-16:
As it'd alter the behavior of the existing UIs we can't add these changes to v8.3.x. :slightly_frowning_face:

closed 2023-03-03 by lanistor