lvgl/lvgl #3994: Atomic not supported on Raspberry PI (GCC 10.2.1 | aarch64)

Chuvi-w, 2023-02-16:
### Introduce the problem
```
[build] /usr/lib/gcc/aarch64-linux-gnu/10/include/stdatomic.h:40:9: error: ‘_Atomic’ does not name a type`
[build]    40 | typedef _Atomic _Bool atomic_bool;
```

### System Info:
OS = Debian GNU/Linux 11 (bullseye)
gcc (Debian 10.2.1-6) 10.2.1 20210110

### Examples and cases

### Suggested solution
Will be added as pull-request

Sh3Rm4n, 2023-02-16:
Do you include lvgl in a C++ project?

Sh3Rm4n, 2023-02-17:
So what I assume, is that you consume the lvgl header with a C++ compiler. But [GCC 10.2.1](https://godbolt.org/#g:!((g:!((g:!((h:codeEditor,i:(filename:'1',fontScale:14,fontUsePx:'0',j:1,lang:c%2B%2B,selection:(endColumn:23,endLineNumber:1,positionColumn:23,positionLineNumber:1,selectionStartColumn:23,selectionStartLineNumber:1,startColumn:23,startLineNumber:1),source:'%23include+%3Cstdatomic.h%3E'),l:'5',n:'0',o:'C%2B%2B+source+%231',t:'0')),k:34.176891489942705,l:'4',m:100,n:'0',o:'',s:0,t:'0'),(g:!((h:compiler,i:(compiler:g102,deviceViewOpen:'1',filters:(b:'0',binary:'0',binaryObject:'1',commentOnly:'0',demangle:'0',directives:'0',execute:'0',intel:'1',libraryCode:'0',trim:'1'),flagsViewOpen:'1',fontScale:14,fontUsePx:'0',j:1,lang:c%2B%2B,libs:!(),options:'',selection:(endColumn:1,endLineNumber:1,positionColumn:1,positionLineNumber:1,selectionStartColumn:1,selectionStartLineNumber:1,startColumn:1,startLineNumber:1),source:1),l:'5',n:'0',o:'+x86-64+gcc+10.2+(Editor+%231)',t:'0')),header:(),k:35.922218641626074,l:'4',m:100,n:'0',o:'',s:0,t:'0'),(g:!((h:output,i:(compilerName:'x86-64+gcc+12.2',editorid:1,fontScale:14,fontUsePx:'0',j:1,wrap:'1'),l:'5',n:'0',o:'Output+of+x86-64+gcc+10.2+(Compiler+%231)',t:'0')),k:29.900889868431236,l:'4',m:100,n:'0',o:'',s:0,t:'0')),l:'2',n:'0',o:'',t:'0')),version:4) does not yet support including the C `stdatomic.h` header, this comes only with [version 12](https://godbolt.org/#g:!((g:!((g:!((h:codeEditor,i:(filename:'1',fontScale:14,fontUsePx:'0',j:1,lang:c%2B%2B,selection:(endColumn:23,endLineNumber:1,positionColumn:23,positionLineNumber:1,selectionStartColumn:23,selectionStartLineNumber:1,startColumn:23,startLineNumber:1),source:'%23include+%3Cstdatomic.h%3E'),l:'5',n:'0',o:'C%2B%2B+source+%231',t:'0')),k:34.176891489942705,l:'4',m:100,n:'0',o:'',s:0,t:'0'),(g:!((h:compiler,i:(compiler:g122,deviceViewOpen:'1',filters:(b:'0',binary:'0',binaryObject:'1',commentOnly:'0',demangle:'0',directives:'0',execute:'0',intel:'1',libraryCode:'0',trim:'1'),flagsViewOpen:'1',fontScale:14,fontUsePx:'0',j:1,lang:c%2B%2B,libs:!(),options:'',selection:(endColumn:1,endLineNumber:1,positionColumn:1,positionLineNumber:1,selectionStartColumn:1,selectionStartLineNumber:1,startColumn:1,startLineNumber:1),source:1),l:'5',n:'0',o:'+x86-64+gcc+12.2+(Editor+%231)',t:'0')),header:(),k:35.922218641626074,l:'4',m:100,n:'0',o:'',s:0,t:'0'),(g:!((h:output,i:(compilerName:'x86-64+gcc+12.2',editorid:1,fontScale:14,fontUsePx:'0',j:1,wrap:'1'),l:'5',n:'0',o:'Output+of+x86-64+gcc+12.2+(Compiler+%231)',t:'0')),k:29.900889868431236,l:'4',m:100,n:'0',o:'',s:0,t:'0')),l:'2',n:'0',o:'',t:'0')),version:4)

But technically, `stdatomic.h` compatibility is only supported since [`C++23`](https://en.cppreference.com/w/cpp/23). So without a `C11` and a `C++23` compiler atomic types can not be shared between C and C++ projects via the header. Also the current approach of determining atomics support via the define is different depending on the compiling C11 compiler and the consuming `C++` compiler, which still could lead to compiler or linker errors in the end. The current approch is not feasible. I probably have to revert https://github.com/lvgl/lvgl/pull/3965. 

Either this approach should only be allowed with a combination of a `C11` and `C++23` compiler or every use of atomic should be implementation detail hidden in the translation units not exposed via the header files. But this would mean, that current types / structs need to be restructured / changed.

Chuvi-w, 2023-02-17:
Yes, I've included lvgl in a C++ project

PR #3997 (2023-02-17): Revert "fix(flush,tick): use atomics instead of volatile for synchronization (#3965)"
### Description of the feature or fix

This reverts commit fc659cdecc0378dd19dd3f8cf03d4ba5156e589c.

I feel the current approach results in more problems than improvements. The reasoning is explained in  https://github.com/lvgl/lvgl/issues/3994#issuecomment-1434162987

Fixes #3994 
As an alternative to #3995 

<details><summary>checkpoints</summary>

### Checkpoints

- [x] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [x] Update th [... cut]
closed 2023-02-17 by kisvegabor