lvgl/lvgl #3996: lv_slider not rendering properly

kdschlosser, 2023-02-16:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
There is pixels that do not get filled in properly when lv_slider is rendered. 

### To Reproduce
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/aaa1b3880e9ca63ac3f577c5de610bbfd32741ba/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/aaa1b3880e9ca63ac3f577c5de610bbfd32741ba/examples/event/lv_example_event_4.py&script_direct=5009ff256b48024e72530b295430f421dbd029e2

### Expected behavior
for it to be rendered properly

### Screenshots or video
follow the red arrow!!!.

[image]

kdschlosser, 2023-02-16:
The same thing also shows up at the top if you set the background of the knob to have 0 opacity

kisvegabor, 2023-02-17:
I can take a deeper look a little bit later, but it seems like some artifacts from LVGL's masking. I could highly mitigate the issue by adding:
```
cont.set_style_radius(0, lv.PART.INDICATOR)
```

kdschlosser, 2023-02-17:
Nope that doesn't work. It's still there.

[image]

kdschlosser, 2023-02-17:
it's not as noticeable but still visible if using black and white as the indicator and background colors.

kdschlosser, 2023-02-20:
fantastic. something really simple. as much as bugs are not wanted getting simple ones like this does make it easier.

commit a95619f5 (2023-02-20): fix(bar): remove masking artifacts

fixes #3996