lvgl/lvgl #4002: large grid uint8_t overflow due to mismatched parameter type in function lv_obj_set_grid_cell

eventlOwOp, 2023-02-19:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
in function `lv_obj_set_style_grid_cell_column_pos` we have the variable `col_pos, col_span, row_pos, row_span` declared in uint8_t.

but we later call `lv_obj_set_style_grid_cell_column_pos` which requires variables to be lv_coord_t (int32_t or int16_t) .

so when i created a lot of columns in a grid, it overflowed.

the overflow also happens in `static inline uint8_t get_col_pos` in file `lv_grid.c`

### To Reproduce
just create a lot of grid columns, more than 256.

### Expected behavior
i changed the uint8_t to lv_coord_t and it was fixed.

kisvegabor, 2023-02-20:
There really shouldn't be such limitation. Could you send a pull request to fix it?

kisvegabor, 2023-03-07:
The PR is merged so I close it.

closed 2023-03-07 by kisvegabor