lvgl/lvgl #4009: lv_roller does not emit events when it is scrolled

kdschlosser, 2023-02-20:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
lv_roller does not emit en event when it is scrolled. the lv_label instance that is a child of lv_roller does not emit events for scrolling either

### To Reproduce
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/df789ed3c7973b44be438436f658c5d55ae2f270/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/df789ed3c7973b44be438436f658c5d55ae2f270/examples/widgets/roller/lv_example_roller_1.py&script_direct=2990bf8796ec9527e7b0d4c8e55f232de8e61789

### Expected behavior
for events to be emitted when scrolling of the widget occurs

kisvegabor, 2023-02-24:
@embeddedt @amirgon We have moved our server to a new web hoisting service provider and now the saved code is not loaded. In the Inspector panel's console I see that there is an issue with a certificate. Do you know how can we fix it?

@kdschlosser 
In the meantime, could you copy paste the code here?

embeddedt, 2023-02-24:
@kisvegabor Judging by the error message I suspect there is no CORS header on `forum.lvgl.io:8443`, which means `sim.lvgl.io` is not allowed to make requests since it is considered a different domain. I'm assuming this got lost in the migration.

kisvegabor, 2023-02-24:
Is this a DNS setting of some kind?

kdschlosser, 2023-02-24:
```python
##### startup script #####

#!/opt/bin/lv_micropython -i

import lvgl as lv
import display_driver

##### main script #####

import fs_driver
import sys

class Lv_Roller_3():

    def __init__(self):
        self.mask_top_id = -1
        self.mask_bottom_id = -1

        #
        # Add a fade mask to roller.
        #
        style = lv.style_t()
        style.init()
        style.set_bg_color(lv.color_black())
        style.set_text_color(lv.color_white())

        lv.scr_act().add_style(style, 0)

        roller1 = lv.roller(lv.scr_act())
        roller1.add_style(style, 0)
        roller1.set_style_border_width(0, 0)
        roller1.set_style_pad_all(0, 0)
        roller1.set_style_bg_opa(lv.OPA.TRANSP, lv.PART.SELECTED)

        #if LV_FONT_MONTSERRAT_22
        #    lv_obj_set_style_text_font(roller1, &lv_font_montserrat_22, LV_PART_SELECTED);
        #endif
        try:
            roller1.set_style_text_font(lv.font_montserrat_22,lv.PART.SELECTED)
        except:
            fs_drv = lv.fs_drv_t()
            fs_driver.fs_register(fs_drv, 'S')
            print("montserrat-22 not enabled in lv_conf.h, dynamically loading the font")
            font_montserrat_22 = lv.font_load("S:" + "../../assets/font/montserrat-22.fnt")
            roller1.set_style_text_font(font_montserrat_22,lv.PART.SELECTED)

        roller1.set_options("\n".join([
            "January",
            "February",
            "March",
            "April",
            "May",
            "June",
            "July",
            "August",
            "September",
            "October",
            "November",
            "December"]),lv.roller.MODE.NORMAL)

        roller1.center()
        roller1.set_visible_row_count(3)
        roller1.add_event_cb(self.mask_event_cb, lv.EVENT.ALL, None)

    def mask_event_cb(self,e):

        code = e.get_code()
        obj = e.get_target()

        if code == lv.EVENT.COVER_CHECK:
            e.set_cover_res(lv.COVER_RES.MASKED)

        elif code == lv.EVENT.SCROLL:
            print('SCROLLING')

        elif code == lv.EVENT.DRAW_MAIN_BEGIN:
            # add mask
            font = obj.get_style_text_font(lv.PART.MAIN)
            line_space = obj.get_style_text_line_space(lv.PART.MAIN)
            font_h = font.get_line_height()

            roller_coords = lv.area_t()
            obj.get_coords(roller_coords)

            rect_area = lv.area_t()
            rect_area.x1 = roller_coords.x1
            rect_area.x2 = roller_coords.x2
            rect_area.y1 = roller_coords.y1
            rect_area.y2 = roller_coords.y1 + (obj.get_height() - font_h - line_space) // 2

            fade_mask_top = lv.draw_mask_fade_param_t()
            fade_mask_top.init(rect_area, lv.OPA.TRANSP, rect_area.y1, lv.OPA.COVER, rect_area.y2)
            self.mask_top_id = lv.draw_mask_add(fade_mask_top,None)

            rect_area.y1 = rect_area.y2 + font_h + line_space - 1
            rect_area.y2 = roller_coords.y2

            fade_mask_bottom = lv.draw_mask_fade_param_t()
            fade_mask_bottom.init(rect_area, lv.OPA.COVER, rect_area.y1, lv.OPA.TRANSP, rect_area.y2)
            self.mask_bottom_id = lv.draw_mask_add(fade_mask_bottom, None)

        elif code == lv.EVENT.DRAW_POST_END:
            fade_mask_top = lv.draw_mask_remove_id(self.mask_top_id)
            fade_mask_bottom = lv.draw_mask_remove_id(self.mask_bottom_id)
            # Remove the masks
            lv.draw_mask_remove_id(self.mask_top_id)
            lv.draw_mask_remove_id(self.mask_bottom_id)
            self.mask_top_id = -1
            self.mask_bottom_id = -1

roller3 = Lv_Roller_3()

```

kdschlosser, 2023-02-24:
Just to let you know that most of the examples are not going to function properly. I had to change `add_event` to `add_event_cb` and I had to change `get_target_obj` to `get_target`

embeddedt, 2023-02-24:
> Is this a DNS setting of some kind?

No, it should be entirely controlled by the HTTP server as far as I know. I believe it was @amirgon who configured this so maybe he has a better idea of what broke in this setup.

amirgon, 2023-02-24:
> @embeddedt @amirgon We have moved our server to a new web hoisting service provider and now the saved code is not loaded. In the Inspector panel's console I see that there is an issue with a certificate. Do you know how can we fix it?

@kisvegabor Looks like the discourse SSL key was replaced on January 3rd.  
Snippetbin uses the same SSL key, so when you replace the key you need to kill the `snippetbin` container and run it again with the new key.
I've prepared a bash script `run.sh` to start snippetbin on the VPS. So all you need (for the next time you replace SSL keys) is:

```bash
cd snippetbin/
sudo docker rm -f snippetbin
./run.sh
```

I've just ran that on the VPS and I think it's working fine now.

amirgon, 2023-02-24:
> Just to let you know that most of the examples are not going to function properly. I had to change `add_event` to `add_event_cb` and I had to change `get_target_obj` to `get_target`

I think this is related to https://github.com/lvgl/lvgl/issues/4011 and it's not really a bug, it's a mismatch between LVGL master and lv_micropython master.
 
The issue is that lv_micropython master branch isn't updated (yet) to latest LVGL master and did not adapt the changes in https://github.com/lvgl/lvgl/issues/4011 yet. On the other hand, the Python examples on LVGL master are already fixed to match latest LVGL.

If you pull lv_micropython and update submodules you'll get an older LVGL with the examples before these fixes, so lv_micropython will work locally. But when using lv_micropython online simulator, it takes the latest examples from LVGL master, and this will work only after I'll merge arch/drivers branch to master on lv_micropython.

kisvegabor, 2023-02-27:
@kdschlosser 
I see the problem now. Really there is no scroll event, as the roller doesn't use LVGL's scroll mechanism but just adjusts the `y` position. Just to better understand the use-case: why do you need the scroll event?

@amirgon 
> Looks like the discourse SSL key was replaced on January 3rd.
Snippetbin uses the same SSL key, so when you replace the key you need to kill the snippetbin container and run it again with the new key.

Thank you, I confirm that the example posted by @kdschlosser is loading correctly now. Although I haven't changed the SSL key on the server, especially not on January 3 (I still was on Christmas holiday)

kdschlosser, 2023-02-27:
> Just to better understand the use-case: why do you need the scroll event?

To extend the roller so it can contain a "virtual" list instead of it being a static list.

so if the roller has 300 or 400 items and those items are created dynamically there is no real need to populate the entire list as items should be able to be added to the list when they are going to be see and removed when they are not seen. This would make the memory footprint smaller.

Without the scroll event there is no way to be able to add or remove the items from the list.

amirgon, 2023-02-27:
> Although I haven't changed the SSL key on the server, especially not on January 3 (I still was on Christmas holiday)

This worries me a little. The certificates definitely changed. This is from Chrome Certificate viewer:

[image]

The certificate looks legit, maybe you have some automatic job that re-generates certificates?  
If you do - this job should also kill and restart the snippetbin container...

kisvegabor, 2023-02-28:
@kdschlosser 

> so if the roller has 300 or 400 items and those items are created dynamically there is no real need to populate the entire list as items should be able to be added to the list when they are going to be see and removed when they are not seen. This would make the memory footprint smaller.

Are you sure it's worth the effort? All roller options are stored in a long string. So e.g. 400 options, each with 10 characters is only 4kB. 

@amirgon 
> This worries me a little. The certificates definitely changed. This is from Chrome Certificate viewer:

It's interesting that now it will expire in 4 moths, however earlier it wasn't expired in a year or so. 

So what we need to do is running
```
cd snippetbin/
sudo docker rm -f snippetbin
./run.sh
```
e.g. in each month?

amirgon, 2023-02-28:
> So what we need to do is running
> 
> ```
> cd snippetbin/
> sudo docker rm -f snippetbin
> ./run.sh
> ```
> 
> e.g. in each month?

But how would you know when exactly to run that? If you don't know exactly when a certificate is replaced, it can still leave a few days or weeks of broken snippetbin.
The best way is to understand what updates the certificate, and add to the same job a snippetbin restart.  
Or sync them together if you can control the date that a certificate is updated, such that snippetbin is restarted right after a certificate change.

kdschlosser, 2023-02-28:
if you then throw in the additional characters if you want to color the text for say a gradient color it then takes up twice that amount of memory.

There was someone of the forum that has requested this feature because of memory limitations which is the reason why I am investigating being able to do it. That is when I noticed there was no event emitted for the scrolling.

kdschlosser, 2023-02-28:
The other benefit is not having to use the large coordinates That could also consume quite a bit of memory because that is a global modifier and all x, y coordinates become 32 bit instead of 16 bit. That would also be a pretty big impact on memory if a users UI has a lot going on.

kisvegabor, 2023-03-02:
@amirgon 
We will look into it!

@kdschlosser 
> There was someone of the forum that has requested this feature because of memory limitations which is the reason why I am investigating being able to do it. That is when I noticed there was no event emitted for the scrolling.

There should be an event, but I think not for this reason but to detect which option is in the middle while the roller is scrolled. It can be used to see the intermediate values too. I think using 4-6 kB for 400 options is reasonable. 

Anyway, I'm absolutely open to discuss how to improve the roller, but than let's do it one discussion to see the whole picture. As a topic starter I see these improvement options:
- Figure out an animation/effect when the selected font is larger or has different font.
- Cylinder effect, that is a transformation on the top and bottom part
- Better handling of infinite roller
- Firing scroll event
- Using more generic option elements where arbitrary content can be added.

What do you think?

kdschlosser, 2023-03-02:
I was looking at extending the widget for the user and not something to be a permanent addition.. There are some things that could use to be updated in it and if there is time to add onto it amongst all of the other things then by all means have at it. The event thing I figured is something that would be fairly easy to make happen.

kisvegabor, 2023-03-04:
> The event thing I figured is something that would be fairly easy to make happen.

True, it can be rather considered as a fix. I'll looking into it next week.

kdschlosser, 2023-03-04:
I'm not in a rush for it. Get the high priority things like the examples and the simulator fixed first. I am sure I can poke about with it and get the event added to the widget and do a PR for it.

kdschlosser, 2023-03-04:
I do have a question regarding performance. The more items that are added to the list the slower it responds. This is probably due to having to position such a large object.

Would it be better from a performance standpoint to pass a 2 dimensional array of null terminated strings?. The roller widget allows the user to set the number of lines that are displayed. That same mechanism can be used to create lv_label objects for each of those lines and as the roller gets scrolled the lv_label object that goes out of view gets recycled to the other side with the next string set as the text.

I think that performance wise this would be better if passing a large number of items to the roller. It also removes the need to use large coordinates. IDK if it is something that is able to be done because of the magnifying glass thing that is over the middle of the roller. I don't know if that would work if using separate lv_label objects.

It would also make it far easier if wanting to add and remove items from the widget while it is in use.

kisvegabor, 2023-03-07:
Actually there is a mechanism in LVGL to speed up the drawing of long labels. The draw function can receive and maintain a `hint` variable in which it save some data to find the lines faster. We could use it in case of the roller too.

Have you really seen the slowdown due to the many options or is it rather a theoretical issue?

kdschlosser, 2023-03-07:
I personally have not seen it but it has been reported by a user on the forum. I have not verified it. I have only used the roller in the simulator which is not a good way to know if there is a performance related issue because of it. I will test it out and show a video of it if there are any performance related issues. I only have ESP32's that I am working with so those are at the higher end of the processor spectrum for the MCUs that LVGL typically runs on.

kisvegabor, 2023-03-08:
Okay, it seems like really there are some todos around the roller :slightly_smiling_face: 
Now I'm totally into the [parallel rendering topic](https://github.com/lvgl/lvgl/issues/4016#issuecomment-1460116443). It will mean a large change in the internals of LVGL so I'd like to do widget level changes only when it's stable.

After that it would be great to discuss how can we improve the roller.

kdschlosser, 2023-03-08:
I hear ya. It is important to get the core up to snuff and then mess with the widgets. There is a mechanism in Github to be able to set things that need to be done for target releases of a project. It is called milestones. It is a good way to add in what should be looked at being done for target version releases.

Easy way to organize what needs to be done and for what version release.

kisvegabor, 2023-03-10:
Now I'm using the [next major](https://github.com/lvgl/lvgl/labels/next%20major) label. If you click a label like this you can see which issues are planned.

kdschlosser, 2023-03-10:
easier to keep track of things using those labels/milestones

kisvegabor, 2023-03-13:
Ah, the progress indicator is really great! I'll look into it.

kisvegabor, 2024-06-20:
It not on the roadmap, so I'm closing this issue to keep the list of open issues shorter.

closed 2024-06-20 by kisvegabor