lvgl/lvgl #4011: Changes in master - new driver API

kisvegabor, 2023-02-22:
A large update was merged to `master` which involves API changes as well. You can use `master` to try out the latest changes but for production we recommend using the `release/v8.3` branch. Below you can see the main changes

### Built in drivers
- The goal of these changes is to have built-in drivers right in the LVGL repo
- Now SDL (supporting window resize and multiple windows), Linux Frame buffer, and a TFT_eSPI wrapper is available
- Later display controllers (e.g. ILI9341) will be added too
- A "vendor layer" will be added as well to make it simpler to use the display or touch controllers e.g. with STM's SPI, or ESP's I2C, etc. 

### General API changes
- `lv_obj_add_event_cb` is renamed to `lv_obj_add_event`

### New color format management
- `LV_IMG_CF_...` was removed and was replaced by `LV_COLOR_FORMAT_...`
- The image converter is not updated yet so you need to manually update the color format of the generated images
- Alpha 1bit, 2bit and 4bit color formats were removed because they can be replaced by Indexed 1 bit, 2 bit and 4 bit (alpha 8 bit is available) 
- `LV_COLOR_DEPTH 24` is supported for RGB888 rendering

### Display API
- `lv_disp_drv_t` and `lv_disp_draw_buf_t` was removed
- To create a display and set it up:
```c
lv_disp_t * disp = lv_disp_create(hor_res, ver_res)
lv_disp_set_flush_cb(disp, flush_cb);
lv_disp_set_draw_buffers(disp, buf1, buf2, buf_size_in_bytes, mode);
```
  - Note that now the buf size is in bytes and not pixels
  - modes can be:
     - `LV_DISP_RENDER_MODE_PARTIAL` This way the buffers can be smaller then the display to save RAM. At least 1/10 sceen size buffer(s) are recommended.
    - `LV_DISP_RENDER_MODE_DIRECT` The buffer(s) has to be screen sized and LVGL will render into the correct location of the buffer.  This way the buffer always contain the whole image. With 2 buffers the buffers' content are kept in sync automatically. (Old v7 behaviur)
    - `LV_DISP_RENDER_MODE_FULL` Just always redraw the whole screen
- Similarly to the widgets, now you can attach events to the display too, using `lv_disp_add_event()`
- `monitor_cb` is removed and a `LV_EVENT_RENDER_READY` event is fired instead
- Instead of having display background color and image, `lv_layer_bottom()` is added where any color can be set or any widget can be created. 
- For more details check out the docs [here](https://docs.lvgl.io/master/porting/display.html) and [here](https://docs.lvgl.io/master/overview/display.html).

#### Color format of the display
- The target color format can be adjusted in the display by calling `lv_disp_set_color_format(disp, LV_COLOR_FORMAT_...)`
- `LV_COLOR_16_SWAP` is removed and `lv_disp_set_color_format(disp, LV_COLOR_FORMAT_NATIVE_REVERSED)` can be used instead
- `disp_drv.scr_transp`  was removed and `lv_disp_set_color_format(disp, LV_COLOR_FORMAT_NATIVE_ALPHA)` can be used instead
- `set_px_cb` is removed, you can add a custom color format converter with `disp->draw_ctx->buffer_convert = my_converter;`. As a reference see the [built-in converter](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/lv_draw_sw.c#L108)

### Indev API
- Similarly to the display `lv_indev_drv_t` was removed and an input device can be created like this:
```c
lv_indev_t * indev = lv_indev_create();
lv_indev_set_type(indev, LV_INDEV_TYPE_...);
lv_indev_set_read_cb(indev, read_cb);
```

---

I'll update this post if there will new changes or if something important is missing.

ImkereiWetzel, 2023-02-22:
I suppose many (simple) projects really need a simple integrated rotary encoder example. Would you mind taking this also into account?

ImkereiWetzel, 2023-02-23:
You need 3 GPIOs: ENC_A, ENC_B and most of the encoders in such projects have a integrated tactile switch - so ENC_BTN.
Many modern µC have "enough" interrupts avail for GPIO. So, in my projects I use interrupts on any of these pins , but you are right, one of the detector pins and maybe the button input should be enough.
I use ESP32Encoder library most of the time, I found it to be quite stable.

kisvegabor, 2023-02-23:
Yeah, true, it's 3 buttons. We can add it for sure. 

Which display controller and driver library (if any) do you use for the display?

ImkereiWetzel, 2023-02-23:
I use TFT_eSPI, with an ILI9341 most of the time - I build some little helper machines for beekeeping with them in my free time - on the bigger displays I use touch (if onboard the display), on the little displays I use the rotary encoder, and maybe some additional buttons and switches

X-Ryl669, 2023-02-23:
The documentation linked above for the bottom layer is confusing and probably wrong. The part that read:
```
In summary, to enable transparent screens and displays for OSD menu-like UIs:

    Set the screen's bg_opa to transparent: lv_obj_set_style_local_bg_opa(lv_scr_act(), LV_OPA_TRANSP, 0)

    Set the bottom layer's bg_opa to transparent: lv_obj_set_style_local_bg_opa(lv_scr_act(), LV_OPA_TRANSP, 0)

    Set the screen's bg_opa to 0: lv_obj_set_style_local_bg_opa(lv_layer_bottom(), LV_OPA_TRANSP, 0)

    Set a color format with alpha channel. E.g. lv_disp_set_color_format(disp, LV_COLOR_FORMAT_NATIVE_ALPHA)

```
has copy and paste error or I'm missing the point.
Also, the "mirror display" part should be titled "duplicate display", because it's not about mirroring but duplicating, IIUC.

GorgonMeducer, 2023-02-23:
@kisvegabor What happens to the previous RGB565A8? is it now the `LV_COLOR_FORMAT_ARGB8565`?

kisvegabor, 2023-02-24:
@X-Ryl669 
Thanks, I've fixed it here https://github.com/lvgl/lvgl/commit/31a0a324b90b6a1cb62fab10bda9ffd1db3c0703

@GorgonMeducer 
I've fixed it too here: b66512183

GorgonMeducer, 2023-02-24:
@kisvegabor Thank you. I can move forward to update the cmsis-pack now.

GorgonMeducer, 2023-02-26:
@kisvegabor cmsis-pack scripts are updated #4020

GorgonMeducer, 2023-03-01:
Found some render issue #4029

amirgon, 2023-03-05:
> ```c
> lv_disp_set_draw_buffers(disp, buf1, buf2, buf_size_in_bytes, mode);
> ```
> 
> Note that now the buf size is in bytes and not pixels
> 

Contradicts:

https://github.com/lvgl/lvgl/blob/67941f4964da065623132ea00ce83c6abc1ab86d/src/core/lv_disp.h#L242-L243

which says `buf_size_px` and not `buf_size_in_bytes`

amirgon, 2023-03-05:
https://github.com/lvgl/lvgl/blob/67941f4964da065623132ea00ce83c6abc1ab86d/src/core/lv_disp_private.h#L126-L130

Why do we have both `user_data` and `driver_data`?  
`user_data` is required for Micropython. What is `driver_data` used for? Why not use `user_data` instead?

amirgon, 2023-03-05:
Trying to port [ESP32 ili9XXX driver](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/esp32/ili9XXX.py) to the new API, looks like `LV_COLOR_FORMAT_NATIVE_REVERSED` is broken...  
I'm using `LV_COLOR_DEPTH=16`, so maybe also related to https://github.com/lvgl/lvgl/issues/4035 ?

[image]

kisvegabor, 2023-03-06:
> I found out the hard way that buf_size_px is correct, when initializing byte size caused crashes.

It should be in bytes and I've fixed the comments. I'll make some test later today.

> Why do we have both user_data and driver_data?
user_data is required for Micropython. What is driver_data used for? Why not use user_data instead?

As it's almost certain that the `lv_disp_t` wants to store something about the driver I didn't want to steal the `user_data` from the user.

> Trying to port [ESP32 ili9XXX driver](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/esp32/ili9XXX.py) to the new API, looks like LV_COLOR_FORMAT_NATIVE_REVERSED is broken...
Were all the color modes tested?

I'll try it out today. The SDL driver uses DIRECT mode and it's probably related to PARTIAL mode.

amirgon, 2023-03-06:
> It should be in bytes and I've fixed the comments. I'll make some test later today.

No, look at the [code references above](https://github.com/lvgl/lvgl/issues/4011#issuecomment-1455199216).  
All LVGL built-in driver still pass size in pixels. Some header/code refer to bytes other to pixels. It's a real mess.

kisvegabor, 2023-03-06:
> No, look at the https://github.com/lvgl/lvgl/issues/4011#issuecomment-1455199216.
All LVGL built-in driver still pass size in pixels. Some header/code refer to bytes other to pixels. It's a real mess.

It's because in case of DIRECT mode with NATIVE color format the buffer size is not used because as buffers are assumed to be screen sized. And the drivers you have referenced are using DIRECT mode. 

We need to know the size in byte because now LVGL can handle changing the output color format runtime. So pixel count as size would be reliable.

amirgon, 2023-03-06:
Well, the only certain thing I can say is that it crashes when using byte size and doesn't when using pixel size, which is opposite to your comment that it should be in bytes.  
Maybe it's also related to color mode issue, I don't know.

kisvegabor, 2023-03-06:
I've fixed the color reverse issue and it can be related to buffer size. LVGL assumed that reversed colors have 3 byte size.

amirgon, 2023-03-06:
> LVGL assumed that reversed colors have 3 byte size.

I'm defining `LV_COLOR_DEPTH=16`, so wouldn't `LV_COLOR_FORMAT_NATIVE_REVERSED` have 2 bytes, same as native?

kisvegabor, 2023-03-06:
>  I'm defining LV_COLOR_DEPTH=16, so wouldn't LV_COLOR_FORMAT_NATIVE_REVERSED have 2 bytes, same as native?

It should. It was the bug that it assumed 3 bytes :slightly_smiling_face:

amirgon, 2023-03-06:
> I've fixed the color reverse issue and it can be related to buffer size. LVGL assumed that reversed colors have 3 byte size.

Ok, it is working now as expected! Thanks :+1:

amirgon, 2023-03-07:
@kisvegabor some drivers use the callbacks `gpu_blend_cb` and `gpu_fill_cb`, but with the new API I don't see how to set them. 

See for example: https://github.com/lvgl/lv_binding_micropython/blob/f960502a3260f795f21bfa90b2e7d7199430cf88/examples/example1.py#L91-L92

kisvegabor, 2023-03-08:
It should be a very old example. `gpu_blend_cb` and `gpu_fill_cb` was already removed in v8 too.

amirgon, 2023-03-09:
> * `monitor_cb` is removed and a `LV_EVENT_RENDER_READY` event is fired instead

On v8:

https://github.com/lvgl/lvgl/blob/9a870b34a9e8a4b55c7d58b69ba3f1eccca1305b/src/hal/lv_hal_disp.h#L121-L123

On v9 when I receive `LV_EVENT_RENDER_READY`, how do I obtain the `time` and `px`?

amirgon, 2023-03-10:
I've merged the MP fixes to master on lv_micropython and lv_binding_micropython.  
I tested what I could on the unix and esp32 ports, but I'll need someone's help to test the rp2 and stm32 ports.

kisvegabor, 2023-03-11:
Amazing! :tada: 
I've already tested the STM32 port and it worked as earlier, just needed to use the new API functions the create the display driver.

amirgon, 2023-03-11:
@embeddedt There's an issue with the JS port, the v9 simulator is not updated.

Looks like the [js merge succeeded](https://github.com/lvgl/lv_micropython/actions/runs/4389056469/jobs/7686217630) but maybe it didn't really merge anything because I can see that `lvgl_javascript_v9` is not aligned to `master` and [Build Javascript v9 port fails](https://github.com/lvgl/lv_micropython/actions/runs/4389082454/jobs/7686273536) looking for `sdl_common.c` which is removed on master...  

Any idea why this happens?

embeddedt, 2023-03-11:
@amirgon The branch seems to be aligned correctly from what I can tell. However, like the Unix port, the Makefile manually referenced `sdl_common.c` and so I had to remove it as was done there.

Now it compiles fine (locally) but I noticed the methods like `lv.sdl_mouse_create` are missing in Python-land. I did define `MICROPY_SDL` to be 1 and the C files do seem to be getting compiled, however there is nothing in `lv_mpy.c` referring to these APIs. Is there a step I forgot?

amirgon, 2023-03-11:
> I did define `MICROPY_SDL` to be 1 and the C files do seem to be getting compiled, however there is nothing in `lv_mpy.c` referring to these APIs. Is there a step I forgot?

On the unix port on the master branch it works fine.  
I'm using `lv.sdl_window_create`, `lv.sdl_mouse_create` etc. See: https://github.com/lvgl/lv_binding_micropython/blob/357d9bd8191a43209a46549502871d989fb8a7ee/lib/display_driver_utils.py#L37-L46

The unis port Makefile defines `MICROPY_SDL` when SDL is installed, which enables `LV_USE_SDL`.  
I don't think there was anything else needed in order to enable SDL.

amirgon, 2023-03-11:
@embeddedt  Are you sure it's merged correctly?

```bash
$ git diff --stat upstream/master upstream/lvgl_javascript_v9
 .gitattributes                                     |   20 +
 .github/workflows/merge-to-js.yml                  |    2 +-
 .gitignore                                         |    1 +
 README.md                                          |   33 +-
 ports/javascript/.browserslistrc                   |    5 +
 ports/javascript/.gitignore                        |    1 +
 ports/javascript/Makefile                          |   72 +-
 ports/javascript/README.md                         |  118 -
 ports/javascript/build_workers.sh                  |    8 +
 ports/javascript/docs/wasm_file_api.js             |  230 +
 ports/javascript/docs/wasm_file_api.txt            |   37 +
 ports/javascript/dummy_header.h                    |    1 +
 ports/javascript/favicon.ico                       |  Bin 0 -> 1406 bytes
 ports/javascript/fdfile.h                          |   40 +
 ports/javascript/file.c                            |  315 +
 ports/javascript/frontend.html                     |   94 +
 ports/javascript/frontend_src/golden-layout.less   |    2 +
 ports/javascript/frontend_src/main.js              |  565 ++
 ports/javascript/frontend_src/styles.css           |  213 +
 ports/javascript/index.html                        |   11 +
 ports/javascript/library.js                        |    4 +-
 ports/javascript/lv_font_heb_16.c                  | 1138 +++
 ports/javascript/lvgl.html                         |  152 +
 ports/javascript/lvgl_editor.html                  |   15 +
 ports/javascript/main.c                            |   16 +-
 ports/javascript/manifest.py                       |    1 +
 ports/javascript/modules/imp.py                    |   92 +
 ports/javascript/modules/imp_empty_pivot_module.py |    1 +
 ports/javascript/modules/imp_pivot.py              |    5 +
 ports/javascript/modules/site.py                   |   10 +
 ports/javascript/modules/types.py                  |   24 +
 ports/javascript/mpconfigport.h                    |   67 +-
 ports/javascript/mphalport.c                       |   31 +-
 ports/javascript/package-lock.json                 | 8464 ++++++++++++++++++++
 ports/javascript/package.json                      |   49 +
 ports/javascript/push.sh                           |    4 +
 ports/javascript/scripts/build_freetype.sh         |   15 +
 ports/javascript/scripts/build_rlottie.sh          |   18 +
 ports/javascript/scripts/rlottie.patch             |   30 +
 ports/javascript/test.py                           |    4 +
 ports/javascript/testcode.py                       |  139 +
 ports/javascript/wasm_file_api.c                   |   38 +
 ports/javascript/webpack/webpack.common.js         |   55 +
 ports/javascript/webpack/webpack.dev.js            |   15 +
 ports/javascript/webpack/webpack.prod.js           |    9 +
 ports/javascript/wrapper.js                        |   18 +-
 py/mkrules.mk                                      |    4 +-
 47 files changed, 12041 insertions(+), 145 deletions(-)
```

embeddedt, 2023-03-11:
Yes, remember there are lots of additional changes on the JavaScript branch for the purposes of making it work in our simulator environment.

amirgon, 2023-03-11:
@embeddedt You need `MICROPY_SDL` added to `CFLAGS_MOD` so it would affect both LVGL header preprocessing and LVGL sources build.

See: https://github.com/lvgl/lv_micropython/blob/5bc85a4f6fd3f24480f3e345519447fc9872aac7/py/py.mk#L46

Maybe `MICROPY_SDL` is defined only for the source build, and is missing when preprocessing LVGL headers?

embeddedt, 2023-03-11:
That's probably why. I think I added it somewhere else. Thanks; I'll try it.

amirgon, 2023-03-11:
> That's probably why. I think I added it somewhere else. Thanks; I'll try it.

You can do something like this: https://github.com/lvgl/lv_micropython/blob/5bc85a4f6fd3f24480f3e345519447fc9872aac7/ports/unix/Makefile#L226-L231

embeddedt, 2023-03-11:
The good news is that it now runs the startup script without issues. The bad news is that the simulated display is now completely blank. I'm guessing something about the event loop got changed. Still investigating.

amirgon, 2023-03-12:
> The bad news is that the simulated display is now completely blank

I believe I know why.   
The previous driver created the event loop by default, the new LVGL built in driver doesn't.
Instead, it relies on the event loop from lv_utils which is much more flexible, but I think we had some issues with it on js (something with timer implementation?)

embeddedt, 2023-03-12:
There is no `uasyncio` support in the JS port (and the stub Timer class I implemented is nonfunctional, I think it was only added to fix an error). Is there a way to opt back in to the old behavior, or do I need to implement my own event loop?

amirgon, 2023-03-13:
> Is there a way to opt back in to the old behavior, or do I need to implement my own event loop?

It's possible to opt back, but I'm not sure it's the best solution.  
There are advantages to using the LVGL built in driver: it ensures consistent behavior between the MP bindings and LVGL, it saves the need to maintain two very similar drivers when fixing bugs/adding features etc.  

How hard is it to implement an event loop in JS?   

One way is to implement the Timer class instead of the stub, but if it's too hard we can simply call `lv.task_handler` and `lv.tick_inc` periodically .

embeddedt, 2023-03-23:
Implementing `Timer` looks like the cleaner solution (as then the behavior would be closer to the other platforms) but I can't reuse the Unix port's version since FFI doesn't exist on the JS port. Hence, I would need to completely rewrite the module, probably in C, which is something I don't have the time for now. For now I'd like to just implement a reasonable and quick solution by hardcoding the calls to `lv_task_handler`/`lv_tick_inc`, since this issue is blocking any use of the simulator with  v9.

It's worth noting that the JS port *does* support `mp_sched_schedule`, as that's the original mechanism used to invoke `lv_task_handler` up until now. So one option could be adding a third mode to `lv_utils` that runs the event loop using those APIs. I am not sure if there is a good way to do that.

Suggestions?

C47D, 2023-03-24:
@kisvegabor  is this change meant to land in v9 or another minor version of v8?

amirgon, 2023-03-24:
> So one option could be adding a third mode to `lv_utils` that runs the event loop using those APIs. I am not sure if there is a good way to do that.

That sounds good, but for an intermediate solution we don't even need that. We can call mp_sched_schedule from C periodically, to invoke lv_task_handler. No need to change any Python code.  How do you call something periodically from js? 
I'm currently traveling for the next two weeks so I won't be able to help for the short term solution.

X-Ryl669, 2023-03-24:
> How do you call something periodically from js?

`setInterval(yourFunction, periodInMilliseconds)`

But if you want this to actually refresh the screen, you should use `requestAnimationFrame(yourFunction)` instead (that's asynchronous, so you don't really know when it happens, just that it'll be called as soon as possible).

embeddedt, 2023-03-24:
There is a helper function we can use to invoke a C function periodically ([reference](https://github.com/lvgl/lv_binding_micropython/blob/6af055286249cb0fb7c36651c3fde3a7a8567434/driver/SDL/modSDL.c#L95) from the old driver). My main concern is making sure `lv_task_handler` doesn't start running before `lv_init`.

X-Ryl669, 2023-03-24:
I meant, for your javascript's Timer stub, it should be like this, I think:
```python
import jswindow
class Timer:

    PERIODIC = 0
    ONE_SHOT = 1

    def __init__(self, id):
        pass

    def init(self, mode=PERIODIC, period=-1, callback=None):
        self.mode = mode
        if mode == PERIODIC:
           self.id = jswindow.setInterval(callback, period)
        else:
           self.id = jswindow.setTimeout(callback, period)

    def deinit(self):
        if self.mode == PERIODIC:
           jswindow.clearInterval(self.id)
        else
           jswindow.clearTimeout(self.id)
        pass
```
As I understand it, you'd need to create a C module (called `jswindow in my example above`) that's hooked into emscripten for javascript linking:
```cpp

#ifdef __emscripten__
int setInterval(func_t cb, int period) {
  return EM_ASM_INT({ return setInterval($0, $1); }, cb, period);  // The part in brace is actual javascript
};
int setTimeout(func_t cb, int period) {
  return EM_ASM_INT({ return setTimeout($0, $1); }, cb, period);
};
etc...
#endif
```

This you don't care about LVGL at all and the synchronization.

embeddedt, 2023-03-24:
Hmm that could work.

MouriNaruto, 2023-03-28:
I am sorry I have noticed it because I was busy in the recent days.

I will port Win32 driver as soon as possible.

Kenji Mouri

kisvegabor, 2023-03-29:
Hi Kenji,

No problem, thank you for working on it! :blush:

espzav, 2023-03-30:
@kisvegabor great news! I am thinking about improving the task which is calling the LVGL timer handler and it seems that `lv_disp_add_event` is a function, which I was looking for. The task now is don't sleep too much, because there isn't a solution to wake up, when `lv_timer_handler` returns an infinite value. 
Please, when do you plan to release version with these updates? 
Thank you

kisvegabor, 2023-03-31:
> Please, when do you plan to release version with these updates?

There is a larger update that we are still working on. It will allow utilizing multiple CPU cores for rendering (I think it wil important on ESP chips). I estimate that we will ready this autumn. So it will still take some time. Until that feel to try out the `master` branch and let us know if you have any ideas or comments.

MouriNaruto, 2023-04-10:
[image]

I have adapted old Win32 support for the new LVGL driver architecture for implementation in https://github.com/lvgl/lv_port_win_visual_studio. But there is lots of refactor work to do before merging to the LVGL mainline.

Here is the reference for refactor work which I want to do for the new implementation: https://github.com/lvgl/lv_port_win_visual_studio/issues/53

Kenji Mouri

kisvegabor, 2023-04-11:
Wow, this FPS is unbelievable! Was it changed compared v8?

MouriNaruto, 2023-04-11:
@kisvegabor 

[image]

Note: The implementation logic of the current win32 support in LVGL v9 is basically as same as LVGL v8's.

Kenji Mouri

kisvegabor, 2023-05-05:
@MouriNaruto I'd like to follow up about the status of the Window driver.  Would it be possible to add at least basic version of it to `master` in the near future?

MouriNaruto, 2023-05-05:
@kisvegabor 

> Would it be possible to add at least a basic version of it to master in the near future?

The basic implementation like the old win32drv is ready. But I'm struggling with how to refactor it to look like the master's SDL implementation. It's a little hard to be as modular as that.

I will try my best to add the relevant implementation to the master with some issue fixes in the next week.

Kenji Mouri

kisvegabor, 2023-05-05:
> But I'm struggling with how to refactor it to look like the master's SDL implementation. It's a little hard to be as modular as that.

Is there something I can help with?

 > But I'm struggling with how to refactor it to look like the master's SDL implementation. It's a little hard to be as modular as that.

Amazing, thank you!

MouriNaruto, 2023-05-11:
@kisvegabor 

> Is there something I can help with?

Yes, I hope LVGL can have a way to get the actually refreshing draw area even not in the LV_DISP_RENDER_MODE_PARTIAL mode in each flush_cb call. It can help me to optimize the Windows GDI rendering performance.

Kenji Mouri

kisvegabor, 2023-05-12:
@MouriNaruto I've just pushed fix, now it works as you have described.

derekstavis, 2023-08-22:
I found several API changes on my port to V9. If you want to take a look at the diff [here it is](https://github.com/lvgl/lv_binding_js/commit/3870e759871b62fda87696217b436e504a8132a2).

My list of changes:

Color:
* `LV_IMG_PX_SIZE_ALPHA_BYTE` -> `LV_COLOR_FORMAT_NATIVE_ALPHA_SIZE`
* `LV_IMG_CF_TRUE_COLOR_ALPHA` -> `LV_COLOR_FORMAT_NATIVE_ALPHA`
* `lv_color32_t[index].ch.r` -> `lv_color32_t[index].r` (same for `b` and `g`)

* Event:
* `lv_obj_add_event_cb` -> `lv_obj_add_event`
* `lv_obj_remove_event_cb` -> `lv_obj_remove_event` (plus some [unwieldy code](https://github.com/lvgl/lv_binding_js/commit/3870e759871b62fda87696217b436e504a8132a2#diff-2ee9bd5e86c8b40d602d3eb521d5cc0e6c5b2377692ae7bc5056cb2999e1283aR32) to find the event index)
* `event->target` -> `event->original_target`

Display:
* `lv_disp->driver->hor_res` -> `lv_disp_get_hor_res(lv_disp)`
* `lv_disp->driver->ver_res` -> `lv_disp_get_ver_res(lv_disp)`

Migration most things was pretty straightforward through educated guesses, other than `lv_obj_remove_event`. The previous API was much cleaner for the developer, and I feel now manual work needs to be done to find the event index and remove it.

kisvegabor, 2023-08-22:
> Migrating most things was pretty straightforward through educated guesses

Amazing! :+1: 

> Migrating most things was pretty straightforward through educated guesses, other than lv_obj_remove_event. The previous API was much cleaner for the developer, and I feel now manual work needs to be done to find the event index and remove it.

Yeah, we really need to add back a helper to remove events.

codewitch-honey-crisis, 2023-10-06:
@kisvegabor , you can use codewitch-honey-crisis/htcw_encoder as a reference implementation.

[5 more replies]

closed 2023-10-20 by kisvegabor