lvgl/lvgl #4012: Naming suggesting

kdschlosser, 2023-02-22:
### Introduce the problem
Recent changes to some of the api naming. 

### Examples and cases
`add_event_cb` was changed to `add_event`

### Suggested solution
This could be a confusing name. Technically speaking you are not adding an event. It makes me think of pushing an event onto a widget for the purposes of notifying some other portion of code that a change took place. An event is not really being added to the widget. hat is being done is a callback is being bound to an event. so maybe using `bind_event` and `unbind_event` might be better suited. add infers adding something to the widget, `add_flag`, `add_style`, etc... thing are being added to the widget. In those cases a flag or a style is being added. `add_event` you are not adding an event to the widget but instead you are adding a callback to an event. it doesn't fit with the naming API all that great. 

Just a thought.

kisvegabor, 2023-02-23:
Hi,

Thanks for pointing it out, it's very true.  I thin the cleanest would be to add back the `_cb` postfix. What do you think?

kdschlosser, 2023-02-24:
as a developer I can appreciate short function names and as small a number of special characters as possible in a function name. the less key strokes the better. That's why I like `bind` and `unbind` really short and no special characters that require the use of additional keys. lol...

It all depends on how concerned you are with API breaking changes. I do not believe that you are that concerned due to the sheer number of API breaking changes that have been made.

Now what I think should be done is a step back needs to be taken and a reevaluation of the naming convention needs to be done, and also take a look at the API in general.

IDK if you know this or not,.. Here is the list of functions that take an lv_obj structure as the first parameter. 

<details>
  <summary>Click me</summary>

add_event_cb
add_flag
add_state
add_style
align
align_to
allocate_spec_attr
area_is_visible
calculate_ext_draw_size
calculate_style_text_align
center
check_type
class_create_obj
class_init_obj
clean
clear_flag
clear_state
del_anim_ready_cb
del_async
del_delayed
delete
dpx
draw_dsc_init
draw_part_check_type
enable_style_refresh
event_base
fade_in
fade_out
get_child
get_child_cnt
get_child_id
get_class
get_click_area
get_content_coords
get_content_height
get_content_width
get_coords
get_disp
get_event_user_data
get_group
get_height
get_index
get_local_style_prop
get_parent
get_screen
get_scroll_bottom
get_scroll_dir
get_scroll_end
get_scroll_left
get_scroll_right
get_scroll_snap_x
get_scroll_snap_y
get_scroll_top
get_scroll_x
get_scroll_y
get_scrollbar_area
get_scrollbar_mode
get_self_height
get_self_width
[... 385 lines cut ...]
    LV_ALIGN_OUT_LEFT_TOP= LV_ALIGN_LEFT | LV_ALIGN_OUT_VERTICAL
    LV_ALIGN_OUT_LEFT_MID= LV_ALIGN_LEFT | LV_ALIGN_OUT_VERTICAL
    LV_ALIGN_OUT_LEFT_BOTTOM= LV_ALIGN_LEFT | LV_ALIGN_OUT_VERTICAL
    LV_ALIGN_OUT_RIGHT_TOP= LV_ALIGN_RIGHT | LV_ALIGN_OUT_VERTICAL
    LV_ALIGN_OUT_RIGHT_MID= LV_ALIGN_RIGHT | LV_ALIGN_OUT_VERTICAL
    LV_ALIGN_OUT_RIGHT_BOTTOM= LV_ALIGN_RIGHT | LV_ALIGN_OUT_VERTICAL

7 bytes vs 22 bytes. 1/3rd the size. my way also takes up less space in the gray matter as well. LOL

It also adds some room for expansion onto it.

    LV_ALIGN_CENTER = 0x00
    LV_ALIGN_TOP = 0x01
    LV_ALIGN_BOTTOM = 0x02
    LV_ALIGN_LEFT = 0x04
    LV_ALIGN_RIGHT = 0x08
    LV_ALIGN_HORIZONTAL = 0x10
    LV_ALIGN_VERTICAL = 0x20
    LV_ALIGN_OUT  = 0x40

This allows for horizontal alignment and vertical alignment so positions are not etched in stone. So if I have a widget that is 100 x 100 in size and I have text that I positioned at 75 pixels down from the top of the parent and I want to keep the y axis position but center the x axis...   LV_ALIGN_CENTER | LV_ALIGN_HORIZONTAL

or say I want to keep the x axis and align the text to the top edge of the parent  LV_ALIGN_TOP  | LV_ALIGN_VERTICAL

¯\_(ツ)_/¯

kdschlosser, 2023-02-24:
I think the best thing to do is to step back and look at it all and come up with a game plan. If you are going to make naming convention changes better to do it a single time and break API instead of doing it over a bunch of different releases. a help doc can be made that links the old to the new so it would be easier for a user to update their code and they would only have to update the code a single time instead of a bunch of times spread out over several releases.

It is hair pulling when this kind of a thing needs to get hammered out later on in development but if a good amount of notice is given to the users letting them know about the changes and the documentation is in place beforehand then when a release comes about a user can have their code updated and ready to go.

kisvegabor, 2023-02-27:
I'm happy that you have opened this issue because unifying and simplifying names is target for v9 (See the [ROADMAP](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md)). 

I agree to do it as one larger step, so let's leave the `event` API as it now. 

As I'm a non native speaker and I has limited experience compared to a larger community, some help with API refactoring would be very welcome. :slightly_smiling_face: 

>  I have never seen a GUI framework that has the alpha level of a color set/get in a separate function.

I've added note in the ROADMAP about using `lv_color32_t` in APIs too.

> With the ability of overloading why are there so many public functions?. 

There is no overloading in C. :slightly_frowning_face: 

> bitwise OR operations can be used to save memory

`enum`s don't  use memory. The compiler just replaces e.g. `LV_ALIGN_CENTER` in the code with `3`.

kdschlosser, 2023-02-27:
I think any recent naming changes should be put back the way they were until a naming convention is hammered out. Then make all naming changes at a single time. 

The naming convention needs to be documented and also checked for consistency when new functions are added. The latter being the hard part because it is easy to forget about.

I have an idea on reducing the number of functions. I am going to type it up so it would be easier to understand. I have to do some looking at the existing code for LVGL to see how things are stored...

kisvegabor, 2023-02-28:
> C code doesn't support overloading?????? Are you sure???

True, but it's a C11 feature. We are still on C99 but maybe C11 is already well-spread enough to migrate to it. I'm not sure.

> I have an idea on reducing the number of functions. I am going to type it up so it would be easier to understand. I have to do some looking at the existing code for LVGL to see how things are stored...

Okay, I'm looking forward to seeing it :slightly_smiling_face:

incity, 2023-06-27:
LV_ASSERT_NULL also makes me confused.
----
The assertNotNull() method means "a passed parameter must not be null": if it is null then the test case fails.
The assertNull() method means "a passed parameter must be null": if it is not null then the test case fails.
-------

kisvegabor, 2023-06-29:
I've also realized it and it really needs to be updated.

incity, 2023-08-12:
[Google C++ Style Guide](https://google.github.io/styleguide/cppguide.html#Naming) says:

>  Use names that describe the purpose or intent of the object. Do not worry about saving horizontal space as it is far more important to make your code immediately understandable by a new reader. Minimize the use of abbreviations that would likely be unknown to someone outside your project (especially acronyms and initialisms). Do not abbreviate by deleting letters within a word. As a rule of thumb, an abbreviation is probably OK if it's listed in Wikipedia. Generally speaking, descriptiveness should be proportional to the name's scope of visibility. For example, n may be a fine name within a 5-line function, but within the scope of a class, it's likely too vague.

There are many abbreviations in lvgl that contain ambiguity, which makes it difficult for me to understand.

For example, 'res' in LVGL seems the abbreviation for result, but for me (and I think it's the same for most people), res should be the abbreviation for resource. As a return value, perhaps ret(urn) is more appropriate.

kisvegabor, 2023-08-15:
I perfectly agree... already. Earlier I thought it's better to abbreviate, but now I'm also convinced that it's better to be clearer.

kisvegabor, 2023-08-30:
Let's warm up this topic and find a consistent standard. There are the names/keywords that are messy now and we should pick one that we will use consistently. See my candidate with **bold**.

- obj: I think it should be `**widget**`. LVGL is too complex now to call something just object.
- **remove**/clear/delete/del: 
   -  `lv_obj_clear_flag` -> `lv_obj_remove_flag` 
   - `lv_obj_del` -> `lv_obj_remove`
- **offset**/ofs, 
- add/create: They are different, so I would keep both. "Add" might mean "assign" as well
   -  `lv_chart_add_series` could be `lv_chart_create_series` or `lv_chart_series_create` 
   - `lv_list_add_btn()` -> `lv_list_btn_create()`
- id/idx/**index**: 
- middle/mid/**center**
- `LV_ALIGN_...`, `LV_FLEX_ALIGN_...`, `LV_GRID_ALIGN_...`, `LV_TEXT_ALIGN_...`: have a single LV_ALIGN_..., accept that not all values are valid in all cases. E.g. `lv_style_set_flex_main_align(&style, LV_ALIGN_BOTTOM_MID)` is invalid.  
- img/**image**
- btn/**button**
- txt/**text**
- **opa**/opacity/alpha
- scr/**screen**
- disp/**display**
- indev: any ideas?
- finished/complete/completed/**ready**
- buf/**buffer**
- res/**resolution**
- res/**result**
- ..._cb
- angle/**rotation**
- zoom/**scale**
- **set/get**: ok
- **dsc**/descriptor
- has: use to check if something is set or exists. E.g. `lv_obj_has_flag()`
- is: use to check 
- enable: I'm not sure. E.g. 
`void lv_textarea_set_text_selection(lv_obj_t * obj, bool en);` -> 
`void lv_textarea_enable_text_selection(lv_obj_t * obj, bool en);`?

Please let me know what you think and if you have other keywords in mind. 

cc @kdschlosser @xiaoxiang781216 @XuNeo @zjanosy

kdschlosser, 2023-08-30:
OK so most phrases in an API will need an opposite phrase.
 
 add : remove
 set : get
focus : unfocus
hide : show
check : uncheck
press : release
mark : clear

I know extra typing sucks but acronyms can suck more especially if you have a function that has 3 acronyms in it. So loose the acronyms if possible, or trim them down to only a single acronym. I know that obj instead of object is really clear and easily remembered. However object is not really the best use in LVGL because an object can really be anything. IDK if it would be wise to change that portion of the naming convention due to it being in place for so long.

This naming convention.

`lv_obj_set_style_bg_color` is a tad bit obscene. Does the "style" really need to be there? It is a given that `lv_set_bg_color` is going to be altering the visual appearance of the widget. remember the acronyms. `lv_obj_set_background_color`. it's a shame that overloaded functions cannot be used... what if a function could be used for dual purposes
`lv_obj_background_color` would be both a getter and a setter. 

getter
`color = lv_obj_background_color(obj, NULL);`

setter
`lv_obj_background_color(obj, color);`

dum question here maybe. C99 has the ability to put functions into a structure yes? well function pointers at least. Does a function need to be defined as a global? or can it be defined as a local?.

What I am getting at is this.

Can something like this be done???

```c
void (*_background_color_t)(lv_color_t * color);

typedef struct {
    _background_color_t set_background_color;
} lv_obj_t;

void _set_bg_color(lv_obj_t * obj, lv_color_t * color) {
    // do color change here
}

lv_obj_t * lv_obj_create(lv_obj_t * parent) {
    lv_obj_t * obj = {};
    
    void s_bg_color(lv_color_t * color) {
        _set_bg_color(obj, color);
    }
    obj->set_background_color = s_bg_color;
    
    return obj;
}
```

and it would get used like this.

```c
lv_obj_t * new_obj = lv_obj_create(NULL);
new_obj->set_background_color(color);
```

I don't know if something like that is able to be done or not. I don't know C all that well and what the limits are. I don't know if wrapping functions like that can be done or if they would even keep obj for the outer function in scope or not or if it could be used in an inner function.

We do know that multiple definitions can exist for a function pointer prototype and not cause issues. so I would think that it shouldn't matter if the function itself is not specifically typed to the prototype so long as it has proper argument types and return type.

If it can be done and making functions to act as getter and setter in one might be a thought 

This would make it by far easier to deal with because the obj structure can be extended and functions added that are specific to a widget so they would only be available  for that specific widget. right now there are functions specific to say an arc that are always declared even if the macro to use arcs has been turned off. all widget specific functions are also have a first parameter type of `lv_obj_t` so extra steps need to be taken to ensure that is what was passed to the function call. This would eliminate the need for those extra steps that have to run each and every single time a function gets called. This only gets done once when the widget is initially constructed so it only has a performance impact a single time. instead of over and over again.

kdschlosser, 2023-08-30:
`lv_obj_del and `lv_obj_remove are 2 completely different things. `lv_obj_remove` is if you want to remove a object from its parent but keep the memory allocation so the object can be used at a later time. `lv_obj_del` should call `lv_obj_remove` and then unallocate the memory that has been used by the object. `lv_obj_del` should be `lv_obj_delete`

kdschlosser, 2023-08-30:
when coming up with an api it is best to type things and see what is easier to type.
example

opacity vs alpha

alpha is the winner here in the easier to type. less keystrokes and the keys are aligned in a manner that makes it easier to type out,

evaluate the phrase against standards. RGBA doesn't stand for Red, Green, Blue and Opacity. It stands for Red, Green Blue and Alpha.. so alpha is a better fit there.

common use is another evaluation that can be done. while alpha and opacity are the same thing in the world of colors which one is more commonly used? I think that alpha wins there as well. 

With using the above there is really not a question of what should be used as it is really clear. It is a matter of changing it. "alpha" is the best when using the methods above to evaluate for API use.

While I believe the API needs to change I think you might want to open a topic on the forum about it. Get the users feelings and opinions before doing it. It's a good idea so you will have to deal with less of them getting their panties all in a bunch because it was changed.

Me personally I would rather see the API change across the board in a single giant alteration. Some people prefer it to be a gradual thing. That is something you would need to take a poll on.

 I do not think that alteration should be included with any other changes if it is going to be done all at once. Do a major release without the changes.. Then do a separate major release with only the API changes. This can very easily be reverted if it gets done this way. the alterations are not going to be buried in a mess of other commits that would need to be sifted through if it needs to be reverted. I would suggest to do this and skip a major number. so if it doesn't go well then you can continue along the version numbering from the version before the changes.

v9.0 - no API changes
v11.0 = API changes

if the API changes don't work then go back to version 9 and the next major version release would be version 10.0 and then 11.0 after that as if the API changes never existed.  If the API changes do work then continue from version 11 onto version 12. The large jump being there is an indicator of a really HUGE alteration and that it will not be a drop in replacement for version 9
I know that you try to keep with doing a major version each year but this is one of those times you may have to consider side stepping that.

kdschlosser, 2023-08-30:
Also by doing the version changes all in a single bunch with no other changes being made it gives developers more time to adapt code to the new changes. The current running code base would be identical between the 2 versions so they get the same features and functions. This way they wont have to make a huge alteration to their software if they are really wanting to use a new featrure that has been added.

I think it's a nice way to go about it and not put a developer into a panic mode because of the enormity of work that has to be done for a single new feature to be added to their software.

kisvegabor, 2023-08-31:
> I personally like obj. It just simply sounds too familiar.

It really bothers me for a while :smile: 

We can provide an optional compatibility layer where possible. E.g.
```c
#define lv_obj_t lv_widget_t
#define lv_widget_set_background_color    lv_obj_stye_style_bg_color
```
So the users don't have to change the API immediately. 

> There's alreay CENTER in current definition, change mid to center could confuse people. E.g.

I don't know. @kdschlosser is a native US person. So Kevin, does `TOP_MID`, `TOP_MIDDLE` or `TOP_CENTER` sound odd?

> input device compared with display device, we can use the wordinput similar to display . E.g.

Ah simply `input`, brilliant! :slightly_smiling_face: 

> OK so most phrases in an API will need an opposite phrase.

I like the idea of thinking in pairs. 

> I know extra typing sucks but acronyms can suck more especially if you have a function that has 3 acronyms in it.

I also agree to not use abbreviations, except a very few things, e.g. bg/background. dsc/descriptor

> lv_obj_set_style_bg_color is a tad bit obscene.

I agree to make these `lv_obj_set_bg_color`-like.

However the setter/getter combo is confusing IMO.

> typedef struct {
    _background_color_t set_background_color;
} lv_obj_t;

This way each widget would have 200+ callbacks, meaning 200 x 4 = 800 byte extra RAM per widget.
 
> lv_obj_del and lv_obj_remove are 2 completely different things. lv_obj_remove is if you want to remove a object from its parent but keep the memory allocation so the object can be used at a later time. 

I see you point, however for widget there is no way to be floating without a parent, but e.g. events are really "removed" from a list.

> evaluate the phrase against standards. RGBA doesn't stand for Red, Green, Blue and Opacity. It stands for Red, Green Blue and Alpha.. so alpha is a better fit there.

Usually Alpha is the name of the color channel and Opacity is basically everything else. E.g. in CSS there is Opacity as well.

> Me personally I would rather see the API change across the board in a single giant alteration. Some people prefer it to be a gradual thing. That is something you would need to take a poll on.

> I would suggest to do this and skip a major number. so if it doesn't go well then you can continue along the version numbering from the version before the changes.

I think the above mentioned compatibility layer will solve most of the issues.

kdschlosser, 2023-08-31:
with respect to Opacity and Alpha. It completely up to you which way you want to go. I can only tell you that "alpha" seems to be the preferred word to use when wanting to alter how translucent a color, image or a GUI element is. To my understanding alpha is the property and opacity is the value. The correct thing would be to name the property you are changing which would be alpha. Naming the value that goes into that property doesn't seem the right thing to do. 

As far as embedding functions into structures. I see what you are saying about the 4 bytes per function and I get that. But can this be done instead?

```c
typedef struct _lv_obj_t {
    void * set_bg_color(struct * _lv_obj_t, lv_color_t * color) {
        //code to set the color... 
    }
} lv_obj_t;
```

again IDK if C99 is able to do that or not. The benifit would be attaching functions to specific widgets and it also make for less typing for the user.

```c
lv_obj_t * obj = lv_obj_create(NULL);
obj->set_bg_color(obj, lv_color_hex(0xFFFFFF));
```

little less typing and I think it makes it a little bit easier to read as well. 

Again excuse my ignorance with C99 and not knowing if that is even possible. 

Middle and Center are easy. Middle is X axis and Center is Y axis. so the center of an object is technically middle center.

```
Left Top       Middle Top       Right Top
Left Center    Middle Center    Right Center
Left Bottom    Middle Bottom    Right Bottom
```

top to bottom left to right. That is how the naming should be. Positioning is X, Y typically. that same thing should be sued with the naming as well.  X axis first and then Y axis. as seen above. 

doing a compatibility header file is a fantastic idea. But I think it should be reversed so things are actually renamed and the compatibility layer is for backward compatibility. This way it will steer people in the direction of using the new API instead of the old but for purposes of being able to be dropped into an already existing project then the header file for backwards compatibility can be included instead.

kisvegabor, 2023-08-31:
> As far as embedding functions into structures. I see what you are saying about the 4 bytes per function and I get that. But can this be done instead?

Sorry, but such  a change is out of scope now. It was already discussed many times earlier and we have decided not to go into this direction.

> Middle and Center are easy. Middle is X axis and Center is Y axis. so the center of an object is technically middle center.

Is it a personal preference, or a common thing? :smile:

kdschlosser, 2023-08-31:
I am a big fan us using vertical and horizontal in conjunction with center as it makes it easier to understand.  But for that to work it would have to be an or'ed design.

center vertically aligning to the right.
center horizontally aligning to the top

Middle gets confusing as most people do not know that center is for vertical and middle is for horizontal.

Another thing that is a little bit confusing is when I set the alignment of a widget to be in the center LVGL changes from a top left corner relative position system to a center relative position system. When you rotate a widget you have the ability to specifically specify the point that is to be used as the pivot location. Same kind of an idea I think would be nice for alignment as well. 0, 0 is always used as the starting position unless the user specifically changes it. Can either use specific x, y coords or instead use alignment flags That would be a decision you would have to make. I think it offers far greater flexibility than the current design. 

It also removed the need to have so many alignment flags. These flags would be obsolete.
```
LV_ALIGN_OUT_TOP_LEFT
LV_ALIGN_OUT_TOP_MID
LV_ALIGN_OUT_TOP_RIGHT
LV_ALIGN_OUT_BOTTOM_LEFT
LV_ALIGN_OUT_BOTTOM_MID
LV_ALIGN_OUT_BOTTOM_RIGHT
LV_ALIGN_OUT_LEFT_TOP
LV_ALIGN_OUT_LEFT_MID
LV_ALIGN_OUT_LEFT_BOTTOM
LV_ALIGN_OUT_RIGHT_TOP
LV_ALIGN_OUT_RIGHT_MID
LV_ALIGN_OUT_RIGHT_BOTTOM
```

```c
lv_obj_set_align_point(obj, LV_ALIGN_MIDDLE_BOTTOM);
lv_obj_set_align(obj, LV_ALIGN_MIDDLE_TOP);
```

is the same thing as LV_ALIGN_OUT_TOP_MID

```c
lv_obj_set_align_point(obj, LV_ALIGN_MIDDLE_TOP);
lv_obj_set_align(obj, LV_ALIGN_MIDDLE_TOP);
```

would align the top edge of the child widget with the top edge of the parent widget right in the middle 

```c
lv_obj_set_align_point(obj, LV_ALIGN_MIDDLE_CENTER);
lv_obj_set_align(obj, LV_ALIGN_MIDDLE_TOP);
```

this would place the widget alignment point to width / 2, height / 2. So the widgets alignment would be the top edge of the parent running through the middle of the child and the child would be centered horizontally in the parent. 

Now If I was to  do this.

```c
lv_obj_set_align_point(obj, LV_ALIGN_LEFT_TOP);
lv_obj_set_align(obj, LV_ALIGN_MIDDLE_TOP);
```
it would place the top left corner of the child on the top so the corner was centered horizontally on the parent. 

Far greater flexibility with that than the current design and you eliminate a bunch of enumerations. 

I think it is also easier to visualize what is happening with it this way...

kisvegabor, 2023-08-31:
I see, and it sounds interesting, but this change would deserve it's own issue. :slightly_smiling_face: 
Here let's focus purely on naming, and open new issue for feature ideas.

kdschlosser, 2023-09-01:
It's kind of there with the naming as it would address the large number of enumerations with a complex naming system. That's the only reason why I mentioned it.

But for the purposes of keeping things how they are and only focus on naming and what should be sued and why I think going with keeping things in an x, y coordinate fashion with center being a vertical or Y alignment and middle being a horizontal or x adjustment would be the proper thing to do.

So here are the alignment enumerations

kdschlosser, 2023-09-01:
question on this tho.

> I see you point, however for widget there is no way to be floating without a parent, but e.g. events are really "removed" from a list.

I am able to do this.

```c
lc_obj_t * hanging_obj = lv_obj_create(NULL);
```

This object has no parent. it is hanging in limbo basically. I would think that if a user wants to keep a widget maybe say a slider resident in memory instead of deleting a creating the thing over and over again they can do that. It would eliminate quite a bit of overhead if there is the ability to reuse a widget across different screens. or different menus.  if it's something simple then the value would only have to be changed. To "Remove" a widget is to take it's parent away but the object still exists it's not rendered because it has no place to be rendered onto. then it can be added to another parent when it needs to be  used again and that could be in a different part of the UI from where it was previously. If an object gets deleted then a recursive delete should be performed on all of the objects under it. If an object gets removed then all of the objects under it stay attached to.

There is technically a "remove" function it's not called that tho.  in order to remove an object you have to set it's parent to `NULL`. The set_parent should remain but maybe adding a helper function called remove that  makes the call to set parent passing NULL is something that should be added. and the obj_del function  should be renamed to `lv_obj_delete` and that is pretty self explanatory as to what it does, It deletes the object so it can never be used again. where as `lv_obj_remove` would remove it from use but it can be put back into use at a latter point in time.

kdschlosser, 2023-09-01:
>>Middle and Center are easy. Middle is X axis and Center is Y axis. so the center of an object is technically middle center.
> Is it a personal preference, or a common thing? 😄

I believe that is how CSS is set up. so maybe not a personal preference.

In css to center text along the x axis you would use `text-align: center;` and in css to"center" an image on the y axis you use `vertical-align: middle;` There is no `vertical-align: center;`  to my knowledge there is no `text-align: middle;` unless something has changed. I don't keep up with CSS changes at all.

kisvegabor, 2023-09-01:
Let's put the align part aside for now. It's really a complicated one. Let's focus choosing the right keywords for now.

This is what I have in mind so far:
- pairs
  - add/remove
  - create/delete
  - set/get

- keywords in general:
  - obj -> widget 
  - rotation
  - scale
  - has
  - is
  - enable
  - ready or finished?
  
- to not abbreviate
  - buffer
  - index
  - button
  - image
  - text
  - result
  - resolution 
  - screen
  - display
  - input
  - middle
  - opacity?

- to abbreviate
  - bg
  - dsc
  - flex
  - opa?

- others
  - use `lv_obj_set_border_color` (omit `style`)

What do you think?

XuNeo, 2023-09-01:
👍

> ready or finished?

Resource is `ready`. Action is `finished`. 
Both should be kept?

> opacity/opa

I vote for opacity.

I against for changing obj to widget, which causes lots of API renaming. Adding an API remaping doesn't solve the problem, because we need to migrate to new APIs eventually.

FASTSHIFT, 2023-09-01:
I don’t want to modify the original concise API abbreviation naming, which will increase a lot of unnecessary migration costs, please keep the original naming style of the API and focus on functionality and performance optimization.

kdschlosser, 2023-09-01:
might be easier to see the list of function names.

<details>
  <summary>Click here for the list, it's long</summary>

```
_lv_anim_core_init
_lv_area_intersect
_lv_area_is_equal
_lv_area_is_in
_lv_area_is_on
_lv_area_is_out
_lv_area_is_point_on
_lv_area_join
_lv_area_set_pos
_lv_bidi_detect_base_dir
_lv_bidi_get_logical_pos
_lv_bidi_get_visual_pos
_lv_bidi_process
_lv_bidi_process_paragraph
_lv_demo_music_album_next
_lv_demo_music_get_artist
_lv_demo_music_get_genre
_lv_demo_music_get_title
_lv_demo_music_get_track_length
_lv_demo_music_list_btn_check
_lv_demo_music_list_create
_lv_demo_music_main_create
_lv_demo_music_pause
_lv_demo_music_play
_lv_demo_music_resume
_lv_disp_get_refr_timer
_lv_disp_refr_timer
_lv_draw_sw_mask_cleanup
_lv_event_mark_deleted
_lv_event_pop
_lv_event_push
_lv_fs_init
_lv_gc_clear_roots
_lv_group_init
_lv_img_buf_get_transformed_area
_lv_img_cache_builtin_init
_lv_img_cache_open
_lv_img_decoder_init
_lv_indev_read
_lv_indev_scroll_handler
_lv_indev_scroll_throw_handler
_lv_inv_area
_lv_ll_chg_list
_lv_ll_clear
_lv_ll_get_head
_lv_ll_get_len
_lv_ll_get_next
_lv_ll_get_prev
_lv_ll_get_tail
_lv_ll_init
_lv_ll_ins_head
_lv_ll_ins_prev
_lv_ll_ins_tail
_lv_ll_is_empty
_lv_ll_move_before
_lv_ll_remove
_lv_obj_destruct
_lv_obj_get_ext_draw_size
_lv_obj_get_layer_type
_lv_obj_scroll_by_raw
_lv_obj_style_apply_color_filter
_lv_obj_style_create_transition
_lv_obj_style_init
_lv_obj_style_state_compare
[... 2351 lines cut ...]
_STYLE_STATE_CMP_DIFF_LAYOUT
_STYLE_STATE_CMP_DIFF_REDRAW
_STYLE_STATE_CMP_SAME
_SWITCH_KNOB_EXT_AREA_CORRECTION
_ZOOM_INV_UPSCALE
qrcodegen_Ecc_HIGH
qrcodegen_Ecc_LOW
qrcodegen_Ecc_MEDIUM
qrcodegen_Ecc_QUARTILE
qrcodegen_Mask_0
qrcodegen_Mask_1
qrcodegen_Mask_2
qrcodegen_Mask_3
qrcodegen_Mask_4
qrcodegen_Mask_5
qrcodegen_Mask_6
qrcodegen_Mask_7
qrcodegen_Mask_AUTO
qrcodegen_Mode_ALPHANUMERIC
qrcodegen_Mode_BYTE
qrcodegen_Mode_ECI
qrcodegen_Mode_KANJI
qrcodegen_Mode_NUMERIC
```
</details>

W-Mai, 2023-09-04:
> ```c
> lv_obj_t * obj = lv_obj_create(NULL);
> obj->set_bg_color(obj, lv_color_hex(0xFFFFFF));
> ```
> 
> little less typing and I think it makes it a little bit easier to read as well.
> 
> Again excuse my ignorance with C99 and not knowing if that is even possible.
> 

I don't think this will make the user input less. You can use lv_set_bg_color(obj, XXX); to replace obj->set_bg_color(obj, XXX); Generally speaking, the IDE supports that if the first parameter is a certain type of object, you can use obj.XXXX to automatically complete the corresponding code, and automatically place obj to the position of the first parameter of the method when the completion is completed.

you type `obj.lv_set_b`
the IDE will complete a code like `lv_set_bg_color(obj, );`

faxe1008, 2023-09-04:
Also throwing my hat into the ring on this one :) . I think generally speaking the naming should be **clear** in the first place.
Brevity is not really a concern. Unless we are using ancient compilers that only allow for 31 char identifiers :^) . Also almost everybody nowadays uses a texteditor with some sort of autocompletion. 

```
lv_obj_t * obj = lv_obj_create(NULL);
obj->set_bg_color(obj, lv_color_hex(0xFFFFFF));
````
On this one I do not really see the benefit, if we put aside the amount of typing to be done. Also its quite unusal in terms of API design from what I would expect from a C style API. Putting functions into structs is usually ever only done in the case of callbacks. (On that note having the old `add_event_cb` method back would be cleaner I guess.)

Regarding the renaming, I like almost all of them except for:
-  renaming `lv_obj_t` to  `lv_widget_t`. This breaks **a ton** of code and at this point I think everybody who used the library for extended amount of time has gotten used to it. Also its more generic tone fits better for certain things, e.g. a `lv_line`. Is a line really a widget? 
- `lv_indev_t` to `lv_input_t`: I'd go for `lv_input_dev_t`  since the "input" is really the "data" you are putting in. But this should be describing the device accepting said inputs. Using `dev` for "device" should be acceptable since its an established abbreviation.

kisvegabor, 2023-09-04:
Thank you all! 
I think we mostly on the same page, only need to agree on a few cases. However seemingly `lv_obj` -> `lv_widget` is something that only I want :smile: 

Isn't is it just a search an replace in your project? `lv_obj` -> `lv_widget` and `LV_OBJ` -> `LV_WIDGET`?

kdschlosser, 2023-09-07:
That is the reason why there is going to be a header file that can be included instead. Either the header file is going to contain all of the new names or it is going to contain all of the old names. I think it should be the latter. This way both API's will be available. This is done using preprocessor macros. so no additional overhead needed.

kisvegabor, 2023-09-08:
I think it's ok to add 20-30 functions in the "v8 compatibility header" but wrapping all the `lv_obj_*`s seems a little bit too much. If a search and replace is not working we can really leave it as it is for now.

Anyway, let's focus on simple wording change first, and we will see the rest. I'll open a PR for it next week.

kdschlosser, 2023-09-08:
I can generate all of the macro code automatically if you plan on having the changes in the alternative header file. If you plan on making the changes to the normal code base I can still have the alternative header generated just need to know the new API names first.  either way there is no need to manually make the alternative API header file. The code is already in place to allow this to be done. just some little tweaks to it is all that would be needed.

kisvegabor, 2023-09-11:
Let start in small. What I did know is renaming "btn" to "button" and "img" to "image" just to see how it looks like. 
See it here: https://github.com/lvgl/lvgl/tree/naming-changes
Actually it's already massive due to the lot of renames in types and API. 

The compatibility layer is not added yet.

I'm planning to continue with:
- result
- resolution
- index
- text
- display 
- middle

kisvegabor, 2023-09-14:
I merged `[naming-changes](https://github.com/lvgl/lvgl/tree/naming-changes)` with "img"->"image" and "btn"->"button" refactoring.

kisvegabor, 2023-10-31:
I merged some API cleanup again. All should be backed with defines in `lv_api_map.h`. 

The only thing that I'm still planing to do is keeping only `LV_ALIGN_*` instead of `LV_FLEX/GRID/TEXT_ALIGN_*`.

kdschlosser, 2023-11-01:
It's a huge amount of work isn't it...

kisvegabor, 2023-11-02:
I already see the light at the end of the tunnel :slightly_smiling_face:

kisvegabor, 2023-11-28:
According to the original suggestion of @kdschlosser I reverted the rename of `add_event_cb` to `add_event`. So for consistence there is `add_event_cb` now.

cc @XuNeo 

This was the last naming change that I think we can add to v9 without changing the logic too much. Let's spare the other ideas for the next release.

closed 2023-11-28 by kisvegabor