lvgl/lvgl #4015: [Parallel rendering] Define an OS API

kisvegabor, 2023-02-25:
### Introduce the problem
So far LVGL could work in a bare metal environment. However if we want to support rendering with multiple CPU cores and/or multiple GPUs, allow yielding (sleeping) while LVGL is waiting for a GPU/CPU core, we needed to add an OS abstraction layer.

It's important that LVGL still should work without an OS too, but in this case we can do only something similar to the current behavior. 

### Examples and cases
I think we need these things:
- Thread create/destroy E.g. to the create new SW render threads
- Basic mutex support to protect common resources (e.g. cache)
- Wait for an event, and sleep until it happens. Now I use `pthread_cond_signal` for it. E.g. CPU1 draw thread sleep until there is a new task. Or the main thread should sleep while the draw units are working. 

### Suggested solution
We need to discuss whether we need other OS components too and figure out an API which can work for
- pthread
- RT-Thread (cc @mysterywolf )
- Zephyr (cc @vanwinkeljan )
- NuttX (cc @acassis)
- FreeRTOS (cc @tore-espressif)

To be perfectly honest I usually work in a bare metal environment, so I hope I can get some feedback from people with more experience in (RT)OSes. 

---

As a topic started here is my idea about the OS API (feel free to suggest something completly different)
```c
lv_thread_t * lv_thread_create(int32_t priority, void (*callback)(void *), void * user_data);
lv_res_t lv_thread_delete(lv_thread_t * thread);

lv_mutex_t * lv_mutex_create(void);
lv_res_t lv_mutex_lock(lv_mutex_t * mutex);
lv_res_t lv_mutex_unlock(lv_mutex_t * mutex);
lv_res_t lv_mutex_del(lv_mutex_t * mutex);

lv_thread_sync_t * lv_thread_sync_create(void);
lv_res_t lv_thread_sync_wait(lv_thread_sync_t * sync);       //Wait/sleep here until a signal
lv_res_t lv_thread_sync_signal(lv_thread_sync_t * sync);   //Wake up
lv_res_t lv_thread_sync_destroy(void);     
```

What do you think?

embeddedt, 2023-02-25:
It seems like a reasonable API to me. I wonder if it would be better to use `lv_thread_t` instead of `lv_thread_t *` and allow the OS layer to define that type. This way the OS' native handle, or a pointer to the OS' native structure can be used directly instead of wrapping it in an LVGL structure.

C47D, 2023-02-26:
TinyUSB has a nice OSAL (OS Abstraction Layer): [TinyUSB OSAL] (https://github.com/hathach/tinyusb/tree/master/src/osal)

mysterywolf, 2023-02-27:
the priority should be given a unified rule, for different OS, the  priority rule is different. Some OS are the number of the priority higher, the priority is higher. Some are the number of the priority smaller, the priority is higher.

nicusorcitu, 2023-02-27:
I would make it:
`lv_res_t lv_thread_create(lv_thread_t *thread, int32_t priority, void (*callback)(void *), void * user_data);`
just because we can not rely on the thread pointer (check if not null) to figure out if the creation succeeded or not. So I would have a return code for success/failure.

kisvegabor, 2023-02-27:
@embeddedt 
> It seems like a reasonable API to me. I wonder if it would be better to use lv_thread_t instead of lv_thread_t * and allow the OS layer to define that type. This way the OS' native handle, or a pointer to the OS' native structure can be used directly instead of wrapping it in an LVGL structure.

Good idea!

@C47D 
> TinyUSB has a nice OSAL (OS Abstraction Layer): [TinyUSB OSAL](https://github.com/hathach/tinyusb/tree/master/src/osal)

Thanks, its really useful. At first look we don't need `msec` parameter, but an `in_isr` flag should be really useful. E.g.
```c
lv_res_t lv_mutex_lock(lv_mutex_t * mutex, bool in_isr);
```

@nicusorcitu 
> [...] just because we can not rely on the thread pointer (check if not null) to figure out if the creation succeeded or not

It'd be really more flexible. 

@mysterywolf 
> the priority should be given a unified rule, for different OS, the priority rule is different. Some OS are the number of the priority higher, the priority is higher. Some are the number of the priority smaller, the priority is higher.

> and stack size should be a parameter.

Both are true. Regarding the priority, I don't think we need so many levels, so what do you think about having enum like this:
```c
typedef enum {
  LV_OS_THREAD_PRIO_LOWEST,
  LV_OS_THREAD_PRIO_LOW,
  LV_OS_THREAD_PRIO_MID,
  LV_OS_THREAD_PRIO_HIGH,
  LV_OS_THREAD_PRIO_HIGHEST,
}lv_os_thread_prio_t;
```
and these value can be assigned to an OS specific value internally.

--- 

I'm updating the example in the first  post based on your comments.

GorgonMeducer, 2023-03-06:
In cmsis, there is an OS abstraction layer called cmsis-rtos2, you can take it into consideration and take it as a reference.

https://www.keil.com/pack/doc/CMSIS/RTOS2/html/index.html

kisvegabor, 2023-03-08:
Thanks!

Actually all looks straight forward except the `sync` API. In `pthread` you can do it with [pthread_cond_signal](https://www.geeksforgeeks.org/condition-wait-signal-multi-threading/) which requires a mutex. AFAIK using a mutex is not needed in other OS-es, so we can either figure out a hack for `pthread` or make the mutex required for other OS-es too.

What do you think?

GorgonMeducer, 2023-03-08:
@kisvegabor in most of the case, you can implement a mutex using semaphore. So, it is safe to add an `lv_mutex_t` and implement it in osa.

kisvegabor, 2023-03-08:
I was thinking about using semaphores to use for synchronization too. It seems simpler and more generic.

Let me make some experiments.

GorgonMeducer, 2023-03-08:
probably using semaphore to provide a default implementation for lv_mutex_t and let people to override it when a true mutex is available in the host OS.

kisvegabor, 2023-03-08:
Each OS should have it's own implementation (similarly to GPUs). So "OS implementers"  can use any technique they want to manage thread synchronization.

I've updated it to work with `pthread` semaphores but it's not the best solution. E.g. if there are 5 rendering threads and all sends a dispatch request the related semaphore will have the value of 5. So LVGL will run the dispatcher 5 times but it'd be enough to do it once.

Anyway, it's working now and we can figure out something better later.

kisvegabor, 2023-03-08:
FYI, updates and summary in [this comment](https://github.com/lvgl/lvgl/issues/4016#issuecomment-1460116443).

GorgonMeducer, 2023-03-08:
Yes, the mutex is somehow called binary semaphore.

kisvegabor, 2023-03-08:
AFAIK the difference between mutex and binary semaphore is that a mutex can be unlocked only in the same thread where it was locked. So it can not be used for signaling between threads. 

As `pthread` doesn't have ready to use binary semaphores I've implemented them based on [this comment](https://stackoverflow.com/a/7478825/1999969) and it seems working well.

nicusorcitu, 2023-03-16:
Is it possible to have the abstraction APIs defined? I would like to look for freertos support but I see the current code on parallel-rendering branch is hacked only with pthread.

kisvegabor, 2023-03-16:
How do you mean it? You just need to create `lv_freertos.c` and `lv_freertos.h` similarly to [`lv_pthread.c`](https://github.com/lvgl/lvgl/blob/feat/parallel-rendering/src/osal/lv_pthread.c) and [`lv_pthread.h`](https://github.com/lvgl/lvgl/blob/feat/parallel-rendering/src/osal/lv_pthread.h).

It's a little bit different approach to what we have done so far with e.g. 
```c
#define LV_MALLOC       malloc
```

For the OS API I tried not implementing any OS functions by default, but let the use select an implementation (e.g. `LV_USE_PTHREAD` or `LV_USE_FREERTOS`). It has one big advantage: you don't need to add function names in `lv_conf.h` which is great as you can't do such a think in KConfig. If it works well with OS API, probably I'll update the stdlib API to work like this as well.

nicusorcitu, 2023-03-16:
Ok, it is already defined in [lv_os.h](https://github.com/lvgl/lvgl/blob/feat/parallel-rendering/src/osal/lv_os.h)
I have seen some pthread code in lv_mem.c and I thought you just used directly the pthread.
Thanks.

kisvegabor, 2023-03-16:
> I have seen some pthread code in lv_mem.c and I thought you just used directly the pthread.

It's just by accident while I was debugging.

Note that `malloc` on desktop seems thread safe, but probably we need to use a mutex on embedded systems around malloc, free and realloc.

nicusorcitu, 2023-03-31:
I add the OS layer for FreeRTOS... i will post them soon, probably sending directly to Gabor.

kisvegabor, 2023-03-31:
@nicusorcitu I can add it from your latest email. Is it ok?

nicusorcitu, 2023-03-31:
@kisvegabor I just sent you the latest version on email. Please use that one.

kisvegabor, 2023-03-31:
I've added the FreeRTOS API. thank you @nicusorcitu 
(1f13eae10)

kisvegabor, 2023-04-17:
I think we can close it. The OSAL is tested and works with `pthread` and `FreeRTOS`. 

If something is still missing we can tweak it later too.

closed 2023-04-17 by kisvegabor

GorgonMeducer, 2023-04-18:
I might contribute an implementation for cmsis-rtos2 later.

nicusorcitu, 2023-04-18:
The freertos thread sync is currently done with semaphores but i found that task notification might be faster. I will keep it in mind to add it later but if someone else wants to improve the freertos then is perfectly fine.
________________________________
De la: Gabriel Wang ***@***.***>
Trimis: Tuesday, April 18, 2023 5:21:20 PM
Către: lvgl/lvgl ***@***.***>
Cc: Nicușor Cîțu ***@***.***>; Mention ***@***.***>
Subiect: [EXT] Re: [lvgl/lvgl] [Parallel rendering] Define an OS API (Issue #4015)

Caution: EXT Email

I might contribute an implementation for cmsis-rtos2 later.

GorgonMeducer, 2023-04-18:
@kisvegabor can you please write simple guidance about files that we need to update/modify when adding new rtos support?

GorgonMeducer, 2023-04-19:
@kisvegabor I planned to check the pthread version, but I am not sure about other parts, like should I define any macros? should I update lv_internal_conf etc.

kisvegabor, 2023-04-19:
You need only one define in `lv_conf_template.h` [here](https://github.com/lvgl/lvgl/blob/feat/parallel-rendering/lv_conf_template.h#L276-L283).

GorgonMeducer, 2023-04-19:
Thank you! @kisvegabor 
I guess also here: https://github.com/lvgl/lvgl/blob/f5db0672f5ef15ebc63b793c76f804a8f1841660/src/osal/lv_os.h#L22-L26

kisvegabor, 2023-04-19:
Ah, yes. We should definitely have a short guide about in the future.

nicusorcitu, 2023-04-19:
> You need to add only one define in lv_conf_template.h [here](https://github.com/lvgl/lvgl/blob/feat/parallel-rendering/lv_conf_template.h#L276-L283).
>
@kisvegabor 
Only one OS define needs to be enabled at a time, correct? How shall be updated the template?

kisvegabor, 2023-04-19:
Good point! 

Actually something like this could work too: 
```c
#defien LV_USE_OS    LV_OS_PTHREAD      //0 for no OS
```

And have a list of defines in `lv_osal.h` like:
```
#define LV_OS_PTHREAD   1
#define LV_OS_FREERTOS  2
```

What do you think?

nicusorcitu, 2023-04-19:
I do not think is a good idea to have all OS defined like this. It will conflict with the changes needed [here](https://github.com/lvgl/lvgl/blob/f5db0672f5ef15ebc63b793c76f804a8f1841660/src/osal/lv_os.h#L22-L26).

I think we can go like this:
```
/*-------------
 * OSAL
 *-----------*/
#define LV_USE_OS   1
#if LV_USE_OS
    #define LV_USE_PTHREAD  0
    #define LV_USE_FREERTOS 1
    #define LV_OS_CUSTOM_INCLUDE <stdint.h>
#endif
```
The question is which OS shall be the default one? Or maybe by default we can disable all, so the user will have to enable one by need.

kisvegabor, 2023-04-19:
We can do this for the includes:
```c
#if LV_USE_OS == LV_OS_PTHREAD
#include "lv_pthread.h"
#elif LV_USE_OS == LV_OS_FREERTOS
#include "lv_freertos.h"
#endif
```

nicusorcitu, 2023-04-19:
True.

So `lv_conf.h` shall have:
```
#define LV_USE_OS   LV_USE_PTHREAD
```

Then on file `lv_os.h` we need to first define:
```
#define LV_USE_PTHREAD  1
#define LV_USE_FREERTOS  2
```
Then next we need to include the proper header in the same `lv_os.h`:
```
#if LV_USE_OS == LV_USE_PTHREAD
#include "lv_pthread.h"
#elif LV_USE_OS == LV_USE_FREERTOS
#include "lv_freertos.h"
#endif
```

Note: I used the current define `LV_USE_PTHREAD` instead of `LV_OS_PTHREAD` from your meesages.

kisvegabor, 2023-04-20:
Done in b501320fe

@nicusorcitu I've just noticed that there is an NXP copyright notice in `lv_freertos.c/h`. As these file are not strictly related to NXP having an NXP copyright notice looks a little it weird. So would you mind if I removed it from these 2 files?

nicusorcitu, 2023-04-20:
Hi Gabor,
Unfortunately the copyright notice is something mandatory when we upstream any code. Any new file needs to have the copyright an licence before any type of "software delivery": official release, upstream code, demos to customers...
I understand what you mean, the licence is MIT, is a non restrictive one, but this are our legal department rules.

As you have the MIT LICENCE.txt file in the root folder of the lvgl, I would say you can remove the the licence identifier and disclaimer. But please leave the line with "Copyright 2023 NXP". I think is harmless, it only say the original author of the file.

nicusorcitu, 2023-04-20:
> Done in https://github.com/lvgl/lvgl/commit/b501320fe3705e95de0ea134a1f2d5f3dac8e625
>
I run into some dependency include issue and I still try to solve it.
Did it worked on pthread?

kisvegabor, 2023-04-20:
>  But please leave the line with "Copyright 2023 NXP". I think is harmless, it only say the original author of the file.

Can we say "Authored by NXP, 2023"?

> I run into some dependency include issue and I still try to solve it.
Did it worked on pthread?

It worked with a simple example, but with the complex animation example it really crashed. There was dependency issue in `lv_malloc_builtin` but I fixed it here:  598d14346

What is the exact problem? Can I help with it?

nicusorcitu, 2023-04-21:
> Can we say "Authored by NXP, 2023"?
>
I would say no, :( but let me ask around, is not something I can decide.

kisvegabor, 2023-04-21:
> I would say no, :( but let me ask around, is not something I can decide.

Thanks!

---

Does FreeRTOS work now?

nicusorcitu, 2023-04-21:
Yes, it works now.

[0001-fix-osal-Fix-osal-header-inclusion.patch](https://github.com/lvgl/lvgl/files/11293871/0001-fix-osal-Fix-osal-header-inclusion.patch)

Please include this patch, most of the issues were because of a typo or a wrong circular inclusion.

kisvegabor, 2023-04-21:
Thank you, added. b2091647d

It seems I'm unable to write 10 lines of code without typos. :sweat_smile:

nicusorcitu, 2023-04-21:
No worries, it took a lot to me to figure out the typo too :)

nicusorcitu, 2023-04-21:
Hi Gabor,
I have asked the legal depart about the removal of the licence header from the files.
The answer is that is not possible as we wrote them.
Changing the copyright string is not possible.
The most we can do is to remove the licence identifier (permission notice) and disclaimer (capital strings) but we need to use instead an SPDX identifier.
```
/**
 * @file lv_freertos.c
 *
 */

/**
 * Copyright 2023 NXP
 *
 * SPDX-License-Identifier: MIT
 *
 */
```

GorgonMeducer, 2023-04-24:
Looks like it is the good time to add cmsis-rtos2 support! Try to finish it this week.

GorgonMeducer, 2023-05-01:
cmsis-rtos2 suport is added: https://github.com/lvgl/lvgl/pull/4193