lvgl/lvgl #4016: [Parallel rendering] General discussion

kisvegabor, 2023-02-25:
A lot of things were already discussed in #3700 and that issue is already quite long.  Let's start a new issue where we can summarize the TODOs for parallel rendering.  I see these main topics:

- [ ] OS API:  #4015 
(Main architecture is stable, the [feat/parallel.rendering](https://github.com/lvgl/lvgl/tree/feat/parallel-rendering) branch works with `pthread` on multiple cores using the new OS API, still limited drawing capabilities. ) 
- [ ] Masking 
(All widgets and features should work after that)

 I think if we all these are ready we can merge it to `master`. After that we still need to deal with these:
- [ ] Vector graphics support  
(Support for path stroke, fill, gradient, etc.,  Abstract vector graphics API (Maybe OpenVG?), Address #3650, #3651, #3652, #3673)
- [ ] Optimization 
(Dispatching policies, Faster dispatching, Benchmarking)
- [ ] Caching of draw tasks or widgets
- [ ] Built-in GPU and OS support for specific vendors/projects 
(FreeRTOS,  Zephyr, NuttX, RT-Thread, Linux, Windows, STM, NXP, ARM GPU)

Let's address these one-by-one starting with the OS API. Once we have a consensus there, I'll integrate it to the [feat/parallel.rendering](https://github.com/lvgl/lvgl/tree/feat/parallel-rendering) branch.

cc @GorgonMeducer @nicusorcitu

GorgonMeducer, 2023-02-27:
Where do you want to start?

kisvegabor, 2023-02-27:
> Where do you want to start?

I was planning:
> Let's address these one-by-one starting with the OS API.

Once it's ready we can continue with the more exciting "Masking". Do you agree?

GorgonMeducer, 2023-02-27:
Yes! I agree.

kisvegabor, 2023-03-08:
I made some updates and now you can try it out [here](https://github.com/lvgl/lv_port_pc_eclipse/tree/feat/parallel-rendering). 

Here is what is working:
- Rendering simple rectangles and labels
- Transformation and opacity  including handling layers
- Using multiple threads for SW rendering
- Yielding while waiting for all tasks to be finished before flushing
- Works with or without OS (`LV_USE_OS`).
- `pthread` OS API is added.

What is not working:
- Masks, and complex drawing
- Image, Arc and Line drawing 
- Most of the widgets are not enabled

Description of the new concepts/types/names:
- [`lv_draw_ctx_t` ](https://github.com/lvgl/lvgl/blob/f7df46708f0e5fbb0c3ad44497118e660054203b/src/draw/lv_draw.h#L71-L131)
     - Now it means simply a context in which LVGL can draw.  It has only a few callbacks and it knows the buffer and the buffer's properties.  E.g. SDL render's `draw_ctx` is not compatible with SW render because it uses textures and the buffer is not just an array. So it's mainly about knowing how to handle the buffer and where the buffer is now.
     - When a display is created a `draw_ctx` is added automatically. Usually the default `draw_ctx` of SW render is fine but if a  GPU is used it can also use it's own `draw_ctx`. Especially `buffer_copy` and similar function should have a GPU versions.    
     - All `draw_ctx` are independent from each other and can be processed in parallel
     - Unlike in v8, multiple `draw_ctx`s can not be chained to have a fallback mechanism for the callbacks. A `draw_ctx` is either for SW or NXP, etc. 
     - When a new layer is created (e.g. for a rotated widget) a new `draw_ctx` is created for it too (using the `draw_ctx_init` function of display being refreshed). When the layer is fully rendered a special draw operation draws it as an image to a parent `draw_ctx`'s buffer. 
- [`draw_task`s](https://github.com/lvgl/lvgl/blob/f7df46708f0e5fbb0c3ad44497118e660054203b/src/draw/lv_draw.h#L49-L56)
    - Each `draw_ctx` have a list from the tasks which should be drawn to the buffer of that `draw_ctx`  
    - If you call e.g. [`lv_draw_rect()`](https://github.com/lvgl/lvgl/blob/f7df46708f0e5fbb0c3ad44497118e660054203b/src/draw/lv_draw_rect.c#L62-L70) it just adds a `draw_task` to `draw_ctx` which will be dispatched later (see below)
- [`draw_unit`](https://github.com/lvgl/lvgl/blob/f7df46708f0e5fbb0c3ad44497118e660054203b/src/draw/lv_draw.h#L62-L68)s
   -  An abstract rendering engine. We called it PE (processing unit) but I found it too generic
   -  A displays can have any number of `draw_unit`s
   -  The `draw_rect` and similar callbacks are removed. There is only a `dispatch` callback and the `draw_unit` can pick a task and draw it as it wishes. 
   - [`LV_DRAW_TASK_TYPE_...`](https://github.com/lvgl/lvgl/blob/f7df46708f0e5fbb0c3ad44497118e660054203b/src/draw/lv_draw.h#L33-L40) is added, which tells the type of draw task. 
   - `draw_unit`s can use [lv_draw_get_next_available_task](https://github.com/lvgl/lvgl/blob/f7df46708f0e5fbb0c3ad44497118e660054203b/src/draw/lv_draw.c#L181) to find a task whose area is independent (i.e. not overlapped)  and can be drawn
   - I created 3 SW renderer [here](https://github.com/lvgl/lv_port_pc_eclipse/blob/feat/parallel-rendering/main.c#L123). Feel free to modify it according to the number of threads in your CPU. Note that there is always a main thread which iterates through the LVGL widgets and calls the `lv_draw_rect/label/etc()` functions. When all widgets is iterated, this thread waits until there is no more `draw_task`s in the main `draw_ctx`.
- dispatcher:
   - The key function is [`lv_draw_dispatch()`](https://github.com/lvgl/lvgl/blob/f7df46708f0e5fbb0c3ad44497118e660054203b/src/draw/lv_draw.c#L75) which first removes the ready draw_tasks from the lists (in `draw_ctx`s) and iterates through all  `draw_ctx`'s and and asks the `draw_unit`s to pick tasks from the list. 
   - Note that both the `draw_unit`s and `draw_ctx` belong to `lv_disp_t`. E.g. there can be 3 draw unit (2 SW core + 1 GPU) and any number of `draw_ctx` (e.g. 1 main `draw_ctx` and 7 other for layer which will be destroyed/created during rendering)
   
Notes:  
- [lv_draw_get_next_available_task](https://github.com/lvgl/lvgl/blob/f7df46708f0e5fbb0c3ad44497118e660054203b/src/draw/lv_draw.c#L181) is slow and can be considered as a bottle neck.
- All SW rectangle drawing are [called 10 times](https://github.com/lvgl/lvgl/blob/f7df46708f0e5fbb0c3ad44497118e660054203b/src/draw/sw/lv_draw_sw_rect.c#L77) to make the performance changes more visible

cc @nicusorcitu @GorgonMeducer 
- All asserts are enabled which makes LVGL much slower
- `lv_draw_get_next_available_task` doesn't consider the transformed areas so overlapping object can have artifacts. (It's great see how two thread are working on the same area though)
[image]

nicusorcitu, 2023-03-13:
Hi @kisvegabor 
1. I see there is not yet support for freertos OS. Do you plan to add it soon? Else I will try to add by myself the freertos apis.

2. The display driver has also changed in LVGL v9. Is there any correspondent for:
`disp_drv.clean_dcache_cb` or it has been removed completely? In this case, shall clean_dcache() be called during the flush_cb()?

3. Can you please give me the lv_conf.h with all the widgets disabled.... I can get rid of all the errors.

Thanks,
Nicu.

kisvegabor, 2023-03-13:
> I see there is not yet support for freertos OS. Do you plan to add it soon? Else I will try to add by myself the freertos apis.

For me `pthread` is enough for me for the development so wasn't planing to focus on FreeRTOS yet. However, it'd be useful to see if it really works with FreeRTOS too. So it'd be great if you could add it.

> The display driver has also changed in LVGL v9. Is there any correspondent for:
disp_drv.clean_dcache_cb or it has been removed completely? In this case, shall clean_dcache() be called during the flush_cb()?

Yes, I think it's cleaner.

> Can you please give me the working lv_conf.h you have for this branch with all the widgets disabled.... Is it [this](https://github.com/lvgl/lv_port_pc_eclipse/blob/feat/parallel-rendering/lv_conf.h) ?

Yes, I use it on PC and it should work on embedded systems too. Except `LV_MEM_SIZE`. :slightly_smiling_face:

nicusorcitu, 2023-03-14:
I am currently trying to run the master branch (not the parallel-rendering branch) just to make the display work. I am running in double buffer mode but I see the display is getting refreshed just once. There is no buffer switch at all.
E.G. call_flush_cb() gets called just once at the beginning.

My code looks like this:
```
    /*------------------------------------
     * Create a display and set a flush_cb
     * -----------------------------------*/
    lv_disp_t * disp = lv_disp_create(DEMO_BUFFER_WIDTH, DEMO_BUFFER_HEIGHT);

    //lv_draw_unit_sw_create(disp, 1);

    lv_disp_set_flush_cb(disp, DEMO_FlushDisplay);

    /*
     * Two buffers screen sized buffer for double buffering.
     * Both LV_DISP_RENDER_MODE_DIRECT and LV_DISP_RENDER_MODE_FULL works, see their comments.
     */
    lv_disp_set_draw_buffers(disp, s_frameBuffer[0], s_frameBuffer[1], DEMO_FB_SIZE, LV_DISP_RENDER_MODE_FULL);
```

kisvegabor, 2023-03-14:
Do you with a debugger where it spins? 
Are you sure `lv_disp_flush_ready` is called?

nicusorcitu, 2023-03-14:
[image]

nicusorcitu, 2023-03-14:
[image]

nicusorcitu, 2023-03-14:
Yes, it gets called once.

kisvegabor, 2023-03-14:
Do you call `lv_tick_inc()`?

nicusorcitu, 2023-03-14:
yes it is incremented

kisvegabor, 2023-03-14:
But `lv_tick_inc` is called too? 

After the first refresh if you pause it, in which function does it spinning?

kisvegabor, 2023-03-14:
I've tested the 2 buffers with full refresh mode using SDL and it worked on my end. On which MCU/board do you test it? If I have that board, can you send a project so that I can try it out as well?

(I'll be on Embedded World until Friday).

kisvegabor, 2023-03-14:
I've open a new issue to discuss masking: #4059

nicusorcitu, 2023-03-14:
The board is RT1170 with a display 720x1280.
I enabled timer and disp traces and the logs shows in a loop these messages:
```
[Trace] (7.380, +15)     lv_timer_handler: begin        (in lv_timer.c line #70)
[Trace] (7.386, +6)      lv_timer_handler: finished (4294967295 ms until the next timer call)   (in lv_timer.c line #145)
```

kisvegabor, 2023-03-14:
Can you send your project? (via email is fine too)

nicusorcitu, 2023-03-15:
Can you please point me to the same application you run (the one that rotates all the rectangles and text - [this](https://github.com/lvgl/lv_port_pc_eclipse/blob/87ec6529cb4c82efa1665b36c6a153e9df9ee3ce/main.c#L70) does not seem to be animated.

codewitch-honey-crisis, 2023-03-16:
Hey, @kisvegabor I have a heck of an ask, but I feel like if I can get what I'm after I'd be a lot more helpful to this conversation:

I'd like to understand the techniques used for software rendering antialiased draw elements. I've looked through the code, and it does all of this strange masking and such, I just can't wrap my head around it, but it's almost like you're rasterizing vectors already, but without floating point.

That's important. I need to know how this works, although admittedly, not just to contribute to this project in this arena, but it would also be helpful to understand it for my own purposes.

Is it possible for me to get some time with the authors, maybe on discord or something to understand how this works?

kisvegabor, 2023-03-17:
@nicusorcitu 
I used something like tihs:
```c

void rotate_cb(void * var, int32_t v)
{
      lv_obj_set_style_transform_angle(var, v, 0);
}

...
//In main.c
  lv_obj_set_flex_flow(lv_scr_act(), LV_FLEX_FLOW_ROW_WRAP);

  for(int i = 0; i < 100; i++) {
      lv_obj_t * panel = lv_obj_create(lv_scr_act());
      lv_obj_set_size(panel, 140, 45);
      lv_obj_set_style_bg_color(panel, lv_color_hex3(0xf00 + i%16), 0);
      lv_obj_set_style_border_color(panel, lv_color_hex3(0x0f0), 0);
      lv_obj_set_style_pad_all(panel, 8, 0);

      lv_obj_t * label = lv_label_create(panel);
      lv_label_set_text_fmt(label, "Hello label\nI'm with the ID of %d", i);

      lv_anim_t a;
      lv_anim_init(&a);
      lv_anim_set_var(&a, panel);
      lv_anim_set_time(&a, lv_rand(1000, 10000));
      lv_anim_set_values(&a, 0, 3600);
      lv_anim_set_exec_cb(&a, rotate_cb);
      lv_anim_set_repeat_count(&a, LV_ANIM_REPEAT_INFINITE);
      lv_anim_start(&a);
  }

```

kisvegabor, 2023-03-17:
@codewitch-honey-crisis 
[Here](https://docs.lvgl.io/master/overview/drawing.html#masking) you can check out an overview about the masking. If you still have any questions I'd be happy to answer them.

codewitch-honey-crisis, 2023-03-17:
Wow thanks! I never expected something that technical to be in the documentation. It didn't even occur to me look. Doh!

kisvegabor, 2023-03-31:
@nicusorcitu found a problem with the layers.  Let's say are rotating 100 rectangles and each need a 20 kB buffer. In total it's 2 MB. Imagine that we have only main thread and a rendering thread. This is what will happen:
1. The main thread start to collect the draw tasks
2. The rendering thread will start to work on the rendering 
3. As adding the draw tasks is much faster then the rendering all tasks will be create before even the background is filled.
4. When a new layer was required (for the rotated rectangles) LVGL already initialized (i.e. allocated) the buffer of that layer. So when all draw tasks were added, already 100 layer buffer were allocated, consuming 2MB.

As we have only one renderer thread, it doesn't make sense to have all the layers allocated. I've fixed it in e98db0a54 by allocating the layer's buffer just before the first rendering on them.

nicusorcitu, 2023-04-01:
I think there is an extra issue while releasing the memory of the layer.
The layer may consist of multiple tasks: in our case a rectangle draw and a label draw. After those two tasks are done, the layer is added for blending into the FB as another task. After the blend is done, the layer memory is released.
The layer blending task should be more priority, otherwise other tasks might take over and we will still run into allocating too much memory.
________________________________
De la: Gabor Kiss-Vamosi ***@***.***>
Trimis: Friday, March 31, 2023 10:30:13 PM
Către: lvgl/lvgl ***@***.***>
Cc: Nicușor Cîțu ***@***.***>; Mention ***@***.***>
Subiect: [EXT] Re: [lvgl/lvgl] [Parallel rendering] General discussion (Issue #4016)

Caution: EXT Email

@nicusorcitu<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fnicusorcitu&data=05%7C01%7Cnicusor.citu%40nxp.com%7C49faa992dd3a465d8bd408db321e5a2d%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C0%7C638158878162227494%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=leRVDAIiU3PRfbJ1vSgW6M0Y2bQcTINhcC%2ByxH0%2Bb7U%3D&reserved=0> found a problem with the layers. Let's say are rotating 100 rectangles and each need a 20 kB buffer. In total it's 2 MB. Imagine that we have only main thread and a rendering thread. This is what will happen:

  1.  The main thread start to collect the draw tasks
  2.  The rendering thread will start to work on the rendering
  3.  As adding the draw tasks is much faster then the rendering all tasks will be create before even the background is filled.
  4.  When a new layer was required (for the rotated rectangles) LVGL already initialized (i.e. allocated) the buffer of that layer. So when all draw tasks were added, already 100 layer buffer were allocated, consuming 2MB.

As we have only one renderer thread, it doesn't make sense to have all the layers allocated. I've fixed it in e98db0a<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Flvgl%2Flvgl%2Fcommit%2Fe98db0a54c8bbd6b4aab0ef0915455c1f12e5b6f&data=05%7C01%7Cnicusor.citu%40nxp.com%7C49faa992dd3a465d8bd408db321e5a2d%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C0%7C638158878162227494%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=IHGWZrSkoaItub3G9mwE2ayiwExBDeExjfKnsn8jvmE%3D&reserved=0> by allocating the layer's buffer just before the first rendering on them.

nicusorcitu, 2023-04-01:
And another request would be if we run with a single draw task, for example in a system with single cpu core and no gpu, or in a sistem we want to draw with gpu only.
There is no point having those layers in the above cases, we can draw directly in fb as we will not need to alocate any layer space and no extra blending of them is required.
________________________________
De la: Nicușor Cîțu ***@***.***>
Trimis: Saturday, April 1, 2023 7:42:12 AM
Către: lvgl/lvgl ***@***.***>; lvgl/lvgl ***@***.***>
Cc: Mention ***@***.***>
Subiect: Re: [EXT] Re: [lvgl/lvgl] [Parallel rendering] General discussion (Issue #4016)

I think there is an extra issue while releasing the memory of the layer.
The layer may consist of multiple tasks: in our case a rectangle draw and a label draw. After those two tasks are done, the layer is added for blending into the FB as another task. After the blend is done, the layer memory is released.
The layer blending task should be more priority, otherwise other tasks might take over and we will still run into allocating too much memory.
________________________________
De la: Gabor Kiss-Vamosi ***@***.***>
Trimis: Friday, March 31, 2023 10:30:13 PM
Către: lvgl/lvgl ***@***.***>
Cc: Nicușor Cîțu ***@***.***>; Mention ***@***.***>
Subiect: [EXT] Re: [lvgl/lvgl] [Parallel rendering] General discussion (Issue #4016)

Caution: EXT Email

@nicusorcitu<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fnicusorcitu&data=05%7C01%7Cnicusor.citu%40nxp.com%7C49faa992dd3a465d8bd408db321e5a2d%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C0%7C638158878162227494%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=leRVDAIiU3PRfbJ1vSgW6M0Y2bQcTINhcC%2ByxH0%2Bb7U%3D&reserved=0> found a problem with the layers. Let's say are rotating 100 rectangles and each need a 20 kB buffer. In total it's 2 MB. Imagine that we have only main thread and a rendering thread. This is what will happen:

  1.  The main thread start to collect the draw tasks
  2.  The rendering thread will start to work on the rendering
  3.  As adding the draw tasks is much faster then the rendering all tasks will be create before even the background is filled.
  4.  When a new layer was required (for the rotated rectangles) LVGL already initialized (i.e. allocated) the buffer of that layer. So when all draw tasks were added, already 100 layer buffer were allocated, consuming 2MB.

As we have only one renderer thread, it doesn't make sense to have all the layers allocated. I've fixed it in e98db0a<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Flvgl%2Flvgl%2Fcommit%2Fe98db0a54c8bbd6b4aab0ef0915455c1f12e5b6f&data=05%7C01%7Cnicusor.citu%40nxp.com%7C49faa992dd3a465d8bd408db321e5a2d%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C0%7C638158878162227494%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=IHGWZrSkoaItub3G9mwE2ayiwExBDeExjfKnsn8jvmE%3D&reserved=0> by allocating the layer's buffer just before the first rendering on them.

kisvegabor, 2023-04-03:
> The layer blending task should be more priority, otherwise other tasks might take over and we will still run into allocating too much memory.

Good point! I'll look into this.

> There is no point having those layers in the above cases, we can draw directly in fb as we will not need to alocate any layer space and no extra blending of them is required.

Just to say a concrete example: let's say there is a widget with some children, which is zoomed and the invalidated area is smaller or equal than the zoomed widget. We can use the "main layer" (i.e. the display's layer) directly for drawing the widget but I don't know how to proceed from here. E.g. in partial rendering mode we need to transform this image and draw it into the display. But how? There is no other buffer where we can store the result.

Could you say an example for this?

nicusorcitu, 2023-04-03:
I see, these layers are mandatory for partial rendering mode. I was thinking only on full double buffering mode.
I thought you created just now those layers just to have independent draw tasks which later will get blended.

kisvegabor, 2023-04-03:
> I thought you created just now those layers just to have independent draw tasks which later will get blended.

It's not the case. Layers are created only if needed. See [here](https://github.com/lvgl/lvgl/blob/e98db0a54c8bbd6b4aab0ef0915455c1f12e5b6f/src/core/lv_refr.c#L869-L872). In the common cases, objects are just drawn in to the active layer (usually the buffers of the display). But if for any transformation, or other magic the objects needs to be on a separate layer, LVGL will create a layer and render the widgets there. 

In the spinning rectangle+label example all rectangles are on a separate layers as they are rotated.

kisvegabor, 2023-04-04:
I've added a nice debug tools
- `LV_USE_LAYER_DEBUG` Draw a red overlay for ARGB layers and a green overlay for RGB layers
- `LV_USE_PARALLEL_DRAW_DEBUG` Draw overlays with different colors for each draw_unit's tasks. Also add the index number of the draw unit on white background. For layers add the index number of the draw unit on black background.

The latter looks like this:
[image]

nicusorcitu, 2023-04-04:
Hi Gabor,

I am trying to enable the VG Lite on parallel rendering thread branch.
So I used for the beginning the same [lv_draw_unit_sw_create](https://github.com/lvgl/lvgl/blob/919829026b24cffa8ed4c2f989f03518c3a5f08f/src/draw/sw/lv_draw_sw.c#L97) in order to create the GPU unit and I changed only the thread routine, I created my own render_thread_gpu_cb() which can draw only rectangles.
```
static void render_thread_gpu_cb(void * ptr)
{
    lv_draw_sw_unit_t * u = ptr;

    while(1) {
        while(u->task_act == NULL) {
            lv_thread_sync_wait(&u->sync);
        }
        if(u->task_act->type == LV_DRAW_TASK_TYPE_RECTANGLE) {
            lv_draw_vglite_rect((lv_draw_unit_t *)u, u->task_act->draw_dsc, &u->task_act->area);

            /*Cleaup*/
            u->task_act->state = LV_DRAW_TASK_STATE_READY;
            u->task_act = NULL;

            /*The draw unit is free now. Request a new dispatching as it can get a new task*/
            lv_draw_dispatch_request();
        }
    }
}
```

But I have the same dispatcher [lv_draw_sw_dispatch](https://github.com/lvgl/lvgl/blob/919829026b24cffa8ed4c2f989f03518c3a5f08f/src/draw/sw/lv_draw_sw.c#L120)

The key is that if the GPU thread is woken because of a task that is not being supported (like LV_DRAW_TASK_TYPE_LABEL), it will loop forever.
1. The problem is believe is that dispatcher does not take account of the feature supported by each unit. E.G. is shall wake up only the unit threads that have are capable of handling that task. Like here shall be added that check.
[lv_draw_dispatch](https://github.com/lvgl/lvgl/blob/919829026b24cffa8ed4c2f989f03518c3a5f08f/src/draw/lv_draw.c#L144)
OR
2. Shall I have a separate lv_draw_gpu_dispatch() for the GPU unit?

kisvegabor, 2023-04-05:
Note that `lv_draw_sw_dispatch` is not a generic dispatcher. For VGLite we should have a custom one (of course ` lv_draw_sw_dispatch` can be a good starting point). Here is an example:
```c
static int32_t lv_draw_vglite_dispatch(lv_draw_unit_t * draw_unit, lv_layer_t * layer)
{
   lv_draw_unit_vglite_t * draw_unit_vglite = (lv_draw_unit_vglite_t *) draw_unit;

    /*Return immediately if it's busy with draw task*/
    if(draw_unit_vglite->task_act) return 0;

    /*Try to get an ready to draw*/
    lv_draw_task_t * t = lv_draw_get_next_available_task(layer, NULL);
    if(t == NULL) return -1; //Signal that there were no areas to draw 
    while(t) {
      if(can_i_draw_it(t)) {
        //Allocate lyer buffer if needed
        ...
        //Init and wake up the draw thread
        ...
        
        //1 draw task was taken
        return 1;      
      }
   
     //Find another area, maybe it will be supported by VG-Lite   
     t = lv_draw_get_next_available_task(layer, t);
   }

   //No draw task was taken
   return 0;
}
```

`lv_draw_get_next_available_task` is very slow now (O(N^2)). An idea is already forming in my mind about how to organize the draw task to easily find independent areas.

nicusorcitu, 2023-04-05:
@kisvegabor Thank you, I did pretty much the same.

kisvegabor, 2023-04-05:
And did you get the infinite loop?

> The key is that if the GPU thread is woken because of a task that is not being supported (like LV_DRAW_TASK_TYPE_LABEL), it will loop forever.

So you shouldn't wake the GPU thread for a task that it can not draw. For a LABEL you can just `return 0` to indicate that you haven't picked any tasks. 

There is one theoretical case when the system can halt: let's say there are 2 draw units but only of them is working on a task now. That task is ready, sends a dispatch requests, but for some reason non of the draw units will pick up any tasks. In this case no one will send a new dispatch request (as there is no draw task will be finished)  and no dispatching will happen.

However at worst case SW render should take any the tasks one by one. The only reason it cannot take a task is that it's busy with a task. But when that task is finished a new dispatch request will be sent.

Maybe it's not fully related to your question but at least I also understand better what happens :slightly_smiling_face:

nicusorcitu, 2023-04-05:
It was actually my exact problem. Then indeed I figured out I shall not wake up GPU thread if is not supported.
So I implemented yesterday my own GPU dispatcher (starting from sw one) and now I run with one SW thread and one GPU thread.
But now I am digging again into a different issue, the rectangles are not rotating anymore. I have to debug it closely.

kisvegabor, 2023-04-05:
> So I implemented yesterday my own GPU dispatcher (starting from sw one) and now I run with one SW thread and one GPU thread.

Amazing!!! :tada: 

> But now I am digging again into a different issue, the rectangles are not rotating anymore. I have to debug it closely.

Let me know if I can help with it.

nicusorcitu, 2023-04-06:
Hold on, is not fully working :) And I enabled only the rectangle draw on the GPU.

kisvegabor, 2023-04-06:
I know, but at least now the new architecture is being evaluated with a GPU too. :slightly_smiling_face:

nicusorcitu, 2023-04-06:
It works now, the problem was that GPU dispatcher shall not return -1 if there is no supported task in the list. It shall return 0 so that SW dispatcher can get called and handle it.

GorgonMeducer, 2023-04-06:
> But maybe the memory bandwidth would be a limiting factor...
Don't worry about this part. At least that means we can keep the bus busy and fully use the potential bandwidth.

kisvegabor, 2023-04-06:
Okay. Actually my idea is to define a `LV_DRAW_MAX_AREA_SIZE` parameter and if a draw task is larges cut it into as many pieces as required to fit into the defined size.

nicusorcitu, 2023-04-10:
> >The display driver has also changed in LVGL v9. Is there any correspondent for:
    disp_drv.clean_dcache_cb or it has been removed completely? In this case, shall clean_dcache() be called during the flush_cb()?
>
>Yes, I think it's cleaner.
>
Before submitting to the GPU (any GPU or smart DMA), the CPU cache needs to be flushed as well.
I know this has nothing to do with the display, is something in between CPU and GPU. Before we had `clean_dcache_cb` and we could make use of it: [here](https://github.com/lvgl/lvgl/blob/53dcb8cd998a6f98701a5702f1c9cc62246b313a/src/draw/nxp/vglite/lv_vglite_utils.c#L142)
I a thinking how to get access now to the "platform code", e.g. the ARM CMSIS code - e.g. how to invalidate the cache from the LVGL draw APIs.

kisvegabor, 2023-04-11:
Can it be done internally in the `lv_nxp_... .c` files if needed. You can do any platform specific function calls there.

nicusorcitu, 2023-04-19:
During the disp->flush_cb we have to clean and invalidate data cache too. So we already have that piece of the code implemented in the platform code.

Implementing the same code in NXP GPU files does not seem reasonable.

The solutions will be:
1. Keep the implementation on the platform code (sdk) but expose it via an global function.
The platform header with definition will need to be included in the NXP GPU files.... not so nice. I would leave the acceleration code independent of the SDK code. The reason is that lvgl can be used independent of the SDK, we are planing to use lvgl under linux for an automotive board.
2. Keep the disp->clean_dcache_cb so that I can have access to it like before.
All the other GPU vendors will make use of it.

I think it is cleaner with 2. If you do not agree then we can try 1.

kisvegabor, 2023-04-20:
> During the disp->flush_cb we have to clean and invalidate data cache too. So we already have that piece of the code implemented in the platform code.

I see now and in this case 2) is ok too. However `clean_dcache` seems like too specific to me. To approach it from a broader perspective, what we need to do is preparing a buffer for a new operation. So we need to be sure that its content is ready/finalized and we can do something with it.

On MCUs it usually mean the invalidating the `dcache` but on PCs with video cards or with some special GPUs it could mean different things too. So it's not necessary about `dcache`.

Besides invalidating the whole `dcache` is probably not a good idea. We should invalidate only the required buffers. 

And finally if there are multiple displays, all should use the same `clean_dcache_cb`. So seemingly it's not related to the display object but to the whole system. 

So what about something like this:
```c
lv_draw_finalize_buf(buf, size);  //finalize is not a good word
```
This function together with `lv_draw_malloc_buf` can be implemented in a platform specific file. 

(The names should be improved :slightly_smiling_face: )

What do you think?

nicusorcitu, 2023-04-20:
It looks like a better option to invalidate only the proper buffer.

Some suggestions:
`lv_draw_flush_buf` - simplest one
`lv_draw_clean_invalidate_dcache_buf` - more suggestive

I have only one concern. The ARM documentation says the buf address need to be aligned to 32 bytes! We do not guarantee that.
```
/**
  \brief   D-Cache Clean and Invalidate by address
  \details Cleans and invalidates D_Cache for the given address
           D-Cache is cleaned and invalidated starting from a 32 byte aligned address in 32 byte granularity.
           D-Cache memory blocks which are part of given address + given size are cleaned and invalidated.
  \param[in]   addr    address (aligned to 32-byte boundary)
  \param[in]   dsize   size of memory block (in number of bytes)
*/
__STATIC_FORCEINLINE void SCB_CleanInvalidateDCache_by_Addr (volatile void *addr, int32_t dsize)
```

kisvegabor, 2023-04-20:
> lv_draw_flush_buf - simplest one

I was thinking about `flush` too but it'd be easy to confuse with the `flush_cb` which updates the display. Maybe we should rename the `flush_cb`? 

> The ARM documentation says the buf address need to be aligned to 32 bytes! So we really need the lv_mem_alloc_buf() with the alignment option discussed https://github.com/lvgl/lvgl/pull/1824#issuecomment-1508259006

Yes, it's nice to see that the pieces are falling into place.

kisvegabor, 2023-04-21:
The same happened to me when I used `layer->clip_area` during drawing.  If you also use it, please use `draw_unit->clip_area` instead.

I need to hide,`layer->clip_area` a little because it's too easy to mix them.

codewitch-honey-crisis, 2023-04-22:
Re: OpenVG support. I've been doing some R&D because I'm about to put together an ARM hardware and firmware package using ZephyrOS and I really need a graphics library that can leverage the Mali GPU. It does OpenGL 2.0. All of this to try to wedge something realtime onto 16MB of flash or less. LVGL would be a godsend. Frankly, I think ARM Apus are the future, so it would really be great to support OpenVG. You'll have to eventually to stay relevant. I'd put mortgage money on it.

kisvegabor, 2023-04-24:
At least OpenGL ES should be supported in v9 and we should it make it simple to be integrated. So it's definitely on the road map.

nicusorcitu, 2023-04-24:
> The same happened to me when I used layer->clip_area during drawing. If you also use it, please use draw_unit->clip_area instead.
>
I used `draw_unit->clip_area` and works now. Just wondering how it coexist together with `layer->clip_area`. 

> I guess it's to early to talk about performance metrics, but maybe... 🙂 Can you test the rotating rectangle UI with v8 and v9? But without actually rotating the rectangles and with ~20 rectangles.
>
I did a quick test by rotating 20 rectangles (rounded corners) with CPU draw thread only and I got 14FPS. This with double buffer mode and FB in external memory (SDRAM).
I enabled the VGLite draw thread and got the same 14FPS. It looks reasonable as GPU is less frequency and we do not have all the optimizations enabled.
It will be interesting to find out the CPU idle percentage, in lvgl v8 I had some patches where I could get the idle time from the freertos idle task hook. We shall have a dedicated discussion for that, I will open one.

Still not tried the same test with v8.3.5 but I will do it.

Just for reference, I tried to change the code to not rotate the rectangles (e.g. to not create any new layers and then blend them to FB) so I changed the callback to only update the y position of the rectangles (e.g. to move them up and down to the screen).  I could not make them work, they simply seem to ignore the APIs `lv_obj_set_y()` or `lv_obj_set_style_y()`.

kisvegabor, 2023-04-24:
> I used draw_unit->clip_area and works now. Just wondering how it coexist together with layer->clip_area.

It's really counter intuitive but I added  a description [here](https://github.com/lvgl/lvgl/blob/a3c5a5960d74eb5b946f9ccd9ccd3c5cbb118e7c/src/draw/lv_draw.h#L96-L103).

 >  I could not make them work, they simply seem to ignore the APIs lv_obj_set_y() or lv_obj_set_style_y().

It's because te screen has a layout (`lv_obj_set_flex_flow(lv_scr_act(), LV_FLEX_FLOW_ROW_WRAP);`) and teh layout is stringer than normal coordinate setting. You can
1. not use layout but position the widget manually
2. translate the widget widgets with `lv_obj_set_style_translate_y(panel, 200, 0);`

nicusorcitu, 2023-04-25:
Thanks. The test to just draw 20 rectangles works now. But it seems to be limited to 28 fps, no matter what I do. I let just 2 rectangles draw and still got 28 fps. I am not sure where does it come this limit of 28 fps.

I am testing with the below test patch.
[0001-fix-perf-Log-FPS-once-per-second-instead-of-creating.patch](https://github.com/lvgl/lvgl/files/11323720/0001-fix-perf-Log-FPS-once-per-second-instead-of-creating.patch)

kisvegabor, 2023-04-26:
By default the time between two calls of the `_lv_disp_refr_timer` is measured. And it's limited by `LV_DEF_REFR_PERIOD`.

I suggest adding an event_cb to display (v9 supports events on the display) like this:
```c

void lv_disp_event_cb(lv_event_t * e)
{

    lv_event_code_t code = lv_event_get_code(e);
    static uint32_t t;
    if(code == LV_EVENT_RENDER_START) {
        t = lv_tick_get();
    }
    if(code == LV_EVENT_RENDER_READY) {
        printf("%d\n", lv_tick_elaps(t));

    }
}

  lv_disp_add_event(disp, lv_disp_event_cb, LV_EVENT_ALL, NULL);
```

nicusorcitu, 2023-04-26:
> By default the time between two calls of the _lv_disp_refr_timer is measured. And it's limited by LV_DEF_REFR_PERIOD.
>
Yes, I know that it was also before.
/*Default display refresh, input device read and animation step period.*/
#define LV_DEF_REFR_PERIOD  33      /*[ms]*/

So basically LVGL will try to satisfy as much as 30FPS.
But it was that calculation, if the frame took only 5ms instead of 30ms. it will assume is capable of 6times more FPS (an assumption that is optimistic) and will conclude is capable of 180FPS.

Now, no matter what I do (even a single rectangle) I get only 28FPS. If it would be 30 then I would say is the limitation of the LV_DEF_REFR_PERIOD but is not.

kisvegabor, 2023-04-26:
> In _lv_disp_refr_timer() the elaps value is exactly the time between two refreshes instead of the difference between RENDER_START and RENDER_READY.

Yes, that's why I suggest adding the events.

nicusorcitu, 2023-04-26:
Instead of declaring myself the event for my display, is not a better common solution to have the FPS every time we need it?
We shall have a common solution for calculating FPS as well as CPU idle (real CPU idle, not the one that assumes that CPU draws everything).

nicusorcitu, 2023-04-26:
I mean I can do it just for my testing, I have done it before also to calculate the CPU idle from the FreeRTOS idle task measurements, but I would like to have a common solution on LVGL upstream for all platforms, all OS-es.

nicusorcitu, 2023-04-26:
For now, what about this fix for FPS calculation while enabling PERFORMANCE_MONITOR?
[0001-fix-perf-Fix-the-fps-elapsed-time-calculation.patch](https://github.com/lvgl/lvgl/files/11332451/0001-fix-perf-Fix-the-fps-elapsed-time-calculation.patch)

kisvegabor, 2023-04-26:
We can figure out something for sure! Probably the OS API needs to have a `lv_os_get_idle_pct()` function too and in `LV_USE_OS != 0` we can use it instead of `lv_timer_get_idle()` in `lv_refr.c`.

Regarding the FPS measurement, for a normal user it might be confusing to see 300 FPS when he presses a button. And it's really not the case as the screen was refreshed only e.g. 30 times per second. What about showing 
```
33 FPS (7ms)
CPU: 35%
``` 

Where "7ms" is the rendering time.

[120 more replies]

closed 2024-01-14 by kisvegabor