lvgl/lvgl #4017: lv_img_get_angle() returns incorrect number on negatively rotated images.

vritzka, 2023-02-26:
### Perform all steps below and tick them with [x]
- [x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I have an image 'myImage'.

The function lv_img_get_angle(myImage) will return 0; 

Now I rotate the image left with lv_img_set_angle(myImage, -200);
When I now run lv_img_get_angle(myImage), it returns 65336.

### To Reproduce

```c

// 1. Create any image
ui_needle1 = lv_img_create(ui_Home_Screen);
lv_img_set_src(myImage, &ui_img_1113527240);
lv_obj_set_width( myImage, LV_SIZE_CONTENT);  
lv_obj_set_height( myImage, LV_SIZE_CONTENT);   
lv_obj_set_x( myImage, 10 );
lv_obj_set_y( myImage, 10 );

// 2. Read angle of image

lv_img_get_angle(myImage);      // 0

// 3. Rotate the image negatively by -200

lv_img_set_angle(myImage, -200);

// 4 Read angle of image again:

lv_img_get_angle(myImage);      // 65336

```
-->

### Expected behavior
After rotating an image by -200 with lv_img_set_angle(), the new value given by lv_img_get_angle() should return 3400. 

### Screenshots or video

kisvegabor, 2023-02-27:
Which LVGL version do you use?

vritzka, 2023-02-27:
I’m using version 8.3.0

kisvegabor, 2023-02-27:
It was already fixed in `master` but I've just fixed it in the `release/v8.3` branch too.
See e8d8f399e

closed 2023-02-27 by kisvegabor