lvgl/lvgl #4018: lv_list arbitrary content?

pr8x, 2023-02-26:
### Introduce the problem

Hey, is it possible to add arbitrary (i.e. any `lv_object`) to a `lv_list`? 

### Examples and cases

### Suggested solution

It would be nice if I could et (for example) `lv_span` in there.

kisvegabor, 2023-02-27:
Yes you can add any widgets. E.g.
```c
   lv_obj_t * sg =  lv_spangroup_create(list1);
   lv_spangroup_set_mode(sg, LV_SPAN_MODE_BREAK);
   lv_obj_set_width(sg, lv_pct(100));
   lv_span_t * s = lv_spangroup_new_span(sg);
   lv_span_set_text(s, "Hello, hello! I'm very long, so you can see word wrapping!");
```

[image]

pr8x, 2023-03-01:
Ah, I see. Thanks.

closed 2023-03-01 by pr8x