lvgl/lvgl #4019: [Unit tests] Running a single test file

C47D, 2023-02-26:
### Introduce the problem
Lately when running the test suite the last test execution takes a long time and I had to Ctrl-C it.

[image]

You can see the 40th test is missing.

At work (and for personal projects) I'm used to run only the test file I'm working on, and do nightly runs of all the test files (like LVGL currently do), so I wondered how feasible it would be to run only one test file. I tried to write a Makefile to run only the lv_chart test runner but I found out it has a lot of dependencies on other source files so I gave up.

### Examples and cases
While working in a test file we should not be worried about breaking other test files.

### Suggested solution
Try to find out if it's possible to run single test runners (they are stored in `src/test_runners` and add an option to the main.py file that launches the tests to run a selected test runner, this way hand made Makefiles would not be needed.

This can lead to running test files in less than a second, like this one that took 80ms
[image]

kisvegabor, 2023-02-27:
`test_style` was slow because there was loop in it with 1000 repetitions. I've just updated in, and now it should be faster. 

Anyway,  it's just a workaround and doesn't solve your original question.

@some00 has added some changed to the CMakeFile recently, so I'd like to ask if you a have some time and interest to add this improvement.

kisvegabor, 2023-02-27:
Assigned, thank you!

C47D, 2023-02-27:
Great, thanks in advance @some00 and also thanks @kisvegabor

PR #4023 (2023-03-02): test(main.py): add cli option to select test suite
Fixes: #4019

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Checkpoints
- [ ] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- [ ] If you adde [... cut]
closed 2023-03-02 by kisvegabor