lvgl/lvgl #4024: Conical gradient for lv_arc

kdschlosser, 2023-02-28:
### Introduce the problem
It's not a problem but a discussion about adding a feature.

### Examples and cases
able to apply a gradient to lv_slider. it would be the same thing.

### Suggested solution
I have been looking at the code for LVGL and most of the bits are in place to be able to add a gradient to lv_arc. It would be as simple as calculating a linear gradient between 2 colors using the first colors angle and the second colors angle as the means to get the number of steps to traverse between the 2 colors. if more colors were to be employed  it would be along the lines of...

This is written in Python

```python

# ============= Calculate Gradient =====================
def build_gradient(begin_rgb, end_rgb, nb):

    def _hue2rgb(v1, v2, vH):
        while vH < 0:
            vH += 1
        while vH > 1:
            vH -= 1

        if 6 * vH < 1:
            return v1 + (v2 - v1) * 6 * vH
        if 2 * vH < 1:
            return v2
        if 3 * vH < 2:
            return v1 + (v2 - v1) * ((2.0 / 3) - vH) * 6

        return v1

    def hsl2rgb(h, s, l):
        if s == 0:
            return l, l, l

        if l < 0.5:
            v2 = l * (1.0 + s)
        else:
            v2 = (l + s) - (s * l)

        v1 = 2.0 * l - v2

        r = _hue2rgb(v1, v2, h + (1.0 / 3))
        g = _hue2rgb(v1, v2, h)
        b = _hue2rgb(v1, v2, h - (1.0 / 3))

        return int(r * 255), int(g * 255), int(b * 255)

    def rgb2hsl(r, g, b):
        red = r / 255.0
        green = g / 255.0
        blue = b / 255.0

        vmin = min(red, green, blue)
        vmax = max(red, green, blue)
        diff = vmax - vmin

        vsum = vmin + vmax

        l = vsum / 2

        if diff < 0.0000005:
            return 0.0, 0.0, l

        if l < 0.5:
            s = diff / vsum
        else:
            s = diff / (2.0 - vsum)

        if red == vmax:
            dg = (((vmax - green) / 6) + (diff / 2)) / diff
            db = (((vmax - blue) / 6) + (diff / 2)) / diff
            h = db - dg
        elif green == vmax:
            dr = (((vmax - red) / 6) + (diff / 2)) / diff
            db = (((vmax - blue) / 6) + (diff / 2)) / diff
            h = (1.0 / 3) + dr - db
        elif blue == vmax:
            dr = (((vmax - red) / 6) + (diff / 2)) / diff
            dg = (((vmax - green) / 6) + (diff / 2)) / diff
            h = (2.0 / 3) + dg - dr
        else:
            raise RuntimeError  # sanity check

        if h < 0:
            h += 1
        if h > 1:
            h -= 1

        return h, s, l

    begin_hsl = rgb2hsl(*begin_rgb)
    end_hsl = rgb2hsl(*end_rgb)

    step = tuple(
        [float(end_hsl[i] - begin_hsl[i]) / nb for i in range(0, 3)]
    ) if nb > 0 else (0, 0, 0)

    def mul(stp, value):
        return tuple([v * value for v in stp])

    def add_v(stp, stp2):
        return tuple([v + stp2[i] for i, v in enumerate(stp)])

    def rgb2hex(r, g, b):
        # r = hex(r)[2:].upper()
        # g = hex(g)[2:].upper()
        # b = hex(b)[2:].upper()
        #
        # if len(r) < 2:
        #     r = '0' + r
        # if len(g) < 2:
        #     g = '0' + g
        # if len(b) < 2:
        #     b = '0' + b
        # return '#' + r + g + b
        return r << 16 | g << 8 | b

    for i in range(0, nb + 1):
        yield rgb2hex(*hsl2rgb(*add_v(begin_hsl, mul(step, i))))

# ================== calculate gradient end =================

draw_ctx = lv.draw_ctx_t()
part_draw_dsc = lv.obj_draw_part_dsc_t()
lv.obj_draw_dsc_init(part_draw_dsc, draw_ctx)

center = lv_point_t()
center.x = 240
center.y = 160

arc_r = 100

arc_dsc = lv.draw_arc_dsc_t()
obj.init_draw_arc_dsc(lv.PART.INDICATOR, arc_dsc)

arc_dsc.width = 30
arc_dsc.opa = 255
arc_dsc.rounded = 0

gradient_stops = ((255, 0, 0, 120), (0, 255, 0, 160), (0, 0, 255, 240))
last_stop = gradient_stops [1]

for stop in gradient_stops[1:]:
    start_angle = last_stop[-1]
    stop_angle = stop[-1]
    steps = stop_angle - start_angle

    last_angle = start_angle
    for color in build_gradient(last_stop[:-1], stop[:-1], steps):
        
        arc_dsc.color = lv.color_hex(color)
        arc_dsc.start_angle = last_angle
        arc_dsc.end_angle = last_angle  + 1
        draw_ctx .draw_arc(
            arc_dsc, 
            center, 
            arc_r, 
            start_angle,
            stop_angle
        )
        last_angle += 1
```

This is pseudo code and wouldn't run but it gives the general idea. It's just a matter of calculating the color for each angle and stepping through them using the the already made functions for rendering the arc and only changing the color for each angle increment. This would be something useful for making something like an arc control for controlling a thermostat or an arc control for displaying the current temperature. It would create a conical gradient which is for lack of better words a bent linear gradient. The gradient feature is available for the indicator with lv_slider and it should be also available for lv_arc.

I believe it would be easier to do then what I had thought and probably others have thought as well. IDK if the performance hit for doing this would make it worth doing over using a static image and setting that as the indicator. depending on the size of the arc widget it could be a pretty large memory hit for storing a static image. In my use case the image is 230 x 230 so the hit is 158,700 bytes (RGBA5658) for the one image. While I have enough memory to store the image in both RAM and also FLASH, other users may not and the performance hit might be worth the tradeoff.

IDK if all of the bits and pieces needed to do this are exposed to MicroPython to even try and see if it works. Python would be a whole lot slower to generate the gradient than in C code so I have not bothered trying.

kisvegabor, 2023-02-28:
I'm a little bit worried about the performance. Drawing these little arcs can be really slow as they need to calculate a lot of anti-aliased lines on each start and end angle. 

I can imagine more effective algorithms, e.g. calculate the color of every 8th pixels in each row and simply interpolate between them. The number of skipped pixels can be calculated adaptively too. If the algorithm sees that there is a too large change in the colors it can skip only 4 pixels. Maybe we could also use a look up table or caching to speed up the calculation.

Anyway, we can work on gradients too (our designer also asks them continuously), but I 'd like to add a dedicated algorithms for them. However, once we start to work on them we should consider skew and circular gradients too.

>  In my use case the image is 230 x 230 so the hit is 158,700 bytes (RGBA5658

In this case you can use an RGB565 image too, which is "only" ~100kB. I'm not sure but a dither indexed 8 bit image might work too (~50kB)

kdschlosser, 2023-03-01:
I wasn't sure how LVGL handled masking the arc. So I created an identical arc that has the gradient and I used the alpha channel. Again I am not in the memory deficient department so no concerns for me but I didn't know if it would be worth the tradeoff to render it.

I also only used the above as an example to show the idea. I would suggest a setting gets added so the user is able to set the gradient stops factor. so instead of having a different color for each degree the factor would be a number that is able to divide the angle range cleanly. so is an arc had a start angle of 135 and a stop angle of 405 the factor supplied would have to be able to break 270 into even parts. So 1, 2, 3, 5, 6, 9, 10, 15, 18, 27, 30, 45, 54, 90, 135, 270 would be valid factors for that angle range.

This would allow a user to tailor the widget to match the performance of the MCU being used.

kdschlosser, 2023-03-01:
I am in agreement with your designer that there needs to be support for more than just the axial/linear gradient.

radial
conical
reflected
repeating
diamond
angle

kisvegabor, 2023-03-03:
"reflected" and "repeating" can be built from line, right?

kdschlosser, 2023-03-03:
reflected and repeated is pretty easy to do. It is a matter of copying the gradient, reversing it and attaching it to the back of the original. If you are using a mirrored conical gradient you would not be able to do that...

how a gradient should be handled is there would be a gradient structure that has 4 fields. a pointer to an array of gradient stops, the number of stops in the array, a pointer to an array of colors (the gradient) and a field for the number of colors in the array of colors. 

A gradient stop structure holds 2 elements. the color and then number of steps between the last color in the array and the color in the gradient stop... The user would populate the array of stops and the number of stops then the structure would get passed to a function that would process the stops and it would fill the array of colors and set the number of colors. 

You would then have functions that would take that gradient and apply it to a canvas. so a function for turning a gradient structure into a conical and one for a linear and another for radial.. etc....

You can add additional fields to the gradient structure to define how the gradient should be applied to a canvas. like angle, start and stop angle. these additions would be in the form of a union and depending on what gradient is going to be rendered would depend on what fields get filled in. 

I believe that giving the user the ability to decide to render directly to the frame buffer or to create a new buffer with the gradient would be a good thing to do. The difference between the 2 is rendering to the frame buffer is going to be better on memory but it would be a slower redraw because the processing of the gradient would have to be done each time that area of the screen changes. rendering the gradient to a new buffer would consume more memory but be faster on the rendering to the screen because the processing of the gradient would not need to be done over and over again it would simply  need to be copied to the frame buffer for the redraw.

Little flexibility does go a long way  with something like this. gradients are pretty intense when you get to the rendering aspect of them depending on what is being done. so for a conical gradient it would be better to draw to a new buffer and copy the buffer each time a redraw is needed. A linear is not as CPU intensive to render so you could have it get processed each time and written to the frame buffer to save on memory.

kdschlosser, 2023-03-03:
reflected is a reverse of the gradient appended to the back of the original gradient. A repeat is a gradient appended to the back of its self so it repeats.

This is something that would get handled at the time the gradient is constructed from the color stops that have been specified.
As you know a gradient is just a list of colors and nothing more than that. The rendering of the gradient is the second part of that which is what is going to be wildly different.

There are a plethora of ways to calculate a gradient and the one that is going to fit the bill for a MCU is going to be one that uses as little memory as possible and also not be crazy in the math involved. There will be a happy balance that you will have to mess about with to find what is best. You are going to more than likely end up with different algorithms based on what MCU is being used.

In my example written in Python above it does pretty good in the memory department and speed is pretty decent as well. You have to consider it is running 200 times slower then it would be in C code. The only thing that needs to be updated with it is removing the use of floats in the HSV calculations. In Python an int takes up the same space as a float so it doesn't matter but in C code the memory footprint can be reduced even further by changing it to integer math instead of floating point.

kisvegabor, 2023-03-06:
I have something similar in mind. Let's get back to this when the major architectural changes are done.