lvgl/lvgl #4025: Adding SVG support to LVGL

codewitch-honey-crisis, 2023-02-28:
### I can add SVG support if it's worthwhile, but please consider the caveats

I have code that I can port over to LVGL eventually, but with work. Currently, I'm tooling it into the next revision of htcw_gfx - my graphics library for IoT. Interestingly enough, it is based on the same rasterizer code as the code tiny_ttf is based on.

It's slow. It parses most or all of SVG but renders a subset of SVG, and until/if I can improve on it, it whips the heap with a lot of little allocations.

### Examples and cases

I was originally looking to add basic anti-aliased draws to htcw_gfx. (I'd love a rundown on how LVGL's software drawing hal works - i can't make heads or tails of it) but after I couldn't find enough information to implement what I needed completely I found nanosvg which is a little SVG processor.

When combined with my fledgling widget library, htcw_uix my plan is to make it so you can "skin" user interfaces with SVG. This also lends itself to my web based visual designer, since you can draw using the same code.

### Suggested solution

I've already brought my truetype port over to LVGL, so I figure why not this? 

The reasons why to implement this are fairly obvious, so instead I'll tell you why not:

It's a heck of a lot of effort, some of which I need to take care of before I can even begin adapting it to LVGL.
The rendering is slow, and not incredibly accurate. For all my tests, it's certainly useable but I can't make guarantees.
The parser needs to be completely retooled. It currently requires you to load the XML into an in memory string temporarily. You can't even use PROGMEM because it writes to the string!
It needs perhaps a "precompiled" format to convert SVG to that can load with less fragmentation.
It's going to be very difficult to test extensively.

With all of that in mind, if there's a fair amount of interest in this then I will schedule it for just after I finish my current task of fixing some of the above while being integrated into my library.

kisvegabor, 2023-03-01:
Hi,

I've opened [this issue](https://github.com/lvgl/lvgl/issues/4016) a few days ago about the rendering related ROADMAP of LVGL. As you can see " Vector graphics support" is a dedicated point there. 

I find this topic very important and I'd like to discuss it very thoroughly. The API of the vector graphics "module" is especially important as it will hide not only an SW rasterizer but 2.5D  (e.g. NXP VGLite) or 3D GPUs  too. 

I still can't see the whole picture here. (Maybe SVG parser -> OpenVG like API -> Rasterizer/GPU?) 
As it's really a large work, I wouldn't recommend to get started with it now, becasue it happen that we can't use the result later. 

Let's get back to this in a few month!

closed 2023-03-24 by lvgl-bot

ljunquera, 2023-08-28:
Do we have a target date/version for SVG support?

codewitch-honey-crisis, 2023-08-28:
No because I think Gabor seems to want to implement it on top of some kind of 2d engine, and I'm not really about that. If you want SVG support you can use https://honeythecodewitch.com/gfx

ljunquera, 2023-08-28:
I want to be able to use the "scalable" part of it.

codewitch-honey-crisis, 2023-08-28:
The graphics library I linked to supports scaling.

ljunquera, 2023-08-28:
Thank you! I will take a look at it now. Hopefully I can find a Zephyr example.

codewitch-honey-crisis, 2023-08-28:
https://www.codeproject.com/Articles/5365397/Porting-EspMon-to-ZephyrOS-on-a-STM32-Boards-or-ot

codewitch-honey-crisis, 2023-08-28:
I should let you know - first that code is explained in the EspMon Reboot article linked to by that article. That article is just about making it go on Zephyr.

I should also let you know that overall LVGL is a lot snazzier than my offering. My offering has better support for e-paper and weird pixel formats, but I'm the only contributor so it lacks a lot of features LVGL has like anti-aliased draws, and the massive collection of widgets. I'm preparing documentation for UIX now. However, half of using it using GFX, which is documented [here](https://honeythecodewitch.com/gfx/wiki/index.md) 

It might be a good idea to peruse it to get an idea of what you're getting into. UIX is a relatively thin wrapper that uses GFX for the heavy lifting. GFX is what actually supports SVG, but UIX is the widget (i call them controls) and draw on demand features like LVGL provides. They're just two separate libraries, the latter dependent on the former.

ljunquera, 2023-08-28:
Ok. Let me go through the materials. I see lvgl is used broadly and does have a lot of capabilities. It's just in this one critical area for me where there seems to be a challenge for me. I'm not sure I see SVG support (specifically resizing) in the immediate future where GFX has it today, correct? I can use other smaller formats, like indexed, without resizing.