lvgl/lvgl #4027: Benchmark Result is always 0

GorgonMeducer, 2023-02-28:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://github.com/lvgl/lv_port_an547_cm55_sim/tree/Using-LVGL9)

### Describe the bug
When running the benchmark, the result is always 0.

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2023-03-01:
Thank you for reporting it.

I've just pushed a fix.

GorgonMeducer, 2023-03-01:
This helps a lot !!! @kisvegabor 
Thank you.

GorgonMeducer, 2023-03-01:
I can confirm this issue is fixed.

commit 175a1427 (2023-03-01): fix(benchmark): fix FPS with LV_DEMO_BENCHMARK_MODE_RENDER_AND_DRIVER

fixes #4027