lvgl/lvgl #4029: Substr. related tests failed when `LV_COLOR_DEPTH` is 16

GorgonMeducer, 2023-03-01:
All Substr. related rendering failed when `LV_COLOR_DEPTH` is 16.
The rendering isn't correct.

```c
    {.name = "Substr. rectangle",            .weight = 10, .create_cb = sub_rectangle_cb},
    {.name = "Substr. border",               .weight = 10, .create_cb = sub_border_cb},
    {.name = "Substr. shadow",               .weight = 10, .create_cb = sub_shadow_cb},
    {.name = "Substr. image",                .weight = 10, .create_cb = sub_img_cb},
    {.name = "Substr. line",                 .weight = 10, .create_cb = sub_line_cb},
    {.name = "Substr. arc",                  .weight = 10, .create_cb = sub_arc_cb},
    {.name = "Substr. text",                 .weight = 10, .create_cb = sub_text_cb},
```

### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

GorgonMeducer, 2023-03-01:
When `LV_COLOR_DEPTH` is `32`, everything works well.

kisvegabor, 2023-03-01:
Thank you, fixed!

commit d4964713 (2023-03-01): fix(color): fix lv_color_format_has_alpha

fixes #4029
closed 2023-03-01 by kisvegabor