lvgl/lvgl #4030: Benchmark Result is always 0 in `LV_DEMO_BENCHMARK_MODE_RENDER_ONLY` mode

GorgonMeducer, 2023-03-01:
Similar issue to #4027 

### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce
```c
lv_demo_benchmark(LV_DEMO_BENCHMARK_MODE_RENDER_ONLY);
```

### Expected behavior

### Screenshots or video

kisvegabor, 2023-03-01:
Fixed. I should have tried it out in #4027... :slightly_frowning_face:

closed 2023-03-01 by kisvegabor

GorgonMeducer, 2023-03-01:
Lol.

Thanks.

commit a02256ff (2023-03-01): fix(benchmark): fix FPS with LV_DEMO_BENCHMARK_MODE_RENDER_ONLY

fixes #4030