lvgl/lvgl #4033: lv_dropdown_create() adds 3 default options

vritzka, 2023-03-03:
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When creating a dropdown with **lv_dropdown_create()**, LVGL adds 3 options by default. These are called Option 1, Option 2, Option 3.

We need to call lv_dropdown_clear_options() to remove these, before we can work with the dropdown.

(https://github.com/lvgl/lvgl/blob/e8d8f399e013aa01f74354d106505c5f7f8847b6/src/widgets/lv_dropdown.c#L598)

### To Reproduce
```c
myDropdown = lv_dropdown_create(myScreen);
lv_obj_set_width( myDropdown, 150);
lv_obj_set_height( myDropdown, LV_SIZE_CONTENT);   /// 1
lv_obj_set_x( myDropdown, 20 );
lv_obj_set_y( myDropdown, 20 );
lv_obj_set_align( myDropdown, LV_ALIGN_CENTER );
lv_obj_add_flag( myDropdown, LV_OBJ_FLAG_SCROLL_ON_FOCUS );

//if you look at this dropdown, it will have 3 default options.
```

### Expected behavior
When creating a new dropdown, it should have no options until we add them.

### Screenshots or video

vritzka, 2023-03-03:
[image]

vritzka, 2023-03-03:
[image]

kisvegabor, 2023-03-03:
It is the expected behavior to have something working by default. 

You just need to call `lv_dropdown_set_options` or `lv_dropdown_set_options_static` to set the options you need. It will overwrite the default options.

vritzka, 2023-03-03:
Ok, understood.  
I did lv_dropdown_add_option(), and it didn't clear from this one.
Thanks

closed 2023-03-03 by vritzka