lvgl/lvgl #4035: Dereferencing type-punned pointer will break strict-aliasing rules

amirgon, 2023-03-03:
### Describe the bug

ESP32 reports `dereferencing type-punned pointer will break strict-aliasing rules` on `lv_color.h`:

```
[11/18] Building C object esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/src/misc/lv_color.c.obj
In file included from /home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/lv_color.c:9:
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/lv_color.h: In function 'lv_color_mix':
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/lv_color.h:590:42: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     uint32_t bg = (uint32_t)((uint32_t)(*(uint16_t *)&c2) | ((uint32_t)(*(uint16_t *)&c2) << 16)) &
                                          ^~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/lv_color.h:590:74: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     uint32_t bg = (uint32_t)((uint32_t)(*(uint16_t *)&c2) | ((uint32_t)(*(uint16_t *)&c2) << 16)) &
                                                                          ^~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/lv_color.h:592:42: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     uint32_t fg = (uint32_t)((uint32_t)(*(uint16_t *)&c1) | ((uint32_t)(*(uint16_t *)&c1) << 16)) & 0x7E0F81F;
                                          ^~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/lv_color.h:592:74: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     uint32_t fg = (uint32_t)((uint32_t)(*(uint16_t *)&c1) | ((uint32_t)(*(uint16_t *)&c1) << 16)) & 0x7E0F81F;
                                                                          ^~~~~~~~~~~~~~~
[12/18] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/__/lv_mp.c.obj
In file included from /home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/../draw/lv_img_buf.h:17,
                 from /home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/../draw/lv_img_decoder.h:19,
                 from /home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/../draw/lv_img_cache.h:16,
                 from /home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/lv_gc.h:22,
                 from ../mpconfigport.h:148,
                 from /home/amirgon/projects/lv_micropython/py/mpconfig.h:62,
                 from /home/amirgon/projects/lv_micropython/py/obj.h:31,
                 from lv_mp.c:20:
```

### To Reproduce

- Install esp-idf 4.4 (probably will happen with any other version as well)
- run `. ./export.sh` on esp-idf dir to setup esp-idf
- Build lv_micropython esp32 port with this command: `make -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=16" BOARD=GENERIC_SPIRAM`

This problem only happens when setting `LV_COLOR_DEPTH`

kisvegabor, 2023-03-06:
Thanks, fixed.

Our CI reported many error like this, and I've fixed all of them when developed the new color format support, but interestingly this one wasn't reported.

amirgon, 2023-03-06:
> but interestingly this one wasn't reported.

Does your CI build also with `LV_COLOR_DEPTH=16` macro?

kisvegabor, 2023-03-06:
It [does](https://github.com/lvgl/lvgl/actions/runs/4343148324/jobs/7584831184) but it was reported somehow.

amirgon, 2023-03-06:
I'm still getting this after the fix:

```
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/../draw/../misc/lv_color.h: In function 'lv_color_mix':
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/../draw/../misc/lv_color.h:589:23: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     uint16_t c1_16 = *(uint16_t *)&c1;
                       ^~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/misc/../draw/../misc/lv_color.h:590:23: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     uint16_t c2_16 = *(uint16_t *)&c2;
                       ^~~~~~~~~~~~~~~
```

kisvegabor, 2023-03-08:
Could you try this?
```c
LV_ATTRIBUTE_FAST_MEM static inline lv_color_t lv_color_mix(lv_color_t c1, lv_color_t c2, uint8_t mix)
{
    lv_color_t ret;

#if LV_COLOR_DEPTH == 16 && LV_COLOR_MIX_ROUND_OFS == 0
    uint8_t * c1_8p = (uint8_t *)&c1;
    uint8_t * c2_8p = (uint8_t *)&c2;
    uint16_t c1_16 = *(uint16_t *)c1_8p;
    uint16_t c2_16 = *(uint16_t *)c2_8p;

    /*Source: https://stackoverflow.com/a/50012418/1999969*/
    mix = (uint32_t)((uint32_t)mix + 4) >> 3;

    /*0x7E0F81F = 0b00000111111000001111100000011111*/
    ...    
```

amirgon, 2023-03-09:
> Could you try this?

Looks like it solves the issue :+1:

The same problem appears in a number of other places:

```c
267/426] Building C object esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/src/draw/sw/lv_draw_sw_blend.c.obj
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/draw/sw/lv_draw_sw_blend.c: In function 'set_px_argb':
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/draw/sw/lv_draw_sw_blend.c:320:17: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     buf[0] = (*((uint16_t *) &res_color)) & 0xff;
                ~^~~~~~~~~~~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/draw/sw/lv_draw_sw_blend.c:321:17: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     buf[1] = (*((uint16_t *) &res_color)) >> 8;
                ~^~~~~~~~~~~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/draw/sw/lv_draw_sw_blend.c: In function 'set_px_argb_blend':
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/draw/sw/lv_draw_sw_blend.c:366:17: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     buf[0] = (*((uint16_t *) &last_res_color)) & 0xff;
                ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/draw/sw/lv_draw_sw_blend.c:367:17: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     buf[1] = (*((uint16_t *) &last_res_color)) >> 8;
                ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[346/426] Building C object esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/src/libs/gif/gifdec.c.obj
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/libs/gif/gifdec.c: In function 'gif_open':
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/libs/gif/gifdec.c:149:37: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
         gif->canvas[i * 3 + 0] = (*((uint16_t *) &c)) >> 8;
                                    ~^~~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/libs/gif/gifdec.c:150:37: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
         gif->canvas[i * 3 + 1] = (*((uint16_t *) &c)) & 0xff;
                                    ~^~~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/libs/gif/gifdec.c: In function 'render_frame_rect':
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/libs/gif/gifdec.c:513:46: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
                 buffer[(i + k) * 3 + 0] = (*((uint16_t *) &c)) & 0xff;
                                             ~^~~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/libs/gif/gifdec.c:514:46: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
                 buffer[(i + k) * 3 + 1] = (*((uint16_t *) &c)) >> 8;
                                             ~^~~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/libs/gif/gifdec.c: In function 'dispose':
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/libs/gif/gifdec.c:549:55: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
                     gif->canvas[(i + k) * 3 + 0] = (*((uint16_t *) &c)) & 0xff;
                                                      ~^~~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/libs/gif/gifdec.c:550:55: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
                     gif->canvas[(i + k) * 3 + 1] = (*((uint16_t *) &c)) >> 8;
                                                      ~^~~~~~~~~~~~~~~~
[347/426] Building C object esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/src/libs/png/lv_png.c.obj
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/libs/png/lv_png.c: In function 'convert_color_depth':
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/libs/png/lv_png.c:257:29: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
         img[i * 3 + 1] = (*((uint16_t *) &c)) >> 8;
                            ~^~~~~~~~~~~~~~~~
/home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/src/libs/png/lv_png.c:258:29: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
         img[i * 3 + 0] = (*((uint16_t *) &c)) & 0xFF;
```

amirgon, 2023-03-10:
> The same problem appears in a number of other places:

@kisvegabor Could you reopen the issue until all these places are fixed?

kisvegabor, 2023-03-11:
I've reopened it, and can fix it on Monday.

amirgon, 2023-03-15:
Related discussion:  
https://forum.lvgl.io/t/since-the-lvgl-api-has-changed-micropython-builds-for-rp2-port-color-depth-16-build-error/11190/5?u=amirgon

kisvegabor, 2023-03-16:
Sorry for the delay. I've just pushed a fix. How does it work?

ngc6589, 2023-03-16:
After modifying lv_color.h, the rp2 port build is complete. The modified parts are the following two lines.

`masahiro@U2204LTS:~/lv_micropython/lib/lv_bindings/lvgl/src/misc$ git diff
diff --git a/src/misc/lv_color.h b/src/misc/lv_color.h
index 66dad68c1..d5da00528 100644
--- a/src/misc/lv_color.h
+++ b/src/misc/lv_color.h
@@ -586,8 +586,10 @@ LV_ATTRIBUTE_FAST_MEM static inline lv_color_t lv_color_mix(lv_color_t c1, lv_co

 #if LV_COLOR_DEPTH == 16 && LV_COLOR_MIX_ROUND_OFS == 0
     /*Source: https://stackoverflow.com/a/50012418/1999969*/
-    uint16_t c1_16 = *(uint16_t *)&c1;
-    uint16_t c2_16 = *(uint16_t *)&c2;
+    /* uint16_t c1_16 = *(uint16_t *)&c1; */
+    uint16_t c1_16 = lv_color_to_int(c1);
+    /* uint16_t c2_16 = *(uint16_t *)&c2; */
+    uint16_t c2_16 = lv_color_to_int(c2);
     mix = (uint32_t)((uint32_t)mix + 4) >> 3;

     /*0x7E0F81F = 0b00000111111000001111100000011111*/`

Unrelated to this issue, but there is a fix in the API, so my application will error with sliders and buttons.
I was able to see the LCD screen.

[image]

amirgon, 2023-03-16:
On ESP32 the problem is gone now. :+1: 
I've updated lv_micropython with latest LVGL.

closed 2023-03-16 by amirgon