lvgl/lvgl #4039: reversing the sidrection of lv_arc and lv_slider

kdschlosser, 2023-03-06:
### Introduce the problem
currently with the user has the ability to alter the way the slider gets rendered with lv_slider the change would be horizontal or vertical and that is controlled by the height being a larger value than the width and with lv_arc by changing the start and the stop angles.

What there is no control over is the direction in which the slider gets actuated. with lv_slider if I wanted the slider to move from right to left or top to bottom. or with lv_arc have it move counter clockwise.

### Examples and cases
purely aesthetic purposes

[image]

in the above image look at the arc furthest to the right. The indicator moves in a clockwise fashion. Aesthetically it would look better if the indicator moved counter clockwise. To balance things it would also look better if the inner most arc also moved counter clockwise as well.

### Suggested solution
Using the set_min and set_max to determine the direction the slider should actuate in. If the user supplies a maximum value that is lower than the minimum value that would indicate that the user wants the direction of the slider to be reversed. so if I supply lv_slider_set_range(50, -50) the slider would then move from right to left or top to bottom.

kdschlosser, 2023-03-08:
I had thought the RTL and LTR applies to text and not to how widgets are rendered. The reason I say this is on every operating system or GUI if a font is used that is "RTL" the GUI doesn't display RTL only the text does. The natural thing of looking at the examples above and below the one you linked to shows text as being part of the bar in the example below the one you linked to. This made me think that the "LTR" and "RTL" would apply to the text and not the actual bar. IDK why it didn't register that it was for the purposes the bar/slider actually being right to left or left to right. 

It would seem more logical to have the directions reversed if the min and max values are reversed The same could also be applied to the arc instead of the angles. Currently the widgets flips the supplied min and max values if the max value is lower than the min value and it does nothing else. Pretty easy since the mechanics are already in place and the check is already being done for the min and max values to add the function call to it. 

You also get the added benefit of having a consistent API between lv_slider and lv_arc instead of 2 different mechanisms that do the same thing for all intents and purposes.

It makes more sense having it that way as there would be no other reason to supply a max value that is lower than the min value unless the bar is to be rendered backwards.

It also falls in line with the horizontal and vertical API for lv_slider. the orientation gets automatically changed if the widget is taller than it is wide and nothing additional needs to be done.

Having the API be consistent is a good thing to have. conceptually the arc rendering clockwise or counter clockwise is the same thing as the slider rendering right to left or left to right. after all the arc is a slider it is bent instead of straight so the API should be close to the same.

Maybe it would be better to only have a slider widget and add in a function for specifying the start and stop angles and that would turn the slider into an arc. The greater of width and height is would would be used for the size.

From a user standpoint this would simplify things.

kdschlosser, 2023-03-08:
Oh the other thing is... RTL and LTR does not work if the var is vertical..

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/0107879924866ad5547ccfcf1102c154590c7e14/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/0107879924866ad5547ccfcf1102c154590c7e14/examples/widgets/bar/lv_example_bar_5.py&script_direct=a6597a59f77918ec8c07655f683c5fda38cd132f

kisvegabor, 2023-03-10:
> It would seem more logical to have the directions reversed if the min and max values are reversed
> Oh the other thing is... RTL and LTR does not work if lv_slider is vertical..

That's correct. However, @amirgon pointed out that bars and switches also feel more natural if the they are growing from the right in RTL context.

So I think what we need is:
- LTR
  - min < max: :arrow_right: 
  - max < min :arrow_left:
- RTL
  - min < max: :arrow_left: 
  - max < min :arrow_right:

kdschlosser, 2023-03-10:
I do not know how to read or write any languages that are right to left. so I don't know if it is just the language that is right to left or if everything in the GUI gets reversed. I don't believe the GUI elements get reversed when a right to left language is used. When I change the language in Windows to a right to left language the controls do not get reversed only the text does. Also I believe that numbers in a right to left language are read and displayed as left to right. Since the slider and also the arc operate on changing numerical values then it's orientation would not change based on a language being right to left...

Another thing is, what about languages that are vertical? the orientation of a control does not change if a vertical language is used does it?

If you wanted to have the controls change direction based on what the language is then what you have outlined would be the idea of how it would work. 

My suggestion is to stick with how traditional operating system GUIs work with respect to this. That is the typical behavior that is to be expected when creating GUI elements. 

Outside of that I do not believe I am going to be able to offer any great insight on how it should be implemented. LOL..

amirgon, 2023-03-10:
> I don't believe the GUI elements get reversed when a right to left language is used.

They definitely do.

For example:  
Microsoft Word in English layout:

[image]

Microsoft Word in Hebrew layout, notice the "File" (קובץ) starts from the right instead:

[image]

kdschlosser, 2023-03-10:
ok yeah.. that's text. does the slider widget in Windows reverse it's operation because Hebrew text is being used? I do not believe it does. I believe it still moves from left to right as it does with a left to right language.

kdschlosser, 2023-03-10:
The menus may reverse and that would make sense because they are reading from right to left so the orientation of the menus would need to be in the same order in which it is for a left to right language. so menu#1 is the File menu and to maintain Menu #1 being the file menu in the Hebrew language it would need to be on the right because that is the order in which the language is read. 

placement changes but operation I do not believe does. It might, I have not thought to specifically check this aspect.

The funny thing is languages like Japanese and Cherokee which are supported in Windows are read top to bottom moving left to right. The GUI does not rotate 90° clockwise for those languages..

I have to tell you that I have spent a lot of time working with the languages API in Windows. It is a huge mess to deal with and I am sure it is the same for any operating system. I say make it however you want it to operate keeping a consistent API across widgets. If someone speaks up and says it is wrong then it can be changed at that time.

The goal here is consistent API between widgets that have the ability to operate in a different manner. In the case of the sliders (linear or arc) making the change in operation work in reverse should be the same. when it be both of those widgets has a "reverse" function that can be called or it is handled internally when the min and max values get set makes no difference. I believe that using the min and max values as the way to determine the direction of operation is a geed mechanism to use and it also fall into alignment with making a slider vertical which is also handled automatically based on the dimensions.

Whatever is easiest for you to do when handling RTL. If it is easier to have the widget flipped if someone sets up the GUI for RTL then do that. If the user doesn't like it they can always flip it a round by calling the set_range function with the min and max values reversed.

amirgon, 2023-03-10:
> does the slider widget in Windows reverse it's operation because Hebrew text is being used?

When the entire layout is aligned to the right, the slider or progress bar is also aligned to the right.  
In such case the "max" value should be on the left and "min" on the right, otherwise it looks wrong.
There can be some exceptions, such as media player controls, but in general it's true for all widgets.

For example:

[image]

kdschlosser, 2023-03-10:
The only place I can see an issue is with a vertical slider. If the original operation of the slider is to be able to move it up then setting RTL will cause it to move down and that will not be correct behavior. This is something that will have to be accounted for. The same kind of an issue exists with lv_arc as well. If I have an arc that has a start and stop that lands in the range of 90 to 270 or in the range of 270 to 450 then reversing the direction is not a behavior that would be wanted. But again it is also easily fixed in the user code by simply reversing the min and max values.

Just make sure there is documentation that addresses how to change the direction the widget needs to be actuated in

kdschlosser, 2023-03-10:
There is no special specification that is followed but everyone that writes GUI code when dealing with this kind of thing. It is all up to the decision of the author. This is why you make the big bucks sir!!!! a decision you and you alone are going to have to make.
😉

kdschlosser, 2023-03-10:
Personally if RTL text is being used that is the only thing that would needs to have automatic handling. You need to provide a mechanism for each widget that allow the widget to be mirrored horizontally and that would have to be used by the user. The user would be responsible of placement of said widget based on the language being used.

I have never thought about it until now but I do have to ask this question....

you have a display that is 480 x 320. in LTR the 0, 0 coords are in the upper left of that display. in RTL they should be in the upper right and the numbers increase moving across the display to the left. In LTR the numbers would decrease if moving left... 

Does this hold true? with LTR vs RTL. If it does then the handling of RTL vs LTR becomes a simple thing to do and it gets done in the frame buffer with a simple flip flopping of the x axis

kdschlosser, 2023-03-10:
Here is a good read on the subject..

https://blog.prototypr.io/dont-be-backwards-2573b117c5d9

In it it states that things like progress bars do not change but input controls do. LOL.  Basically that puts you up shits creek without a paddle., There is nothing that you are going to be able to do to fully automate the reversal of the GUI for a user because ultimately you have no idea what the user is intending the control to be used for.

I think the easiest thing for you to do is to flip the X axis in it's entirety leaving only the text and images as they have been provided. If an image is intended to be display the other way around then it would  be up to the user to do that.

I am like you when it comes to programming. I strive to provide the most seamless way of doing something so the user doesn't have to deal with it. Unfortunately with RTL vs LTR there is no magic code that can be written to make the change seamless. You can do as much as possible and the user will be responsible for the rest. You also don't want to bloat you code up dealing with this because the number of RTL languages is not that large and some of them like Japanese is able to be understood LTR or RTL and those people have adapted to using an LTR GUI displaying only the text RTL and in some cases the text is even displayed as LTR.

kdschlosser, 2023-03-10:
And in this blurb is states the exact opposite regarding progress indicators. It states to flip them.

https://developer.apple.com/design/human-interface-guidelines/foundations/right-to-left/

kisvegabor, 2023-03-13:
If we couldn't decide something usually we took web as reference. On web all layouts, progress bars, etc are swapped in case of RTL.

See this for example: https://stackoverflow.com/questions/39439032/how-to-render-a-progress-element-from-right-to-left

kisvegabor, 2023-12-02:
To swap direction when min >max is aleady working for slider and bar. For arc the reverse mode can be used.

With these I think the original issue is resolved,, so close it now.

closed 2023-12-02 by kisvegabor