lvgl/lvgl #4040: Discolouration around transparent objects

windorey, 2023-03-06:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Discolouration around transparent objects

### To Reproduce
```c
    ui_Button1 = lv_btn_create(ui_callnum_top_nav);
    lv_obj_set_width(ui_Button1, 52);
    lv_obj_set_height(ui_Button1, 20);
    lv_obj_set_x(ui_Button1, -10);
    lv_obj_set_y(ui_Button1, 0);
    lv_obj_set_align(ui_Button1, LV_ALIGN_LEFT_MID);
    lv_obj_add_flag(ui_Button1, LV_OBJ_FLAG_SCROLL_ON_FOCUS); 
    lv_obj_clear_flag(ui_Button1, LV_OBJ_FLAG_SCROLLABLE); 
    lv_obj_set_style_bg_color(ui_Button1, lv_color_hex(0x000000), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_Button1, 64, LV_PART_MAIN | LV_STATE_DEFAULT);
```

### Expected behavior
Uniform background colour of the buttons and objects

### Screenshots or video
It doesn't show up very well on camera, but in person, it's very easy to spot.
[image]
Notice the different colour surrounding the border, when it should be uniform

[image]
Digitally enhanced image to make the issue more visible.

https://user-images.githubusercontent.com/65610123/223226934-4fc3a93e-256f-46ca-bf7b-5c7b560e0bca.mov
Horizontal bar-like shape running through this button and gets taller when pressed.

This UI is exported from SquareLine Studio, there, there is no problem.

windorey, 2023-03-08:
Hi, I am already using the latest release. Here is my PlatformIO settings file:
```
lib_deps = 
        (...)
	lvgl=https://github.com/lvgl/lvgl/archive/refs/tags/v8.3.5.zip
        (...)
```
The colour depth is 16. My MCU does have a GPU, but I don't use it.

windorey, 2023-03-08:
Just found out you are talking about something else, I will try it and let you know the result. I'm so sorry.

windorey, 2023-03-08:
Tried out using `https://github.com/lvgl/lvgl/archive/refs/heads/release/v8.3.zip` as the library dependency, it does not help. The issue still remains.

kisvegabor, 2023-03-08:
v8.3.5 should be fine.

I've trued to reproduce it wit this code:
```c
  lv_obj_set_style_bg_color(lv_scr_act(), lv_color_hex(0x555555), LV_PART_MAIN | LV_STATE_DEFAULT);

  lv_obj_t * ui_Button1 = lv_btn_create(lv_scr_act());
  lv_obj_set_width(ui_Button1, 52);
  lv_obj_set_height(ui_Button1, 20);
  lv_obj_set_x(ui_Button1, -10);
  lv_obj_set_y(ui_Button1, 0);
  lv_obj_set_align(ui_Button1, LV_ALIGN_LEFT_MID);
  lv_obj_add_flag(ui_Button1, LV_OBJ_FLAG_SCROLL_ON_FOCUS);
  lv_obj_clear_flag(ui_Button1, LV_OBJ_FLAG_SCROLLABLE);
  lv_obj_set_style_bg_color(ui_Button1, lv_color_hex(0x000000), LV_PART_MAIN | LV_STATE_DEFAULT);
  lv_obj_set_style_bg_opa(ui_Button1, 64, LV_PART_MAIN | LV_STATE_DEFAULT);
  lv_obj_set_style_radius(ui_Button1, 20, LV_PART_MAIN | LV_STATE_DEFAULT);
  lv_obj_set_size(ui_Button1, 300, 100);
```

but can't see the issue.

Could you add a full code that reproduce the issue?

Other question: What is the value of `LV_COLOR_MIX_ROUND_OFS` in `lv_conf.h`. Please try wit 0, 128, and 254. Does it make any difference?

windorey, 2023-03-08:
Changing `LV_COLOR_MIX_ROUND_OFS` does indeed seem to fix it.
I'm so sorry to bother you, but I don't know where to ask, can you give a simple example how to define keymap for `LV_KEYBOARD_MODE_USER_1`? I cant find info anywhere for it.

windorey, 2023-03-08:
Figured it out on my own. Thanks.

closed 2023-03-08 by windorey

kisvegabor, 2023-03-08:
> Changing LV_COLOR_MIX_ROUND_OFS does indeed seem to fix it.

What was the original value (which showed the issue) and what is the new value (which solved it)?

> I'm so sorry to bother you, but I don't know where to ask, can you give a simple example how to define keymap for LV_KEYBOARD_MODE_USER_1? I cant find info anywhere for it.

You can take a look at how other maps are implemented [here](https://github.com/lvgl/lvgl/blob/release/v8.3/src/extra/widgets/keyboard/lv_keyboard.c#L49-L60).

You can also take a look at the docs [here](https://docs.lvgl.io/8.3/widgets/extra/keyboard.html#new-keymap).

windorey, 2023-03-08:
Original value was the default from LV_CONF_INCLUDE_SIMPLE.

windorey, 2023-03-08:
New value is 128 or 254, both fixed it.