lvgl/lvgl #4041: glitches using stm32 dma2d for lvgl v8.3

HaoboGu, 2023-03-07:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

recently I updated lvgl version from v8.2 to v8.3, but the display has some glitches if stm32 dma2d is enabled. I used dma2d in version v8.2, all things are good. Disable dma2d in v8.3, the display is also good, so I suspect that the problem is introduced by stm32 dma2d updates in v8.3. 

[image]

### Expected behavior

### Screenshots or video

kisvegabor, 2023-03-07:
@tdjastrzebski Could you take a look at it?

tdjastrzebski, 2023-03-08:
I am afraid without source code I can not even try to make a guess.

HaoboGu, 2023-03-10:
I'm using FMC to send data to LCD using 8080 interface. In this case, DMA2D is used only for processing frames, am i right? The only difference I see is that I use FMC to send data to LCD, not DMA2D.

tdjastrzebski, 2023-03-11:
I have no way of knowing.
If you do NOT use DMA(2D) to move data from draw buffer e.g. to 8080 try invalidating draw buffer first (call SCB_InvalidateDCache_by_Addr()) to avoid reading dirty cache bytes, make sure both draw buffer length and start address are 32 byte (byte, not bit) aligned. See [this line](https://github.com/tdjastrzebski/STM32F769I-DISCO-LVGL/blob/c5aa36a3de62e464c07bce401cbcb5f3ac5ff3e3/master.cpp#L33)

HaoboGu, 2023-03-13:
CleanDCache worked, but only by calling `SCB_CleanInvalidateDCache_by_Addr((uint32_t *)buf, buffer_length);` before and after flushing. Here is my `disp_flush`:

```c
static void disp_flush(lv_disp_drv_t *disp_drv, const lv_area_t *area, lv_color_t *color_p) {
#if LV_USE_GPU_STM32_DMA2D
  // Clean and invalidate dcache before and after display flushing
  SCB_CleanInvalidateDCache_by_Addr((uint32_t *)buf_1, buffer_length);
#endif

  // Write display data using fmc
  lcd_fill_data_fmc(area->x1, area->x2, area->y1, area->y2, color_p);

#if LV_USE_GPU_STM32_DMA2D
  SCB_CleanInvalidateDCache_by_Addr((uint32_t *)buf_1, buffer_length);
#endif

  lv_disp_flush_ready(disp_drv);
}
```

tdjastrzebski, 2023-03-13:
That is interesting. I would say Invalidating alone an only before the transmitting should be sufficient abut anyway, I glad it works.
Do not forget to align the draw buffer, otherwise you may experience random crashes as you may clean invalidate bytes which should not be touched.

HaoboGu, 2023-03-13:
Adding `SCB_InvalidateDCache_by_Addr((uint32_t *)buf, buffer_length);` fixes almost all cases except scenes having this picture when running lvgl benchmark: 
[image]

The interesting thing is, only the picture is affected, the FPS monitor does not:

[image]

HaoboGu, 2023-03-13:
I guess `SCB_InvalidateDCache_by_Addr((uint32_t *)buf, buffer_length);` should be also added to somewhere else when rendering images? Is there a separate buffer for images in lvgl? @kisvegabor

tdjastrzebski, 2023-03-13:
Yeah, it is a bit more complicated since, unfortunately, draw buffer is not always filled-in using DMA2D when it is enabled.
Hopefully this will be some how fixed in rev 9.0  
See my comments [here](https://github.com/lvgl/lvgl/pull/3904#issuecomment-1398193372)

tdjastrzebski, 2023-03-13:
Do you use 16 or 32bit color?

HaoboGu, 2023-03-13:
I use 16bit color, RGB565

tdjastrzebski, 2023-03-13:
That's what I thought. Especially in 16b mode sometimes images are painted without DMA2D when internal image format does not match any of the available DMA2 modes. I may want to take another look to see if anything could be SAFELY improved.

kisvegabor, 2023-03-14:
> Is there a separate buffer for images in lvgl? @kisvegabor

Actually there are two main cases:
- As @tdjastrzebski pointed out, during software rendering ARGB images are splint into an RGB and and alpha buffer
- If there some special effects on the images (e.g. additive blending, `style_transform_...`, not `img_angle` or `img_zoom`) then a widget an all its children are drawn into a separate buffer first and the "snapshot" is rendered as an image.

tdjastrzebski, 2023-03-14:
Regardless of special effects, if at the end bytes are in a buffer and color format can be handled by dma2d, there is no problem.
I believe currently the most problematic case is 16b color with opacity, 24b total (ARGB8565). There is no way to draw it directly with dma2d, unless buffer is first translated to ARGB32 (or ARGB4444) - which is an option. Another case which dma2d cannot handle is subtractions.

tdjastrzebski, 2023-03-15:
@kisvegabor Maybe, the right solution to the above problem would be NOT to use separate 8b bitmap masks and 16b color buffers and instead always use 32b/pixel buffers, even when effective color format is 24b? This way 24b color (argb8565) could be easily converted to argb8888 just before dma2d rendering, without having to copy bytes to a separate buffer, which would involve allocating extra memory and then having to free it.

kisvegabor, 2023-03-17:
@tdjastrzebski 
We are talking about using RGB565 + A8 format in v9. See [this comment](https://github.com/lvgl/lvgl/issues/4059#issuecomment-1469699345). What do you think? To keep the conversation in one place, please comment in the linked issue.

closed 2023-04-09 by lvgl-bot

maxpromer, 2023-10-05:
> ```c
> SCB_CleanInvalidateDCache_by_Addr((uint32_t *)buf, buffer_length);
> ```

What is `buf` and `buffer_length` ? it frame buffer like `color_p` ?