lvgl/lvgl #4043: Linear meter

qpmr, 2023-03-08:
### Introduce the problem

We have the meter widget which only has a round form. There were several proposals to implement linear widget next to the round ([Forum](https://forum.lvgl.io/t/linear-meter-bar-with-ticks/10986)). As a result in Roadmap this issue is: #[10986]https://github.com/lvgl/lvgl/commit/389e32d72f119e42b391833310e3107236a6c1d9)

### Examples and cases

The linear widget can be useful for visualize thermometer, ruler, linear gauge, etc.

### Suggested solution

Based on the current meter widget (v8.3.5) I added the second "mode" - "linear representation" of the meter. This works for all current code that already uses the actual meter API. I just added few additional options: mode (linear, round), scale direction, label/needle swap.
My PR is [now based on  the last v.8.3.5 release](https://github.com/lvgl/lvgl/pull/4036), but I will rebase it to the master in another PR

qpmr, 2023-03-09:
I created another branch based on the master branch([commit](https://github.com/lvgl/lvgl/commit/bdf3d4e66606135e84a49ce47da22df2b66120a7) ). Because this is the last commit on which I can launch the simulator and test the results. [My branch preliminary branch](https://github.com/qpmr/lvgl/tree/feature/linear_meter_master_bdf3d4e666)
The current master commit (lvgl : master : f336ea7cb1b7a70a9a8f6694008b218bb6391726) is unstable, and building for the simulator has failed.
I will wait for the master branch to stabilize before creating a PR

kisvegabor, 2023-03-10:
Thank you for opening this issue.

I think what we really need is a scale widget like these:
[image]

[image]

and their circle form.

This is my idea about how to make a couple widgets more flexible and easier to use by having a scale widget. For bar, slider, arc, meter, and chart we can have function like 
```c
lv_scale_t * lv_<widget_type>_add_scale(obj, <some_settings_if_needed>)
``` 

After that we will have a custom scale widget which can have it's own generic API but it's still bound to the widget that created it. E.g. if the range of the bar changes it can keep track of it's `scale`s and update their range accordingly.

A concrete example for the chart: 
```c
lv_scale_t * scale1 =  lv_chart_add_scale(chart1, LV_CHART_AXIX_PRIMARY_X, fixed_OR_scroll_with_the_chart);

const char * txts[] = {"first", "second", ..., NULL};
lv_scale_set_text_src(scale1, txt);
lv_scale_set_text_rotation(scale1, 450); //45.0 deg
lv_scale_set_major_tick_count(scale1, 10); //AUTO can be a possible value too which sets it to equal to the div. line count 
...
```

The required features I see:
- vertical, horizontal, round forms
- min, max values
- tick settings (major, minor, length, width, label (top, bottom?))
- tick recoloring
- else?

What do you think?

qpmr, 2023-03-12:
Looks good, but it is about scale. What about form (round/vertical/horizontal)?
I think there is are two issues related to this widgets (arc, bar, slider, meter, chart): form(linear/round) and scale.
1) scale: chart + scale = good, meter + scale = good, do we really need scale on the **bar**, **slider** and **arc**?
2) form: **arc** looks like linear **bar**, **slider** is different, linear **meter** similar to round **meter**

kisvegabor, 2023-03-13:
> do we really need scale on the bar, slider and arc?

I can easily imagine e.g. a thermostat with an arc to set the target temperature and a scale around it to with the degrees. Or and arc to adjust pressure with the `bar` units on the scale.

E.g. lv_bar and lv_slider as a thermometer with a scale is also quite standard.

 ---

By having a scale widget (which can be round or linear) we'd have a modular and flexible solution to extend any widget.

qpmr, 2023-03-14:
I get your idea and I will think about it

closed 2023-04-06 by lvgl-bot

kisvegabor, 2023-04-06:
FYI, it's added to the ROADMAP so won't be forgotten.