lvgl/lvgl #4044: LVGL is stuck in endless loop during partial redraw when lv_disp_flush_ready() is called outside of flush_cb()

cocasema, 2023-03-08:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
While updating my working code to switch to the asynchronous drawing mode I ended up with LVGL being stuck forever in a `while` loop (this is not related to any specific display driver or communication method).

Initially to get a working prototype faster I used blocking SPI write inside the `flush_cb()` and everything worked fine:

`listing 1`
```c++
enum { DISP_BUFFER_SIZE = 240 * 240 };
lv_color_t disp_buffer[DISP_BUFFER_SIZE];

lv_disp_draw_buf_t draw_buf{};

void display_flush(lv_disp_drv_t* disp, lv_area_t const* area, lv_color_t* color_p)
{
  auto width  = area->x2 - area->x1 + 1;
  auto height = area->y2 - area->y1 + 1;
  display.driver.update(area->x1, area->y1, width, height, color_p);
  lv_disp_flush_ready(disp);
}

void init()
{
  ...
  lv_init();

  lv_disp_draw_buf_init(&draw_buf, disp_buffer, nullptr, DISP_BUFFER_SIZE);

  lv_disp_drv_init(&disp_drv);
  disp_drv.hor_res  = 240;
  disp_drv.ver_res  = 240;
  disp_drv.draw_buf = &draw_buf;
  disp_drv.flush_cb = display_flush;
  lv_disp_drv_register(&disp_drv);
  ...
}

void loop()  {
  ...
  if (gui_update_timer.execute()) {
     lv_task_handler();
  }
  ...
}
```
Note: I defined `LV_TICK_CUSTOM & LV_TICK_CUSTOM_SYS_TIME_EXPR` in my `lv_conf.h` and do not call `lv_tick_inc` manually.

Then I tried to move  `lv_disp_flush_ready` out of `display_flush` callback to utilize DMA, and got LVGL stuck after a few redraws in `lv_task_handler` call. After spending some time debugging and to exclude potential DMA related issues, I got back to the original non-DMA version, and modified it to behave more like the asynchronous:

`listing 2`
```c++
enum { DISP_BUFFER_SIZE = 240 * 240 };
lv_color_t disp_buffer[DISP_BUFFER_SIZE];

lv_disp_draw_buf_t draw_buf{};

lv_disp_drv_t* disp_drv = nullptr;  // <<<<<< added

void display_flush(lv_disp_drv_t* disp, lv_area_t const* area, lv_color_t* color_p)
{
  auto width  = area->x2 - area->x1 + 1;
  auto height = area->y2 - area->y1 + 1;
  display.driver.update(area->x1, area->y1, width, height, color_p);
  //  lv_disp_flush_ready(disp);   <<<<<< commented out
  disp_drv = disp;  // <<<<<< added
[... 205 lines cut ...]
     * In full double buffered mode wait here while the buffers are swapped and a buffer becomes available*/
    bool full_sized = draw_buf->size == (uint32_t)disp_refr->driver->hor_res * disp_refr->driver->ver_res;
    if((draw_buf->buf1 && !draw_buf->buf2) ||
       (draw_buf->buf1 && draw_buf->buf2 && full_sized)) {
        while(draw_buf->flushing) {
            printf("%i: Waiting for flush to complete\r\n", __LINE__);   // <<<<<<<<<<<<<<<<<<<<<<<< added this line
            if(disp_refr->driver->wait_cb) disp_refr->driver->wait_cb(disp_refr->driver);
        }

        /*If the screen is transparent initialize it when the flushing is ready*/
#if LV_COLOR_SCREEN_TRANSP
        if(disp_refr->driver->screen_transp) {
            if(disp_refr->driver->clear_cb) {
                disp_refr->driver->clear_cb(disp_refr->driver, disp_refr->driver->draw_buf->buf_act, disp_refr->driver->draw_buf->size);
            }
            else {
                lv_memset_00(disp_refr->driver->draw_buf->buf_act, disp_refr->driver->draw_buf->size * LV_IMG_PX_SIZE_ALPHA_BYTE);
            }
        }
#endif
    }
```

### Screenshots or video
N/A

kisvegabor, 2023-03-09:
Your approach(es) seems correct but I doubt that it's an LVGL issue as probably it would have been already reported by many other people too. 

Please try adding logs to these places:
1. beginning of `display_flush`
2. end of `display_flush`
3. before `lv_disp_flush_ready`

cocasema, 2023-03-09:
I created MVP with repro here: https://github.com/cocasema/mvp-lvgl-4044/commit/41c49f4d3001aae8a3f68a140b848c2f3b0eb4d4
It's on a branch created from v8.3.5 label, with 2 commits, where the first one is simply `cp lv_conf_template.h lv_conf.h` to make it easier to see diff later.

It's buildable and runnable on Linux with just one `make` command. Different CASEs can be run as `make CASE=0` etc.

Could you please check if I'm doing something stupid there? :)

```c++
#ifndef CASE
#define CASE 0
#endif

#if CASE == 0
enum { DISP_BUFFER_SIZE = 240 * 240 };
lv_color_t disp_buffer[DISP_BUFFER_SIZE];
#elif CASE == 1 || CASE == 2
enum { DISP_BUFFER_SIZE = 240 * 60 };
lv_color_t disp_buffer[DISP_BUFFER_SIZE];
#endif

lv_disp_draw_buf_t draw_buf{};

lv_disp_drv_t disp_drv{};

lv_disp_drv_t *flush_disp_drv = nullptr;

void display_flush(lv_disp_drv_t *disp, lv_area_t const *, lv_color_t *) {
  printf("display_flush {\n");
#if CASE == 2
  lv_disp_flush_ready(disp);
#else
  // do nothing
  flush_disp_drv = disp;
#endif
  printf("} display_flush\n");
}

void init() {
  lv_init();

  lv_disp_draw_buf_init(&draw_buf, disp_buffer, nullptr, DISP_BUFFER_SIZE);

  lv_disp_drv_init(&disp_drv);
  disp_drv.hor_res = 240;
  disp_drv.ver_res = 240;
  disp_drv.draw_buf = &draw_buf;
  disp_drv.flush_cb = display_flush;
  lv_disp_drv_register(&disp_drv);
}

Timer gui_update_timer(500);

void update() {
  printf("\n--- %u ---\n", millis());
  printf("update {\n");

  if (flush_disp_drv) {
    printf("lv_disp_flush_ready {\n");
    lv_disp_flush_ready(flush_disp_drv);
    flush_disp_drv = nullptr;
    printf("} lv_disp_flush_ready\n");
  }

  if (gui_update_timer.execute()) {
    printf("lv_task_handler {\n");
    lv_task_handler();
    printf("} lv_task_handler\n");
  }

  printf("} update\n");
[... 98 lines cut ...]
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
627: while(draw_buf->flushing)
```

kisvegabor, 2023-03-13:
I've tried it out an see the problem with `CASE == 1`.

LVGL waits for `lv_disp_flush_ready` [here](https://github.com/lvgl/lvgl/blob/9a870b34a9e8a4b55c7d58b69ba3f1eccca1305b/src/core/lv_refr.c#L625).

So it spins in the a `while` loop until `lv_disp_flush_ready` is called. Therefore the code outside of `lv_task_handler` won't be executed.

So you should call `lv_disp_flush_ready` from the `flush_cb` or from an interrupt.

cocasema, 2023-03-22:
Thanks for your update.