lvgl/lvgl #4045: get_content_width and get_content_height returns zero

kdschlosser, 2023-03-10:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I am using lv_label and I am setting the text and using the default font. I set the size to (LV_SIZE_CONTENT, LV_SIZE_CONTENT) and then call get_content_width and get_content_height and they both return zero. get_width and get_height also return zero. If I set up a timer and in the callback for the timer I call those functions then they will return a value. I am not sure why the widget has to be rendered in order for it to return the size. I did try get_self_height and get_self_width and those do return a value without the widget needing to be rendered first. I am not sure what the purpose of get_self_width and get_self_height is and if the returned values could be considered concrete.

### To Reproduce
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/28d4bff4408a282304181a09e787a3d462956737/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/28d4bff4408a282304181a09e787a3d462956737/examples/get_started/lv_example_get_started_1.py&script_direct=8d566656f877654e5981115aeed17f03496c207d

### Expected behavior
for get_content_width and get_content_height to return a non zero value

kisvegabor, 2023-03-11:
Hi,

Take look at this: https://docs.lvgl.io/master/overview/coords.html#postponed-coordinate-calculation

kdschlosser, 2023-03-11:
OK I'll buy that.

But what is the purpose for `get_self_height` and `get_self_width` and they do return the correct values without needing to recalculate the layout?

kdschlosser, 2023-03-11:
also that little nugget of  information about needing to calculate the layout in order to get the correct size should be in the function documentation. or a link pointing to that page at the very least should be in the function docs.

Oh wait. There is no information about `lv_obj_get_width` or `lv_obj_get_height`. It's like those functions don't even exist. Tho there is documentation in the lv_obj_pos header file. I thik I now know why there are things that are not in the documentation. It is because of the type of the first parameter. For `lv_obj_get_height` the first parameter type is `_lv_obj_t` where as `lv_obj_add_flag` has a first parameter of `lv_obj_t` and that function does show up in the documentation. It's the underscore that is causing so many functions to not have the documentation generated.

kdschlosser, 2023-03-11:
here is a list of all of the functions that use `_lv_obj_t` as the first parameter

<details>
  <summary>Click me</summary>

lv_obj_class_init_obj
lv_obj_is_editable
lv_obj_is_group_def
lv_obj_get_child_id
lv_obj_add_style
lv_obj_replace_style
lv_obj_remove_style
lv_obj_remove_style_all
lv_obj_refresh_style
lv_obj_get_style_prop
lv_obj_set_local_style_prop
lv_obj_set_local_style_prop_meta
lv_obj_get_local_style_prop
lv_obj_remove_local_style_prop
lv_obj_fade_in
lv_obj_fade_out
lv_obj_set_style_pad_all
lv_obj_set_style_pad_hor
lv_obj_set_style_pad_ver
lv_obj_set_style_margin_all
lv_obj_set_style_margin_hor
lv_obj_set_style_margin_ver
lv_obj_set_style_pad_gap
lv_obj_set_style_size
lv_obj_get_style_space_left
lv_obj_get_style_space_right
lv_obj_get_style_space_top
lv_obj_get_style_space_bottom
lv_obj_calculate_style_text_align
lv_obj_init_draw_rect_dsc
lv_obj_init_draw_label_dsc
lv_obj_init_draw_img_dsc
lv_obj_init_draw_line_dsc
lv_obj_init_draw_arc_dsc
lv_obj_calculate_ext_draw_size
lv_obj_refresh_ext_draw_size
lv_obj_set_scrollbar_mode
lv_obj_set_scroll_dir
lv_obj_set_scroll_snap_x
lv_obj_set_scroll_snap_y
lv_obj_get_scrollbar_mode
lv_obj_get_scroll_dir
lv_obj_get_scroll_snap_x
lv_obj_get_scroll_snap_y
lv_obj_get_scroll_x
lv_obj_get_scroll_y
lv_obj_get_scroll_top
lv_obj_get_scroll_bottom
lv_obj_get_scroll_left
lv_obj_get_scroll_right
lv_obj_get_scroll_end
lv_obj_scroll_by
lv_obj_scroll_by_bounded
lv_obj_scroll_to
lv_obj_scroll_to_x
lv_obj_scroll_to_y
lv_obj_scroll_to_view
lv_obj_scroll_to_view_recursive
lv_obj_is_scrolling
lv_obj_update_snap
lv_obj_get_scrollbar_area
lv_obj_scrollbar_invalidate
lv_obj_readjust_scroll
lv_obj_set_pos
lv_obj_set_x
[... 854 lines cut ...]
lv_msgbox_get_close_btn
lv_msgbox_get_text
lv_msgbox_get_content
lv_msgbox_get_btns
lv_msgbox_get_active_btn_text
lv_msgbox_close
lv_msgbox_close_async
lv_sdl_keyboard_create
lv_draw_img_dsc_init
lv_draw_img_decoded
lv_strlen_builtin
lv_strncpy_builtin
lv_rlottie_create_from_file
lv_rlottie_create_from_raw
lv_rlottie_set_play_mode
lv_rlottie_set_current_frame
lv_style_register_prop
lv_style_transition_dsc_init
lv_style_set_size
lv_style_set_pad_all
lv_style_set_pad_hor
lv_style_set_pad_ver
lv_style_set_pad_gap

</details>

kdschlosser, 2023-03-13:
There is a lot that is documented but does not show up in the documentation...

Here is the API list for lv_obj

https://docs.lvgl.io/master/widgets/obj.html?highlight=lv_obj#api

Look at the number of functions... There is only 17 functions listed.. That is no where near the correct number of functions available.

kisvegabor, 2023-03-13:
Take a look at [this md file](https://github.com/lvgl/lvgl/blob/a19ad9bd1d40d1cfba208cbbaeba6238dc12170b/docs/widgets/obj.md#api) where you can see that really only `lv_obj.h` is added to the API section.

kdschlosser, 2023-03-13:
That could be the case. If it is then it does in fact need to be corrected. You guys have put in a lot of effort in the documentation and it not ending up getting used is almost hurtful I would think.,.

You are going to have to write a custom script for Sphinx that will sort the functions in a header file to properly organize where they actually go. I know there are some header files that have functions in them that do not pertain to lv_obj_t and those you do not want to show up in the API list for lv_obj_t. They need to be put where they belong.

kisvegabor, 2023-03-14:
> You are going to have to write a custom script for Sphinx that will sort the functions in a header file to properly organize where they actually go. 

Do you mean having a script to combine e.g. `lv_obj.h`, `lv_obj_pos.h`, `lv_obj_draw.h`, etc?

kdschlosser, 2023-03-14:
You might have to write a script to get the API to properly generate. Sphinx groups things together by module so unless that behavior is altered so it will instead group items together by the first parameter type it may not generate the documentation properly.

kisvegabor, 2023-03-16:
I see. I won't have enough time in the near future to dive into it for sure. :slightly_frowning_face: 

So if someone  is interested in contributing with such a thing, please let me know!

kdschlosser, 2023-03-16:
I would be more than happy to investigate what is going on with it.

The documentation NEEDS to be right. There is a heap load of bits and pieces to LVGL and without the documentation it makes it more difficult to get the information that is needed.

kdschlosser, 2023-03-17:
OK so here is the output of what I have so far.

If you can check it for accuracy that would be great.

it is broken apart by the filenames used in `docs/widgets` Basically what would happen is the directives in the list below would be added to the bottom of each of those markdown files for the widgets. This gets done before Sphinx runs. The build script would run Doxygen and then a copy will be made of the `docs` folder to a temporary directory. At that point my script gets imported and told to run passing the temporary directory name to my script. My script will then do its thing to create the information seen below and it will alter the markdown files in the temporary directory. Then Sphinx will get run and create the documentation from what is in that temporary folder and write it to`html_out` like it does right now. Once finished the temporary directory will get removed. Gotta make sure things get cleaned up 😉 

basically this is going to target the objects specific to a widget instead of files. as you know lv_obj is spread across many files and in some of those files are things that shouldn't be married to lv_obj. It could be something that actually gets used on some other component or even behind the scenes and shouldn't be added to the docs.

Now remember this output is only based on the names in the widgets folder. The script does parse EVERYTHING so if you wanted to add additional sections it's just a matter of creating the markdown file for the new section and pointing the script to the folder the file is located in. Adding widgets all you have to do is drop a new MD file in the widgets folder with the name matching the name used in the objects and the script will do it's thing without you having to do anything else.

Make sure you click on the expanding thing below, I put it into a that so it wouldn't become a pain to scroll down the issue.

<details>
  <summary>Click to view output</summary>

```
animimg
========================================

typedefs
.. doxygentypedef:: lv_anim_t
   :project: lvgl

.. doxygentypedef:: lv_obj_t
   :project: lvgl

structures
.. doxygenstruct:: lv_animimg_t
   :project: lvgl
   :members:
   :protected-members:
   :private-members:
   :undoc-members:

.. doxygenstruct:: lv_img_t
   :project: lvgl
   :members:
   :protected-members:
   :private-members:
   :undoc-members:

functions
.. doxygenfunction:: lv_animimg_create
   :project: lvgl

.. doxygenfunction:: lv_animimg_get_src
   :project: lvgl

.. doxygenfunction:: lv_animimg_get_src_count
   :project: lvgl

.. doxygenfunction:: lv_animimg_get_repeat_count
   :project: lvgl

.. doxygenfunction:: lv_animimg_start
   :project: lvgl

.. doxygenfunction:: lv_animimg_get_duration
   :project: lvgl

.. doxygenfunction:: lv_animimg_set_duration
   :project: lvgl

.. doxygenfunction:: lv_animimg_set_src
   :project: lvgl

.. doxygenfunction:: lv_animimg_set_repeat_count
   :project: lvgl

arc
[... 3651 lines cut ...]
   :project: lvgl
   :members:
   :protected-members:
   :private-members:
   :undoc-members:

functions
.. doxygenfunction:: lv_win_add_btn
   :project: lvgl

.. doxygenfunction:: lv_win_create
   :project: lvgl

.. doxygenfunction:: lv_win_get_header
   :project: lvgl

.. doxygenfunction:: lv_win_get_content
   :project: lvgl

.. doxygenfunction:: lv_win_add_title
   :project: lvgl

```

</details>

kisvegabor, 2023-03-20:
Thank you @kdschlosser! Could you attache the `html_out` directory to compare the result?

kdschlosser, 2023-03-20:
I am still working on the HTML portion of things. I got rid of most of the warnings from Sphinx, The Doxygen warnings are only for undocumented items and duplicate objects/functions.

What I have done with the script is I used it to iterate over the entire src directory locating all of the header files. I then created an API folder and in that folder I duplicated the structure from the src directory. The script creates an rst file for each of the header files. This is going to make a "master" API documentation in HTML. The API sections that are in each of the categories will link to the appropriate section of the master API.

I am also working on converting all of the markdown files to RST. This will allow for use of the sphinx directives and mainly the roles

* :c:member:
* :c:data:
* :c:var:
* :c:func:
* :c:macro:
* :c:struct:
* :c:union:
* :c:enum:
* :c:enumerator:
* :c:type:

The roles above are a fantastic thing to use. As an example there are statements like this ``lv_draw_ctx_t`` all over the documentation. While this is nice because it changes the appearance of of the text to indicate that it's code we can do one better.
By changing that to :c:struct:`lv_draw_ctx_t` you get the appearance change but it will also link to the item in the API section as well. 

I also wrote a script that will collect the version information so now building documentation is cross platform instead of only being able to build it on a nix system. The way it was there was a shell script that was ran to collect the version information. I am running Windows so it is easier for me to work in that environment then to have to mess about with WSL. Tho compiling on a nix system I am able to add -j 32 to the command that runs sphinx and that causes it to use all 32 logical processors my machine has. I use the OS module to determine how many logical processors a machine has so that is an automatic setting as well. On Windows the -j switch is ignored.

kdschlosser, 2023-03-20:
Now if I can just figure out how to add the API bits to the other pages without causing duplicate warnings.. I tried adding the page as raw HTML and I added some javascript to load the HTML from that page into another..... it doesn't work 😞 browser blocks it by default. so that is not a viable way to go about it.

I can add a link to the page, but I really wanted to have the API in there and not a link to it.

kisvegabor, 2023-03-21:
> I am also working on converting all of the markdown files to RST. This will allow for use of the sphinx directives and mainly the roles

I don't think it's a good idea. Although rst has more features it has a steeper learning curve and I'm afraid it wouldn't welcome contribution. IMO some local rst-in-md hack would be better to rst only if it's really required. 

---

Could you share your current work? Maybe someone will have any idea how to solve it.

kdschlosser, 2023-03-21:
also... just to let you know... the python library (recommonmark )that does all of the markdown to restructured text evaluations so the docs will build is no longer being maintained.

https://github.com/readthedocs/recommonmark

There is no drop in replacement. so eventually you are going to have to do something because it is not a matter of "if" it breaks it's a matter of "when" it breaks. It is 100% guaranteed that using markdown is not going to function unless you use old stale versions of the libraries that compile it and that includes python. right now using any python version > 3.8 is not going to happen because of library incompatibility with newer python versions. The libraries that have issues are the ones that surround the use of markdown.

kdschlosser, 2023-03-23:
There is a bug in Doxygen and I have opened an issue for it.

https://github.com/doxygen/doxygen/issues/9944

I have coded a way around the problem until this gets fixed in Doxygen. When this bug gets fixed the workaround code I wrote will become a dead stick and do nothing. So it can be removed whenever. Doxygen getting fixed will not break the documentation build.

kdschlosser, 2023-03-23:
You are gonna love this.

```console
****************
Building
****************
Running doxygen

cd ../scripts && doxygen Doxyfile
-------------------------------------
skipping latex build as requested

sphinx-build -b html . ../out_html -E -j 32
-------------------------------------
Running Sphinx v6.1.3
building [mo]: targets for 0 po files that are out of date
writing output...
building [html]: targets for 375 source files that are out of date
updating environment: [new config] 375 added, 0 changed, 0 removed
reading sources... [100%] widgets/win
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
preparing documents... done
writing output... [100%] widgets/win
generating indices... genindex done
writing additional pages... search done
copying images... [100%] misc/align.png
copying static files... done
copying extra files... done
dumping search index in English (code: en)... done
dumping object inventory... done
sphinx-sitemap: sitemap.xml was generated for URL https://docs.lvgl.io/master/en/html/ in C:\Users\drsch\PycharmProjects\lvgl\out_html/sitemap.xml
build succeeded.

The HTML pages are in ..\out_html.
```

how's that for a nice clean output? Not 1 error or warning. It's all cleaned up and running good. I still have formatting that needs to be done and some tweaking to the API portion. 

Here is a link to the branch I made.
https://github.com/kdschlosser/lvgl/tree/fix-docs

kisvegabor, 2023-03-23:
> You are gonna love this.

Wow, is this real? :open_mouth: 

I've checked out a few rst files and they are really digestible. There is  only one thing I don't really like: e.g. the first link in [slider.rst](https://raw.githubusercontent.com/kdschlosser/lvgl/fix-docs/docs/widgets/slider.rst) looks like this: 
```
`Bar </widgets/bar>`__
```

[Here](https://stackoverflow.com/questions/37553750/how-can-i-link-reference-another-rest-file-in-the-documentation) the comments suggest 
```
:doc:`path/to/document`
```
 should work too, which is more user friendly IMO.

What do you think about it?

kdschlosser, 2023-03-23:
There are things that do need to be done that's for sure. The biggest hurdle was getting it to compile with zero warnings and zero errors. There was a lot that needed to be done to get to that point. All of the markdown files I did a preliminary conversion using pandoc. I used that which took care of getting a basic conversion to eliminate the errors. pandoc does not know sphinx directives and roles and the formatting of the reST files is not right. They need to be gone through and corrected. the :doc: directive is a sphinx directive and the reason why it was not used. I went over a bunch of the files and fixed the formatting and added C specific roles.

Now.. I did add the information to the API sections in the specific areas. 

take `docs/overview/anim.rst` as an example.

```rest
==========
Animations
==========

You can automatically change the value of a variable between a start and
an end value using animations. Animation will happen by periodically
calling an "animator" function with the corresponding value parameter.

The *animator* functions have the following prototype:

.. code:: c

   void func(void * var, lv_anim_var_t value);

This prototype is compatible with the majority of the property *set*
functions in LVGL. For example :c:expr:`lv_obj_set_x(obj, value)` or
:c:expr:`lv_obj_set_width(obj, value)`

Create an animation
*******************

To create an animation an :c:struct:`lv_anim_t` variable has to be initialized
and configured with ``lv_anim_set_...()`` functions.

.. code:: c

   /* INITIALIZE AN ANIMATION
    *-----------------------*/

   lv_anim_t a;
   lv_anim_init(&a);

   /* MANDATORY SETTINGS
    *------------------*/

   /*Set the "animator" function*/
   lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t) lv_obj_set_x);

   /*Set target of the animation*/
   lv_anim_set_var(&a, obj);

   /*Length of the animation [ms]*/
   lv_anim_set_time(&a, duration);

   /*Set start and end values. E.g. 0, 150*/
   lv_anim_set_values(&a, start, end);

   /* OPTIONAL SETTINGS
    *------------------*/

   /*Time to wait before starting the animation [ms]*/
   lv_anim_set_delay(&a, delay);

   /*Set path (curve). Default is linear*/
   lv_anim_set_path(&a, lv_anim_path_ease_in);

   /*Set a callback to indicate when the animation is ready (idle).*/
   lv_anim_set_ready_cb(&a, ready_cb);

   /*Set a callback to indicate when the animation is deleted (idle).*/
   lv_anim_set_deleted_cb(&a, deleted_cb);

   /*Set a callback to indicate when the animation is started (after delay).*/
[... 147 lines cut ...]
then run the following command

`python -m pip install --no-cache -r requirements.txt`

This will ensure the latest versions of the libraries are gotten.

Also check and make sure the doxygen version is 1.9.6
`doxygen --version`

if not then upgrade it.

then run 
`python build.py skip_latex`

The build commands have not changed at all. 

If I can see a copy of the output that would help. if the build was successful how long did it take to compile? There should have been a noticeable speed improvement (excluding Windows).

If there are any errors let me know what they are.

I would like to make a request. If you can make a branch specifically for the documentation updates that I can do pull requests against that would be helpful. This way other people have the ability to help out or make suggestions if they want... I will break apart this single giant change into smaller changes so it is easier to follow what has been changed and why it has been changed.

There is a massive number of modifications to this and it needs to be organized. I did what I did because honestly I didn't know if I was going to be able to get to the goal of no errors and warnings. so I didn't bother with keeping it organized as I did it because I was trying all kinds of things and the commit history would have been a complete disaster. This way I can take the existing changes and separate them to keep it clean.

and I do agree with you about using the `:doc:` role. That will be what gets used.

kdschlosser, 2023-03-23:
I just did another large commit. I am working on fixing the roles and directives that are being used.

while the code of LVGL is C code for whatever reason the roles and directives being used are for CPP. which is fine I have to go in and change all of the `:c:` roles to `:cpp:` instead. How things are handled between the 2 different role types is different. While I can do a bulk change on all files form the `:c:` to `:cpp:` a lot of errors became of it. I am sorting out those errors.

So to give you some background on how this works.

* `:cpp:func:`: This is as it states, it is for functions. only the function name. no parameters or `()`
* `:cpp:strut:`: This is for structures, be careful because LVGL has a lot of typedefs for structures and a typedef is not a structure, it is a type.
* `:cpp:union`: same as structures
* `:cpp:type:`: all typedefs use this role.
* `:c:macro:`: This is the only "c" role. it is for preprocessor directives and macros
* `:cpp:enum:`: this is for an enum, not an enum item but the actual enumeration contain itself. Remember to watch out because a lot of LVGL enums are actually typedefs
* `:cpp:enumerator:`: this is for enumeration items.
* `:cpp:expr:`: this one is for expressions. The expression has to be a correct C/CPP expression. it cannot contain anything that is not normally seen in C/CPP code. so no `"..."` or `"<TYPE>"` kind of things. 

Now that I have done this last commit you will really be able to see the benefits of switching to using reST. if you compile the docs and go into the Overview->Events section. Then hover over one of the specially formatted boxes for one of the above roles you will find that it is a link. That link brings you to the API section of the documentation where that item is declared. It provides you will all of the documentation contained for that specific item.

kdschlosser, 2023-03-23:
I made a code change to the source of LVGL surrounding the enumerations. I am attempting to see if I can get the documentation to link correctly to an enumeration from a function parameter.

because LVGL uses typedefs to uint*_t for may of it's enumerations there is no link between a enumeration and a function parameter. instead it links to the typedef. I have to check and see but I made a change that would still link the parameter to the typedef but the typedef would then link to the enumeration. There would not be a path the user would be able to follow to know what actually gets placed for a function parameter.

This code change only effects the documentation and not change how LVGL runs.

kdschlosser, 2023-03-25:
OK so I have done a slew of commits. you should have a look at it. It's getting better with each commit. I am working over the links that reference other areas in the documentation. It's going to take be a bit to get those all ironed out. I have all of the sphinx roles done for the cpp and c code. YAY!

kisvegabor, 2023-04-11:
Why was it closed?

kdschlosser, 2023-04-11:
I only looked at the title and when I looked at that I closed it because the correct size gets returned when updating the layout on the object before calling the functions.

We took a major detour from the original  reason for this being opened up. don't ya think?? LOL
can always reopen it.

kisvegabor, 2023-04-12:
I've updated the title. 

What is the current status of the md->rst conversation? It'd be great to merge what we have now, else if someone updates the docs it will conflict with your version.

kdschlosser, 2023-04-12:
not too worried about version conflicts at all. I will open a PR for what I have up to date so you can give it a test drive and see if anything needs to be changed/tweaked.

kdschlosser, 2023-04-12:
#4129

kdschlosser, 2023-04-27:
I am going to finish up the PR for this this evening.

closed 2023-05-13 by kdschlosser