lvgl/lvgl #4050: Why change lv_event_get_target/current_target API?

FASTSHIFT, 2023-03-11:
### Introduce the problem
@kisvegabor 
Hi, I noticed that the event-related API has been modified in this commit: https://github.com/lvgl/lvgl/commit/680d6d1a7dae46c73b2313cd0692711c9f4e8117. What is the reason for changing this API?

### Examples and cases
https://developer.mozilla.org/en-US/docs/Web/API/Event/target
https://developer.mozilla.org/en-US/docs/Web/API/Event/currentTarget

### Suggested solution
Keep the definition of these two APIs, the first reason is that it can correspond to the definition of Web, and the second reason is that it can maintain compatibility :)

kisvegabor, 2023-03-13:
Let's examine two use cases:
- `lv_msgbox`: you add a `VALUE_CHANGE`   event to the msgbox, but it's bubbled from the btmmatrix of the buttons
- `lv_calendar`: you add a `VALUE_CHANGE`   event to the calendar, but it's bubbled from the btmmatrix of the days

I'd expect if I call `lv_event_get_target` I'll get the object to which the event is actually attached (msgbox or calendar) and I can call related functions. (E.g. `lv_calendar_get_...` or `lv_msgbox_close()`).

And if in some rare cases I need the original target I can call the longer `lv_event_get_original_target`.

Can you mention use case(s) when `get_target` meaning `get_original_target` is more intuitive?

W-Mai, 2023-03-14:
@kisvegabor 
Changing the names of event properties could lead to confusion and compatibility issues, as these properties are already widely used in many web applications and have been implemented and used for many years across different browsers and JavaScript libraries. Therefore, even though some may feel that modifying the names of these properties would make more sense, it's unlikely that these changes would be widely adopted and supported in the near future.

Furthermore, the names of `currentTarget` and `target` are already well-established in web development and easy to understand. currentTarget refers to the element currently handling the event, while target refers to the original element that triggered the event. These names are intuitive and straightforward, and changing them could take time for people to get used to and accept.

In summary, while some may argue for renaming `target` to `originalTarget` and `currentTarget` to `target`, doing so could cause confusion and compatibility issues, and therefore is unlikely to be widely adopted and supported in the short term.

FASTSHIFT, 2023-03-14:
@kisvegabor 
Yes, I don't think there's enough reason to change the API name just to make the API shorter or more intuitive. I have two reasons for not changing the API: 
1. It will confuse developers who are familiar with LVGL, and it may cause problems in the future. I will think of a shorter name and modify it again, which makes the upper-layer application generate a lot of unnecessary workload when migrating the large version of LVGL.
2. Modern IDEs have strong completion capabilities, and you can quickly find the API you want to find by simply typing a few letters.

kisvegabor, 2023-03-16:
I see your points. Just to better understand it, please mention a few life-like cases when you need to get the original target. I can imagine only one: when you have a lot of buttons (or similar  widgets) and instead of adding an event handler to each buttons one-by-one, you rather add it to the parent.

FASTSHIFT, 2023-03-17:
> I see your points. Just to better understand it, please mention a few life-like cases when you need to get the original target. I can imagine only one: when you have a lot of buttons (or similar [...]

I think we may have some misunderstanding, I have no problem with your `lv_msgbox` and `lv_calendar` registering only one callback.
I am confused by the name change of these two APIs:
`lv_event_get_current_target` -> `lv_event_get_target`
`lv_event_get_target` -> `lv_event_get_original_target`

kisvegabor, 2023-03-20:
Ok, let's follow the JS way. If anyone asks it later, why it is this way we can tell because we follow the JS convention. 
See 1faa4c8d7

commit 1faa4c8d (2023-03-20): revert(event): use original_target and target to follow JS conventions

fixes #4050