lvgl/lvgl #4051: Remove LV_USE_USER_DATA configuration

FASTSHIFT, 2023-03-11:
### Introduce the problem
In most GUI applications, `user_data` is basically relied on to bind some private data, and `LV_USE_USER_DATA` is almost a configuration item that must be enabled. So the meaning of this configuration reservation is actually not great.

### Examples and cases

### Suggested solution
Remove the `LV_USE_USER_DATA `configuration, which can reduce the macro judgment in many data structures and make the code more elegant.

kisvegabor, 2023-03-13:
The only advantage of having `LV_USE_USER_DATA` is to save 4 bytes in some cases. 
But as time passes an MCUs get better, it's really less important.

Could you send PR to remove it?

FASTSHIFT, 2023-03-14:
> The only advantage of having `LV_USE_USER_DATA` is to save 4 bytes in some cases. But as time passes an MCUs get better, it's really less important.
> 
> Could you send a PR to remove it?

Done

kisvegabor, 2023-03-14:
The PR is merged, closing. 
Thank you!

closed 2023-03-14 by kisvegabor