lvgl/lvgl #4052: Styles are acting funny

kdschlosser, 2023-03-12:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
2 sliders are created identically except for the position. Setting the knob opacity to 0 the border and outline widths to 0 for both MAIN and INDICATOR. When the widgets get rendered to the screen the first slider has a border and the second does not. If you click on either of the sliders to move the bar then the border around the first slider disappears

### To Reproduce
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/126fbee5ae0f568ed35d03736de00ef093b9bd0a/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/126fbee5ae0f568ed35d03736de00ef093b9bd0a/examples/widgets/bar/lv_example_bar_5.py&script_direct=ad8dea7d648357165e93fa31e9dcad3a8e194a4f

### Expected behavior
for the styles to not have odd behavior.

kdschlosser, 2023-03-13:
if it was a focus issue then once the widget lost it's focus the outline would go away but then when it regained focus again the outline should come back. That is not the case. Once the focus goes away the outline goes away and if i focus the widget again the outline never comes back. Also the second slider never gets an outline when focus is given to it.

kisvegabor, 2023-03-13:
It's really a tricky case. This is what happens:
1. `lv_group_set_default` is called in MicroPython somewhere so controllable widget will be added to the default group automatically. This makes possible to use key navigation (TAB, arrows, etc) without any extra action.
2. When the first widget is added to the default group, it will be focused. LVGL sees that there is a KEYPAD or ENCODER indev so besides `LV_STATE_FOCUSED`, `LV_STATE_FOCUS_KEY` is added too. `FOCUS_KEY` indicates that the widget was focused by a key (i.e. not touch) and you can add different styles in this case. 
3. The default theme indicates focused by key state with an outline. That's what you see by default
4. When you click the other widget, it won't get the `FOCUSED_KEY` state as you clicked it by touchpad and not focused by a key.
5. If you press TAB and use arrows you can navigate among the sliders and adjust them, and you will aslo see the outline as you use a key to focus the sliders.

It's quite the same as how it works on web.  At the end a clicked so the outline was disappeared. 
[image]

In CSS it's the [:focus-visible](https://developer.mozilla.org/en-US/docs/Web/CSS/:focus-visible) pseudo class

kdschlosser, 2023-03-13:
This is the first time this has happened,... 

So if a click of the mouse doesn't change the focus then how come the focus disappears from the first widget (and the outline along with it)

kisvegabor, 2023-03-14:
The mouse changes the focus, but doesn't add the `FOCUS_KEY` state (as it's not a key). therefore the outline (which exists only in `FOCUSE_KEY` state) is not shown.

https://user-images.githubusercontent.com/7599318/224914160-b4649fef-0d42-4fe4-815b-882f5b6e3876.mp4

newbthenewbd, 2023-03-18:
Perhaps the default, or an available option, should be to only, as on the web, be to only focus something like that once a key is pressed?

newbthenewbd, 2023-03-20:
Yup :) I just fell into the very same rabbit hole earlier, unable to figure out as to why the "currently focused" textarea doesn't focus anymore once I click away, and only luckily finding this issue provided that "ooh"...

newbthenewbd, 2023-03-20:
I mean- making "nothing focused" a working option within a group would also elegantly solve this, I suppose :) Seems like it's been a roadmap idea for a while...

kdschlosser, 2023-03-20:
what I have seen in the past with other GUI frameworks is if a widget is clicked but not in an area that does anything then the widget gets focus. If the widget is clicked in an area where it changes the state of the control then it does not get focus.

kisvegabor, 2023-03-21:
@newbthenewbd 
> I mean- making "nothing focused" a working option within a group would also elegantly solve this, I suppose :) Seems like it's been a roadmap idea for a while...

Oh, yes. It should really help too.

@kdschlosser 
I think it's not always true. E.g. on my notebook now a text editor is focused but if I click the full screen button of Chrome, Chrome will be focused.

kdschlosser, 2023-03-21:
That is a full screen button and not a "widget" but is an application. clicking on an application is going to pretty much implying that the application needs to be focused.

So when you have a windowed application that you are working in and then you go and click on the button to full screen chrome the application focus is changing to chrome.

kisvegabor, 2023-03-22:
That's true. It's really different. 

However it seems (at least on web) it works differently:
[image]

When I click on the text "commented" it becomes focused. Although the focus is no visible but we can know that it'was focused by pressing the tab key. Due to tab the next widget gets focused.

closed 2023-04-06 by kdschlosser

newbthenewbd, 2023-04-06:
Has a solution been reached that closes the issue? I've been looking forward to one, kinda... :)

kdschlosser, 2023-04-06:
idk if we actually figured out if there was a problem or not. The issue is there is no real set standard for the behavior of when focus changes and it appears like the behavior is completely up to the developer.

kisvegabor, 2023-04-11:
Finally, I agree that the ring shouldn't be displayed by default. I can update v9 accordingly, if you also agree.

kdschlosser, 2023-04-11:
i do agree!!

one less thing I have to do in my code.

kisvegabor, 2023-04-12:
Added

kdschlosser, 2023-04-12:
cool.

commit d6cc2ff7 (2023-04-12): fix(group): do not prefer KEYPAd and ENCODER indevs

fixes #4052