lvgl/lvgl #4054: Demos not installable

daniu-de, 2023-03-12:
Hi
I tried to copy files and folders around but I just cannot get this library running.

To make debugging easier I started from scratch. I did this:
(according to: https://docs.lvgl.io/master/get-started/platforms/arduino.html?highlight=demo)

1. setup TFT_eSPI - had that before already and running
2. editted user_setup.h and moved it to /library/
3. uncommented "lv_demo_widgets();"
4. uncommented "#include <lv_examples.h>" on the very top (even it is not written in the linked description) 

without No 4 I get:
```
Arduino\libraries\lv_examples\src\lv_demo_widgets\assets\img_clothes.c:4:10: fatal error: lvgl/lvgl.h: No such file or directory
 #include "lvgl/lvgl.h"
          ^~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```
I totally do not understand that - there is a "lvgl.h in /libraries/lvgl - and - looking into "img_clothes.c" I did not define: "LV_LVGL_H_INCLUDE_SIMPLE" because I put the "lv_conf.h" directly next to the lvgl folder

with No 4 I get:
```
xxxxx.ino:8:10: fatal error: lv_examples.h: No such file or directory
 #include <lv_examples.h>
          ^~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: lv_examples.h: No such file or directory
```

I assume that this Important note wants to say "move" instead of copy?
"IMPORTANT Due to some the limitations of Arduino's build system you need to copy lvgl/examples to lvgl/src/examples. Similarly for the demos lvgl/demos to lvgl/src/demos."

I would highly appreciate direct help that does not make me move around files and folders like crazy ;-)

daniu-de, 2023-03-13:
HI
found my problem: do not install lv_examples!

I even got the samples running with GT911 - will post the code soon. Looks really great even though if I would really want to do something fancy I would use a parallel connected display

kisvegabor, 2023-03-13:
> found my problem: do not install lv_examples! Maybe there would be a way to prevent people from doing that?

Do you have any idea how to do it? Maybe add comments at top of the INO file to warn people to not install lv_examples?

daniu-de, 2023-03-13:
Yes - that might help (not sure)

Other odeas:
* put a hint in tutorials wherever possible
* delete the files from arduino ide library or at least mark them obsolete somehow?

daniu-de, 2023-03-13:
since I don't know where to put the code for GT911 - I just put it here - people might find it in search of GT911:

on the top somehwere:
```
// for GT911 Touch
#include "TAMC_GT911.h"

#define TOUCH_SDA 21
#define TOUCH_SCL 22
#define TOUCH_INT 2
#define TOUCH_RST 13
#define TOUCH_WIDTH 480
#define TOUCH_HEIGHT 320

TAMC_GT911 tp = TAMC_GT911(TOUCH_SDA, TOUCH_SCL, TOUCH_INT, TOUCH_RST, TOUCH_WIDTH, TOUCH_HEIGHT);
```
my_touchpad_read becomes:
```
void my_touchpad_read( lv_indev_drv_t * indev_driver, lv_indev_data_t * data )
{
    uint16_t touchX, touchY;

    // bool touched = tft.getTouch( &touchX, &touchY, 600 );
    
    // for GT911
    tp.read();
    bool touched = tp.isTouched;
    touchX = tp.points[0].x;
    touchY = tp.points[0].y;

    if( !touched )
    {
        data->state = LV_INDEV_STATE_REL;
    }
    else
    {
        data->state = LV_INDEV_STATE_PR;

        /*Set the coordinates*/
        data->point.x = touchX;
        data->point.y = touchY;

        Serial.print( "Data x " );
        Serial.println( touchX );

        Serial.print( "Data y " );
        Serial.println( touchY );
    }
}
```
in setup comment this out:
```
/*Set the touchscreen calibration data,
     the actual data for your display can be acquired using
     the Generic -> Touch_calibrate example from the TFT_eSPI library*/
    //uint16_t calData[5] = { 275, 3620, 264, 3532, 1 };
    //tft.setTouch( calData );
```

and add this lines:
```
    //for GT911 Touch 
    tp.begin();
    tp.setRotation(ROTATION_LEFT);
```
make sure to change width and height values for the touchpad as well when changes are necessary:

rotation tanslates as:
  // Values for TFT | Touch:
  // 1 - ROTATION_RIGHT
  // 2 - ROTATION_NORMAL
  // 3 - ROTATION_LEFT
  // 4 - ROTATION_INVERTED

but don't forget the pixels as described further up

kisvegabor, 2023-03-14:
I added notes in the docs and the ino file.

Thank you for sharing the driver.

commit e26a46c4 (2023-03-14): docs(arduino): add note to not use lv_examles library

fixes #4054
closed 2023-03-14 by kisvegabor