lvgl/lvgl #4055: alignment of objects stops working properly

kdschlosser, 2023-03-13:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
This is going to be a tricky one because of hot it is going to need to be handled.
The issue is when `set_style_transform_angle` gets used on an object the alignment of the object no longer works properly. In the code example below I am aligning an lv_label object to the center of the screen and then rotating it 180°. The object should still be centered in the screen just upside down now. This is not the case. The object gets shifted off to the left side of the screen and any attempts to align it to the center of the screen do not work.

The reason why I say this gets tricky is because the starting corner of the object never changes when performing a a transform angle change on an object. so at 0 degrees rotation the upper left corner is the corner that is used to set the position. if you rotate the object 90° the corner that gets used is now the upper right hand corner. Rotate to 180° and the corner is now the lower right and at 270° the corner is now the lower left. The rotation of the object needs to be taken into consideration when the alignment gets set.

I would use the bounding box of the object when setting the alignment of a rotated object. This simplifies the process I would think and it would also give the expected behavior across the board for all of the alignment flags

Here is an example in Python of what I mean by the bounding box. It's a loose example and not exact but it provides the general concept.

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/126fbee5ae0f568ed35d03736de00ef093b9bd0a/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/126fbee5ae0f568ed35d03736de00ef093b9bd0a/examples/event/lv_example_event_3.py&script_direct=7dd1e9b29deaf00c02e17eeb02722cdb8cdbba52

### To Reproduce
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/126fbee5ae0f568ed35d03736de00ef093b9bd0a/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/126fbee5ae0f568ed35d03736de00ef093b9bd0a/examples/widgets/arc/lv_example_arc_1.py&script_direct=5d56bbf5ec415f016e3671da78a07f1ad6941e59

### Expected behavior
the alignment of the object to work correctly.

kdschlosser, 2023-03-13:
I did want to mention that I am aware that I could make the rotated object a child of an lv_obj_t instance and get it centered out properly inside of that instance and then align that instance. That would work but it would be at the expense of more processing and more memory use.

kdschlosser, 2023-03-13:
Or I could set the size of the instance to be CONTENT_SIZE and it should center the child correctly (would hope).

kdschlosser, 2023-03-14:
That is only for a single use case. What happens if I don't want to center the object but instead I want to align to the bottom left. The alignment is still not going to work correctly. The other thing is if i create an lv_obj instance and I set that as the parent of another object.. say lv_label and I set some text to the label and then I rotate the label. After that I set the width and height of the parent lv_obj instance to LV_SIZE_CONTENT it should size the lv_obj instance like it is a bounding box. It does not do this.

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/126fbee5ae0f568ed35d03736de00ef093b9bd0a/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/126fbee5ae0f568ed35d03736de00ef093b9bd0a/examples/widgets/arc/lv_example_arc_1.py&script_direct=fb133b381fc76bbc830382df0bca095194e5387e

kdschlosser, 2023-03-14:
It shouldn't matter what the pivot point is when using the example above or setting the alignment. It should create a proper bounding box and that is what ends up getting used to set the alignment and LV_SIZE_CONTENT should make the parent a proper bounding box..

kdschlosser, 2023-03-14:
If you look at this example the widget is exactly what is in the green box. That box is not even close to being centered on the screen.

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/126fbee5ae0f568ed35d03736de00ef093b9bd0a/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/126fbee5ae0f568ed35d03736de00ef093b9bd0a/examples/widgets/bar/lv_example_bar_5.py

kdschlosser, 2023-03-14:
I am not sure how the back end of LVGL works in terms of size and position. but it should be "rectangle" based.

Now remember that I am not proficient in C code so I am sure that I have the pointer aspect of this all mucked up.
when a widget is created the top left coords, size and angle should be stored in an lv_rect_t structure. This structure can then be passed to a function that will return the bounding box. from that bounding box you can determine where the widget should be located when doing an alignment.

This same bounding box can also be used when LV_SIZE_CONTENT is used as the width and height.

I would also have a function that is able to determine offset values for the top left corner. So if you rotate a widget say 180° the 0,0 corner of that widget is actually going to be on the bottom right so the offset factors would be (-width, -height). By using the rect and locating the coordinates for the 4 corners it makes it easier to determine what those offset values should be by doing a comparison of the coords to find out which corner left most and also the top most then take the (0, 0) corner position coords and subtract the coords for the corner that has a positional placement being at the top left.

again excuse my code below, it is only pseudo code for the purposes of showing the idea. 
```c
typedef struct {
    lv_point_t pos;
    lv_point_t size;
    lv_point_t * top_left;
    lv_point_t * top_right;
    lv_point_t * bottom_left;
    lv_point_t * bottom_right;
    lv_point_t * center;
    int16_t angle;
} lv_rect_t;

void _lv_get_end_point(lv_point_t * end_point, lv_point_t * start_point, int16_t angle, int16t length) {
    end_point->x = length * lv_trigo_cos(angle) + start_point->x;
    end_point->y = length * lv_trigo_sin(angle) + start_point->y;
}

uint16_t _lv_get_angle_between_points(lv_point_t * start_point, lv_point_t * end_point) {
    return lv_atan2(end_point->y - start_point->y, end_point->x - start_point->x)
}

void _lv_get_distance_between_points(lv_point_t * start_point, lv_point_t * end_point, lv_sqrt_res_t * res) {
    lv_sqrt(
        (uint32_t) (lv_pow((int64_t)(end_point->x - start_point->x), 2) +  lv_pow((int64_t)(end_point->y - start_point->y), 2)),
        res,
        0x8000
    )
}

void _lv_get_bounding_box(lv_rect_t * in_rect, lv_rect_t * out_rect) {
        tl_x, tl_y = self.top_left
        tr_x, tr_y = self.top_right
        bl_x, bl_y = self.bottom_left
        br_x, br_y = self.bottom_right

        out_rect->pos.x = min(in_rect->top_left.x, in_rect->top_right.x, in_rect->bottom_left.x, bin_rect->bottom_right.x)
        out_rect->pos.y = min(in_rect->top_left.y, in_rect->top_right.y, in_rect->bottom_left.y, bin_rect->bottom_right.y)
        out_rect->size.x = max(in_rect->top_left.x, in_rect->top_right.x, in_rect->bottom_left.x, bin_rect->bottom_right.x) - out_rect->pos.x
        out_rect->size.y = max(in_rect->top_left.y, in_rect->top_right.y, in_rect->bottom_left.y, bin_rect->bottom_right.y) - out_rect->pos.y

        _lv_populate_rect(out_rect)
}

void _lv_populate_rect(lv_rect_t * rect) {    
    rect->top_left->x = rect->pos.x;
    rect->top_left->y = rect->pos.y;

    _lv_get_end_point(rect->pos, rect->top_right, rect->angle, rect->size.x);
    _lv_get_end_point(rect->top_right, rect->bottom_right, rect->angle + 900, rect->size.y);
    _lv_get_end_point(rect->bottom_right, rect->bottom_left, rect->angle + 1800, rect->size.x);
    _lv_get_end_point(
        rect->bottom_right,
        rect->center,
        _lv_get_angle_between_points(rect->pos, rect->bottom_right),
        _lv_get_distance_between_points(rect->pos, rect->bottom_right) / 2
    )
}
```

kdschlosser, 2023-03-14:
If you have an object that is a child and the parent has the size set to LV_SIZE_CONTENT if the child object gets rotated the parent's size should expand in order to fit the child. This can now be done fairly easily because you have the ability to get the bounding box of the child.

It gets tricky because technically speaking an objects size does not change when the angle changes but once the object has been rotated you are not going to be able to use the width and height to be able to determine the center for alignment purposes. However you will be able to use the bounding box for the object to set the alignment. The important thing is the results are going to be predictable. 

Right now if I set the angle of a widget to 45° and center it the widget will have a different placement than setting the object to a 135° angle an centering it. They should be located in the exact same place on the screen except one will be upside down compared to the other.

Getting this to work properly will allow for widgets like this to be made.

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/126fbee5ae0f568ed35d03736de00ef093b9bd0a/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/126fbee5ae0f568ed35d03736de00ef093b9bd0a/examples/widgets/arc/lv_example_arc_1.py&script_direct=fea1735d77c85c287d95a0e3c867b5ec4c6586cc

That is just an example and it's not polished. You get the idea tho. Need to be able to align the widgets exactly to each other in order for it to work and making the whole thing dynamic based on the size would be exceedingly hard to do without the alignment portions working correctly on an angled object.

kdschlosser, 2023-03-14:
This was actually an annoyance to get coded.

[image]

Look at the labeling around the arcs. I gave up using camel case lettering and went with all capitals because the height of the font is more consistent and I didn't have to mess about with the positioning based on the height of a single letter. I did pretty good with getting the letter spacing right.

kisvegabor, 2023-03-16:
Note that it's obvious that always the transformed coordinates should be used as the basis of alignments and layouts. Imagine 5 buttons in a row aligned by a flex layout. Let"s say if you press a button it will be zoomed a little bit. Now you might want 2 things and both are equally valid:
1. as the pressed button grows push the adjacent buttons (i.e. consider the transformed coordinates)
2. as the pressed button grows do nothing with  adjacent buttons (i.e. consider the un-transformed coordinates)

It's also important to add that web also uses the un-transformed coordinates. Besides I can imagine cases where using the transformed coordinates result in infinite loops or hard to calculate/predict behavior. E.g. let's say we rotate button around an eternal pivot point (e.g. -100;0). IF the transformed coordinates are uses as basis, the rotated object will stay in place, but one might expect to see it rotated around circle with 200px diameter. 

I still say that we should support it, but the normal coordinate based layouts should be supported too.

closed 2023-04-08 by lvgl-bot