lvgl/lvgl #4058: Add drivers in the LVGL repository

kisvegabor, 2023-03-14:
As [some larges changes](https://github.com/lvgl/lvgl/issues/4011) are added to master we can start to add more drivers into LVGL. As an experiment I've already added SDL, linux/fb, and TFT_eSPI. It seems working well, but we still don't have any display controllers and vendor APIs.

This is how I imagined it:
- All display controllers have its own create function. E.g. the SDL driver has [lv_sdl_window_create](https://github.com/lvgl/lvgl/blob/master/src/dev/sdl/lv_sdl_window.c#L66). The create functions can have any prototype with any parameters we need. 
- We should have an `lv_disp_drv_t` struct like:
```c
typedef struct
{
   lv_res_t (*send_cmd_cb)(uint8_t * cmd,  size_t cmd_len, uint8_t * data,  size_t data_len);
   lv_res_t (*read_data_cb)(uint8_t * cmd,  size_t cmd_len, uint8_t * read_buf,  size_t read_buf_len);
   lv_res_t (*send_image_cb)(const lv_area_t * area, void * buf);
   void * user_data;  //For extra data if needed
}lv_drv_t
```
It's a high level description of the drivers. All the pins and SPI, I2C etc handling is hidden inside the these callbacks. In other words the same struct is used for SPI, I2C, parallel, etc
- Once we have an `lv_disp_drv_t` initialized with the callbacks we can create e.g. an ILI9341 display with `lv_ili9341_create(320, 240, &drv)`
- Having `lv_drv_t` is one layer but we can push it one step further by having e.g. an ESP API like `lv_esp32_spi_init(&drv, &spi_cfg)` where `spi_cfg` is something like:
```c
typedef struct {
   int spi_instance; //e.g. HSPI
   int clk_io;
   int mosi_io;
   int miso_io;
   int cs_io;
   int clk_speed; //E.g. 20 MHz
}lv_esp32_spi_cfg_t
``` 
It initializes a driver with the given settings. Handles DMA, CS pin, etc. Ideally we should be able to use the same SPI driver for the display and touch too.  
- To change resolution, orientation, etc we can use LVGL's API like `lv_disp_set_rotation();` which will send a `LV_EVENT_RESOLUTION_CHANGED`. So the `lv_disp_t` created in `lv_ili9341_create` should subscribe to events (similarly to `lv_obj`s) . See the SDL driver as a [reference](https://github.com/lvgl/lvgl/blob/master/src/dev/sdl/lv_sdl_window.c#L100).

I think the first step is to agree on a display controller and vendor API, and add an example display controller for a vendor (e.g. ILI9341 + ESP).

What do you think? Please let me know if you have ideas or remarks.

I don't have too many boards with ILI or similar controllers so I hope you can help with the integration and testing of this drivers. We have a lot of display controller and touch drivers in the [lvgl_esp32_drivers](https://github.com/lvgl/lvgl_esp32_drivers/) repo which can used as a starting point.

amirgon, 2023-03-15:
> What do you think? Please let me know if you have ideas or remarks.

To make the drivers portable to Micropython, it's important to set all parameters (pins, clks, spi etc. as suggested in `lv_esp32_spi_cfg_t`), and avoid using compile-time defines for configuration.

This is a great step forward and I'm looking forward to seeing display drivers as part of LVGL!

kisvegabor, 2023-03-16:
@amirgon It's great that you that like this direction. 

I was on the Embedded World fair yesterday and discussed this topic with @igrr (Ivan who is VP at Espressif). He explained that IDF v5 has built in support for many display controllers and an abstraction layer for a generic display device. So we concluded that the best thing we can do is having an `lv_esp_display` "module" which is just a wrapper around the display configured in the IDF. Espressif would like to use their own display drivers (and not rely on LVGL's drivers) to allow people using IDF with any UI library, not only LVGL.

I had the same discussion with Zephyr. They are also intended to ship their own display drivers for the same reason. 

Regardless to it we can decide to implement an SPI/I2C/parallel interface for Zephyr and ESP and use our own display drivers too. 

However, for chip vendors (like NXP, STM, Infineon, Renesas, etc) we should have the  SPI/I2C/parallel drivers for sure.

HowardAtHome, 2023-04-02:
" @igrr (Ivan who is VP at Espressif). He explained that IDF v5 has built in support for many display controllers and an abstraction layer for a generic display device"

How timely - I have just spent a week (I do not get much spare time, so a few hours in reality) trying to get the ESP32 RGB Panel working on the Arduino PlatformIO framework. Turns out to be very hard as it is based on the ESP IDF v4.4.4 which has many issues solved in v5.0.1.

Simple message - great idea to leave the drivers with the chip vendors - but make it clear what works now and when the vendor get it working let the community know.

kdschlosser, 2023-04-29:
I am thinking something along these lines.

```c

enum {
    LV_IO_BUS_BYTE_ORDER_MSB;
    LV_IO_BUS_BYTE_ORDER_LSB;
};

typedef uint8_t lv_io_bus_byte_order_t;

typedef struct _lv_io_bus_handle_t lv_io_bus_handle_t;
typedef struct _lv_disp_handle_t lv_disp_handle_t;

typedef lv_res_t (* lv_disp_reset_cb_t)(lv_disp_handle_t * handle);
typedef lv_res_t (* lv_disp_init_cb_t)(lv_disp_handle_t * handle);
typedef lv_res_t (* lv_disp_del_cb_t)(lv_disp_handle_t * handle);
typedef lv_res_t (* lv_disp_flush_cb_t)(lv_disp_handle_t * handle, lv_area_t * area, lv_color_t *buf);
typedef lv_res_t (* lv_disp_mirror_cb_t)(lv_disp_handle_t * handle, bool mirror_x, bool mirror_y);

// the mirror and swap_axis callbacks are the mechanism used to set the screen orientation
typedef lv_res_t (* lv_disp_swap_axis_cb_t)(lv_disp_handle_t * handle, bool swap_axes);
typedef lv_res_t (* lv_disp_offset_cb_t)(lv_disp_handle_t * handle, uint16_t x, uint16_t y);
typedef lv_res_t (* lv_disp_invert_cb_t)(lv_disp_handle_t * handle, bool invert);
typedef lv_res_t (* lv_disp_power_cb_t)(lv_disp_handle_t * handle, bool state);
typedef lv_res_t (* lv_disp_brightness_cb_t)(lv_disp_handle_t * handle, uint8_t brightness);

typedef struct {
    uint16_t hres;
    uint16_t vres;
    uint_16_t offset_x;
    uint_16_t offset_y;
    uint8_t mirror_x: 1;
    uint8_t mirror_y: 1;
    uint8_t swap_axis: 1;
    uint8_t power: 1;
    uint_8t brightness;
    uint8_t invert: 1;
    lv_disp_reset_cb_t reset_cb;
    lv_disp_init_cb_t init_cb;
    lv_disp_del_cb_t del_cb;
    lv_disp_flush_cb_t flush_cb;
    lv_disp_mirror_cb_t mirror_cb;
    lv_disp_swap_axis_cb_t swap_axis_cb;
    lv_disp_offset_cb_t offset_cb;
    lv_disp_invert_cb_t invert_cb;
    lv_disp_power_cb_t power_cb;
    lv_disp_brightness_cb_t brightness_cb;
    lv_io_bus_handle_t * io_bus
} lv_disp_handle_t

typedef lv_res_t (* lv_io_bus_trans_done_cb_t)(lv_io_bus_handle_t * handle);
typedef lv_res_t (* lv_io_bus_tx_cmd_cb_t)(lv_io_bus_handle_t * handle, uint32_t cmd);
typedef lv_res_t (* lv_io_bus_rx_param_cb_t)(lv_io_bus_handle_t * handle, uint8_t * param);
typedef lv_res_t (* lv_io_bus_tx_param_cb_t)(lv_io_bus_handle_t * handle, uint8_t *param);
typedef lv_res_t (* lv_io_bus_tx_buf_cb_t)(lv_io_bus_handle_t * handle, uint8_t *buf);
typedef lv_res_t (* lv_io_bus_del_cb_t)(lv_io_bus_handle_t * handle);
typedef lv_res_t (* lv_io_bus_power_cb_t)(lv_io_bus_handle_t * handle, bool state);
typedef lv_res_t (* lv_io_bus_reset_cb_t)(lv_io_bus_handle_t * handle);
typedef lv_res_t (* lv_io_bus_brightness_cb_t)(lv_io_bus_handle_t * handle, uint8_t level);

typedef struct {
    lv_io_bus_trans_done_cb_t trans_done_cb;
    lv_io_bus_tx_cmd_cb_t tx_cms_cb;
    lv_io_bus_rx_param_cb_t rx_param_cb;
    lv_io_bus_tx_param_cb_t tx_param_cb;
    lv_io_bus_tx_buf_cb_t tx_buf_cb;
    lv_io_bus_del_cb_t del_cb;
    lv_io_bus_brightness_cb_t brightness_cb;
    lv_io_bus_reset_cb_t reset_cb;
[... 219 lines cut ...]
{
    return disp->power_cb(disp, state);
}

lv_res_t lv_disp_set_brightness(lv_disp_handle_t disp, uint8_t level)
{
    return disp->brightness_cb(disp, state);
}

```

there would be functions for each display that is supported. so as an example you would have

```c
typedef lv_disp_handle_t *lv_disp_st7789_t;
lv_res_t lv_disp_st7789_create(lv_io_bus_handle_t * bus_handle, lv_disp_st7789_t disp_handle);
```

you use one of the bus create functions to create the bus. the relitive bus structure needs to be populated with the information needed to initilize the bus and other anscellary GPIO pins.

When the bus is being created that is when it is going to populate the bus callbacks and it will use the rest of the informatiom to set up the bus.

Then the display structure gets pupolated by the user except for the callbacks. when the bus handle, display structure and display handle get passed to a display creation function the display callbacks will be populated and the display will get set up. the display handle contains a merging of the display structure and also the bus structure this is done so if the user oses one of the functions to say turn the power on to the display the callback the display registered will get called and the display clalback will then call the io function to turn the power on and that will call the callback that was registered when the io was created. then the appropriate things can be done to turn the GPIO on or off.

the io callbacks are going to be specific to the MCU being used.

kisvegabor, 2023-05-01:
My concern with this approach is that the driver API layer of the vendors  are not standard. Even an SPI bus can have many different features and approaches. Therefore developing a unified bus structure is quite complicated. Actually I think it's not really required because we want to do only a few things with the bus: send a few data/command, and send the rendered image.  So we can approach it from this higher level point of view and let the implementation hide the HW/vendor specific things. 

Instead of having many callbacks for the display (e.g. `lv_disp_set_brightness/mirror/etc`) we can use events. In `master`  display can subscribe to events too. So we can have a high level function, e.g.:
```c
void lv_disp_mirror(lv_disp_t * disp, bool hor_mirror, bool ver_mirror)
{
   disp->mirror_hor = mirror_hor;
   disp->mirror_vre = mirror_ver;
   lv_disp_send_event(disp, LV_EVENT_DISP_MIRROR_CHANGED);
}
```

And the drivers can subscribe to this event, and to required config changes.

kdschlosser, 2023-05-01:
the problem with the event is they run on a timer based system. If a user wants to rotate the screen they are going to want it done right them and there and not wait until lv_task_handler gets called so the event can get processed. It would cause an issue if the layout has to change along with the rotation because of the differnece in the amount of time it takes between a refresh of the display is done and when the event gets run. 

In the end sending an event is timer with a callback. still callbacks just in a different area of the code. The reason for using callbacks is so you can have multiple displays and they can have different IC's but there is a single public entry point to change things on any display no matter what kind of display it is.

The same thing goes for the bus. so when the flush callback gets called for the display there is a single set of entry points to read or write to the bus doesn't matter if the bus is an i8080 or an SPI. The methodology above is a general outline. You set up the bus structures so that the common items across manufacturers that would be the same like data pins, cs, dc and wr pins spi modes those kinds of things are in the structure. You have a "driver_data" field in lv_disp_t. the purpose for it is to store information that is specific to the display that is being used. It holds the things that are not common across all display manufacturers. the same thing can also be done with the IO structures as well. so when the bus structure gets passed to one of those public entry point functions the callback to the function that is specific for that bus type will get called and then the stored driver data can get cast to the proper type which will be able to be read because the code that is bring run for the IO is going to be specific to the MCU.

basically you would have a mechanism in place that would set the MCU being used and that would load the include files for the IO BUS related things where the structures needed for that MCU would be defined and the callback functions doe performing the different IO tasks would also be defined.

kdschlosser, 2023-05-01:
the only difference between this code

```c
void lv_disp_mirror(lv_disp_t * disp, bool hor_mirror, bool ver_mirror)
{
   disp->mirror_hor = mirror_hor;
   disp->mirror_vre = mirror_ver;
   lv_disp_send_event(disp, LV_EVENT_DISP_MIRROR_CHANGED);
}
```

and this code

```c
lv_res_t lv_disp_mirror(lv_disp_handle_t disp, bool mirror_x, bool mirror_y)
{
    return disp->mirror_cb(disp, mirror_x, mirror_y);
}
```

is the latter gets run right away where the top one gets called later. both still need to have a callback assigned in the first parameter (disp) and that is what gets called. 

no overloading in C99 so you cannot set a single callback to handle multiple things. no keywords in function calls to assign data to specific params in the function call like Python.

kisvegabor, 2023-05-01:
> the problem with the event is they run on a timer based system. 

That's not the case. The events are sent immediately. They are sent immediately on touch as well, but the touch is read periodically in a timer. 

Regarding the callbacks, I think the only difference between your and  my view is that I think we should have vendor specific driver descriptors. So instead of `lv_io_bus_spi_t` I suggest `lv_esp32_spi_cfg_t`. I feel like `lv_io_bus_spi_t` is an unnecessary abstraction. If someone uses ESP32 they can chosse the ESP32 specific data type, where we list the config options as ESP wants to see them. 

For example `lv_io_bus_pin_t` is not really working for STM32 because the pins are described by a port and a pin (e.g. C7). Probably we could figure out something more generic, but why? For STM32 people can just use an STM32 specific SPI descriptor.

kdschlosser, 2023-05-01:
the way I am thinking about it is it would be compile time specific to the board. so if it is compile time specific there is no need to have different container names because all the different boards can share the same container name. just have to make sure the common parameters have the field names. anything that is specific to a board would be added to the already present names. This keeps code more portable between different boards. simply a matter of some small tweaks with the data that needs to go into the structure but no need to use a macro to change the structure completely.

kdschlosser, 2023-05-01:
with something like this it is not something that is going to be able to be chosen at runtime it MUST be done at compile time.

kdschlosser, 2023-05-01:
Now my question about the whole event thing is this. How to you register the callback function? You either have to register a callback to a specific event which would require the callback to be stored for each event or if I only wanted say 2 event but there are a total of 10 that can occur the function would have to be called for each of the 10 events even tho I only needed 2 of them. The events cannot be OR'ed together can they? If they could that would be a solution to the problem. a separate event system would have to be set up to do that because of the number of events that already exist for the typical events that occur in LVGL. you would need 17 bits to be able to OR the events together. 9 bits for the IO/GPIO related parts and 8 for the display relayed pieces.

It actually should be a separate system because for the most part they are "private" and should not be accessible through the public event system.

I am not saying to make it 100% private because it does need to be accessible to someone that makes a custom driver. But here is an issue. if you have say an ESP32 SPI structure and you have an STM32 structure and you have an NXP structure how would you go about passing that through the same event system as the display for starters and how would you be able to set the type for all of those different structures?

kisvegabor, 2023-05-01:
> here is no need to have different container names because all the different boards can share the same container name.

Good idea. We can use the `lv_spi_cfg_t` for ESP, STM32, NXP, etc, but the struct will have a different content. 

> with something like this it is not something that is going to be able to be chosen at runtime it MUST be done at compile time.

Yes, but it's reasonable to decide  in compile time which vendor's platform you are using :slightly_smiling_face:  It's not something that you can change in runtime.

> The events cannot be OR'ed together can they? 

They cannot, because we have too many events. You can use `LV_EVENT_ALL` to catch all the events sent to that widget/display.

> It actually should be a separate system because for the most part they are "private" and should not be accessible through the public event system.

I don't get it. What is private and why is it problem?

>  if you have say an ESP32 SPI structure and you have an STM32 structure and you have an NXP structure how would you go about passing that through the same event system as the display for starters and how would you be able to set the type for all of those different structures?

From the high level perspective all these are `lv_disp_t` with an `void * drv_data`. The `drv_data` can store any drive specific thing, the driver need on the low level. We don't need from LVGL. All we need to do is having high level `lv_disp_()` functions sending events and if the driver is interested in those events it can subscribe and do the required changes accordingly.  See the [SDL driver](https://github.com/lvgl/lvgl/blob/master/src/dev/sdl/lv_sdl_window.c) as a reference.

kdschlosser, 2023-05-01:
ok you have the current event system. it handles events that get generated for GUI related things. There should be a second event system that is for hardware related things. the 2 really should be separate. The hardware one it would make sense to be able to OR what events are wanting to be gotten. This is because not every display supports brightness, or hardware rotation or even a hardware reset. so having to do a check to test for an event that there is not going to be anything done for is a waste of time to make the function call in the first place. a simple bit check in the event handler to determine if the event even needs to get past the event handler.

the functions for changing things a display may not support will be accessible and usable too. if something gets changed that shouldn't get changed it may cause issues. so ion your example the arguments are not passed in a function call. they get set and then the function gets called. If the display doesn't support what is being changed and what was set needed to be the way it was your design could cause an issue. There needs to be a return value from the even call. a simple boolean value to let you know if the things that were changed need to be changed back to what they were.

Here is a great example. the rotation. if the rotation is not supported because of some kind of a glitch or a goofy design so the values that are stored in mirror_x, mirror_y and swap_xy are set because that is what works properly then the user starts mucking about because it's not rotating and then they try resetting the display after the new setting are place and that causes even more issues for the user.

ideally having the user not go bald prematurely (from pulling their hair out) is a better things to do. So a warning about lack of support in the log and the settings get reverted.

kdschlosser, 2023-05-01:
as far as lv_disp_t is concerned. Yes it does have the driver_data field this field should be changed to a specific type. the lv_disp_t structure contains mostly the top most layer of information. the things that user will see. mostly software related things. That driver_data should have it's type change to lv_display_driver_t and that would be structure that contains the next layer down which is the hardware related settings for the display.  anything that is driver IC specific there can be a "manufacturer_specific" field and that field would hold well settings that are unique to that display (or a small handful of displays..

The lv_disp_t structure would hold all the information from the top down. it should be something along there lines.

lv_disp_t: LVGL/software related settings.
lv_disp_t.driver: hardware display settings
lv_disp_t.driver.mfg_specific: settings that are specific to the display that is being used
lv_disp_t.driver.io_bus: hardware bridge between the MCU and the display driver IC
lv_disp_t.driver.io_bus.mfg_specific: settings that would be specific to the MCU manufacturer

reset, power and backlight control should be handled in the IO section. that's because it is input/output after all.

so when starting the bus the config for the bus and also a pointer to a structure for the hardware display settings would get passed to the function. that function once it has initialized everything would then place the IO config settings into the field in the display hardware structure. once that has been done then a call to initialize the display hardware is made, passing lv_disp_t and the display hardware struct to the function. Once the display IC has been initialized the display hardware struct gets placed into the lv_disp_t structure.

That keeps everything in a single package so when events/callbacks are made things are able to be accessed easily. the display hardware is able to pull the structure for the IO and make the call that needs to be made to the IO to set the DC pin before sending the data. the IO bus structure would get passed in that function call so the function now has the GPIO information on what pin is set as the DC pin.

anything that handles the frame buffer should be done by use of direct callbacks instead of the event system. This is because of not being able to pass the pointer to the frame buffer in the function call..

I do have a question regarding that aspect of it. can the buffers be stored in lv_disp_t and would they contain the data that needs to be written? I would imagine the answer to be yes. If that is the case then all that would be needed is the lv_area_t structure and some kind of a marker indicating which buffer has the data that needs to be written..

kdschlosser, 2023-05-02:
the IO is what is going to do the sending. the IO is a separate entity from the display driver and is a separate entity from lvgl's "display driver"

lvgl's display driver contains information specific to lvgl and how the attached display is to be used. I would not join any of these things together. While I do believe that the IC drivers and the IO drivers should be bundled with LVGL I do not believe they should be intertwined in any way. If they are intertwined in a manner where LVGL is dependent on some form of a driver being available it makes it more difficult for existing projects to update to a newer version if they have a custom display. IC drivers and IO drivers should be optional. If I wanted to be able to keep the IC drivers and not use any of the bus drivers that should be able to be done

kisvegabor, 2023-05-03:
Sounds good. My main goal is to give the highest level of freedom for the vendor's drivers. Of course, they can (probably will implement) some kind of bus/io abstraction but it's completely hidden from us and related only to the given driver/vendor.

kisvegabor, 2023-06-20:
Hi Arafat,

It's nice to see you here! :slightly_smiling_face: 

I summarized my idea in the [first comment of this issue](https://github.com/lvgl/lvgl/issues/4058#issue-1622879369). Please take a look at it and let us know what you think.

arafat877, 2023-06-20:
For the concept, it's ok, but we'll have an issue in the future, for each display/controller we should implement it's own driver, that should be done from scratch, what I suggest is to have an abstract implemention, that each vendor can implement easily

kisvegabor, 2023-06-21:
What parts could be abstracted? As I see if we have a generic communication interface (something like that I have proposed in the first comment) the rest is controller specific commands.

nylnx, 2023-07-30:
I got the idea to implement the correct rotation handling for the SDL simulation as a PR, because it was hinted at by @kisvegabor in my feature request [here](https://forum.lvgl.io/t/pre-rotated-display-mode/12415) and I thought it can't be too hard 😁

I looked at the code for the SDL simulation and I realized that `LV_EVENT_RESOLUTION_CHANGED` for a `lv_disp_set_rotation(...)` is not enough. Since `lv_disp_t` is now opaque, the display driver needs the rotation information as well. At least that is my conclusion. Are there already plans how this will be solved?

kisvegabor, 2023-07-31:
Hi @nylnx,

The display can subscribe to any display events too. E.g. in  `my_driver.c`
```c
lv_disp_add_event(disp, LV_EVENT_RESOLUTION_CHANGED, my_disp_update_cb) ;
```

nylnx, 2023-07-31:
@kisvegabor Yes, that also happens in the SDL display driver. The problem is that a screen rotation is not a resolution change. A resolution change just says you need to adjust the frame buffer and nothing more. From a resolution change you cannot infer the rotation that happened. You could use `lv_disp_get_rotation()` to determine if the rotation changed, but that's also not working and generally a bad idea. This is because the the display  driver, as a subscriber of this event, does not need to act upon every rotation event, because there is also `sw_rotate`. And this property is not part of the public API as far as I can tell. In my opinion, there should also be a `LV_EVENT_DISPLAY_ROTATION` and some way to know who has the responsibility for the rotation. Which also fits nicely with the above discussion about capabilities of devices.
Imho, the correct behavior is as follows:

1. `lv_disp_set_rotation()`
2. Event `LV_EVENT_DISPLAY_ROTATION` distribution
3. Display driver is subscribed to the event and gets called (or if it doesn't support rotation anyway it can also not be subscribed)
4. Determine if the framebuffer needs adjusting and do it accordingly
5. If `sw_rotate == false`, use the rotation capability of the display, if `sw_rotate == true`, do nothing.

And in case of the SDL simulation, the actual desktop window size should never change, regardless of the rotation currently set, because the window is like actually holding a display. It does not physically turn, just because you set a rotation on the device. That's why there is a distinction between the physical rotation and the rotation inside LVGL.

kdschlosser, 2023-07-31:
rotation should be handled at the hardware end of things. I am sure that SDL has built in mechanics to handle this kind of thing it's just a matter of instructing SDL to do the rotation. As far as rotation when using an MCU like an ESP with an attached display most displays allow for setting the orientation and will handle doing that internally. For the cases where software is needed to perform the rotation I think this should be done in the flush function for the frame buffer. There would need to be a second buffer allocated to do this and LVGL can provide a function to handle it.

Think about it, how many users would use a software based rotation? I don't believe there would be that many. no when dealing with larger displays. small displays, maybe.. but the performance impact on those smaller displays wouldn't be too bad. Most people developing for larger displays and need to have rotation available to them are going to opt for hardware rotation vs software because it is able to be done faster.

So having LVGL check to see if the buffer needs to be rotated for each and every single UI element needs to be rendered is doing to add up. The question becomes is it worth doing for the small number of cases that need to have it available? and can it be done in a manner that would not impact performance for all of the users that have hardware rotation or don't need rotation at all??

setting the rotation using a macro is not the way to go because statically fixing the rotation is not going to allow things like wearables or handhelds to change the orientation based on the way the device is being held.

nylnx, 2023-07-31:
> Think about it, how many users would use a software based rotation? I don't believe there would be that many. not when dealing with larger displays. small displays, maybe.. but the performance impact on those smaller displays wouldn't be too bad. Most people developing for larger displays and need to have rotation available to them are going to opt for hardware rotation vs software because it is able to be done faster.

I dream of a world, where every rotation happens in hardware, but it's just not reality. I may reference the thread I have with @kisvegabor about a [Pre-rotated display mode](https://forum.lvgl.io/t/pre-rotated-display-mode/12415). I, myself, am in this situation (no hw rotation, large display) currently and there is nothing I can do or could have done about it. Especially for cost-reasons, people can, and will, end up there. As a reminder: The whole STM32 Cortex-M line has nothing that facilitates hardware rotation.

>  I am sure that SDL has built in mechanics to handle this kind of thing it's just a matter of instructing SDL to do the rotation.

Don't worry about SDL... SDL, as least for me, is a simulation. Restrictions barely apply here and my suggestions are most certainly not focused towards SDL. It's just the easiest example and the code is already there.

> For the cases where software is needed to perform the rotation I think this should be done in the flush function for the frame buffer. There would need to be a second buffer allocated to do this and LVGL can provide a function to handle it.

Handling the rotation as part of the flush callback would be fine I think, as long as LVGL provides well-optimized and tested rotation utility functions. The minimum would be a rotation function that rotates a buffer of any size.

> setting the rotation using a macro is not the way to go because statically fixing the rotation is not going to allow things like wearables or handhelds to change the orientation based on the way the device is being held.

There's more about this idea in the forum thread above. But if you wanted to do something with static code, it would probably be something like defining a new default rotation and then have rotation defined as relative to this rotation. But as discussed in the forum thread, there should be better compromises.

kdschlosser, 2023-08-01:
Static rotation would not be something that is needed if there is a utility function available in LVGL to do the rotation of the entire buffer. Rotation is already possible on a widget by widget bases so maybe we look at the simple solution. The "screen" is an `lv_obj` and an `lv_obj` is a widget so the ability to rotate is a matter of rotating the bottom most object the amount that is wanted. All items within that object should transform into the correct locations and having rotated themselves as well. IDK if that holds true but it should. If a parent object is rotated then everything within the object should move with it as well.

kisvegabor, 2023-08-01:
In v9 we will approach it slightly differently. I've just merged the `draw_buf` update into master which unifies how draw buffers are handles (width, height, color format, stride, etc) and we can do operation on these buffers as well. 

Besides clearing or coping the buffer, there will be a `lv_draw_buf_rotate()` function too which can be implemented in a platform specific way. We provide a SW rotation implementation but if you have some GPU you can have a custom `lv_draw_buf_rotate()` function. 

You will be able to call thin function in `flush_cb`, or maybe the driver will call it automatically if it can't rotate the display from hardware. This way there won't be `sw_rotation`, but the driver should solve it somehow: either by calling `lv_draw_buf_rotate()` or by reconfiguring the display.

Regarding the event, I think there is a resolution change. If the display is 240x320 and you rotate it, it will be 320x240. Of course there is no change if a 240x240 display is rotated. So maybe we should fire `LV_EVENT_RESOLUTION_CHANGED` only when really there is a resolution change and `LV_EVENT_DISP_ROTATION_CHANGED` on well... rotation change.

nylnx, 2023-08-01:
> Besides clearing or coping the buffer, there will be a lv_draw_buf_rotate() function too which can be implemented in a platform specific way. We provide a SW rotation implementation but if you have [...]

I support that. That seems reasonable. Will take a look at how this code looks now :)

> Regarding the event, I think there is a resolution change. If the display is 240x320 and you rotate it, it will be 320x240. Of course there is no change if a 240x240 display is rotated. So maybe we [...]

I would compare this situation to the famous rectangle/square inheritance example. It looks quite similar, but the constraints are very different. For me, these two things are very different. With a resolution change, the rotation stays the same and the buffer can get any new size. With a rotation the rotation changes, the buffer size may change, but **only** by swapping width and height. I would advise against being too clever and give the event consumers clear instructions what to do. Yes, it takes a bit more work to have `LV_EVENT_RESOLUTION_CHANGED` and `LV_EVENT_DISP_ROTATION_CHANGED` work correctly together, but it's worth it for the clear semantics and also possible optimizations that could be done based upon more context on how a particular resolution change happened.

kisvegabor, 2023-08-02:
> With a rotation the rotation changes, the buffer size may change, but only by swapping width and height

It's not always true. For example in v9 we support stride (The number of bytes in a line of a buffer. It can be larger than the visible width. Some GPUs has such requirements, e.g. to have the width always multiple of 64). With e.g. 64 byte stride alignment and RGB565 color format a 240 x 320 display we need a 512x320=163,840 byte buffer. With 320x240 we need a 640x240 = 153,600 byte buffer.

nylnx, 2023-08-03:
Okay, the resolution calculation can be a bit more complicated, but that's all the more reason to extract this calculation to a separate function inside the driver. 

My point still stands though. If anybody _ever_ has to/wants to know why a particular resolution change event happened, it will be a major pain in the a** to correlate `LV_EVENT_RESOLUTION_CHANGED` to a potential `LV_EVENT_DISP_ROTATION_CHANGED` which might have come before, come after or will never come. 

I can think of one plausible scenario why somebody wants to know if it was a rotation or a resolution change: If somebody wants to keep the frame buffer intact while rotating without a full redraw. If it is a split event, he has to clip the picture to the new resolution and the rotate it in a second step, but then part of the picture is already lost to the clipping. This could be avoided with dedicated events.

Could I maybe help to implement some of it? I would certainly fix the SDL simulation, but I could also try to implement these events or something 😊

nylnx, 2023-08-04:
> So this is the plan, right?
> 
> 1. Send LV_EVENT_DISP_ROTATION_CHANGED only on rotation change
> 2. Send LV_EVENT_RESOLUTION_CHANGED only on "real" resolution change

This is your decision 😄 If you approve it, then we should try to implement it in code and see if it works or not. I just reasoned from a theoretical standpoint and what I would do. Maybe I just don't understand LVGL enough and it's wrong. I could totally accept that 😊

kisvegabor, 2023-08-06:
I just wanted to summarize it to be sure we meant the same thing.   If you meant it as well, I'm ok with it. If you have some time, please send a PR :slightly_smiling_face:

nylnx, 2023-08-16:
> > This is because not every display supports brightness, or hardware rotation or even a hardware reset. so having to do a check to test for an event that there is not going to be anything done for is a waste of time to make the function call in the first place. a simple bit check in the event handler to determine if the event even needs to get past the event handler.
> 
> It's a good idea, but I think we shouldn't mix it with the events but store the capabilities of the display in `lv_disp_t`. E.g.
> 
> ```c
> disp->features = LV_DISP_SUPPORT_BRIGHTNESS | LV_DISP_SUPPORT_HOR_MIRROR;
> ```

I am currently trying to get the rotation handling correct. My current struggle is with the resolution change. A resolution change is not a simple bit mask feature and requires a two-way communication to be done properly. The `lv_disp_*` interface would have to ask the display driver, if the requested resolution works and then synchronously execute this change and return a status code for it. There are things where a simple feature bitmask + event works (like rotation), but for a resolution change, it's not that simple. At least that is my interpretation of the situation.

I would also like if the `lv_disp_*` interface was split up by user and driver functions. There are functions only a driver should ever use like `lv_disp_get_hor_res()` or `lv_disp_set_draw_buffers()`, while others like `lv_disp_set_rotation` obviously have to be used by a user.

And as a sidenote: I'm not sure if the current code for single/double-buffering supports all the buffering methods there are. It may be too inflexible. Maybe take a look at it again and prevent future pain if anyone comes asking for them 😅 Embedded Wizard has a good list of the methods that should be supported: [Framebuffer Concepts](https://doc.embedded-wizard.de/framebuffer-concepts)

kisvegabor, 2023-08-17:
> The lv_disp_* interface would have to ask the display driver, if the requested resolution works and then synchronously execute this change and return a status code for it. 

My way of thinking was that, the user shouldn't change the resolution to a not supported value. :smile:  But, yeah, it would be great if we couldn't set the resolution to a not supported value at all. What should happen in this case? `lv_disp_set_res` should return an error code?

>  I'm not sure if the current code for single/double-buffering supports all the buffering methods there are. It may be too inflexible.

With 1 or 2  buffer and `PARTIAL`, `DIRECT`, and `FULL` modes and the flexibility to do any magic in the `flush_cb` I think we have a pretty good coverage. Embedded Wizard's list is really a good one, but it seems to me we can support all of them. Is there particular option which worries you?

kdschlosser, 2023-08-17:
Remember when I said that all functions should return a way to know if the function succeeded or failed? This is exactly why they should be doing that. All of the `set` functions should return a 1 or a 0.. 1 would be a success and 0 a fail... Tho if you flip the so 0 is success and 1 is a fail you can return different integer values to let the user know what the exact failure is.

IDK how c99 does a boolean evaluation of a negative number but in Python they evaluate to True, only a zero evaluates to False.

kisvegabor, 2023-08-18:
> All of the set functions should return a 1 or a 0

Yeah, we can really work on the gradually. It won't be API braking thing, so let's finalize the main things of v9 first. :slightly_smiling_face: 

> IDK how c99 does a boolean evaluation of a negative number but in Python they evaluate to True, only a zero evaluates to False.

It's the same in C.

kisvegabor, 2024-01-14:
The first driver are added here: https://github.com/lvgl/lvgl/tree/master/src/dev/display

Closing this issue.

closed 2024-01-14 by kisvegabor