lvgl/lvgl #4059: [Parallel rendering] Masking

kisvegabor, 2023-03-14:
Part of #4016

### Introduce the problem
In the current stage of parallel rendering an lot of drawing related features are disabled. We should reconsider and re-implement them. This issue is to discuss how to handle masks, effects, etc in the new parallel render architecture.

### Examples and cases

Here are a few use-case for which we should have a solution in the end. I also add my proposed solution (see their description in the next section)

**Mixed use cases**
- Alpha gradient: Use `lv_color32_t` in APIs and make gradients deal with the alpha channel too.
- Music player demo stripes around the cover: Use triangles (now they are trapezoids but in v9 Sw render shouldn't support polygons)
- Chart faded area (like in the widgets demo): Use right angle triangles with gradient
- Transform widget parts (e.g. slider knob rotation) : Allow easily creating and managing new layers manually  in draw events
- Masking
   - Clip corner: Built-in solution for better optimization, render children to a new layer if they overlap the corner and apply mask
   - Bitmap or text mask to any drawing or layer: There are special "mask draw tasks". If they are added the widget are drawn to a new ARGB layer and that layer is masked by the masking draw tasks.
- Blur and other filters: `post process cb`
- Backdrop filter (e.g. blur): ???
- Drop shadow on ARGB images: `post process cb`
- Real time outline for ARGB images: `post process cb`
- GPU effects (e.g. contrast): `post process cb`
- Perspective transform: just implement it
- SVG/vector graphics: generic API  + 3rd party lib 

**Software rendered primitives** 
- Line:
    - idea: use A8 image rotation 
    - dash skewed: dash the A8 image
    - square and rounded line ending: add rounded caps to the A8 image
    - line joint: not supported, use SVG instead  
- Arc
  - square and rounded line ending: custom mask, some local masking magic 
  - process in 16x16 blocks to easily ignore chunks in the middle 
- Rounded rectangle and border
  - different radius on corners
  - process large radius in 16x16 chunks to have simplify squares on not radius parts.
- Triangle
   - simple anti-aliasing
   - polygons are not supported: use SVG lib if needed 
- Image
  - Transform the image in 8x8 squares to easily discard 64 px if they are out of the original image

**Not strictly related**
- Text on path
- Text shadow
- Gradient on border and shadow
- Multiple border,s outlines and shadows

### Suggested solution
- Have built-in mask-related draw tasks applied only on ARGB images (layers). It's like `LV_DRAW_TASK_TYPE_IMAGE` but it performs masking. The following mask sources should be supported:  bitmap,  text, rounded rectangle
- `post_process cb`:
  - A style property which can add new draw tasks e.g. for masking, blurring, etc. It's called when all children's draw tasks are added. If there is a `post_process cb` the widget with its children are drawn on a new layer and that layer can be manipulated independently of the other parts of the screen.  
  - An other option is to add new draw tasks in `LV_EVENT_DRAW_POST_END`. The object needs to know to create a new layer. 
- clip corner:  A special mask (i.e. rectangle mask with radius). It should be applied only on the direct children if they are on a radius area. Not called from `post_process_cb` as LVGL needs to do some special considerations with it and enable it only if needed.
- new layer during drawing: Make it simple to manually create nee layers. E.g. in a draw begin event create a layer and continue the drawing there. In a draw end event merge the layer back to the original (parent) layer. It can be done automatically if a part has transformations, or anything requiring a new layer.

---

I'm thinking about these for months and I believe it's hard to see how the pieces fit together, especially as it's heavily related to the new parallel rendering  architecture. However, if you have any idea or remark about, I'd be more than happy the hear it :slightly_smiling_face: 

I can take care most of the implementation but it'd be amazing if you could bring up other use cases that we should consider.

GorgonMeducer, 2023-03-14:
> Alpha gradient: Use lv_color32_t in APIs and make gradients deal with the alpha channel too.

Do you plan to use ARGB8888 for internal processing?

kisvegabor, 2023-03-14:
> Do you plan to use ARGB8888 for internal processing?

Definitely not. Just in the API to set the colors.   E.g. before filling an area with a color the ARGB8888 color should be converted to e.g. RGB565 (according to `LV_COLOR_DEPTH`).

GorgonMeducer, 2023-03-14:
If that's the case, I fully agree with you that using lv_color32_t in API is definitely a good idea.

kisvegabor, 2023-03-15:
> About gradient, I notice that LVGL only supports two gradient modes, horizontal and vertical. I would suggest you adding support to points based gradient, for example:

You can set the number of max gradients point with [`LV_GRADIENT_MAX_STOPS`]( https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L317).

In a minor version I agree to add more gradients, like skew, circle or conic.

> For ARGB8565, I highly recommend deprecating this color format and moving to RGB565A8.

We are using it internally when a new layer is created and it needs to have alpha channel too. I'll check if we can switch to `RGB565A8`. I'm concerned that it will cause too many branches due to the not tightly packed pixels.

nicusorcitu, 2023-03-15:
The new VGLite GPU will support more features than RT1170 - almost everything is needed for drawing:
**- It will support A8 masking as well as path masking**
- It will support gaussian blur - so we can draw also the shadows
- draw letters (path letters and raster letters)

I believe we shall target the LVGL re-architecture not only for simpler blitters (like PXP) but also for complex GPUs (like VGLite). Maybe at some point the GPU will take care of all of the draw request. Just splitting them in very small parts and queuing them (even as a raw) will not be efficient.

kisvegabor, 2023-03-17:
It sounds amazing! It seems to me these features can be realized with the planned architecture as we describe the draw tasks from a quite high level.

The only thing is not clear to me is how to handle RGB565 + alpha. Now LVGL uses the non standard ARGB9565 color format but @GorgonMeducer [suggested](https://github.com/lvgl/lvgl/issues/4059#issuecomment-1468230452) using RGB565+A8 (all colors first and all alpha bytes after that). I think RGB565+A8 would be easier to handle with VG-Lite too. What to do you think?

tdjastrzebski, 2023-03-17:
I think this discussion ought to be put into broader perspective.  
What is the current goal of LVGL?  
Is it the best performance on current mainstream hardware or, maybe, features are even more important than performance?  
If so, what hardware is currently the most popular or what features are missing the most?  
What improvements are needed the most and how do you know that?

kisvegabor, 2023-03-17:
Great questions!

> What is the current goal of LVGL? What is the ambition and motivation?

I couldn't say a  single stunning marketing sentence but I see these:
- remain the most popular free and open source embedded GUI library
- keep the platform independent and easy to integrate nature of the library
- keep its easy to use nature
- while supporting low end devices and have low memory footprint, support high-end devices too (powerful GPUs, video cards, multi core MCUs, etc)
- have features allowing to create stunning UIs
- encourage contribution
- reach financial  stability, i.e. find some funds, ideally have some full time employees too (I'm already on it) 

> Is it the best performance on current mainstream hardware or, maybe, features are even more important than performance?

As the performance of the hardware is growing rapidly, on the long run features will be more important. However, the low end devices will be with us too for many years and we should support them too. So we should have highly modular ans scalable architecture.

> If so, what hardware is currently the most popular or what features are missing the most? What new hardware is coming?
Or, maybe what is needed the most is stability and quality? What improvements are needed the most and how do you know that?

I think many things are already working well: we have many widgets, great style and layout systems, multi language support, low memory footprint. However as I see  two main things are missing and we try to address these in v9:
- have drivers in LVGL to make it easier to get started
- more effective support for MCU GPUs and 3D GPUs too. 

I'd be very curious about your opinion too. Maybe we should open a new off-topic issue for it as this is not strictly related to how we should implement masking.

tdjastrzebski, 2023-03-17:
Let's take goals 1 and 5, because others kind of only support them:
* **have features allowing to create stunning UIs**
* **remain the most popular**  

Now, among whom? Enthusiasts or professionals?
If financial stability is considered, professionals may count more, but it is only my blind assumption and could be wrong.  
But let's assume I am correct. What do professionals want? What hardware do they use these days designing `new` solutions?

E.g. how would you go about creating packet mp3 player with a stunning UIs? I mean, on embedded device, because DirectX, Unreal Engine etc. or even Chromium is not something to try to compete against. What hardware would you choose? - especially knowing that GUI, more often than not, although important, is just a GUI and device must (e.g.) primarily smoothly play mp3, with no hiccups when rendering screen? Do you see my point?

tdjastrzebski, 2023-03-17:
Today my hardware pick would be STM32U5 - see [this video I just made](https://youtu.be/WbMabPVxtAw).
Surely NXP VGLite GPU is another one. What are the other best options?

GorgonMeducer, 2023-03-17:
STM32U5 with DMA2D is an appropriate choice as a mainstream solution. It is based on Cortex-M33 (slightly better than Cortex-M4F in terms of performance). LVGL should deliver 30FPS performance for most of the scenarios. 

If we target M7-based devices (and VGLite style of GPU) and adjust our resource and performance expectations, the mainstream devices, i.e. M4/M33-based devices will suffer.

kisvegabor, 2023-03-18:
I've just opened https://github.com/lvgl/lvgl/issues/4073 and summarized my thought in more structured way. Could you comment on it?

tdjastrzebski, 2023-03-18:
Note that new STM32U5xx does not only have DMA2D, which good support, I believe, should be among TOP priorities.  
STM32U5 also comes with NeoChrom GPU - ***GPU2D***! I think currently it is only supported by by TouchGFX.
Besides DMA2D, this would be my my next target because this is where technology is going.

nicusorcitu, 2023-03-20:
> The only thing is not clear to me is how to handle RGB565 + alpha. Now LVGL uses the non standard ARGB9565 color format but @GorgonMeducer https://github.com/lvgl/lvgl/issues/4059#issuecomment-1468230452 using RGB565+A8 (all colors first and all alpha bytes after that). I think RGB565+A8 would be easier to handle with VG-Lite too. What to do you think?
>
The old VGlite GPUs does not have support for neither A8RGB565 or RGB565A8. They will not support masking as was till present.

But the new VGLite will have support for both, not sure which one will be more convenient. Probably is the same performance.

> - It will support A8 masking as well as path masking
>
What I am trying to say is the new VGLite GPU will support A8 masking, which is different the way LVGL approach it.
My understanding is that you desire to apply masking at the end of drawing, e.g. if you want to draw this
[image]
you will draw a full rectangle with gradient, then you will apply (blend the result with) an A8 mask so that the left up corner will become transparent. Am I right?
With A8 VGLite masking, you do not have to do that in two steps. And you do not have to draw the full rectangle.
You can pass the A8 mask to the GPU so that it will draw only the visibile part of the gradient. So more efficient.

Having that, is it possible to take into consideration to keep the rectangle task with gradient and mask? At least for the beginning until we find out more possibilities on how to approach masks in general.

kisvegabor, 2023-03-20:
> FYI, arm-2d uses the same method, it is called filling-colour-with-mask-and-opacity. You can consider to update the render and take advantage of those features.

Besides filling with a color what other operation can VGLite and arm-2d do with A8 masks? If it's only filling we can have a special draw task for it.

Note that, I wouldn't like to force any GPU to approach rendering as the SW render does. (In the SW render we have a lot of limitations and trade offs.) Instead we should have high level and abstract draw tasks that can be drawn as the GPU wishes.

E.g.  this special chart section is now drawn by masks which is very specific. 
[image]

Instead these should be right angled triangles with a gradient. A triangle is much more generic and it's up to the GPU how to draw it.

Another option is to make LVGL (SW render and GPUs) capable of rendering in A8 too (besides the selected native color depth). This way LVGL could render masks and could say to use these masks for filling. However it'd have some disadvantages:
-larger code size
-the mask needs RAM 
-it's less generic as we say that the chart's triangle should be drawn with a mask, but maybe the GPU could do it faster/better

So in general I'd say having generic primitives seems reasonable. If we need special masks or effects for any reason (clip corner, fading all children, blur, etc), LVGL will create a layer (image) and do the operations on it.

GorgonMeducer, 2023-03-20:
> Besides filling with a color what other operation can VGLite and arm-2d do with A8 masks? If it's only filling we can have a special draw task for it.

For arm-2d, we have:
- copy-with-mask-and-opacity (already used in accelerating LVGL). 
- tile-fill-with-mask-and-opacity
- filling-colour-with-mask-opacity-and-transform

kisvegabor, 2023-03-20:
Thank you. 

I'm thinking about if it's worth to add an A8 mask parameter to all primitive drawings but I can not really find a use case when a simple draw task should have a bitmap mask. E.g. to render a rounded rectangle arm-2d can get a rounded rectangle mask from the SW render's mask engine. But VG-Lite can just draw a rounded rectangle on its own without any masks.

But what would be use case when the bitmap mask is already known on draw task level. Or just leave this opportunity open for any custom drawing?

tdjastrzebski, 2023-03-20:
Form ST DMA2D perspective I think it would be the best to internally always use ARGB8888 interim buffer.
Although DMA2D can blend A8 with RGB656 and produce ARGB8888 or ARGB4444, it cannot combine bitmap with a mask and blend with a background in one operation.
It needs to do it in two steps using interim buffer, which every time has to be created and then released unless one permanent buffer is kept which could only be increased if/when needed but then more total memory is required and two blending operations are needed while when always using ARGB8888 buffer no extra memory and blending would be necessary.

GorgonMeducer, 2023-03-20:
at least, @tdjastrzebski and I don't like ARGB8565.

tdjastrzebski, 2023-03-20:
@GorgonMeducer I think I disagree, AHB in not the only factor and I believe other reasons to use ARGB888 prevail.
Btw, maybe it would be a good idea to merge PR #3904 (stm32 dma2d rework) with rev 9?

tdjastrzebski, 2023-03-21:
@GorgonMeducer Besides, I think that the decision which platform currently (e.g. M4/M7/M33/M55) should have the most attention should be based on strong evidence. By no means I am saying that I disagree, but how do we actually know?

@kisvegabor I guess, what I really suggest is implementing more formal Product Management practices. In my view, LVGL reached the stage where it is a must to successfully move forward.

nicusorcitu, 2023-03-21:
> Besides filling with a color what other operation can VGLite and arm-2d do with A8 masks?
>
The mask enablement is global, once it it set it will be valid for all kind of operations blit/draw until is disabled.

[image]

> E.g. this special chart section is now drawn by masks which is very specific.
> Instead these should be right angled triangles with a gradient. A triangle is much more generic and it's up to the GPU how to draw it.
> 
For this specific case, it is fine if we draw a rectangle and a rectangle with gradient. Only that you have to recalculate the gradient for triangle. But should be fine even thought it can be done as a single time.

> I'm thinking about if it's worth to add an A8 mask parameter to all primitive drawings but I can not really find a use case when a simple draw task should have a bitmap mask. E.g. to render a [...]

I can give you two more examples of masking:
1. the meeter
[image]
All the lines from are draw by having two points (one is the center of the circe and the other one on the circle circumferince). These lines are "shorted" because in the middle of the meter there is a mask "cutting" those lines. Without the mask the lines would have been till the center of the circle.
With next VGLite for example we can set that mask once and then draw every line till the end when we can disable the mask. We don't have to draw the full line and then blend with A8.

2. slide bar
[image]
Without the mask it would have looked like this:

[image]

With VGLite we can also rotate widgets. We can draw lines, rectangles, etc and rotate them at any degree.

tdjastrzebski, 2023-03-21:
@kisvegabor Always using internal 32b buffer (ARGB8888/xRGB8888+A8) is just a thought, but maybe not such a crazy one.
I think it would simplify many things. The real question is whether or not the **target technology** is already there to handle it without scarifying performance. If not yet, it will be there - it is just a question of time, but how do we know when this time arrives? How will we know that it no longer makes sense to spend effort on maintaining both 16 and 32b graphic? What is the **threshold**? Converting from ARGB8888 to RGB565 in flush_cb() is a simple task if/when needed, with or without DMA2D.

GorgonMeducer, 2023-03-21:
> Besides, I think that the decision which platform currently (e.g. M4/M7/M33/M55) should have the most attention should be based on strong evidence. By no means I am saying that I disagree, but how do we actually know?

You can start a survey and find the platform with the highest volume. Or you can check with silicon vendors, ST for example, why they keep adding 2D GPU to the mainstream Cortex-M processors while DMA2D is already there.

I cannot provide you with strong evidence. I cannot. But I can put some hints.

tdjastrzebski, 2023-03-21:
@GorgonMeducer How about a survey among LVGL users? What projects do they start, what MPUs do they use/consider consider?
>  Using RGBA8888 as internal color format isn't as efficient as RGB565 + A8 in terms of bus bandwidth.

I do not disagree, but how much less efficient? That's what I meant by hard data.

GorgonMeducer, 2023-03-21:
> How about a survey among LVGL users? What projects do they start, what MPUs do they use/consider consider?

It's good idea, and we need the support from @kisvegabor .

> I do not disagree, but how much less efficient? That's what I meant by hard data.

Just run the benchmark (set depth to 16bit) and compare the result of topics that involve ARGB8565 or RGB565A8.

tdjastrzebski, 2023-03-21:
@GorgonMeducer I would add one more yet important factor: DMA2D and similar 'accelerators'. If used efficiently, they offload CPU. It may make a lot of sense to use it even if FPS is actually lower, but much more cycles can be used for stuff other than graphic in exchange. Let's do not forget that nice graphics often is only "nice to have", but it is NOT a top feature. That is why I advocate for LVGL architecture which at least enables efficient DMA2D usage. GPU2D support is next, but it may not be such a 'low hanging fruit'.

GorgonMeducer, 2023-03-22:
@tdjastrzebski I agree with you about the "good enough" statement. Based on this, reducing the memory footprint is much more reasonable.

tdjastrzebski, 2023-03-22:
@GorgonMeducer One of the troubles I discovered reimplementing ST DMA2D support (#3904) was that blending RGB565+A8 with the background requires additional interim memory buffer, which is not the case when xRGB8888+A8 is used. Neither RGB565+A8 nor xRGB8888+A8 can be used by DMA2D directly but xRGB8888+A8 is easier to handle, because it is easier to produce ARGB8888 from xRGB8888+A8 than from RGB565+A8 before DMA2D can take it over. Note that DMA2D does not support 24b ARGB8565 so RGB565+A8 must be converted to ARGB8888 first and that is where extra RAM is required.

kisvegabor, 2023-03-22:
@tdjastrzebski 

I seems to me the main motivation behind suggesting to use ARGB8888 internally is to allow DMA2D (and similar GPUs) work more effectively.

I believe the  architecture of v9 is flexible enough to handle this. Assume these cases
- The display has normal RGB565 format but the user want to use an RGB image: No problem DMA2D also knows RGB565, so RGB565 images can be used directly.
- The display has normal RGB565 format but the user wants to use ARGB image. 
    - Option 1: Use ARGB8565 or RGB565 + A8 image. DMA2D cannot be used
    - Option 2 : Use ARGB8888 images and let DMA2D do the pixel format conversation
- The display has RGB565 + alpha format: Not typical as hardware doesn't support ARGB8565 and ARGB8888 is used instead 
 - An internal LVGL  layer has ARGB8565 format: It can really happen e.g. when a rectangle with rounded corners are rotated. DMA2D can't be used for this format at all. Unless, we allow controlling the color format of the layers too. So the layer's color format could be set to ARGB8888 if we want (based on some policies). The problem with that is  `LV_COLOR_DEPTH` is `16` but we will want the SW render to work in 32 bit mode. So we need use different color depths at run time. I wanted to avoid it for the sake of simplicity and code size but let's discuss it and we will see.
  
> Note that DMA2D does not support 24b ARGB8565 so RGB565+A8 must be converted to ARGB8888 first and that is where extra RAM is required or DMA2D cannot be used in such case - and currently it is not.

It's very important that the images can have any format. The the only thing matter is that in the end they need to rendered in a the color format related to `LV_COLOR_DEPTH`. So it's absolutely possible to have an ARGB8888 image and render it to an RGB565 buffer by letting DMA2D make the color conversation on the fly.

GorgonMeducer, 2023-03-22:
> RGB565 + A8 is really an option as target rendering color format too.

OK, I didn't notice this before. In terms of memory usage, it is the same as ARGB8565, and it is more efficient (because of fully aligned memory access). I would suggest using RGB565-A5 most of the time.

nicusorcitu, 2023-03-22:
> In terms of memory usage, RGB565-A8 it is the same as ARGB8565, and it is more efficient (because of fully aligned memory access).
>

The reason you suggest is more efficient is that ARGB8565 is packed as 3 bytes, while RGB565 is packed 2 bytes and A8 is packed separated as1byte.

When the GPU needs to fetch a pixel, it will fetch twice for RGB565 A8. It needs to read from two different addresses.

Instead, when GPU fetch a pixel from format ARGB8565, it will read only once. And the remaining 1 byte will be cached for the next operation. Am I right?

How is the  RGB565 A8 more efficient?

GorgonMeducer, 2023-03-22:
@nicusorcitu You are talking about THE GPU... and only about the THE GPU you are talking.

I can raise a different example: 

other GPU or 2D-capable DMA might have a different understanding. Just like the DMA-350 can take RGB565 array, background pixels and A8 array with different channels and compose them together with a stream interface (so you can plug an alpha-blending logic to that stream interface) and store the result to a output channel. 

It is more efficient. 

If we go back to software processing, for those MCU without DMA2D, RGB565-A8 is more efficient than ARGB8565.

tdjastrzebski, 2023-03-22:
@kisvegabor 
> the main motivation behind suggesting to use ARGB8888 internally is to allow DMA2D (and similar GPUs) work more effectively.

corect

ST DMA2D can handle RGB565/RGB888 conversions back and forth, with or without alpha. That is not an issue.
The problem occurs when bending of RGB565+A8 image **with existing background** is required. In this case RGB565+A8 image+mask needs to be converted to ARGB8888 first. The same is true for xRGB8888+A8, but RGB565+A8 requires an extra buffer while xRGB8888+A8 does not since xRGB8888 has one unused byte where A8 can be simply copied.

GorgonMeducer, 2023-03-22:
> always use ARGB8888 instead of RGB565+A8, but only for the relatively small internal buffer, which keeps images to be blended, not for anything else like flash-stored graphics etc.

I think you are right. I agree with you now.

kisvegabor, 2023-03-23:
@tdjastrzebski 
> The problem occurs only when blending of RGB565+A8 image with existing background is required. In such case RGB565+A8 image+mask needs to be converted to ARGB8888 first.

However the user always have the opportunity to use ARGB8888 images even if `LV_COLOR_DEPTH` is `16`. We just need to make it clear using ARGB8888 image need more memory but they can be accelerated better with GPU.

kisvegabor, 2023-03-23:
Something just came to my mind. We can mix to the two ideas (ARGB8888 everywhere and `LV_COLOR_DEPTH` related color format).

Here are the possible cases again:
- **Display has RGB565 format**: no problem, both GPU and CPU can handle RGB565 images, GPU can convert ARGB8888 on the fly to RGB565, CPU can deal with RGB565+A8, ARGB8565 or whatever.
- **Display needs to have transparency** (e.g. LVGL is used to render an OSD menu): TFT driver supports only ARGB8888 for this case (maybe ARGB4444) so no need for supporting ARGB8565 or other similar "hacks". 
- **An LVG layer has RGB565 format**: no problem, same as display with RGB565 format
- **An LVG layer needs to be rendered with alpha** (e.g. a widget with corner radius needs to be drawn on a layer): we can say to support only ARGB8888 in this case. So if a layer needs to have alpha, it's color format will be set the ARGB8888. If it doesn't need to have alpha the native color format will be used (e.g. RGB565).

In summary:
- Use the native format where possible (most of the cases), so we same RAM, flash and memory bandwidth
- For transparent screens support only ARGB8888 (maybe ARGB4444 later to save memory bandwidth)
- For layers with alpha support only ARGB8888. Now there is a dedicated darw_task (see [here](https://github.com/lvgl/lvgl/blob/feat/parallel-rendering/src/draw/sw/lv_draw_sw_img.c#L49-L65) and [here](https://github.com/lvgl/lvgl/blob/feat/parallel-rendering/src/draw/lv_draw.h#L37)) which blends the layers into an other layer (e.g. a rounded rectangle to a the main layer of the screen, maybe also applying transformation), so the whole ARGB8888 to whatever conversation problem is local to the implementation of that draw task. 

What do you think?

GorgonMeducer, 2023-03-23:
@kisvegabor No problem.

kisvegabor, 2023-03-23:
To approach it even simpler: 
- we can support any source format (regardless to the destination format) assuming that there is at least one draw unit that can handle it.
- for output formats we support RGB??? (based on `LV_COLOR_DEPTH`)  and ARGB8888.

And that's it.

If you all agree, I'll update the `feat/parallel-rendering` branch accordingly.

tdjastrzebski, 2023-03-23:
@kisvegabor this sounds good

kisvegabor, 2023-03-24:
@nicusorcitu, do you also agree with this approach?

nicusorcitu, 2023-03-24:
Yes, this is fine to me also.

kisvegabor, 2023-03-31:
I've added the discussed updated. Here is what have been changed:
- These are the native color formats for each possible `LV_COLOR_DEPTH`:
  - 8: L8
  - 16: RGB565
  - 24: RGB888
  - 32: XRGB8888
- So all are without alpha
- If LVGL needs to render with alpha it will always use ARGB8888 (we can add an option to support ARGB4444 later). So when a layer with alpha is required an ARGB8888 layer will be created which later needs to be merged to an other layer with either ARGB8888 or native color format.  
- As dealing with ARGB8888 source and destination buffers will be common they need to optimized in SW image draw functions. (native to ARGB8888, ARGB8888 to native, ARGB8888 to ARGB8888)
- I removed the ARGB8565 color format. As we need RGB565 + alpha only for images RGB565+A8 is enough. In summary these images formats can be used with each color depth if alpha is required too:
  - 8: A8  or L8A8 
  - 16: RGB565+A8
  - 24: ARGB8888
  - 32: ARGB8888

GorgonMeducer, 2023-04-01:
This is elegant.

tdjastrzebski, 2023-04-03:
@GorgonMeducer yep, elegant is the right word :)

kisvegabor, 2023-04-03:
Great! ^^

kisvegabor, 2023-04-17:
I've pushed a larger update and now most of the widgets work. 

This is what is still missing:
- line and shadow drawing
- `lv_color32_t` on APIs and alpha gradient
- remove all sw masks from the widget level
- image drawing with bitmap mask and/or parametric mask (details below) 
- animimg, bar, canvas, line, slider, colorwheel, chart is still missing
- clip corner
- support stride
- make all demos work
- make all tests pass

All seems clear but "image drawing with bitmap mask and parametric mask" should be discussed. My view about masking is that draw tasks can't receive mask parameters to keep things simple. These are the cases I considered:
- Label, line, and arc masking: not common. 
- A8 masked plain color: consider the A8 mask as an image and draw it with the given color
- Rounded corner mask on a plain color: draw a rounded rectangle

I think we can support A8 mask and rounded rectangle masks only on the images. It has the following advantages: 
- If label, line, or arc needs to be masked (rarely) LVGL can draw them to a layer (image) which can be masked. 
- With clip corner enabled LVGL can draw the children on the corners to a layer and the image can be masked during drawing.
- Text masking from images can be easily supported too as in v9 the font engines always return A8 bitmaps. So maybe we can add support for text masks too. 

In the end IMO it' will have good performance as most to things can be drawn "natively" and masking will be required only once in the last step.

Just to reiterate about the tricky masking cases and their planned solutions:
- Long tick lines on meter: increase the precision of line drawing so short lines can be drawn
- Gradient mask on bar: recalculate the gradient during drawing
- Chart faded area: use right angle triangles and rectangles below them  

It will be different topic these needs to be optimized later:
- dispatching: find independent areas
- break large areas into smaller once (needs to be discusses and benchmarked)
- non masked SW image drawing: 
   - RGB image to RGB layer (memcpy)
   - RGB image to ARGB8888 layer (color conversation might be needed)
   - ARGB8888 image to RGB layer (color conversation and masking)
   - ARGB8888 image to ARGB8888 layer (complex alpha calculation)  
- same masked SW image drawing

kisvegabor, 2023-04-25:
I've added `lv_color32_t` to style and other APIs but I don't like it. My main problem is that now RGB and A are bound together and it's difficult to set RGB and A separately. For example I want to set only the border opacity to 50% regardless to the actual color. But if we store the color together with the opacity value, we can't change only the opacity (as a value needs to written for the RGB part too in `lv_color32_t`). 

So I think having the color and opa separately  is more flexible. 

To support alpha gradients all we need to do is adding an `lv_opa_t opa` value [here](https://github.com/lvgl/lvgl/blob/38a9c2a8e2cafb3ff3960c7d8a1804ca0b6fd7f7/src/misc/lv_style.h#L152).

What do you think?

GorgonMeducer, 2023-04-25:
We can define lv_color32_t as follows:
```c
typedef union lv_color32_t {
    struct {
        uint8_t chn[3];
        lv_opa_t opa;
    };
    uint32_t val;
} lv_color32_t;
```

kisvegabor, 2023-04-25:
I don't see how it helps. 

So problem is this:
1. I set the border color in the default state to 0xFFAAAAAA (0xFF is the alpha)
2. I'd like to change ONLY the border alpha in the presses state, so I need to set 0x80AAAAAA

The problem is that in 2) when I change the alpha the RGB will be added too. And RGB part will determine a color,  but I wanted to change only the alpha. If in a 3rd step someone changes the default border color to  0xFF111111, the border color in pressed will be still 0x80AAAAAA.

This way styles, events, or animations cannot change only the alpha as they will set the RGB too.

GorgonMeducer, 2023-04-26:
> I change the alpha the RGB will be added too...

I don't understand. What does this "added" mean?

GorgonMeducer, 2023-04-26:
> you can't change only the opacity or the color as they are stored together.

I don't understand why you cannot change only the opacity.

```c
typedef union lv_color32_t {
    struct {
        uint8_t chn[3];
        lv_opa_t opa;
    };
    uint32_t val;
} lv_color32_t;
```
You can modify the opacity by using color.opa = xxxx without modify the RGB part. 
Based on your description, I guess what I said didn't ease your concern. 
But the fact is, I still don't see what the problem is.

kisvegabor, 2023-04-26:
I will try my best to explain it because I really would like to discuss this topic :slightly_smiling_face: 

In v8, the border color is composed from parts: the RGB part and the Opacity part. They are independent. Therefore it's possible to overwrite only the opacity of the border in pressed state. 

You can modify the opacity in ARGB888, it's fine. The problem is that if we store the RGB and the A part together and we want to set ONLY a different opacity value, you need store the RGB part too, as RGB+A are stored together.  

And once the RGB part is there as well, it will also affect the border color. But I wanted to set only the alpha.

In other words, in v8 if a style stores only `border_opa` we know that the opacity of the border needs to be set accordingly. However with ARGB8888 colors if we want to set only the border opacity we need to set `border_color = 0xffaabbcc`. How will LVGL know that it should ignore the `aabbcc` part and use on the `ff` part for the opacity?

GorgonMeducer, 2023-04-26:
> How will LVGL know that it should ignore the aabbcc part and use on the ff part for the opacity?

> How will LVGL know that it should ignore the aabbcc part and use on the ff part for the opacity?

How about considering RGB and OPA are CCC + OPA in the same data structure? Or why cannot LVGL just use `ff` part for opacity? What stop you to do that? 

On they other hand, how about add an extra opa for the whole border? So when dealing with ARGB8888, this opa is also calculated together with alpha part of each pixel.

nicusorcitu, 2023-04-26:
It means that when you change the border color again, you need to take account of the existing alpha and set:
```lv_obj_set_style_border_color(btn, color32(0x88AABBCC), LV_STATE_DEFAULT);```

nicusorcitu, 2023-04-26:
But why on gradient structure? Does every border have a gradient struct? Is this happening only on borders or are other cases too?

kdschlosser, 2023-04-26:
I do not understand why the opa would not be able to get changed separately if it is attached to the color? if you have a single 32 bit union for the color and only functions to set the color and not the opacity you set up the function so a pointer to the color is what gets passed. 

I am not proficient in C so I am sure the example below is not correct in syntax. But it shows the way to do it.

```c

lv_color_t* color = {.r=255, 'g=255, 'b=255, .a=255};

lv_obj_set_style_bg_color(obj, color);

color->a = 50;
lv_obj_invalidate(obj);
```

The ability to change the opacity is able to be done. It's not done through the use of a function but it still is able to be done,

The side benefit of setting it up this way is you gain the ability to do color comparisons.

```
lv_color_t c1 = lv_color_hex(0x555555);
uint32_t c2 = 0x555555;

if (c2 == (uint32_t) c1) {}
```

The above code will all evaluate to `false` 100% of the time. 

```

lv_obj_set_style_bg_color(obj, lv_color_hex(0x555555), 0);
lv_obj_set_style_bg_opa(obj, 0, 0);

lv_color_t c1 = lv_get_style_bg_color(obj, 0);
uint32_t c2 = 0x00555555;

if (c2 == (uint32_t) c1) {}
```

This also evaluates to `false`

```c
 lv_color_hex(0xFF555555)
```

That just plain fails all together. 

The single largest upside of using the 32 bit union all the time is the ability to set the color depth at runtime. Things like lv_canvas will now work properly. a single function can be made that would handle the color conversions when the buffer gets written to. So the only real expense is the additional 2 bytes used for each of the colors the widgets store. This could be reduced by use of styles so only a single set of colors is stored and they are shared across all widgets. There is no real large expense of doing the conversions at runtime, after all it's only bit shifting being done and processors are very efficient at such a simple math operation. Just like the flush callback function works a function can be made that would handle the bulk writing to the buffer. areas of the screen that are the same color get written to by this bulk writer so a color conversion is only done a single time for an area. calculate the bounding box 3where the color is all the same pass that information along with the color and the buffer to the function and let the function do the conversion and write the data.

IDK if the total program size would end up smaller because of the removal of the get/set opa functions and any other code that handles the opa.

You actually only have a cost of a single byte of memory not 2. I made a boo boo in that statement. so 1 byte of additional memory per set color. on a 32 bit screen you actually have a memory savings not a cost. because you have the 32bit color and then an additional byte for the opa. this way the program would have a smaller footprint. 

It also makes it easier for the user to deal with. the idea of 16 bit color, 24 bit color and 32 bit color causes confusion. it should be a transparent thing for the user. 

On a side note: It would make it possible to supply binaries for the MicroPython binding if the color depth and byte order was able to be set at runtime.

I am working on a CPython binding because of this little guy.

https://onion.io/omega2/

which runs Open WRT and is able to run CPython.

Not all displays that would get connected would use SDL2 and that is a fantastic use case where it would not. So having the ability to set the color depth and byte order at runtime would be beneficial

kdschlosser, 2023-04-26:
so right now in LVGL it check all of the different states for an object to see if the color matches and if it does it keeps a record of what states match each other and only stores a single color for multiple states?

sounds like a lot of overhead and memory use to do that. first you have to have a container that would be able to store all of the individual colors and opa's along with the states. then checking needs to be done each time a color gets changed.

You also have a waste of memory if the display is a 32 bit display. now you are actually using 40 bits to store a 32 bit color. colors should not be shared unless using a style and the share is only being done between widgets for the exact same attribute. so borders across all widgets, or shadows across all widgets. that is only if the same style is shared between widgets. If a color is set specifically on a widget and a style has been applied then the color gets stored and that gets used instead of the style. So out of the gate when LVGL starts up all widgets should have a style set to it and that style is shared across all widgets by default. This style should be able to be gotten by the user and the style updated if the user wants to make a change to the color across all widgets.

color propagation should take place when an object has a child created for it. When the first child gets made and has a parent the parent would check and see if there are any custom colors that have been set and if there has been a new style gets created and shared between the parent and the children of that parent. This propagation would NOT take place if the child was created without a parent and the child had it's parent set after it's creation. 

If the child's parent changes then the parents style and the child's style would get decoupled from each. colors that are the same as the new parent would be shared with the new parent via a style and all other colors get stored as custom colors. If the style that is set to the child is the same style as the one that is set to the new parent then nothing needs to be done

kisvegabor, 2023-04-26:
LVGL's style system was inspired by CSS. You can read about the styles in LVGL in more detail [here](https://docs.lvgl.io/master/overview/style.html). 

I think [this comment](https://github.com/lvgl/lvgl/issues/4059#issuecomment-1523332411) summarizes the problem the best way. We can change the alpha channel in ARGB8888 separately. It's fine. The problem is that we can not store only the Alpha value in a style to tell LVGL  to change only the alpha value e.g. in pressed state. If store the the alpha as part of an ARGB8888 color the RGB part will there too, and it will overwrite the color in pressed state (not only the alpha).

GorgonMeducer, 2023-04-26:
As I guessed, others also don't fully understand the issue.
And I now finally see the problem too. My suggestion is to add a dedicated `opa` property.

[28 more replies]

closed 2023-07-05 by kisvegabor