lvgl/lvgl #4060: The function to rotate the display does not work.

mbrazdil, 2023-03-14:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The lv_disp_set_rotation() function to rotate the display does not work.

### To Reproduce

I'm working on a project that uses a development kit with an iMX RT 1060 processor + RK043FN02H display. I'm trying a basic example from the NXP SDK 2.13.0, which contains the LVGL library version 8.3.2. If I leave the example in its original state the image displayed on the display is in landscape and fine.
The project I am working on assumes a portrait display. According to the documentation, just set the lv_disp_set_rotation() function. Unfortunately, this does not work. If I set LV_DISP_ROT_90, the image seems to blur. Interestingly, the button that appears in this case responds to a press in the place where it should actually be.
I also tried the settings directly in the lv_port_disp_init() function, where I set disp_drv.sw_rotate = 1, disp_drv.rotated = 1 | 2 | 3. In this setting it doesn't work at all (black screen). I had to set disp_drv.full_refresh = 0 to at least rotate the image 180 degrees. For the 90 and 270 scale angle it also doesn't work. 
I don't know if it's a library bug or something else, but I posted it here just to be sure.

### Expected behavior
Correctly oriented content shown on the display.

### Screenshots or video

Correct display without code intervention
[image]

Result after calling the function lv_disp_set_rotation(NULL, LV_DISP_ROT_90) after library initialization
[image]

kisvegabor, 2023-03-16:
Probably the driver of that board uses `disp_drv->full_refresh=1` which doesn't support screen rotation.

mbrazdil, 2023-03-16:
Yes, the lv_port_disp_init() initialization function is set to full_refresh = 1.
When I set full_refresh = 0, the display flickers when the button is pressed and creates nonsense at the top of the display.
[image]

mbrazdil, 2023-03-17:
Maybe I got it all wrong. If I have the full_refresh = 1 flag set, no display can be rotated during software rotation (sw_rotate = 1)? I'm basing this on the draw_buf_rotate() function.
What about HW rotation? Here also does it matter if full_refresh = 1?

kisvegabor, 2023-03-17:
You can do two things:
1. use `full_refresh=1` and rotate the rendered screen sized image with e.g. PXP to an other screen sized buffer and display it. 
2. Use e.g. an1/10 screen sized buffer from rendering and `lv_memcpy` it to the frame buffer in the `flush_cb`. In this configuration `LV_DISP_ROT` should work.

mbrazdil, 2023-03-21:
1. I don't seem to understand exactly how the full_refresh flag works. When it is set to 1, does the whole display always refresh or is it something else?
2. I would like to take advantage of HW acceleration when my processor allows it (PXP, VG lite). According to the manual, PXP support should be implemented directly in the library, so why doesn't it work? I've been solving this problem for more than a week now and I can't find a universal solution and most importantly a functional one. I even read that maybe the processor I'm using doesn't support this, even though the HW acceleration of PXP is included ...
3. If I change the frame buffer to 1/10 of the display size, in my case I have a display resolution of 800x480 = 384,000 b / 10 = 38,400 b. Will I have to modify the flush function somehow? I would appreciate some complete example using such a frame buffer.

kisvegabor, 2023-03-22:
> I don't seem to understand exactly how the full_refresh flag works. When it is set to 1, does the whole display always refresh or is it something else?

Yes, exactly.

> I would like to take advantage of HW acceleration when my processor allows it (PXP, VG lite). According to the manual, PXP support should be implemented directly in the library, so why doesn't it [...]

The `release/v8.3` branch should contain the some updates for PXP and VG-Lite. Do you use this branch? 

> If I change the frame buffer to 1/10 of the display size, in my case I have a display resolution of 800x480 = 384,000 b / 10 = 38,400 b. Will I have to modify the flush function somehow? I would appreciate some complete example using such a frame buffer.

Yes, the flush_cb needs to modified too. You need to implement something like [this](https://github.com/lvgl/lv_drivers/blob/master/sdl/sdl.c#L163-L167).

(cc @nicusorcitu)

@nicusorcitu

mbrazdil, 2023-03-22:
Thank you for your answers.
I would like to come back to the possibility of using HW acceleration to rotate the image on the display. I've been trying to wade through the library, I currently have version 8.3.5. I came across the NXP folder where there are some functions for controlling the PXP peripheral.
If I set the LV_USE_GPU_NXP and LV_USE_GPU_NXP_AUTO_INIT flags to 1, it should set HW acceleration. I am using 2 frame buffers in my project. My current flush function looks like this:
```
static void DEMO_FlushDisplay(lv_disp_drv_t *disp_drv, const lv_area_t *area, lv_color_t *color_p)
{
    DCACHE_CleanInvalidateByRange((uint32_t)color_p, DEMO_FB_SIZE);

    ELCDIF_SetNextBufferAddr(LCDIF, (uint32_t)color_p);

    s_framePending = true;

    if (xSemaphoreTake(s_frameSema, portMAX_DELAY) == pdTRUE)
    {
        /* IMPORTANT!!!
         * Inform the graphics library that you are ready with the flushing*/
        lv_disp_flush_ready(disp_drv);
    }
    else
    {
        PRINTF("Display flush failed\r\n");
        assert(0);
    }
}
```
Can you advise me how to adjust this function after using the PXP function to rotate the image?

mbrazdil, 2023-03-22:
Yes, I use NXP SDK. That's more or less the basis of my question.
There are use cases in the SDK for using the LVGL library, but there is no use of PXP for image manipulation. The function mentioned above is directly from the SDK. That's why I was asking how to modify this function to use the functions for PXP that are directly in the LVGL library.

zejiang0jason, 2023-03-23:
Hi @mbrazdil , do you want to 1. Use portrait panel (for example: height 480, width 272), or 2. use the landscape panel, but show portrait images?
If 2, then the EVK-MIMXRT1170 LVGL project can be reference, the panel is portrait panel (height 1280, width 720), it uses PXP rotation to show the landscape mode project (lvgl_demo_widgets).
Thanks

mbrazdil, 2023-03-23:
I need to have the display physically in portrait and also display in portrait like a mobile phone.

mbrazdil, 2023-03-24:
So I'm reporting a shift in what I've been dealing with. Finally, after constant testing, I figured out a few things that helped me solve my problem.
1. Set the macros ```LV_USE_GPU_NXP_PXP``` and ```LV_USE_GPU_NXP_PXP_AUTO_INIT``` to 1.
2. I left the original 800x480 resolution setting.
3. I modified the ```DEMO_flushDisplay()``` function (the original version is listed above):
```
static void DEMO_FlushDisplay(lv_disp_drv_t *disp_drv, const lv_area_t *area, lv_color_t *color_p)
{
    lv_gpu_nxp_pxp_blit(s_frameBuffer[2], area, LCD_WIDTH, color_p, area, LCD_HEIGHT, LV_OPA_100, disp_drv->rotated);

    color_p = s_frameBuffer[2];

    DCACHE_CleanInvalidateByRange((uint32_t)color_p, sizeof(s_frameBuffer[0]));

    ELCDIF_SetNextBufferAddr(LCDIF, (uint32_t)color_p);

    s_framePending = true;

    if (xSemaphoreTake(s_frameSema, portMAX_DELAY) == pdTRUE)
    {
        /* IMPORTANT!!!
         * Inform the graphics library that you are ready with the flushing*/
        lv_disp_flush_ready(disp_drv);
    }
    else
    {
        PRINTF("Display flush failed\r\n");
        assert(0);
    }
}
```
Using the ```lv_gpu_nxp_pxp_blit()``` function, I achieved display rotation. 

I have a question for the creator of the LVGL library:
1. Is it correct that I call this function ```lv_gpu_nxp_pxp_blit()``` manually in the ```DEMO_flushDisplay()``` function? Shouldn't this function be called automatically when setting up the ```LV_USE_GPU_NXP``` and ```LV_USE_GPU_NXP_AUTO_INIT``` macros?
2. In the ```lv_conf_internal.h``` file I came across the macro ```CONFIG_LV_USE_GPU_NXP_PXP```. Where is it set? If I set ```LV_USE_GPU_NXP_PXP``` to ```1``` in the ```lv_conf.h``` configuration file, it is set to ```0``` in this file ```lv_conf_internal.h```. Is this ok? The same is true for the macros ```LV_USE_GPU_NXP_AUTO_INIT``` and ```CONFIG_LV_USE_GPU_NXP_AUTO_INIT```.

nicusorcitu, 2023-03-27:
@mbrazdil 
I am not the "creator of the LVGL" but I can give you some answers as I am dealing with the NXP GPU support:

1. Yes. You are implementing a DEMO_flushDisplay() specific to NXP platform, so it fine to call directly the PXP code over here.
Anyway you should have guarded that ```lv_gpu_nxp_pxp_blit()``` like this:
```
#if LV_USE_GPU_NXP_PXP /* Use PXP to rotate the panel. */
    lv_gpu_nxp_pxp_blit(s_frameBuffer[2], area, LCD_WIDTH, color_p, area, LCD_HEIGHT, LV_OPA_100, disp_drv->rotated);
#else /* Use CPU to rotate the panel. */
   // Use CPU for rotation
#endif
```

@kisvegabor I see that if we set the rotation mode by
```lv_disp_set_rotation()```
Later this will not be taken into consideration while rotating in FULL MODE:
```
static void draw_buf_rotate(lv_area_t * area, lv_color_t * color_p)
{
    if(disp_refr->render_mode == LV_DISP_RENDER_MODE_FULL && disp_refr->sw_rotate) {
        LV_LOG_ERROR("cannot rotate a full refreshed display!");
        return;
    }
```
Next, for the other refresh modes, there is no code for supporting it with a GPU. All the time it is used CPU as rotating buffers.
See ```draw_buf_rotate_90()``` for example.

2. lv_conf_internal.h is used by Kconfig, a graphical configuration tool. You don't need to change those defines.

kisvegabor, 2023-03-28:
Thank your for answering these questions @nicusorcitu .

> Next, for the other refresh modes, there is no code for supporting it with a GPU. All the time it is used CPU as rotating buffers.

Yes, we should definitely allow using GPUs here. There are 3 cases:
- `PARTIAL_MODE`: there is a not too large, buffer containing the rendered image of a smaller area. GPU could be used in the `flush_cb` too, but we need a `draw_task` as the GPU can be used for drawing, so we can't simply use it.
- `FULL_REFRESH_MODE`: Would be hard to manage in SW as we'd need a screen sized buffer for the result too.
- `DIRECT_MODE`: I think we cannot support rotation here, as rendering happens directly into the frame buffer.

The rotation should be a feature of `lv_disp_t` (and not `lv_layer_t`). Maybe a simple but flexible callback or display event could be used, like `PRE_FLUSH` or so, where anything could be done with the rendered image. (even color correction, or other effects)

What do you think?

nicusorcitu, 2023-03-28:
> Maybe a simple but flexible callback or display event could be used, like PRE_FLUSH or so, where anything could be done with the rendered image. (even color correction, or other effects).
>
It looks good to me, any way that can trigger the GPU shall be enough.

I am just wondering how we can do it more efficient especially in FULL_REFRESH_MODE.
Rotating the whole framebuffer at the end of drawing for each frame is performance loosing even if we do it with GPU.
1280x720x4bytes is ~4MB buffer that we need to rotate for every frame.

But imagine that we might deal with 4k screens in the future...

So, I am thinking that if we know the screen is rotated, we shall rotate each and every image/widget at the moment of blit/drawing?! Would that be a solution?

nicusorcitu, 2023-03-29:
> Fortunately, it's not a real rotation but only a smart copy. Reading 1 Mpx and writing 1 Mpx (2 Mpx access) on a 200 MHz RAM really takes 10 ms per frame. 
>
Is not this extremely much to have an extra load of 10ms for each frame? Suppose we expect 30FPS it means one frame for each 33ms. If we have a penalty of 10ms it means 30% slower.

kisvegabor, 2023-03-30:
I meant it's not catastrophic, like 100 ms would be. 
What could we do?

closed 2023-04-21 by lvgl-bot