lvgl/lvgl #4063: I broke the simulator!!!

kdschlosser, 2023-03-15:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I think I broke the simulator. Or maybe found it's limits? I am not sure but I get an error and all it says is to check the browser console. see code block below for browser console output.

Windows 11
Firefox 108.0.1 x64 

### To Reproduce
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/98ced0114cf169eb86bd23ae3fa46d1fc39acecd/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/98ced0114cf169eb86bd23ae3fa46d1fc39acecd/examples/event/lv_example_event_4.py&script_direct=fd05e4fe43b93639a0ba8c8bbe172301f274bdd9

### Expected behavior
not to get the error would be nice....

### Screenshots or video
```
This page is in Quirks Mode. Page layout may be impacted. For Standards Mode use “<!DOCTYPE html>”. app.c1bea19f5aaafe2a9941.js:2:3085079
Iframe src: https://sim.lvgl.io/v9.0/micropython/ports/javascript/lvgl.html top 
Window https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/98ced0114cf169eb86bd23ae3fa46d1fc39acecd/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/98ced0114cf169eb86bd23ae3fa46d1fc39acecd/examples/event/lv_example_event_4.py
app.c1bea19f5aaafe2a9941.js:2:3085552
Attempts to open() files using relative paths will also try: https://raw.githubusercontent.com/lvgl/lvgl/98ced0114cf169eb86bd23ae3fa46d1fc39acecd/examples/event/ wasm_file_api.js:1:318
running init script lvgl.html:60:21
running main script lvgl.html:66:21
INFO: getting online local version of :display_driver.py micropython.js:1:1323
wasm_file_open -> remote host[:display_driver.py] micropython.js:1:1323
Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user’s experience. For more help http://xhr.spec.whatwg.org/ wasm_file_api.js:1:1387
XHRGEThttps://raw.githubusercontent.com/lvgl/lvgl/98ced0114cf169eb86bd23ae3fa46d1fc39acecd/examples/event/display_driver.py
[HTTP/2 404 Not Found 149ms]

INFO: try to get online remote version of https://raw.githubusercontent.com/littlevgl/lv_binding_micro…3260f795f21bfa90b2e7d7199430cf88/driver/js/display_driver.py micropython.js:1:1323
wasm_file_open -> remote host[https://raw.githubusercontent.com/littlevgl/lv_binding_micropython/f960502a3260f795f21bfa90b2e7d7199430cf88/driver/js/display_driver.py] micropython.js:1:1323
XHRGEThttps://raw.githubusercontent.com/littlevgl/lv_binding_micropython/f960502a3260f795f21bfa90b2e7d7199430cf88/driver/js/display_driver.py
[HTTP/2 404 Not Found 115ms]

INFO: try to get online remote version of https://raw.githubusercontent.com/littlevgl/lv_binding_micro…60502a3260f795f21bfa90b2e7d7199430cf88/lib/display_driver.py micropython.js:1:1323
wasm_file_open -> remote host[https://raw.githubusercontent.com/littlevgl/lv_binding_micropython/f960502a3260f795f21bfa90b2e7d7199430cf88/lib/display_driver.py] micropython.js:1:1323
WARNING: this python implementation lacks vars() micropython.js:1:1323
INFO: getting online local version of :display_driver_utils.py micropython.js:1:1323
wasm_file_open -> remote host[:display_driver_utils.py] micropython.js:1:1323
XHRGEThttps://raw.githubusercontent.com/lvgl/lvgl/98ced0114cf169eb86bd23ae3fa46d1fc39acecd/examples/event/display_driver_utils.py
[HTTP/2 404 Not Found 133ms]

INFO: try to get online remote version of https://raw.githubusercontent.com/littlevgl/lv_binding_micro…95f21bfa90b2e7d7199430cf88/driver/js/display_driver_utils.py micropython.js:1:1323
wasm_file_open -> remote host[https://raw.githubusercontent.com/littlevgl/lv_binding_micropython/f960502a3260f795f21bfa90b2e7d7199430cf88/driver/js/display_driver_utils.py] micropython.js:1:1323
XHRGEThttps://raw.githubusercontent.com/littlevgl/lv_binding_micropython/f960502a3260f795f21bfa90b2e7d7199430cf88/driver/js/display_driver_utils.py
[HTTP/2 404 Not Found 126ms]

INFO: try to get online remote version of https://raw.githubusercontent.com/littlevgl/lv_binding_micro…3260f795f21bfa90b2e7d7199430cf88/lib/display_driver_utils.py micropython.js:1:1323
wasm_file_open -> remote host[https://raw.githubusercontent.com/littlevgl/lv_binding_micropython/f960502a3260f795f21bfa90b2e7d7199430cf88/lib/display_driver_utils.py] micropython.js:1:1323
INFO: getting online local version of :lv_utils.py micropython.js:1:1323
wasm_file_open -> remote host[:lv_utils.py] micropython.js:1:1323
XHRGEThttps://raw.githubusercontent.com/lvgl/lvgl/98ced0114cf169eb86bd23ae3fa46d1fc39acecd/examples/event/lv_utils.py
[HTTP/2 404 Not Found 122ms]

INFO: try to get online remote version of https://raw.githubusercontent.com/littlevgl/lv_binding_micro…60502a3260f795f21bfa90b2e7d7199430cf88/driver/js/lv_utils.py micropython.js:1:1323
wasm_file_open -> remote host[https://raw.githubusercontent.com/littlevgl/lv_binding_micropython/f960502a3260f795f21bfa90b2e7d7199430cf88/driver/js/lv_utils.py] micropython.js:1:1323
XHRGEThttps://raw.githubusercontent.com/littlevgl/lv_binding_micropython/f960502a3260f795f21bfa90b2e7d7199430cf88/driver/js/lv_utils.py
[HTTP/2 404 Not Found 123ms]

INFO: try to get online remote version of https://raw.githubusercontent.com/littlevgl/lv_binding_micro…hon/f960502a3260f795f21bfa90b2e7d7199430cf88/lib/lv_utils.py micropython.js:1:1323
wasm_file_open -> remote host[https://raw.githubusercontent.com/littlevgl/lv_binding_micropython/f960502a3260f795f21bfa90b2e7d7199430cf88/lib/lv_utils.py] micropython.js:1:1323
INFO: getting online local version of :lv_timer.py micropython.js:1:1323
wasm_file_open -> remote host[:lv_timer.py] micropython.js:1:1323
XHRGEThttps://raw.githubusercontent.com/lvgl/lvgl/98ced0114cf169eb86bd23ae3fa46d1fc39acecd/examples/event/lv_timer.py
[HTTP/2 404 Not Found 127ms]

[... 29 lines cut ...]
    r https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    instrumentWasmExports/t[r]< https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    createExportWrapper https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    invoke_ii https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    r https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    instrumentWasmExports/t[r]< https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    createExportWrapper https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    ccall https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    cwrap https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    decompressScript https://sim.lvgl.io/v9.0/micropython/ports/javascript/lvgl.html:67
    startRunning https://sim.lvgl.io/v9.0/micropython/ports/javascript/lvgl.html:101
    mainProgram https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    setTimeout handler*mainProgram https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    t https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    run https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    e https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    removeRunDependency https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    t https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    n https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    promise callback*createWasm/< https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    promise callback*createWasm https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
    <anonymous> https://sim.lvgl.io/v9.0/micropython/ports/javascript/micropython.js:1
lvgl.html:103:29

```

amirgon, 2023-03-17:
There are many ways you can break the simulator.  We didn't attempt to make it "unbreakable".

The MP bindings is a thin layer above LVGL and does not prevent you from doing bad things that would crash the application, the same way that referencing an invalid pointer in C would crash your application without any warning.

In your case, you defined:
```python
class _tab_bar(lv.obj):
...
        super().__init__(parent)

class tab_view(lv.obj, limit_methods_mixin):

    def __init__(self, parent):
...
        self._tab_bar = _tab_bar(self)
```

`_tab_bar` base class is `lv.obj` which is a native LVGL object, and you are trying to create an instance of `lv.obj` whose parent is a Python object of type `tab_view`.  
Creating an LVGL native object with a Python parent is not supported unless you create a custom widget properly with `lv.obj_class_t()`.

Unfortunately I didn't find a guide to show how to create an LVGL custom widget (@kisvegabor is there something like that?), but we do have a Micropython example: https://github.com/lvgl/lv_binding_micropython/blob/master/examples/custom_widget_example.py

kdschlosser, 2023-03-18:
I found the problem. I am a ding dong and forgot about MRO

closed 2023-04-02 by kdschlosser