lvgl/lvgl #4064: lv_chart_get_point_pos_by_id() with LV_CHART_TYPE_BAR growing x offset to what is ploted

mhgue, 2023-03-15:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

If using LV_CHART_TYPE_BAR with a cursor, there is a growing offset along the x axis for bars on the right hand side.
The cursor point is correct for data point `id=0` but is left of the center of the bar for increasing  values of `id`.
The x coordinate as provided by `lv_chart_get_point_pos_by_id()` is approximately 8 px per id too small.

### To Reproduce

Just add two lines at the end of [Demo 'Axis ticks and labels with scrolling'](https://docs.lvgl.io/8.3/widgets/extra/chart.html?highlight=chart#axis-ticks-and-labels-with-scrolling). But this issue doesn't depend on zoom or scroll.
```python
cursor = chart.add_cursor(lv.palette_main(lv.PALETTE.BLUE), lv.DIR.LEFT | lv.DIR.BOTTOM)
chart.set_cursor_point(cursor,ser1,10)
```
Resulting in:
```python
##### startup script #####

#!/opt/bin/lv_micropython -i

import lvgl as lv

import display_driver

##### main script #####

def draw_event_cb(e):

    dsc = lv.obj_draw_part_dsc_t.__cast__(e.get_param())
    if dsc.part == lv.PART.TICKS and dsc.id == lv.chart.AXIS.PRIMARY_X:
        month = ["Jan", "Febr", "March", "Apr", "May", "Jun", "July", "Aug", "Sept", "Oct", "Nov", "Dec"]
        # dsc.text is defined char text[16], I must therefore convert the Python string to a byte_array
        dsc.text = bytes(month[dsc.value],"ascii")
#
# Add ticks and labels to the axis and demonstrate scrolling
#

# Create a chart
chart = lv.chart(lv.scr_act())
chart.set_size(200, 150)
chart.center()
chart.set_type(lv.chart.TYPE.BAR)
chart.set_range(lv.chart.AXIS.PRIMARY_Y, 0, 100)
chart.set_range(lv.chart.AXIS.SECONDARY_Y, 0, 400)
chart.set_point_count(12)
chart.add_event_cb(draw_event_cb, lv.EVENT.DRAW_PART_BEGIN, None)

# Add ticks and label to every axis
chart.set_axis_tick(lv.chart.AXIS.PRIMARY_X, 10, 5, 12, 3, True, 40)
chart.set_axis_tick(lv.chart.AXIS.PRIMARY_Y, 10, 5, 6, 2, True, 50)
chart.set_axis_tick(lv.chart.AXIS.SECONDARY_Y, 10, 5, 3, 4,True, 50)

# Zoom in a little in X
chart.set_zoom_x(800)

# Add two data series
ser1 = lv.chart.add_series(chart, lv.palette_lighten(lv.PALETTE.GREEN, 2), lv.chart.AXIS.PRIMARY_Y)
ser2 = lv.chart.add_series(chart, lv.palette_darken(lv.PALETTE.GREEN, 2), lv.chart.AXIS.SECONDARY_Y)

# Set the next points on 'ser1'
chart.set_next_value(ser1, 31)
chart.set_next_value(ser1, 66)
chart.set_next_value(ser1, 10)
chart.set_next_value(ser1, 89)
chart.set_next_value(ser1, 63)
chart.set_next_value(ser1, 56)
chart.set_next_value(ser1, 32)
chart.set_next_value(ser1, 35)
chart.set_next_value(ser1, 57)
chart.set_next_value(ser1, 85)
chart.set_next_value(ser1, 22)
chart.set_next_value(ser1, 58)

# Directly set points on 'ser2'
ser2.y_points = [92,71,61,15,21,35,35,58,31,53,33,73]

cursor = chart.add_cursor(lv.palette_main(lv.PALETTE.BLUE), lv.DIR.LEFT | lv.DIR.BOTTOM)
chart.set_cursor_point(cursor,ser1,10)

chart.refresh()  # Required after direct set
```
[Try here and scroll to the cursor position](https://sim.lvgl.io/v8.3/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/0df09db23e165f157fb8003bf6b914d635dc111e/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/0df09db23e165f157fb8003bf6b914d635dc111e/examples/widgets/chart/lv_example_chart_3.py&script_direct=ce60840fc6c789c55c06ad79744c97f80b2b73d5)

### Expected behaviour

Cursor on center of bar.

### Screenshots or video

[image]

### Quick and dirty fix
Change the line `p_out->x = (int32_t)((int32_t)w * id) / chart->point_cnt;' in 'lv_chart_get_point_pos_by_id()` to `p_out->x = (int32_t)((int32_t)(w+8) * id) / chart->point_cnt;`. No idea why 8 py per id.

kisvegabor, 2023-03-17:
Hi,

It was rounding error when chart was zoomed. I've just pushed a fix! :blush:

closed 2023-03-17 by kisvegabor

mhgue, 2023-03-17:
Thank you for that great graphics library.

commit d855ba88 (2023-03-17): fix(chart): fix rounding error in lv_chart_get_point_pos_by_id

fixes #4064