lvgl/lvgl #4066: esp32 lvgl v8.2 system error after using lv_gif for a period of time

small-zz, 2023-03-16:
The platform is esp32, and the lvgl version number is V8.2; when I use lv_gif to display a gif, sometimes it runs wrong!
Here is my code:
`gif_obj = lv_gif_create(arc_tomato);
    lv_gif_set_src(gif_obj,&gif_test);
    lv_obj_align(gif_obj, LV_ALIGN_CENTER,0,-55);
    lv_obj_set_size(gif_obj,80,50);
`
This is the error log of esp32
`Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4203e362  RA      : 0x4202d084  SP      : 0x3fcc8f60  GP      : 0x3fc90a00
0x4203e362: f_gif_close at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/gif/gifdec.c:650
 (inlined by) gd_close_gif at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/gif/gifdec.c:603

0x4202d084: lv_gif_destructor at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/gif/lv_gif.c:124

TP      : 0x3fb36048  T0      : 0x4005890e  T1      : 0x4038ee3a  T2      : 0x00000000
0x4038ee3a: lldesc_setup_link_constrained at F:/esp/esp-adf/esp-idf/components/soc/lldesc.c:6

S0/FP   : 0x3fcc9edc  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x3fcc9edc
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x4202d06e
0x4202d06e: lv_gif_destructor at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/gif/lv_gif.c:119

A6      : 0x0000010c  A7      : 0x00000006  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x0000ffff  S5      : 0x00000026  S6      : 0x3fc9a000  S7      : 0x3fc998a8
S8      : 0x3fc9d000  S9      : 0x3fc998c9  S10     : 0x3fc998c8  S11     : 0x00000000
T3      : 0x80000000  T4      : 0x00003c00  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000010
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
3fcc8f60: 0x3fcc55a8 0x3fcc9edc 0x3fcc9edc 0x4202d084 0x00000000 0x00000000 0x3fcc9edc 0x4202330a
0x4202d084: lv_gif_destructor at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/gif/lv_gif.c:124

0x4202330a: _lv_obj_destruct at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/core/lv_obj_class.c:137

3fcc8f80: 0x00000000 0x00000000 0x3fcc9edc 0x4202814a 0x3fcc55a8 0x00000000 0x3fcc9d98 0x4202818a
0x4202814a: obj_del_core at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/core/lv_obj_tree.c:391
 (inlined by) obj_del_core at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/core/lv_obj_tree.c:356

0x4202818a: obj_del_core at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/core/lv_obj_tree.c:365

3fcc8fa0: 0x3fcc55a8 0x00000000 0x3fcc55a8 0x4202818a 0x3fcc55a8 0x00000000 0x3fcb2818 0x42028244
0x4202818a: obj_del_core at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/core/lv_obj_tree.c:365

0x42028244: lv_obj_del at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/core/lv_obj_tree.c:67

3fcc8fc0: 0x3fc998ac 0x3fcc4874 0x3fc9d6f0 0x3fc998a9 0x3fc998ac 0x3fc9d000 0x3fca0270 0x4202eab8
0x4202eab8: anim_ready_handler at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/misc/lv_anim.c:432
 (inlined by) anim_timer at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/misc/lv_anim.c:391

3fcc8fe0: 0x3fb36048 0x4005890e 0x0000000f 0x3fc9a000 0x3fc9a000 0x00055c0f 0x3fc9d000 0x00000001
3fcc9000: 0x3fc998b0 0x3fc9d000 0x3fca2d28 0x420308f6 0x000000ef 0x00000000 0x3fca2cd0 0xffffffff
0x420308f6: lv_timer_exec at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/misc/lv_timer.c:316
 (inlined by) lv_timer_handler at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/misc/lv_timer.c:109

3fcc9020: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcc9040: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000001 0x3fc99000 0x3fc995a4 0x4200626e
0x4200626e: gui_task at F:\MC\SpaceMan\weather\weater-c3\build/../components/gui_task/disp_page/taskgui.c:128

3fcc9060: 0x00000000 0x00000000 0x00000000 0x4038b884 0x00000000 0x00000000 0x00000000 0x00000000
0x4038b884: _prvTaskExitError at F:/esp/esp-adf/esp-idf/components/freertos/port/riscv/port.c:164
 (inlined by) prvTaskExitError at F:/esp/esp-adf/esp-idf/components/freertos/port/riscv/port.c:178

3fcc9080: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154
3fcc90a0: 0x3fcc8ec0 0x00055e06 0x3fc95720 0x3fc95720 0x3fcc90a0 0x3fc95718 0x0000000f 0x78b386c4
3fcc90c0: 0x2b13cecb 0x3fcc90a0 0x00000000 0x0000000a 0x3fcc609c 0x5f697567 0x6b736174 0x2a597000
3fcc90e0: 0x004b7788 0x00000000 0x3fcc9090 0x0000000a 0x00000001 0x00000000 0x00000000 0x00000002
3fcc9100: 0x3fc9e0a8 0x3fc9e110 0x3fc9e178 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
3fcc9120: 0x3fcc9c44 0x420d1074 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0x420d1074: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

3fcc9140: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcc9160: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcc9180: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcc91a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcc91c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcc91e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x3f000000 0x00000800 0xa5a5a5a5 0xa5a5a5a5
3fcc9200: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcc9220: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcc9240: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcc9260: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcc9280: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcc92a0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcc92c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcc92e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcc9300: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcc9320: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcc9340: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x4038af86
0x4038af86: pvTaskIncrementMutexHeldCount at F:/esp/esp-adf/esp-idf/components/freertos/tasks.c:5339

ELF file SHA256: 0ec70e38e3e47985

I (22752) esp_core_dump_flash: Save core dump to flash...
I (22759) esp_core_dump_flash: Erase flash 12288 bytes @ 0x6f1000
I (22986) esp_core_dump_flash: Write end offset 0x2fc4, check sum length 4
I (22986) esp_core_dump_flash: Core dump has been saved to flash.
CPU halted.
`

kisvegabor, 2023-03-17:
Hi,

Please enable the [logging](https://docs.lvgl.io/master/porting/log.html) of LVGL.

It can also be a stack overflow issue, so please try increasing the stack size of `taskgui`.

small-zz, 2023-03-20:
The memory resources requested by lv_gif are too large. Is there any way to reduce the resource usage of gif?
```c
I (5673) MAIN: Func:mem_task, Line:79, MEM Total:74380 Bytes

[User]  (6.179, +5719)   lv_scroll_label_create: font_heigh :67 width:41        (in scroller_label.c line #38)
[User]  (6.186, +7)      lv_scroll_label_create: font_heigh :67 width:41        (in scroller_label.c line #38)
[User]  (6.193, +7)      lv_scroll_label_create: font_heigh :67 width:41        (in scroller_label.c line #38)
[User]  (6.201, +8)      lv_scroll_label_create: font_heigh :67 width:41        (in scroller_label.c line #38)
[User]  (6.209, +8)      lv_scroll_label_create: font_heigh :34 width:22        (in scroller_label.c line #38)
[User]  (6.218, +9)      lv_scroll_label_create: font_heigh :34 width:22        (in scroller_label.c line #38)
[Error] (6.275, +57)     lv_mem_realloc: couldn't allocate memory       (in lv_mem.c line #199)
[Error] (6.289, +14)     lv_mem_realloc: couldn't allocate memory       (in lv_mem.c line #199)
I (7878) MAIN: Func:mem_task, Line:79, MEM Total:22020 Bytes

[User]  (6.394, +105)    scroll_event_cb: 1111  (in scroller_label.c line #17)
[User]  (6.587, +193)    scroll_event_cb: 1111  (in scroller_label.c line #17)
[User]  (6.589, +2)      scroll_event_cb: 1111  (in scroller_label.c line #17)
[User]  (6.591, +2)      scroll_event_cb: 1111  (in scroller_label.c line #17)
[User]  (6.597, +6)      scroll_event_cb: 1111  (in scroller_label.c line #17)
[User]  (6.603, +6)      scroll_event_cb: 1111  (in scroller_label.c line #17)
[User]  (6.610, +7)      scroll_event_cb: 1111  (in scroller_label.c line #17)
I (8415) page_weather: 今天是1970-1-1
I (8415) page_weather: 己酉年[鸡]
[User]  (7.620, +1010)   scroll_event_cb: 1111  (in scroller_label.c line #17)
I (9879) MAIN: Func:mem_task, Line:79, MEM Total:29520 Bytes

[User]  (8.529, +909)    scroll_event_cb: 1111  (in scroller_label.c line #17)
[User]  (9.571, +1042)   scroll_event_cb: 1111  (in scroller_label.c line #17)
[User]  (9.572, +1)      scroll_event_cb: 1111  (in scroller_label.c line #17)
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420924e2  RA      : 0x42092464  SP      : 0x3fcce230  GP      : 0x3fc91c00  
0x420924e2: read_image_data at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/gif/gifdec.c:432
 (inlined by) read_image at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/gif/gifdec.c:473
 (inlined by) gd_get_frame at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/gif/gifdec.c:574

0x42092464: read_image_data at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/gif/gifdec.c:420
 (inlined by) read_image at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/gif/gifdec.c:473
 (inlined by) gd_get_frame at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/gif/gifdec.c:574

TP      : 0x3fb21b28  T0      : 0x3fe02046  T1      : 0x4038a0a2  T2      : 0x0000c102
0x4038a0a2: prvIsQueueEmpty at F:/esp/esp-adf/esp-idf/components/freertos/queue.c:2477

S0/FP   : 0x3fcd1520  S1      : 0x3fcca360  A0      : 0x00000004  A1      : 0x00000000
A2      : 0x00000fff  A3      : 0x00000007  A4      : 0x3fcd90ac  A5      : 0x00001000
A6      : 0x420919be  A7      : 0x00000006  S2      : 0x00000084  S3      : 0x00000008
0x420919be: fs_read at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/fsdrv/lv_fs_posix.c:146

S4      : 0x00000084  S5      : 0x00000000  S6      : 0x00000008  S7      : 0x00000001
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x8749e1c6  T4      : 0x16605018  T5      : 0xa0806009  T6      : 0x00001000
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x3fe02046
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
3fcce230: 0x00000000 0x00000001 0x3fcce264 0x420721c2 0x3fc9ae80 0x3fc9e000 0x00000009 0x00000040
0x420721c2: event_send_core at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/core/lv_event.c:469

3fcce250: 0x00003454 0x09bf3c00 0x0000000a 0x42072236 0xa6dbb02c 0x00000000 0x0000ff07 0x3fc9ae7d
0x42072236: lv_event_send at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/core/lv_event.c:78

3fcce270: 0x3fc9ae80 0x3fcd038c 0x3fc9ed80 0x3fcd1068 0x3fc9aea0 0x3fc9aea1 0x3fc9e000 0x3fc9b000
3fcce290: 0x3fc9b000 0x00002570 0x3fc9e000 0x00000001 0x3fc9ae88 0x3fc9e000 0x3fcd108c 0x4207f3ca
0x4207f3ca: next_frame_task_cb at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/extra/libs/gif/lv_gif.c:137

3fcce2b0: 0x00000000 0x00000001 0x3fcd10e4 0x420833ee 0x000000eb 0x000000d6 0x3fcbce5c 0xffffffff
0x420833ee: lv_timer_exec at F:\MC\SpaceMan\weather\weater-c3\build/../components/lvgl/src/misc/lv_timer.c:316
[... 30 lines cut ...]
3fcce530: 0x00000001 0x00000000 0x6600ffff 0x00000000 0xb33fffff 0x00000000 0x00000054 0x00000000
3fcce550: 0x3fcce54c 0x00000000 0x00000000 0x00000000 0x3fcce564 0xffffffff 0x3fcce564 0x3fcce564
3fcce570: 0x00000000 0x3fcce578 0xffffffff 0x3fcce578 0x3fcce578 0x00000001 0x00000001 0x00000000
3fcce590: 0x8200ffff 0x00000000 0xb33fffff 0x00000000 0x0000000c 0x3fcce5b4 0x00000000 0x3fcce59c
3fcce5b0: 0x0000000c 0x071a0008 0x00000000 0x00000000 0x00000028 0x42010ab4 0x42010b14 0x42010b90
[... cut ...]
3fcce5f0: 0x00088002 0x00088002 0x00088002 0x00088002 0x00088002 0x00088002 0x00088002 0x00088002
3fcce610: 0x00028008 0x00028008 0x00028008 0x00028008 0x00028008 0x00028008 0x00028008 0x00028008

ELF file SHA256: 735333306f04cd4c

I (4774) esp_core_dump_flash: Save core dump to flash...
I (4780) esp_core_dump_flash: Erase flash 16384 bytes @ 0x6f1000
I (5091) esp_core_dump_flash: Write end offset 0x3ee4, check sum length 4
I (5091) esp_core_dump_flash: Core dump has been saved to flash.
CPU halted.
```

kisvegabor, 2023-03-20:
> The memory resources requested by lv_gif are too large. Is there any way to reduce the resource usage of gif?

Unfortunately no. :slightly_frowning_face: We could make it smaller if we expected to not have local palettes in gif frames.

small-zz, 2023-03-21:
The chip uses esp32c3, and the product needs to be dynamically switched between BLE and WIFI; once the memory is switched, it will be interrupted, resulting in serious memory fragmentation; so the gif cannot be created; the problem is quite difficult, thank you for your answer anyway

small-zz, 2023-03-25:
> > The memory resources requested by lv_gif are too large. Is there any way to reduce the resource usage of gif?
> 
> Unfortunately no. 🙁 We could make it smaller if we expected to not have local palettes in gif frames.

The GIF displayed by LVGL is faster than the original GIF original? Can it be adjusted?

kisvegabor, 2023-03-27:
Nope, it should be the same. Unless `lv_tick_inc` is not used correctly. 

So how do you handle LVGL's tick?

small-zz, 2023-03-28:
lv_tick_inc The time is 1ms, the specific code is as follows
```c
static esp_timer_handle_t lvgl_timer_handle = NULL;

IRAM_ATTR static void lv_timer_cb(void *arg)
{
    (void) arg;

    lv_tick_inc(1);
}
static esp_timer_create_args_t lvgl_timer = {
    .callback = &lv_timer_cb,
    .arg = NULL,
    .name ="lvgl_timer",
    .dispatch_method = ESP_TIMER_TASK
};

esp_err_t err = esp_timer_create(&lvgl_timer, &lvgl_timer_handle);
err = esp_timer_start_periodic(lvgl_timer_handle, 1000); // start 1ms timer
if (err != ESP_OK)
{
      ESP_LOGE(TAG, "lvgl timer err");
}
``` 
The picture below is the gif I used for testing
[image]

small-zz, 2023-03-28:
> 
[image]
File size: 97.49KiB, width: 80px, height: 80px, frames: 50
It can be displayed normally if the gif file size is within 20Kb; if the file size is too large, an error will be reported
```c
W (7719) page_weather: create_gif_obj lv mem total:59392 free:43892
I (7817) GUI_TASK: lv mem total:59392 free:15428
[Error] (5.388, +180)    lv_mem_realloc: couldn't allocate memory:24588         (in lv_mem.c line #199)
[User]  (5.392, +4)      scroll_event_cb: 1111  (in scroller_label.c line #17)
``` 
According to the documentation, the ram requested by gif is only related to the size of gif
```c
Memory requirements
To decode and display a GIF animation the following amount of RAM is required:

LV_COLOR_DEPTH 8: 3 x image width x image height

LV_COLOR_DEPTH 16: 4 x image width x image height

LV_COLOR_DEPTH 32: 5 x image width x image height
```

small-zz, 2023-03-31:
> 
Let me ask another question, why can't the gif output format binary be displayed with the Image Converter tool?
```c
    gif_obj = lv_gif_create(parent);
    // lv_gif_set_src(gif_obj,&ball_16);
    lv_gif_set_src(gif_obj,"/spiffs/ball_16.bin");
    lv_obj_align(gif_obj, LV_ALIGN_BOTTOM_MID,x_ofs,y_ofs);
```

kisvegabor, 2023-04-03:
You can display it,  however you need to use LVGL's file system API. See [here](https://docs.lvgl.io/8.3/overview/file-system.html).

closed 2023-04-25 by lvgl-bot