lvgl/lvgl #4068: Calling lv_disp_set_draw_ctx to change GPU driver breaks "color_format" field

KarolSaja-AlifSemi, 2023-03-16:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
This issue came up when developing my own GPU driver. Inside lv_disp_create, color_format field in draw_ctx structure is modified after lv_disp_set_draw_ctx is called. This on its own is not a problem, however once application later invokes lv_disp_set_draw_ctx to redirect drawing calls, it will clear the draw_ctx structure (as is shown in lv_draw_sw_init_ctx) and its value (LV_COLOR_FORMAT_NATIVE) will not be restored.

### To Reproduce
Following can be done to simulate changing display driver after lv_disp_create
```c
    lv_init();
    lv_disp_t * disp = lv_disp_create(DISP_HOR_RES, DISP_VER_RES);
    lv_disp_set_draw_ctx(disp, lv_draw_sw_init_ctx, lv_draw_sw_deinit_ctx, sizeof(lv_draw_sw_ctx_t));
```
Afterwards, running any demo application will result in broken visuals due to zero'ed value of color_format.

### Expected behavior
The "color_format" value either should not be part of lv_draw_ctx_t OR it should be explicitly initialized in the software and other renderers in their respective init functions.

### Screenshots or video
[image]

kisvegabor, 2023-03-17:
You are right. I was thinking about what would be the way to fix it, but then realized that this part will overwritten by the `feat/parallel-render` branch in v9. So I wouldn't add a temporary solution. 

If you are using your own `draw_ctx`, can you set the color format in your `draw_ctx_init()` function?

KarolSaja-AlifSemi, 2023-03-17:
Gabor,

I’m currently using the method you proposed, explicitly setting `color_format` to `LV_COLOR_FORMAT_NATIVE` inside my `...init_ctx` function.

If `color_format` is set by the `lv_disp_create` function, it may make more sense to move it inside the `lv_disp_t` structure. Unless it defines a color format specific that renderer (and not a display), in which I case don’t think `lv_disp_create` should touch it at all.