lvgl/lvgl #4069: MSVC: unknown 'LONG_MAX'

gvanem, 2023-03-17:
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

Compiling `extra/libs/fsdrv/lv_fs_win32.c` using MSVC or clang-cl, causes this error:
```c
libs/fsdrv/lv_fs_win32.c(340,36): error: use of undeclared identifier 'LONG_MAX'
        if(file_pointer.QuadPart > LONG_MAX) {
                                   ^
1 error generated.
```

when `<stdlib.h>` (or `<limits.h>`) is **not** in `lv_conf.h` somehow. 
AFAICS, on MinGW, `<limits.h>` gets included via `<windows.h>`.
But not for the Win-Kit/UCRT used by MSVC/clang-cl.

A simple fix for me was:
```diff
--- a/libs/fsdrv/lv_fs_win32.c 2022-10-03 11:10:15
+++ b/libs/fsdrv/lv_fs_win32.c 2023-03-17 07:40:42
@@ -12,6 +12,7 @@

 #include <windows.h>
 #include <stdio.h>
+#include <limits.h>
```

*BTW*. Why this construct at the top `#if LV_USE_FS_WIN32 != '\0'`. A `#if LV_USE_FS_WIN32 != 0` seems more normal.

kisvegabor, 2023-03-17:
Hi

> A simple fix for me was:

After some Googling I found te same solution. Could you send a Pull request?

> Why this construct at the top #if LV_USE_FS_WIN32 != '\0'. A #if LV_USE_FS_WIN32 != 0 seems more normal.

As far as I know `'\0'` is the preferred notion for a zero character.

gvanem, 2023-03-17:
>  Could you send a Pull request?

I'm a *n00b* when it comes to git & PR and such.

> As far as I know '\0' is the preferred notion for a zero character.

Character?! Not an integer; `0` or `!0`?

kisvegabor, 2023-03-17:
> I'm a n00b when it comes to git & PR and such.

No problem, I've added it.

> Character?! Not an integer; 0 or !0?

No, normally it's a character here. E.g. `LV_USE_FS_WIN32 'W'`

commit cdcd6cf5 (2023-03-17): fix(win_fs): add #include <limits.h>

fixes #4069
closed 2023-03-17 by kisvegabor