lvgl/lvgl #4070: MSVC: assert in 'test_fs.c'

gvanem, 2023-03-17:
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

Compiling and running the test-case `test_fs.exe` on Windows (built using MSVC or clang-cl), causes the built-in
*invalid parameter handler* (Dr. Watson) til kick in. Call-stack:
```
ucrtbase!invoke_watson+0x18
ucrtbase!_invalid_parameter+0x12d
ucrtbase!invalid_parameter_noinfo+0x9
ucrtbase!__acrt_stdio_parse_mode<char>+0x27f
ucrtbase!common_openfile<char>+0x22
ucrtbase!common_fsopen<char>+0x9c
test_fs!test_read+0x5b
test_fs!run_test+0x4c
test_fs!main+0x2f
test_fs!invoke_main+0x22
...
```
Since the CRT does not like the `"rd"` parameter to `fopen()`. I fixed this by:
```diff
--- a/test_cases/test_fs.c 2023-03-14 09:11:58
+++ b/test_cases/test_fs.c 2023-03-17 08:10:11
@@ -25,6 +25,12 @@
 #include <errno.h>
 #include <unistd.h>

+#ifdef _WIN32
+#define FOPEN_RD "rb"
+#else
+#define FOPEN_RD "rd"
+#endif
+
 void test_read(void)
 {
     lv_fs_res_t res;
@@ -32,7 +38,7 @@
     char cur[512];
     getcwd(cur, 512);
     errno = 0;
-    void * a = fopen("src/test_files/readtest.txt", "rd");
+    void * a = fopen("src/test_files/readtest.txt", FOPEN_RD);
     printf("%s, %d, %p\n", cur, errno, a);
     fclose(a);
```

So what is this file-mode `"rd"`? A typo?

kisvegabor, 2023-03-17:
I think `"rd"` is a typo. Maybe like the abbreviation of "read"? 

Anyway, does `"rb"` work for you too?

gvanem, 2023-03-17:
> Anyway, does "rb" work for you too?

Yes. But I think there are other issues with binary-mode on Windows. 
AFAICS, there is not a single use of the `O_BINARY` flag in `open()`.
E.g. `test_font_loader.exe` fails on some .fnt-files. But not sure which FS-driver that uses.

kisvegabor, 2023-03-17:
> Yes. But I think there are other issues with binary-mode on Windows.

I've just tried it out and the simple `"r"` is working well too. So I've just pushed an update: c7b26a27b 

> E.g. test_font_loader.exe fails on some .fnt-files. But not sure which FS-driver that uses.
It uses both:
https://github.com/lvgl/lvgl/blob/d855ba8830551ba9f8d1ee9cff14637598e9d638/tests/src/test_cases/test_font_loader.c#L48-L72

closed 2023-04-09 by lvgl-bot