lvgl/lvgl #4071: Building the docs fails on release/v8.3 branch

kisvegabor, 2023-03-17:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I've removed `lv_drivers` from [lv_web_emscripten](https://github.com/lvgl/lv_web_emscripten) and since then the building of the docs fails on lvgl's `release/v8.3` branch because it doesn't find the lv_drivers repo. See [here](https://github.com/lvgl/lvgl/actions/runs/4445704225/jobs/7805143723):
[image]

The obvious solution would be to have `release` branched in `lv_web_emscripten` too and make docs CI use these branches. What do you think @embeddedt ?

embeddedt, 2023-03-17:
@kisvegabor Depending on how often we will run into this problem, a simpler solution could be to just checkout a fixed version of `lv_web_emscripten` within [the script itself](https://github.com/lvgl/lvgl/blob/adcf1663820216a0ade3ab0706139d669f8b84db/scripts/build_html_examples.sh#L7) (assuming we won't need to update the Emscripten port for 8.3 again, which is probably the case).

Creating a branch would also work; to change what branch the CI checks out just requires modifying that script.

kisvegabor, 2023-03-17:
> a simpler solution could be to just checkout a fixed version of lv_web_emscripten within [the script itself](https://github.com/lvgl/lvgl/blob/adcf1663820216a0ade3ab0706139d669f8b84db/scripts/build_html_examples.sh#L7)

Sounds reasonable. This [commit](https://github.com/lvgl/lv_web_emscripten/commit/45e0bc5c8d3e55cfbcaf8214361d2335b8b9a7b4) is the last with v8.3. So I just need to add 
```git
git checkout 45e0bc5c8d
```

to `build_html_examples.sh` on the `release/v8.3` branch, right?

embeddedt, 2023-03-17:
I think that will work fine, yes. (If you wanted to be extremely future-proof, using the full-length hash is technically safer.)

kisvegabor, 2023-03-17:
Thank you, it fixed the issue. Upload to CodeCov still fails, but it's unrelated.

commit 501230e0 (2023-03-17): docs: use a fixed commit of lv_web_emscripten

fixes #4071
closed 2023-03-17 by kisvegabor