lvgl/lvgl #4072: test_calendar and start-of-week

gvanem, 2023-03-17:
I was having a `#define LV_CALENDAR_WEEK_STARTS_MONDAY 1` in my settings (since I'm a Norwegian).
But this caused `test_calendar.exe` to fail. On the 1st button (**Mo**) I assume.
Is there something to do about this? 2 set of `ref_imgs/calendar_*.png` files perhaps?

I'm Win-10 using MSVC-2022 + clang-cl ver. 15.0.3.

gvanem, 2023-03-17:
>  Your lv_conf.h settings shouldn't affect the tests.

Then which config should the tests use? `lv_test_conf.h` only?

All these  `#ifdef` are very confusing.

kisvegabor, 2023-03-17:
Let's approach it from a different angle. Why are the tests compiled for you at the first place? :slightly_smiling_face:  

There shouldn't be `test_calendar.exe` if you are just using a LVGL for UI development.

Which IDE/build system/environment do you use?

gvanem, 2023-03-17:
>  Why are the tests compiled for you at the first place? 

Because I wanted to. Via a hacked up GNU-makefile.

> There shouldn't be test_calendar.exe if you are just using a LVGL for UI development.

I fail to parse this. A `test_calendar.exe` is supposed to work on a specific platform? Which?

> Which IDE/build system/environment do you use?

None besides GNU-make.

kisvegabor, 2023-03-20:
> A test_calendar.exe is supposed to work on a specific platform? Which?

People usually use it on Linux or WSL in Windows. Our CI is Linux based too. TBh, I don't if it works on Windows too, but if so, it's great! :slightly_smiling_face: 
  

Going back to your original question, the tests should use the `lv_test_conf_*.h` files. The build environment for the tests is `cmake` and the `lv_test_conf` file is set in our CMakeList file [here](https://github.com/lvgl/lvgl/blob/master/tests/CMakeLists.txt#L85).

closed 2023-04-12 by lvgl-bot