lvgl/lvgl #4073: Understand to goals of LVGL

kisvegabor, 2023-03-18:
Continue https://github.com/lvgl/lvgl/issues/4059#issuecomment-1474187601

It came up to clarify the goals, directions, target hardware of LVGL. I think it will be a longer discussion  so I opened a dedicated issue for it.

This is how I see it:

- be platform independent to make it suitable in a lot of use cases
   - keep C language
   - keep the simple build system integration
   - have low memory footprint
   - support high-end devices too
   - have bindings to more programming languages
- make LVGL easy to use and easy to get started with 
   - integrate display controllers and platform related drivers 
   - integrate GPUs
   - have ready to use projects
   - support integration with (RT)OS-es
   - ensure SquareLine Studio compatibility
- have features to create amazing UIs
  - visual features similar to what web/CSS has
  - support 3D
  - ensure great performance
- make it sustainable and stable
  - support hobbist too to have a larger user base
  - welcome contribution
  - collaborate with vendors
  - have founds
  - ideally have employees 

These goal are out of scope:
- support creating websites and mobile apps (might be interesting though)
- support super low end MCUs with 64kB flash or so
 
Target hardware
- Entry level: Low power Bluetooth chips with 256kB flash, 64kB RAM, 64 MHz clock speed (e.g. Nordic nrf modules)
- Mid-level MCUs, M4/M33+DMA2D, 1-2 MB flash, 512kB RAM, 100-200 MHz, maybe external RAM. 
- Crossover MCUs, many MBs of flash and RAM, 4-500 MHz, VGLite-like GPU (vector drawing)
- Processors: GHz range CPU with multiple cores, usually with embedded Linux, e.g. NXP i.MX, maybe with OpenGL ES 
- Desktop: video cards

It's important to keep in mind that as LVGL is free we don't have a direct benefit from having two times more users. It includes both hobbiest and professionals. The indirect advantage of having more users are
- probably having more contributors
- have more feedback
- make LVLG more appealing for vendors, who might support us

I'm very open to hear your opinion. I also have meetings with vendors to find mutually beneficial ways to collaborate. It's still in an early stage but I hope we will have an active connection to the vendors and they will also express they needs to shape the future of LVGL.

I'll update the points in this post as discussion goes.

amirgon, 2023-03-18:
How about the following goal: 

_Continue to keep LVGL **a free software with permissive licensing** (MIT). This will help maintain the accessibility and versatility of LVGL for both hobbyists and professionals, and foster a collaborative community of contributors and users._

The benefits of keeping LVGL free under permissive license are:

- Accessibility: By keeping the software free, it becomes accessible to a wider range of users, including hobbyists, students, and developers who may not have the budget for commercial software licenses.

- Collaboration: A permissive license encourages collaboration and contributions from a diverse range of users, which can lead to faster development and innovation.

- Flexibility: The permissive MIT license allows for flexibility in how the software is used, modified, and distributed. This can lead to a wider range of applications and use cases for LVGL.

- Community: A free software license fosters a community of users and contributors who are passionate about the software and invested in its success.

- Reputation: A commitment to free software can enhance the reputation of LVGL and its developers, as it demonstrates a commitment to openness, transparency, and collaboration.

newbthenewbd, 2023-03-18:
What about outright attempting to conquer the non-native desktop app UI sphere?

We here are giving it a little try, and with the likes of Electron on the one side of the scale - much heavier frameworks that require actual web browsers and integrate poorly with simple C code, and on the other side the likes of Nuklear, Dear ImGui or RmlUI, which seem less flexible at this time and have a more difficult API... LVGL, although it has quite many little quirks to polish to be fit (PRs on the way), seems extremely promising!

tdjastrzebski, 2023-03-18:
@kisvegabor  I would further clarify what is a GOAL and what is practice or means to the goal - those seem to be mixed.  
E.g. being pure C written or have ready-to-use projects is good, but what higher purpose/goal does it server? 
What LVGL is supposed to accomplish? Why do you do it? What exactly do you envision as a great success in 2-3 years?
I would list those high-level goals first.
I do not believe that "mission statement" is always a must, but I think in case of LVGL it is.

XuNeo, 2023-03-18:
In case of `have bindings to more programming languages`, I have created the lua binding. It's still ongoing but if any interest, [here](https://github.com/XuNeo/luavgl) it is.

jonbarnettk, 2023-03-18:
> Continue #4059 (comment)

There is a ridiculous amount of potential  for lvgl in many ways.

 I've used it for a long time to make custom de gui in c and it only gets
better and better and is best for really custom gui imo in which few truly
good solutions exist.

I've said a long time ago in another topic about commercial potential there
are many possibilities and directions to go if you wanted that as a goal.

Someone mentioned licensing and I think license should be mpl like zeromq

Just my 2 cents

amirgon, 2023-03-18:
> @amirgon, Collaborations and Community as goal itself? OK, but then why embedded dev graphics? I am sure there are better topics to accomplish that. Reputation? Well, again, there are cheaper and faster ways to gain self-esteem.

@tdjastrzebski I think you misunderstood my post.
The only goal/direction I suggested was "_Continue to keep LVGL a free software with permissive licensing (MIT)_".  
The points below that (Collaborations, Community, Reputation etc.) are **not** goals. These are benefits or reasons for having this one goal, as I've clearly mentioned above them. So no, I didn't suggest Collaborations and Community as goal itself.

> Someone mentioned licensing and I think license should be mpl like zeromq

@jonbarnettk In fact, we already had a discussion of MIT vs. MPL license.
see: https://forum.lvgl.io/t/lvgl-licensing-model-review/6538  
Personally I think MIT is best for LVGL.

tdjastrzebski, 2023-03-18:
@amirgon, my point is that goal and direction are two much different things.
It is like saying: my goal is to go this direction, but where do you want to get? That is unknown.
I am not trying to start a purely philosophical debate.
I really do not know where LVGL is currently heading. It seems that some see it as new universal low-level graphics engine, like Qt or gtk. Some people bring concepts from CSS and probably they already think of XAML. I thought it was going to be excellent, very promising library aimed at embedded devices, but I already find myself looking for alternatives since I am afraid LVGL is not going there.

@jonbarnettk, Exactly, and this huge potential is a serious problem because no reasonable, consistent product can expand in several opposite directions simultaneously. I mean, it can for a while but then it inevitably is going to collapse and become a history.

jonbarnettk, 2023-03-19:
Theres certainly arguments for both, but license is a factor when it comes
to overall goals and license aligns with whatever those goals may be.

I would still agree mpl makes a lot of sense that would benefit the future
of lvgl.  Many mature projects use mpl and it doesn't stop users from using
it, and would arguably keep it free but give lvgl a commercial advantage.

I am unsure the business goals of lvgl, however it definitely deserves some
commercial success if that is an objective and I would argue mit is not the
right license.

There are many objectives of a project, both technical and business goals
(if any), and those should be separated and planned.

And I think that is basically what @tdjastrzebski
<https://github.com/tdjastrzebski> is saying and he is not wrong.

nicusorcitu, 2023-03-19:
I will not touch the license policy here, as this seems to be a different story that the original post about the LVGL rearchitecture goal.
It all starts from the question: What shall LVGL become? Shall it target the DMA2D hardware or a bigger GPU hardware like VGLite?
Inside NXP products we have PXP (which is a "smart DMA" - knowing just fill, blit, transformation) and we also have a series of GPUs increasing in capabilities GC255 on RT500, GC355 on RT1170 and next generation will fallow.
Based on our experience, the expectations from customers always grows, the graphics becomes even more complex and the needs just for a simple DMA seems like will end soon. Is not me saying that we shall fallow a GPU friendly rearchitecture because is more convenient for me, but reworking now the LVGL with only DMA2D feature list support is like stopping the evolution.

I liked very much the idea of reworking the LVGL described in the original post [here](https://github.com/lvgl/lvgl/issues/3700#issue-1382408372) with the dispatcher being able to split the task group into simpler task if none of PE is capable of it.

Draw task descriptor shall be clear and specific.

If we need a simple blit - create a task for it -> both DMA2D and GPU will be able to handle it.
If we need a blit and rotate with opa - create a task for it -> both DMA2D and GPU will be able to handle it.
If we need to draw a rectangle - create a task for it -> both DMA2D and GPU will be able to handle it.
If we need to draw a rectangle border/outline - create a task for it -> **only GPU or CPU will be able to handle it.**

My point here is, do not try to use the DMA2D for tasks that was not designed to.
Do not try to split the border into 4 lines - each line is a very tinny rectangle right? So you might be tempted to draw those 4 lines like filling the rectangle. But then the lines can have rounded endings or can be dashed lined (with gaps). So a single line might become 10 very short lines. Do we really think we will get a performance by filling those 10 * 4 very small lines with a DMA2D? I believe not.
Instead, is we just split that border into very small tasks, we will create a bottleneck for the really performant GPU (like VGLite) which can create that dashed border, with rounded corners etc in a single shot.

Nicu.

tdjastrzebski, 2023-03-19:
Let me present this from yet another perspective. I just came back from Embedded World expo in Nuremberg where ST gave me a new toy - [STM32U5A9J-DK](https://www.st.com/en/evaluation-tools/stm32u5a9j-dk.html) board. Look at [this movie I made](https://www.youtube.com/watch?v=WbMabPVxtAw). This graphics really kicks the ass.  
LVGL is not there yet and, sadly, I am not sure if it **wants to be**.  
Despite my efforts (PR #3904), ST DMA2D cannot be always used for blending due to architectural constraints.

What really makes this graphics so stunning is not DMA2D. It is the new GPU2D (NeoChrom GPU) supporting texture decompressing, triangles, quadrilateral drawing, blit support and vector graphics, image transformation and more. For the complete list of features refer to [RM0456, 45.2](https://www.st.com/content/ccc/resource/technical/document/reference_manual/group0/f3/60/ca/d2/98/c8/47/88/DM00477635/files/DM00477635.pdf/jcr:content/translations/en.DM00477635.pdf).  

My question is simple. Is LVGL going to support it? Is it one of the top priorities?  
If not, I really hate to say it, but I will have to migrate to Touch GFX.

newbthenewbd, 2023-03-20:
@tdjastrzebski
> @amirgon, my point is that goal and direction are two much different things. It is like saying: my goal is to go this direction, but where do you want to get? That is unknown. I am not trying to start a purely philosophical debate. I really do not know where LVGL is currently heading. It seems that some see it as new universal low-level graphics engine, like [Qt](https://en.wikipedia.org/wiki/Qt_(software)) or [gtk](https://en.wikipedia.org/wiki/GTK). Some people mention CSS and probably they already think of XAML. 3D graphics has been brought up. I thought LVGL was going to be an excellent, very promising library aimed at embedded devices, but I already find myself looking for alternatives since I am afraid LVGL is not going there.
> 
> @jonbarnettk, Exactly, and this huge potential is a serious problem because no reasonable, consistent product can expand in several opposite directions simultaneously. I mean, it can for a while, but then it inevitably is going to implode and become a history.

I felt that uneasy gaze going my way 🙂 but I don't think that I agree with this assessment of the situation (i.e. the harm in diversity). Let me express it thus: if Qt or gtk were specifically what some thought of when reviewing LVGL for not-embedded-at-all uses... they would use Qt or gtk and not waste time. The folks are there despite of the currently uncertain fit, because they're looking for something [_much better_](https://handmade.network/manifesto).

The universal appeal that LVGL has stems from its design goals, versatility being one of them, and the small overhead, another: they're in the very name, and clearly implemented in practice, too. From my POV, there indeed exists a place where the immediate interests of the embedded and the desktop diverge: namely, it is in the interests of the desktop for there to be _less_ code. Where embedded runs on the bare metal and requires LVGL to supply the input conversion (e.g. keypresses/screen keyboard to text), and manage the drawing in batches, and allocate memory, and support many low color rendering modes, and the event loop, and...

The desktop tends to already have perfectly good functions for these, and would prefer a bare add-interface-pipe-input-get-draw-function-calls, together with the shorter compile time and smaller binaries. **But would supporting that kind of workflow mean pulling a 180 and dropping all the embedded?** Is this plausible at this point? Rather, perhaps with a very powerful toggle in the config, the versatility of the library could only increase, and the limits of the current architecture shrink. At the end of the day, first-class support for embedded devices together with the indeed modern and expansive design goals clearly pursued feel like a shield against both the kinds of "microoptimizations" that sacrifice usability, and... let's call it "macrooptimizations", which in turn sacrifice the micro, or, as tends to happen, everything older or weaker than the dev's personal computer.

The actual quirks, however, that make LVGL need some work for any big UI usage, are really in the area of polish: stuff that, I suppose, might be hardly a concern or easily hackable around when you're designing for a watch or some other HMI, but teeth-grinding for the big UIs. Stuff like text selection being half a character off and conflicting with scrolling, some elements not refreshing just right on live resize, escape key getting entered as a character... I mean, you may take a look at the little fork under my profile. It's effectively an attempt to improve the quality for everybody. :)

kisvegabor, 2023-03-20:
Wao, it's great that this topic got such a great interest. Thank you! :pray: 

@amirgon 
> Continue to keep LVGL a free software with permissive licensing (MIT). This will help maintain the accessibility and versatility of LVGL for both hobbyists and professionals, and foster a collaborative community of contributors and users.

True, I should have added it. Updated.

@newbthenewbd 
> How about outright attempting to conquer the non-native desktop app UI sphere?

I also have this goal in mind! First thing we need is effective rendering on desktops too. After that I also think that only a few features needs to be added. 

@tdjastrzebski 
> What LVGL is supposed to accomplish? Why do you do it? What exactly do you envision as a great success in 2-3 years?
What is the objective?

I think we can not set a fixed target (goal) and when we reach it we can lay back. To phrase it shortly I'd say the goal is to "Have an **available** and **good** UI library mainly for embedded". Where "available" means platform independent, free, etc, and "good" means feature rich, easy to use, etc.

As these are not absolute terms and the environment is always changing, hardware, trends, etc it also means a continues work.    

> Exactly, and this huge potential is a serious problem because no reasonable, consistent product can expand in several opposite directions simultaneously. I mean, it can for a while, but then it inevitably is going to implode and become a history.

Note that the embedded world is changing too. GPUs are common now, 3D and vector graphics started to appear, and in the few years they will be common too. Let's say in 2030 embedded UI will mean something very different than today. And we can't stuck at 320x240 display and 100 MHz MCU. 

As I see the target platforms are from monochrome displays with basic graphics to full-HD (or even larger) on desktop with 3D. Mobile app and web are not targets.

> I am unsure the business goals of lvgl, however it definitely deserves some
commercial success if that is an objective and I would argue mit is not the
right license.

Now there are no employees in LVGL at all. It's great that we gut that far but it needs to change in the future to make it stable and sustainable. That's why I'm trying to establish some collaboration with vendors. We will see how it works. 

@nicusorcitu 
> Based on our experience, the expectations from customers always grows, the graphics becomes even more complex and the needs just for a simple DMA seems like will end soon. Is not me saying that we [...]

I agree, we need great support for GPU, as they will be mandatory for to UIs of the future.

@tdjastrzebski 
> My question is simple. Is LVGL going to support it? If so, when? What is the roadmap? Is it now one of the top priorities?

Yes, it will, for sure. Actually, this is what I try to reach with the vendors: as LVGL is free, we need to find alternative revenue sources. What I have in mind is for a price offering for vendors to integrate and maintain their GPUs, drivers and boards in LVGL. If they pay a fee, we can have employees who take care of it and ensure that LVGL always support the latest technologies/products. Without it everything will be supported only at a best-effort level.

tdjastrzebski, 2023-03-21:
@kisvegabor What I really suggest is implementing more formal Product Management practices.  
In my view, LVGL reached the stage where it is a must to successfully move forward.

kisvegabor, 2023-03-22:
> What I really suggest is implementing more formal Product Management practices.

Can you mention some examples what you have in mind?

kisvegabor, 2023-04-06:
As I mentioned I hope we can step a level with LVGL's background in the near future. I'll comment here if I know more.

closed 2023-04-28 by lvgl-bot