lvgl/lvgl #4075: Simulating Out Of Memory (OOM) in unit tests

C47D, 2023-03-19:
### Introduce the problem

I'm writing the initial unit tests for `lv_meter` and found 2 issues:
- In `lv_meter_add_needle_line`, `lv_meter_add_needle_img`, `lv_meter_add_arc` and `lv_meter_add_scale_lines` when malloc fails the functions keep going, shouldn't they return NULL?
- Is there a way to force an OOM in a particular set of tests?

### Examples and cases

Testing OOM issues, e.g. testing `indic` is NULL on OOM issues or with the given parameters when it's allocated successfully.

```c
lv_meter_indicator_t * lv_meter_add_scale_lines(lv_obj_t * obj, lv_color_t color_start,
                                                lv_color_t color_end, bool local, int16_t width_mod)
{
    LV_ASSERT_OBJ(obj, MY_CLASS);
    lv_meter_t * meter = (lv_meter_t *)obj;
    lv_meter_indicator_t * indic = _lv_ll_ins_head(&meter->indicator_ll);
    LV_ASSERT_MALLOC(indic);
    lv_memzero(indic, sizeof(lv_meter_indicator_t));
    indic->opa = LV_OPA_COVER;

    indic->type = LV_METER_INDICATOR_TYPE_SCALE_LINES;
    indic->type_data.scale_lines.color_start = color_start;
    indic->type_data.scale_lines.color_end = color_end;
    indic->type_data.scale_lines.local_grad = local;
    indic->type_data.scale_lines.width_mod = width_mod;

    lv_obj_invalidate(obj);
    return indic;
}
```

### Suggested solution

Ideally being able to mock/fake `_lv_ll_ins_head` but I think it's not currently possible because all the source files are compiled in order to run tests, it would require to only compile the file under test.

kisvegabor, 2023-03-20:
> In lv_meter_add_needle_line, lv_meter_add_needle_img, lv_meter_add_arc and lv_meter_add_scale_lines when malloc fails the functions keep going, shouldn't they return NULL after LV_ASSERT_MALLOC?

Yes, malloc's result should be checked, and NULL should be returned. 

> Ideally being able to mock/fake _lv_ll_ins_head but I think it's not currently possible because all the source files are compiled in order to run tests, it would require to only compile the file under test.

That's true. It'd complicate the tests further that I'd like to avoid if possible.

C47D, 2023-03-20:
Is there a way to control the allocation? I kind of remember there was a way to do it, maybe @embeddedt knows?

kisvegabor, 2023-03-21:
> Is there a way to control the allocation?

We can replace `lv_malloc` in `lv_conf.h` with a test `malloc` which can be controlled externally to work normally or return a special value in some cases.

C47D, 2023-03-21:
@kisvegabor I've done a dirty implementation of a malloc stub and it seems to be working as expected, see #4078

closed 2023-03-24 by C47D