lvgl/lvgl #4080: Roller Selected part height

ashrafkamel5, 2023-03-22:
### Perform all steps below and tick them with [x]
- [ x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

The roller selected part height can't be set.
and is only affected by the font size and nothing else.

### To Reproduce

```c
lv_obj_t * roller1 = lv_roller_create(lv_scr_act());
    
    lv_roller_set_options(roller1,
                          "January\n"
                          "February\n"
                          "March\n"
                          "April\n"
                          "May\n"
                          "June\n"
                          "July\n"
                          "August\n"
                          "September\n"
                          "October\n"
                          "November\n"
                          "December",
                          LV_ROLLER_MODE_INFINITE);

    lv_roller_set_visible_row_count(roller1, 1);
   
    
    lv_obj_center(roller1);
    //lv_obj_add_event_cb(roller1, event_handler, LV_EVENT_ALL, NULL);
    lv_obj_set_style_text_font(roller1, &lv_font_montserrat_24, LV_PART_SELECTED);
    //lv_obj_set_style_height(roller1, 100, LV_PART_SELECTED);
    //lv_obj_set_style_text_line_space(roller1, 120, LV_PART_SELECTED);
    
    
```

### Expected behavior

widget parts behavior with style properties should be consistent.

roller1 example
but making `    lv_roller_set_visible_row_count(roller1, 1);`
for only one option to be visible.
then you can't change the widget height without displaying other options.
even after using`   lv_obj_set_style_text_line_space(roller1, 100, LV_PART_SELECTED);`
it only changes the travel distance, not the selected part height.
on the other hand 
` lv_obj_set_style_text_line_space(roller1, 100, LV_PART_MAIN);`
it works fine but affects all options and my intent is to make the selected part only have more height.

and this line has no effect.
`    //lv_obj_set_style_height(roller1, 100, LV_PART_SELECTED);`

kisvegabor, 2023-03-22:
The size of the selection rectangle is set based on ` lv_obj_set_style_text_line_space(roller1, value, LV_PART_MAIN);` Its size is set to be halfway between the slected option, and the options above and below it.

ashrafkamel5, 2023-03-22:
Many thanks for your fast reply.
As I mentioned, I want to make the selected part size to be different. "same idea as different font only for the selected part".

using LV_PART_MAIN changes all option sizes which is not desirable.

do not you see it weird that the selected part accepts some but not all style properties that they are just ignored? 
including padding, it doesn't affect the text placement in the options "they are always centered ".

I want to mimic the behavior of the roller at[ this example](https://www.embedded-wizard.de/demo/washing-machine). 
which should be easy to implement
what do you think?

kisvegabor, 2023-03-23:
> do not you see it weird that the selected part accepts some but not all style properties that they are just ignored?

It's normal that not all style properties are applied for all widget and all parts. E.g. the arc related style properties are also ignored by any part of the roller.

> I want to mimic the behavior of the roller at[ this example](https://www.embedded-wizard.de/demo/washing-machine).

Note that Embedded Wizard has very few built-in widgets. See [here](https://doc.embedded-wizard.de/using-widgets?v=12.00). However, similarly to LVGL it's also flexible enough to create custom widgets. 

This kind of roller is not supported out of the box in LVGL, but you can put it together yourself from simpler building blocks. 

> which should be easy to implement

It's a custom widget where colors are changing dynamically, two objects' scroll is chained together but with different   speed, there are in and out animations, snapping, etc. All these are doable but not "easy" for sure.

ashrafkamel5, 2023-03-25:
Ok many thanks for the clarification.

My thoughts whey I initially interpreted this design.
That it is a simple roller
That shows only the selected option
And in the pressed event call back it is changed to view more options and after a time out event or deactivate event
We set it back to one option

What I couldn't make myself
Was making a different size for the selected option only

Can you please share your experience with how would I solve this?

And another question

Is how to get started with the custom widgets
It's really interesting topic but I didn't find any documentation for it

closed 2023-04-10 by ashrafkamel5